jblance/mpp-solar #113: "no response" when trying to connect to Daly BMS

SirkoVZ, 2021-05-18:
Hi,

I connected my Daly BMS using the shipped RS485 USB Adapter, which seems to be fine:
dmesg:
```
[ 4849.385451] usb 1-1.1.2: new full-speed USB device number 7 using xhci_hcd
[ 4849.521650] usb 1-1.1.2: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.64
[ 4849.521671] usb 1-1.1.2: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[ 4849.521688] usb 1-1.1.2: Product: USB Serial
[ 4849.526542] ch341 1-1.1.2:1.0: ch341-uart converter detected
[ 4849.532007] usb 1-1.1.2: ch341-uart converter now attached to ttyUSB1
```
lsusb:
```
...
Bus 001 Device 007: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
...
```
 lsmod | grep ch341
```
ch341                  16384  0
usbserial              36864  2 cp210x,ch341
```

I tried mpp-solr (0.7.55):
```
mpp-solar -p /dev/ttyUSB1 --porttype daly -b 9600 -P DALY -c cellVoltages -I -D
but I get "no response"
INFO:root:Solar Device Command Utility, version: 0.7.55, recent changes: add extra pi17 commands from lonertic
INFO:root:Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB1 (porttype=daly)" using protocol "DALY"
DEBUG:root:device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
DEBUG:device:mppsolar __init__ args ()
DEBUG:device:mppsolar __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB1', 'protocol': 'DALY', 'baud': 9600, 'porttype': 'daly', 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
INFO:device:set_port: Port overide - using port 'daly'
DEBUG:device:get_port_type: port matches daly
INFO:device:set_port: Using dalyserialio for communications
DEBUG:device:set_protocol: Protocol DALY
DEBUG:device:mppsolar __init__ name unnamed, port <mppsolar.io.dalyserialio.DalySerialIO object at 0xb5eb9e30>, protocol <mppsolar.protocols.daly.daly object at 0xb5eb9c70>
DEBUG:root:Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0xb5eb9f90>, 'cellVoltages', 'cellVoltages', 'screen', None, None)]
INFO:root:Looping 1 commands
INFO:root:Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.dalyserialio.DalySerialIO object at 0xb5eb9e30>, protocol: <mppsolar.protocols.daly.daly object at 0xb5eb9c70> for command: cellVoltages, tag: cellVoltages, outputs: screen
INFO:device:run_command: Running command cellVoltages
INFO:daly:get_full_command: Using protocol b'DALY' with 7 commands
DEBUG:AbstractProtocol:get_command_defn: Processing command 'cellVoltages'
DEBUG:AbstractProtocol:get_command_defn: Found command cellVoltages in protocol b'DALY'
DEBUG:daly:get_full_command: full command: b'\xa5\x80\x95\x08\x00\x00\x00\x00\x00\x00\x00\x00\xc2\n'
INFO:device:run_command: full command b'\xa5\x80\x95\x08\x00\x00\x00\x00\x00\x00\x00\x00\xc2\n' for command cellVoltages
DEBUG:AbstractProtocol:get_command_defn: Processing command 'cellVoltages'
DEBUG:AbstractProtocol:get_command_defn: Found command cellVoltages in protocol b'DALY'
DEBUG:DalySerialIO:send_and_receive: port /dev/ttyUSB1, baudrate 9600
DEBUG:DalySerialIO:send_and_receive: Executing command via dalyserialio...
DEBUG:DalySerialIO:send_and_receive: serial response was: b''
DEBUG:device:run_command: Send and Receive Response b''
INFO:AbstractProtocol:decode: response passed to decode: b''
INFO:AbstractProtocol:No response
INFO:device:run_command: Decoded response {'ERROR': ['No response', '']}
DEBUG:root:results: {'ERROR': ['No response', '']}
INFO:root:attempting to create output processor: screen
DEBUG:screen:processor.screen __init__ kwargs {}
DEBUG:root:Using output filter: None
INFO:screen:output: Using output processor: screen
DEBUG:screen:output: kwargs {'data': {'ERROR': ['No response', '']}, 'tag': 'cellVoltages', 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None, 'mqtt_topic': 'mpp-solar', 'filter': None, 'excl_filter': None, 'keep_case': False}
Parameter                       Value           Unit
DEBUG:helpers:key_wanted: No filter and key error not excluded by excl_filter None so key wanted
error                           No response
DEBUG:root:Not daemon, so not looping

```
What do I wrong?

jblance, 2021-05-18:
I'm not sure - what version of the BMS are you using?
It has worked for some people, but some forums have noted that different models have different protocols.
Can you try sending `b'\xDD\xA5\x05\x00\xFF\xFB\x77\n'` and/or `b'\xa5\x40\x63\x08\x00\x00\x00\x00\x00\x00\x00\x00\x50'` and see what response is given

refs: 
https://diysolarforum.com/threads/daly-bms-rs232-communication-protocol-crc-calculation.21853/
https://diysolarforum.com/threads/decoding-the-daly-smartbms-protocol.21898/

jblance, 2021-05-18:
Use a script like
```python
import serial
import time

## Vars ################
port = "/dev/ttyUSB1"
baud = 9600
########################

cmd = b'\xDD\xA5\x05\x00\xFF\xFB\x77\n'
#cmd = b'\xa5\x40\x63\x08\x00\x00\x00\x00\x00\x00\x00\x00\x50'
print (f"sending command {cmd}")

with serial.serial_for_url(port, baud) as s:
    s.timeout = 1
    s.write_timeout = 1
    s.flushInput()
    s.flushOutput()
    bytes_written = s.write(cmd)
    print (f"wrote {bytes_written} bytes")

    for _ in range(10):
        response_line = s.readline()
        time.sleep(10)
        print (f"Got response: {response_line}")
```

SirkoVZ, 2021-05-19:
Thanks for your response.
in first case I got:

```
[07:40:16][pi@raspi:~/solar]$ python daly_test.py
sending command b'\xdd\xa5\x05\x00\xff\xfbw\n'
wrote 8 bytes
Got response: b''
Got response: b''
Got response: b''
^CTraceback (most recent call last):
  File "daly_test.py", line 24, in <module>
    time.sleep(10)
KeyboardInterrupt
```
in second case I got:
```
[07:41:22][pi@raspi:~/solar]$ python daly_test.py
sending command b'\xa5@c\x08\x00\x00\x00\x00\x00\x00\x00\x00P'
wrote 13 bytes
Got response: b'\xa5\x01c\x08\x01DL-BMS-\xde\xa5\x01c\x08\x02R32-01E\x9d'
Got response: b''
Got response: b''
^CTraceback (most recent call last):
  File "daly_test.py", line 24, in <module>
    time.sleep(10)
KeyboardInterrupt
```
I have abrand of Daly BMS: MGod BMS
https://de.aliexpress.com/item/1005001971361691.html?spm=a2g0s.9042311.0.0.4aa24c4d3z4a8C
16s LiFePo 150A
It is just the different Color, manual and so on is the same as for Daly BMS, Windows Software is also the same...

jblance, 2021-05-19:
it looks like your BMS is responding to a different 'address' compared to the others that have tested the code
I've made a quick change and added a protocol DALY40 that is basically the same as DALY but using address 0x40 for the BMS
Can you grab the latest code from git and try the commands using `-P DALY40`
ie `mpp-solar -p /dev/ttyUSB1 --porttype daly -b 9600 -P DALY40 -c cellVoltages -I`

SirkoVZ, 2021-05-19:
Thanks, looks much better:
```
[11:57:04][pi@raspi:~]$ mpp-solar -p /dev/ttyUSB1 --porttype daly -b 9600 -P DALY40 -c cellVoltages -I
INFO:root:Solar Device Command Utility, version: 0.7.56, recent changes: add daly40
INFO:root:Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB1 (porttype=daly)" using protocol "DALY40"
INFO:device:set_port: Port overide - using port 'daly'
INFO:device:set_port: Using dalyserialio for communications
INFO:root:Looping 1 commands
INFO:root:Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.dalyserialio.DalySerialIO object at 0xb5f25f50>, protocol: <mppsolar.protocols.daly40.daly40 object at 0xb5f25c30> for command: cellVoltages, tag: cellVoltages, outputs: screen
INFO:device:run_command: Running command cellVoltages
INFO:daly40:get_full_command: Using protocol b'DALY40' with 7 commands
INFO:device:run_command: full command b'\xa5@\x95\x08\x00\x00\x00\x00\x00\x00\x00\x00\x82\n' for command cellVoltages
INFO:AbstractProtocol:decode: response passed to decode: b'\xa5\x01\x95\x08\x01\x0c\x86\x0c\x89\x0c\x89\x00\x00\xa5\x01\x95\x08\x02\x0c\x89\x0c\x83\x0c\x87\x00\xfc\xa5\x01\x95\x08\x03\x0c\x88\x0c\x87\x0c\x88\x00\x01\xa5\x01\x95\x08\x04\x0c\x89\x0c\x89\x0c\x87\x00\x04\xa5\x01\x95\x08\x05\x0c\x88\x0c\x89\x0c\x88\x00\x05\xa5\x01\x95\x08\x06\x0c\x86\x0c\x89\x0c\x88\x00\x04'
INFO:daly40:is multiframe response - assuming ok for now
INFO:daly40:Multi frame response with 6 frames
INFO:AbstractProtocol:decode: Processing response of type MULTIFRAME-POSITIONAL
INFO:device:run_command: Decoded response {'raw_response': ['¥\x01\x95\x08\x01\x0c\x86\x0c\x89\x0c\x89\x00\x00¥\x01\x95\x08\x02\x0c\x89\x0c\x83\x0c\x87\x00ü¥\x01\x95\x08\x03\x0c\x88\x0c\x87\x0c\x88\x00\x01¥\x01\x95\x08\x04\x0c\x89\x0c\x89\x0c\x87\x00\x04¥\x01\x95\x08\x05\x0c\x88\x0c\x89\x0c\x88\x00\x05¥\x01\x95\x08\x06\x0c\x86\x0c\x89\x0c\x88\x00\x04', ''], '_command': 'cellVoltages', '_command_description': 'Cell Voltages Information', 'Cell 01 Voltage': [3.206, 'V'], 'Cell 02 Voltage': [3.209, 'V'], 'Cell 03 Voltage': [3.209, 'V'], 'Cell 04 Voltage': [3.209, 'V'], 'Cell 05 Voltage': [3.203, 'V'], 'Cell 06 Voltage': [3.207, 'V'], 'Cell 07 Voltage': [3.208, 'V'], 'Cell 08 Voltage': [3.207, 'V'], 'Cell 09 Voltage': [3.208, 'V'], 'Cell 10 Voltage': [3.209, 'V'], 'Cell 11 Voltage': [3.209, 'V'], 'Cell 12 Voltage': [3.207, 'V'], 'Cell 13 Voltage': [3.208, 'V'], 'Cell 14 Voltage': [3.209, 'V'], 'Cell 15 Voltage': [3.208, 'V'], 'Cell 16 Voltage': [3.206, 'V'], 'Cell 17 Voltage': [3.209, 'V'], 'Cell 18 Voltage': [3.208, 'V']}
INFO:root:attempting to create output processor: screen
INFO:screen:output: Using output processor: screen
Command: cellVoltages - Cell Voltages Information
------------------------------------------------------------
Parameter                       Value           Unit
cell_01_voltage                 3.206           V
cell_02_voltage                 3.209           V
cell_03_voltage                 3.209           V
cell_04_voltage                 3.209           V
cell_05_voltage                 3.203           V
cell_06_voltage                 3.207           V
cell_07_voltage                 3.208           V
cell_08_voltage                 3.207           V
cell_09_voltage                 3.208           V
cell_10_voltage                 3.209           V
cell_11_voltage                 3.209           V
cell_12_voltage                 3.207           V
cell_13_voltage                 3.208           V
cell_14_voltage                 3.209           V
cell_15_voltage                 3.208           V
cell_16_voltage                 3.206           V
cell_17_voltage                 3.209           V
cell_18_voltage                 3.208           V

```

jblance, 2021-05-19:
Oh, thats weird. The response is definitely 6 frames with 3 results in each
Can you get a screenshot of the app and the comand results as close to the same time?

The decode _should_ be the same...
Can you try some of other commands to see if there are other anomalies

SirkoVZ, 2021-05-19:
ok, the first 16 voltages are fine, I don't know what voltage 17 and 18 is:
```
[14:14:22][pi@raspi:~]$ mpp-solar -p /dev/ttyUSB1 --porttype daly -b 9600 -P DALY40 -c cellVoltages
Command: cellVoltages - Cell Voltages Information
------------------------------------------------------------
Parameter                       Value           Unit
cell_01_voltage                 3.142           V
cell_02_voltage                 3.142           V
cell_03_voltage                 3.142           V
cell_04_voltage                 3.142           V
cell_05_voltage                 3.138           V
cell_06_voltage                 3.14            V
cell_07_voltage                 3.14            V
cell_08_voltage                 3.142           V
cell_09_voltage                 3.141           V
cell_10_voltage                 3.141           V
cell_11_voltage                 3.142           V
cell_12_voltage                 3.14            V
cell_13_voltage                 3.142           V
cell_14_voltage                 3.142           V
cell_15_voltage                 3.142           V
cell_16_voltage                 3.142           V
cell_17_voltage                 3.142           V
cell_18_voltage                 3.142           V
```
[image]

The temperatures show some more, there is only one I guess: 22°C

```
[14:27:14][pi@raspi:~]$ mpp-solar -p /dev/ttyUSB1 --porttype daly -b 9600 -P DALY40 -c cellTemperatures -I
INFO:root:Solar Device Command Utility, version: 0.7.56, recent changes: add daly40
INFO:root:Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB1 (porttype=daly)" using protocol "DALY40"
INFO:device:set_port: Port overide - using port 'daly'
INFO:device:set_port: Using dalyserialio for communications
INFO:root:Looping 1 commands
INFO:root:Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.dalyserialio.DalySerialIO object at 0xb5eb7fb0>, protocol: <mppsolar.protocols.daly40.daly40 object at 0xb5eb7c90> for command: cellTemperatures, tag: cellTemperatures, outputs: screen
INFO:device:run_command: Running command cellTemperatures
INFO:daly40:get_full_command: Using protocol b'DALY40' with 7 commands
INFO:device:run_command: full command b'\xa5@\x96\x08\x00\x00\x00\x00\x00\x00\x00\x00\x83\n' for command cellTemperatures
INFO:AbstractProtocol:decode: response passed to decode: b'\xa5\x01\x96\x08\x01>8\x0c9\x0c8\x10T'
INFO:daly40:Multi frame response with 1 frames
INFO:AbstractProtocol:decode: Processing response of type MULTIFRAME-POSITIONAL
INFO:device:run_command: Decoded response {'raw_response': ['¥\x01\x96\x08\x01>8\x0c9\x0c8\x10T', ''], '_command': 'cellTemperatures', '_command_description': 'Cell Temperature Information', 'Cell 01 Temperature': [22, '°C'], 'Cell 02 Temperature': [16, '°C'], 'Cell 03 Temperature': [-28, '°C'], 'Cell 04 Temperature': [17, '°C'], 'Cell 05 Temperature': [-28, '°C'], 'Cell 06 Temperature': [16, '°C'], 'Cell 07 Temperature': [-24, '°C']}
INFO:root:attempting to create output processor: screen
INFO:screen:output: Using output processor: screen
Command: cellTemperatures - Cell Temperature Information
------------------------------------------------------------
Parameter                       Value           Unit
cell_01_temperature             22              °C
cell_02_temperature             16              °C
cell_03_temperature             -28             °C
cell_04_temperature             17              °C
cell_05_temperature             -28             °C
cell_06_temperature             16              °C
cell_07_temperature             -24             °C
```

The Status seems to be fine:
```
[14:27:37][pi@raspi:~]$ mpp-solar -p /dev/ttyUSB1 --porttype daly -b 9600 -P DALY40 -c status -I
INFO:root:Solar Device Command Utility, version: 0.7.56, recent changes: add daly40
INFO:root:Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB1 (porttype=daly)" using protocol "DALY40"
INFO:device:set_port: Port overide - using port 'daly'
INFO:device:set_port: Using dalyserialio for communications
INFO:root:Looping 1 commands
INFO:root:Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.dalyserialio.DalySerialIO object at 0xb5f78f10>, protocol: <mppsolar.protocols.daly40.daly40 object at 0xb5f78c90> for command: status, tag: status, outputs: screen
INFO:device:run_command: Running command status
INFO:daly40:get_full_command: Using protocol b'DALY40' with 7 commands
INFO:device:run_command: full command b'\xa5@\x94\x08\x00\x00\x00\x00\x00\x00\x00\x00\x81\n' for command status
[... 20 lines cut ...]
[14:28:26][pi@raspi:~]$ mpp-solar -p /dev/ttyUSB1 --porttype daly -b 9600 -P DALY40 -c SOC -I
INFO:root:Solar Device Command Utility, version: 0.7.56, recent changes: add daly40
INFO:root:Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB1 (porttype=daly)" using protocol "DALY40"
INFO:device:set_port: Port overide - using port 'daly'
INFO:device:set_port: Using dalyserialio for communications
INFO:root:Looping 1 commands
INFO:root:Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.dalyserialio.DalySerialIO object at 0xb5ec8f10>, protocol: <mppsolar.protocols.daly40.daly40 object at 0xb5ec8cb0> for command: SOC, tag: SOC, outputs: screen
INFO:device:run_command: Running command SOC
INFO:daly40:get_full_command: Using protocol b'DALY40' with 7 commands
[... cut ...] : [1.7, '%']}
INFO:root:attempting to create output processor: screen
INFO:screen:output: Using output processor: screen
Command: SOC - State of Charge
------------------------------------------------------------
Parameter                       Value           Unit
battery_bank_voltage            49.8            V
acquistion                      0.0             V
current                         1.7             A
soc                             1.7             %
```

Are there some commands else?

SirkoVZ, 2021-05-19:
I made several tests and it seems that
cell_17_voltage = cell_14_voltage
and
cell_18_voltage = cell_15_voltage

jblance, 2021-05-19:
There are 7 commands implemented so far. You can see a list by using `-c` with no command name

It looks like the BMS sends info for cells and temp sensors that are not present (probably to pad the result)
I think the gui is taking the earlier results that show the number of cells and Temps into account. This kinda breaks the logic of mppsolar so far (which is send a command and decode the result)

Options as I see it
1. Fix it by ignoring the unnecessary results
2. Use filters to exclude the unwanted results 
3. Add an option to tell how many cells etc are present
4. Change internal logic to make commands rely on other commands

Thoughts from your side?

jblance, 2021-05-19:
Note to self:
status command's `battery_string` should be `cell_count` or `number_of_cells`
status command's `temperature` should be `temperature_count` or `number_of_temperature_sensors`

SirkoVZ, 2021-05-20:
Other commands also work:
```
[07:40:10][pi@raspi:~]$ mpp-solar -p /dev/ttyUSB1 --porttype daly -b 9600 -P DALY40 -q screen -c
Command: command help - List available commands for protocol DALY40
------------------------------------------------------------
Parameter                       Value           Unit
SOC                             State of Charge
cellMinMaxVoltages              Cell Minimum and Maximum Voltages
cellMinMaxTemps                 Cell Minimum and Maximum Temperatures
mosStatus                       mosStatus
status                          Status Information
cellVoltages                    Cell Voltages Information
cellTemperatures                Cell Temperature Information
[07:40:33][pi@raspi:~]$ mpp-solar -p /dev/ttyUSB1 --porttype daly -b 9600 -P DALY40 -q screen -c cellMinMaxVoltages
Command: cellMinMaxVoltages - Cell Minimum and Maximum Voltages
------------------------------------------------------------
Parameter                       Value           Unit
maximum_cell_voltage            3.141           V
maximum_cell_number             14
minimum_cell_voltage            3.133           V
minimum_cell_number             5
[07:40:57][pi@raspi:~]$ mpp-solar -p /dev/ttyUSB1 --porttype daly -b 9600 -P DALY40 -q screen -c cellMinMaxTemps
Command: cellMinMaxTemps - Cell Minimum and Maximum Temperatures
------------------------------------------------------------
Parameter                       Value           Unit
maximum_cell_temperature        22              °C
maximum_cell_number             1
minimum_cell_temperature        22              °C
minimum_cell_number             1
[07:41:30][pi@raspi:~]$ mpp-solar -p /dev/ttyUSB1 --porttype daly -b 9600 -P DALY40 -q screen -c mosStatus
Command: mosStatus - mosStatus
------------------------------------------------------------
Parameter                       Value           Unit
charge_status                   charged
charging_mos_tube_status        01
discharging_mos_tube_status     01
bms_life                        212             cycles
residual_capacity_(todo)        00000d48        (HEX) mAH
[07:42:09][pi@raspi:~]$ mpp-solar -p /dev/ttyUSB1 --porttype daly -b 9600 -P DALY40 -q screen -c status
Command: status - Status Information
------------------------------------------------------------
Parameter                       Value           Unit
battery_string                  16
temperature                     01
charger_status                  disconnected
load_status                     disconnected
flags_(todo)                    02
charge/discharge_cycles         0               cycles
reserved                        48

```

I would ignore all cell_voltages that are more than battery_string (resp. cell_count).

Another question: As I call several commands the sum of time needed to respond is pretty high.
Is there a way to get the results faster?

jblance, 2021-05-21:
try testing all 3 commands at once with
`mpp-solar -p /dev/ttyUSB1 --porttype daly -b 9600 -P DALY40 -c cellTemperatures,SOC,cellVoltages`
and see what the time is like, otherwise i could add times to the log output to see which part is taking all the time

jblance, 2021-05-21:
Also, in one of your outputs you got 
`residual_capacity_(todo)        00000d48        (HEX) mAH` 
what is the correct capacity here? (is it 3400mAH?)

SirkoVZ, 2021-05-21:
the three commands are about 10% faster, but still more than 34 seconds.
As I use MQTT messages with own topics for cellTemperatures, Soc, ... I guess the combined 
-c cellTemperatures,SOC,cellVoltages
is not an option?
```
mpp-solar -p /dev/ttyUSB1 --porttype daly -b 9600 -P DALY40 -o json_mqtt -q localhost --mqtttopic /SmartHome/DalyBMS/Temperatur/tele -c cellTemperatures
mpp-solar -p /dev/ttyUSB1 --porttype daly -b 9600 -P DALY40 -o json_mqtt -q localhost --mqtttopic /SmartHome/DalyBMS/Status/tele -c SOC
mpp-solar -p /dev/ttyUSB1 --porttype daly -b 9600 -P DALY40 -o json_mqtt -q localhost --mqtttopic /SmartHome/DalyBMS/Voltage/tele -c cellVoltages
mpp-solar -p /dev/ttyUSB1 --porttype daly -b 9600 -P DALY40 -o json_mqtt -q localhost --mqtttopic /SmartHome/DalyBMS/MinMaxVoltages/tele -c cellMinMaxVoltages

```
Debugging output with times would be helpful I guess...

The residual_capacity_(todo) seems to me plausible.

Thanks again for your efforts.

jblance, 2021-05-21:
latest commits have changes to logging to add times
A `-I` run would be a good place to start for timings

SirkoVZ, 2021-05-21:
it´seems that the BMS wastes the time:

```
[13:56:44][pi@raspi:~]$ mpp-solar -p /dev/ttyUSB1 --porttype daly -b 9600 -P DALY40 -c SOC -I -D
2021-05-21 13:57:28,680:INFO:__init__:main@235: Solar Device Command Utility, version: 0.7.57, recent changes: change logging format
2021-05-21 13:57:28,680:INFO:__init__:main@331: Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB1 (porttype=daly)" using protocol "DALY40"
2021-05-21 13:57:28,700:DEBUG:__init__:main@334: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2021-05-21 13:57:28,700:DEBUG:device:__init__@31: __init__ args ()
2021-05-21 13:57:28,700:DEBUG:device:__init__@32: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB1', 'protocol': 'DALY40', 'baud': 9600, 'porttype': 'daly', 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
2021-05-21 13:57:28,701:INFO:device:set_port@130: Port overide - using port 'daly'
2021-05-21 13:57:28,701:DEBUG:device:get_port_type@77: port matches daly
2021-05-21 13:57:28,701:INFO:device:set_port@166: Using dalyserialio for communications
2021-05-21 13:57:28,719:DEBUG:device:set_protocol@96: Protocol DALY40
2021-05-21 13:57:28,768:DEBUG:device:__init__@36: __init__ name unnamed, port <mppsolar.io.dalyserialio.DalySerialIO object at 0xb5ef8fd0>, protocol <mppsolar.protocols.daly40.daly40 object at 0xb5ef8df0>
2021-05-21 13:57:28,768:DEBUG:__init__:main@381: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0xb5ef8dd0>, 'SOC', 'SOC', 'screen', None, None)]
2021-05-21 13:57:28,768:INFO:__init__:main@386: Looping 1 commands
2021-05-21 13:57:28,769:INFO:__init__:main@397: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.dalyserialio.DalySerialIO object at 0xb5ef8fd0>, protocol: <mppsolar.protocols.daly40.daly40 object at 0xb5ef8df0> for command: SOC, tag: SOC, outputs: screen
2021-05-21 13:57:28,769:INFO:device:run_command@263: Running command SOC
2021-05-21 13:57:28,769:INFO:daly40:get_full_command@207: Using protocol b'DALY40' with 7 commands
2021-05-21 13:57:28,770:DEBUG:abstractprotocol:get_command_defn@42: Processing command 'SOC'
2021-05-21 13:57:28,770:DEBUG:abstractprotocol:get_command_defn@44: Found command SOC in protocol b'DALY40'
2021-05-21 13:57:28,770:DEBUG:daly40:get_full_command@226: full command: b'\xa5@\x90\x08\x00\x00\x00\x00\x00\x00\x00\x00}\n'
2021-05-21 13:57:28,771:INFO:device:run_command@289: full command b'\xa5@\x90\x08\x00\x00\x00\x00\x00\x00\x00\x00}\n' for command SOC
2021-05-21 13:57:28,771:DEBUG:abstractprotocol:get_command_defn@42: Processing command 'SOC'
2021-05-21 13:57:28,771:DEBUG:abstractprotocol:get_command_defn@44: Found command SOC in protocol b'DALY40'
2021-05-21 13:57:28,772:DEBUG:dalyserialio:send_and_receive@19: port /dev/ttyUSB1, baudrate 9600
2021-05-21 13:57:28,783:DEBUG:dalyserialio:send_and_receive@22: Executing command via dalyserialio...
2021-05-21 13:57:39,803:DEBUG:dalyserialio:send_and_receive@39: serial response was: b'\xa5\x01\x90\x08\x02\x02\x00\x00u(\x00 \xff'
2021-05-21 13:57:39,806:DEBUG:device:run_command@308: Send and Receive Response b'\xa5\x01\x90\x08\x02\x02\x00\x00u(\x00 \xff'
2021-05-21 13:57:39,806:INFO:abstractprotocol:decode@137: response passed to decode: b'\xa5\x01\x90\x08\x02\x02\x00\x00u(\x00 \xff'
2021-05-21 13:57:39,806:DEBUG:daly40:check_response_valid@245: checking validity of b'\xa5\x01\x90\x08\x02\x02\x00\x00u(\x00 \xff'
2021-05-21 13:57:39,806:DEBUG:daly40:check_response_valid@258: DALY Checksum matches response 'b'\xa5\x01\x90\x08\x02\x02\x00\x00u(\x00 \xff'' checksum:255
2021-05-21 13:57:39,807:DEBUG:abstractprotocol:get_command_defn@42: Processing command 'SOC'
2021-05-21 13:57:39,807:DEBUG:abstractprotocol:get_command_defn@44: Found command SOC in protocol b'DALY40'
2021-05-21 13:57:39,807:DEBUG:abstractprotocol:decode@174: trimmed and split responses: [b'\xa5', b'\x01', b'\x90', b'\x08', b'\x02\x02', b'\x00\x00', b'u(', b'\x00 ', b'\xff']
2021-05-21 13:57:39,807:INFO:abstractprotocol:decode@181: Processing response of type POSITIONAL
2021-05-21 13:57:39,807:DEBUG:abstractprotocol:decode@325: Processing POSITIONAL type responses
2021-05-21 13:57:39,807:DEBUG:abstractprotocol:process_response@86: Processing data_type: discard for data_name: start flag, raw_value b'\xa5'
2021-05-21 13:57:39,808:DEBUG:abstractprotocol:process_response@93: Discarding start flag:b'\xa5'
2021-05-21 13:57:39,808:DEBUG:abstractprotocol:decode@325: Processing POSITIONAL type responses
2021-05-21 13:57:39,808:DEBUG:abstractprotocol:process_response@86: Processing data_type: discard for data_name: module address, raw_value b'\x01'
2021-05-21 13:57:39,808:DEBUG:abstractprotocol:process_response@93: Discarding module address:b'\x01'
2021-05-21 13:57:39,808:DEBUG:abstractprotocol:decode@325: Processing POSITIONAL type responses
2021-05-21 13:57:39,809:DEBUG:abstractprotocol:process_response@86: Processing data_type: discard for data_name: command id, raw_value b'\x90'
2021-05-21 13:57:39,809:DEBUG:abstractprotocol:process_response@93: Discarding command id:b'\x90'
2021-05-21 13:57:39,809:DEBUG:abstractprotocol:decode@325: Processing POSITIONAL type responses
2021-05-21 13:57:39,809:DEBUG:abstractprotocol:process_response@86: Processing data_type: discard for data_name: data length, raw_value b'\x08'
2021-05-21 13:57:39,809:DEBUG:abstractprotocol:process_response@93: Discarding data length:b'\x08'
[... cut ...]
current                         -0.8            A
2021-05-21 13:57:39,820:DEBUG:helpers:key_wanted@22: key_wanted: No filter and key soc not excluded by excl_filter None so key wanted
soc                             3.2             %
2021-05-21 13:57:39,820:DEBUG:__init__:main@426: Not daemon, so not looping

```

However, the Windows software shows the first values within a second and updates updates about once a second.
Probably the commands are different?
I can do some dump using the serial port sniffer, if helpful...

jblance, 2021-05-21:
That looks like a bug in the io handler. I'll try coding a fix and let you know when I'm done

SirkoVZ, 2021-05-27:
Hi,
as a workaround(?) I set 
s.timeout = 0.1
https://github.com/jblance/mpp-solar/blob/019b9b023dedd54de32074a8f7102e2b2f5a73f2/mppsolar/io/dalyserialio.py#L23
But I'm not sure whether this is a good idea ;-)

jblance, 2021-05-27:
Not ideal, it's the readline function need to change that and make sure that it exits correctly. But if it works ok.....

Sorry I haven't had a chance to make changes lately

jblance, 2021-06-02:
added a change that _should_ improve performance in 0.7.60
can you try when you get a chance and post a debug run, thanks

jblance, 2021-06-02:
> Edit: Is the sent command really
> \xa5@\x90\x08\x00\x00\x00\x00\x00\x00\x00\x00
> as shown in the log above?
> I can repeat this command in Windows without success.
> Also it seems that all requests I sniffed hava 40 as second byte.

0x40 is 64 decimal which is @ as an ascii char, ie
```python
>>> 0x40
64
>>> chr(64)
'@'
```

SirkoVZ, 2021-06-02:
oh, the cell voltages are not shown completely, sometimes 5 sometimes 9:
```
 mpp-solar -p /dev/ttyUSB1 --porttype daly -b 9600 -P DALY40 -q screen -c cellVoltages -I -D
2021-06-02 11:35:28,021:INFO:__init__:main@235: Solar Device Command Utility, version: 0.7.61, first try at pi17 setter
2021-06-02 11:35:28,022:INFO:__init__:main@339: Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB1 (porttype=daly)" using protocol "DALY40"
2021-06-02 11:35:28,041:DEBUG:__init__:main@342: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2021-06-02 11:35:28,042:DEBUG:device:__init__@31: __init__ args ()
2021-06-02 11:35:28,042:DEBUG:device:__init__@32: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB1', 'protocol': 'DALY40', 'baud': 9600, 'porttype': 'daly', 'mqtt_broker': 'screen', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
2021-06-02 11:35:28,042:INFO:device:set_port@130: Port overide - using port 'daly'
2021-06-02 11:35:28,043:DEBUG:device:get_port_type@77: port matches daly
2021-06-02 11:35:28,043:INFO:device:set_port@166: Using dalyserialio for communications
2021-06-02 11:35:28,065:DEBUG:device:set_protocol@96: Protocol DALY40
2021-06-02 11:35:28,123:DEBUG:device:__init__@36: __init__ name unnamed, port <mppsolar.io.dalyserialio.DalySerialIO object at 0xb5f60fd0>, protocol <mppsolar.protocols.daly40.daly40 object at 0xb5f60dd0>
2021-06-02 11:35:28,123:DEBUG:__init__:main@389: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0xb5f60db0>, 'cellVoltages', 'cellVoltages', 'screen', None, None)]
2021-06-02 11:35:28,124:INFO:__init__:main@394: Looping 1 commands
2021-06-02 11:35:28,124:INFO:__init__:main@405: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.dalyserialio.DalySerialIO object at 0xb5f60fd0>, protocol: <mppsolar.protocols.daly40.daly40 object at 0xb5f60dd0> for command: cellVoltages, tag: cellVoltages, outputs: screen
2021-06-02 11:35:28,124:INFO:device:run_command@263: Running command cellVoltages
2021-06-02 11:35:28,125:INFO:daly:get_full_command@208: Using protocol b'DALY40' with 7 commands
2021-06-02 11:35:28,125:DEBUG:abstractprotocol:get_command_defn@42: Processing command 'cellVoltages'
2021-06-02 11:35:28,125:DEBUG:abstractprotocol:get_command_defn@44: Found command cellVoltages in protocol b'DALY40'
2021-06-02 11:35:28,126:DEBUG:daly:get_full_command@225: full command: b'\xa5@\x95\x08\x00\x00\x00\x00\x00\x00\x00\x00\x82\n'
2021-06-02 11:35:28,126:INFO:device:run_command@289: full command b'\xa5@\x95\x08\x00\x00\x00\x00\x00\x00\x00\x00\x82\n' for command cellVoltages
2021-06-02 11:35:28,126:DEBUG:abstractprotocol:get_command_defn@42: Processing command 'cellVoltages'
2021-06-02 11:35:28,127:DEBUG:abstractprotocol:get_command_defn@44: Found command cellVoltages in protocol b'DALY40'
2021-06-02 11:35:28,127:DEBUG:dalyserialio:send_and_receive@19: port /dev/ttyUSB1, baudrate 9600
2021-06-02 11:35:28,135:DEBUG:dalyserialio:send_and_receive@22: Executing command via dalyserialio...
2021-06-02 11:35:28,237:DEBUG:dalyserialio:send_and_receive@32: bytes waiting 39
2021-06-02 11:35:28,237:DEBUG:dalyserialio:send_and_receive@32: bytes waiting 0
2021-06-02 11:35:28,237:DEBUG:dalyserialio:send_and_receive@38: serial response was: b'\xa5\x01\x95\x08\x01\r1\r3\r3\xa0\xa2\xa5\x01\x95\x08\x02\r3\r1\r3\xa0\xa3\xa5\x01\x95\x08\x03\r4\r2\r4\xa0\xa7'
2021-06-02 11:35:28,241:DEBUG:device:run_command@308: Send and Receive Response b'\xa5\x01\x95\x08\x01\r1\r3\r3\xa0\xa2\xa5\x01\x95\x08\x02\r3\r1\r3\xa0\xa3\xa5\x01\x95\x08\x03\r4\r2\r4\xa0\xa7'
2021-06-02 11:35:28,241:INFO:abstractprotocol:decode@137: response passed to decode: b'\xa5\x01\x95\x08\x01\r1\r3\r3\xa0\xa2\xa5\x01\x95\x08\x02\r3\r1\r3\xa0\xa3\xa5\x01\x95\x08\x03\r4\r2\r4\xa0\xa7'
2021-06-02 11:35:28,241:DEBUG:daly:check_response_valid@244: checking validity of b'\xa5\x01\x95\x08\x01\r1\r3\r3\xa0\xa2\xa5\x01\x95\x08\x02\r3\r1\r3\xa0\xa3\xa5\x01\x95\x08\x03\r4\r2\r4\xa0\xa7'
2021-06-02 11:35:28,242:INFO:daly:check_response_valid@248: is multiframe response - assuming ok for now
2021-06-02 11:35:28,242:DEBUG:abstractprotocol:get_command_defn@42: Processing command 'cellVoltages'
2021-06-02 11:35:28,243:DEBUG:abstractprotocol:get_command_defn@44: Found command cellVoltages in protocol b'DALY40'
2021-06-02 11:35:28,243:INFO:daly:get_responses@279: Multi frame response with 3 frames
2021-06-02 11:35:28,244:DEBUG:abstractprotocol:decode@174: trimmed and split responses: [[b'\xa5', b'\x01', b'\x95', b'\x08', b'\x01', b'\r1', b'\r3', b'\r3', b'\xa0', b'\xa2'], [b'\xa5', b'\x01', b'\x95', b'\x08', b'\x02', b'\r3', b'\r1', b'\r3', b'\xa0', b'\xa3'], [b'\xa5', b'\x01', b'\x95', b'\x08', b'\x03', b'\r4', b'\r2', b'\r4', b'\xa0', b'\xa7']]
2021-06-02 11:35:28,244:INFO:abstractprotocol:decode@181: Processing response of type MULTIFRAME-POSITIONAL
2021-06-02 11:35:28,245:DEBUG:abstractprotocol:decode@294: Processing MULTIFRAME-POSITIONAL type responses
2021-06-02 11:35:28,245:DEBUG:abstractprotocol:decode@298: got 3 frames
2021-06-02 11:35:28,245:DEBUG:abstractprotocol:decode@325: Processing POSITIONAL type responses
2021-06-02 11:35:28,245:DEBUG:abstractprotocol:process_response@86: Processing data_type: discard for data_name: start flag, raw_value b'\xa5'
2021-06-02 11:35:28,246:DEBUG:abstractprotocol:process_response@93: Discarding start flag:b'\xa5'
2021-06-02 11:35:28,246:DEBUG:abstractprotocol:decode@325: Processing POSITIONAL type responses
[... cut ...]
cell_07_voltage                 3.38            V
2021-06-02 11:35:28,293:DEBUG:helpers:key_wanted@22: key_wanted: No filter and key cell_08_voltage not excluded by excl_filter None so key wanted
cell_08_voltage                 3.378           V
2021-06-02 11:35:28,293:DEBUG:helpers:key_wanted@22: key_wanted: No filter and key cell_09_voltage not excluded by excl_filter None so key wanted
cell_09_voltage                 3.38            V
2021-06-02 11:35:28,294:DEBUG:__init__:main@434: Not daemon, so not looping

```
but should be 16 (or 18)

jblance, 2021-06-02:
ok, so outpacing the serial data i think
have a go with v0.7.62+

SirkoVZ, 2021-06-02:
much better now:
```
mpp-solar -p /dev/ttyUSB1 --porttype daly -b 9600 -P DALY40 -q screen -c cellVoltages -I -D
2021-06-02 14:40:20,212:INFO:__init__:main@235: Solar Device Command Utility, version: 0.7.62, dalyio add wiat to read loop
2021-06-02 14:40:20,213:INFO:__init__:main@339: Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB1 (porttype=daly)" using protocol "DALY40"
2021-06-02 14:40:20,219:DEBUG:__init__:main@342: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2021-06-02 14:40:20,220:DEBUG:device:__init__@31: __init__ args ()
2021-06-02 14:40:20,220:DEBUG:device:__init__@32: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB1', 'protocol': 'DALY40', 'baud': 9600, 'porttype': 'daly', 'mqtt_broker': 'screen', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
2021-06-02 14:40:20,221:INFO:device:set_port@130: Port overide - using port 'daly'
2021-06-02 14:40:20,221:DEBUG:device:get_port_type@77: port matches daly
2021-06-02 14:40:20,221:INFO:device:set_port@166: Using dalyserialio for communications
2021-06-02 14:40:20,238:DEBUG:device:set_protocol@96: Protocol DALY40
2021-06-02 14:40:20,259:DEBUG:device:__init__@36: __init__ name unnamed, port <mppsolar.io.dalyserialio.DalySerialIO object at 0xb5e79770>, protocol <mppsolar.protocols.daly40.daly40 object at 0xb5e653f0>
2021-06-02 14:40:20,260:DEBUG:__init__:main@389: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0xb5f3d790>, 'cellVoltages', 'cellVoltages', 'screen', None, None)]
2021-06-02 14:40:20,260:INFO:__init__:main@394: Looping 1 commands
2021-06-02 14:40:20,261:INFO:__init__:main@405: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.dalyserialio.DalySerialIO object at 0xb5e79770>, protocol: <mppsolar.protocols.daly40.daly40 object at 0xb5e653f0> for command: cellVoltages, tag: cellVoltages, outputs: screen
2021-06-02 14:40:20,261:INFO:device:run_command@270: Running command cellVoltages
2021-06-02 14:40:20,261:INFO:daly:get_full_command@208: Using protocol b'DALY40' with 7 commands
2021-06-02 14:40:20,262:DEBUG:abstractprotocol:get_command_defn@42: Processing command 'cellVoltages'
2021-06-02 14:40:20,262:DEBUG:abstractprotocol:get_command_defn@44: Found command cellVoltages in protocol b'DALY40'
2021-06-02 14:40:20,263:DEBUG:daly:get_full_command@225: full command: b'\xa5@\x95\x08\x00\x00\x00\x00\x00\x00\x00\x00\x82\n'
2021-06-02 14:40:20,263:INFO:device:run_command@296: full command b'\xa5@\x95\x08\x00\x00\x00\x00\x00\x00\x00\x00\x82\n' for command cellVoltages
2021-06-02 14:40:20,263:DEBUG:abstractprotocol:get_command_defn@42: Processing command 'cellVoltages'
2021-06-02 14:40:20,264:DEBUG:abstractprotocol:get_command_defn@44: Found command cellVoltages in protocol b'DALY40'
2021-06-02 14:40:20,264:DEBUG:dalyserialio:send_and_receive@19: port /dev/ttyUSB1, baudrate 9600
2021-06-02 14:40:20,271:DEBUG:dalyserialio:send_and_receive@22: Executing command via dalyserialio...
2021-06-02 14:40:20,372:DEBUG:dalyserialio:send_and_receive@32: bytes waiting 39
2021-06-02 14:40:20,473:DEBUG:dalyserialio:send_and_receive@32: bytes waiting 39
2021-06-02 14:40:20,574:DEBUG:dalyserialio:send_and_receive@32: bytes waiting 0
2021-06-02 14:40:20,574:DEBUG:dalyserialio:send_and_receive@38: serial response was: b'\xa5\x01\x95\x08\x01\r.\r6\r8\xab\xb2\xa5\x01\x95\x08\x02\r2\r.\r,\xab\xa3\xa5\x01\x95\x08\x03\r3\r6\r6\xab\xb7\xa5\x01\x95\x08\x04\r2\r/\r2\xab\xac\xa5\x01\x95\x08\x05\r2\r2\r0\xab\xae\xa5\x01\x95\x08\x06\r1\r2\r0\xab\xae'
2021-06-02 14:40:20,578:DEBUG:device:run_command@315: Send and Receive Response b'\xa5\x01\x95\x08\x01\r.\r6\r8\xab\xb2\xa5\x01\x95\x08\x02\r2\r.\r,\xab\xa3\xa5\x01\x95\x08\x03\r3\r6\r6\xab\xb7\xa5\x01\x95\x08\x04\r2\r/\r2\xab\xac\xa5\x01\x95\x08\x05\r2\r2\r0\xab\xae\xa5\x01\x95\x08\x06\r1\r2\r0\xab\xae'
2021-06-02 14:40:20,579:INFO:abstractprotocol:decode@137: response passed to decode: b'\xa5\x01\x95\x08\x01\r.\r6\r8\xab\xb2\xa5\x01\x95\x08\x02\r2\r.\r,\xab\xa3\xa5\x01\x95\x08\x03\r3\r6\r6\xab\xb7\xa5\x01\x95\x08\x04\r2\r/\r2\xab\xac\xa5\x01\x95\x08\x05\r2\r2\r0\xab\xae\xa5\x01\x95\x08\x06\r1\r2\r0\xab\xae'
2021-06-02 14:40:20,579:DEBUG:daly:check_response_valid@244: checking validity of b'\xa5\x01\x95\x08\x01\r.\r6\r8\xab\xb2\xa5\x01\x95\x08\x02\r2\r.\r,\xab\xa3\xa5\x01\x95\x08\x03\r3\r6\r6\xab\xb7\xa5\x01\x95\x08\x04\r2\r/\r2\xab\xac\xa5\x01\x95\x08\x05\r2\r2\r0\xab\xae\xa5\x01\x95\x08\x06\r1\r2\r0\xab\xae'
2021-06-02 14:40:20,580:INFO:daly:check_response_valid@248: is multiframe response - assuming ok for now
2021-06-02 14:40:20,580:DEBUG:abstractprotocol:get_command_defn@42: Processing command 'cellVoltages'
2021-06-02 14:40:20,581:DEBUG:abstractprotocol:get_command_defn@44: Found command cellVoltages in protocol b'DALY40'
2021-06-02 14:40:20,581:INFO:daly:get_responses@279: Multi frame response with 6 frames
2021-06-02 14:40:20,581:DEBUG:abstractprotocol:decode@174: trimmed and split responses: [[b'\xa5', b'\x01', b'\x95', b'\x08', b'\x01', b'\r.', b'\r6', b'\r8', b'\xab', b'\xb2'], [b'\xa5', b'\x01', b'\x95', b'\x08', b'\x02', b'\r2', b'\r.', b'\r,', b'\xab', b'\xa3'], [b'\xa5', b'\x01', b'\x95', b'\x08', b'\x03', b'\r3', b'\r6', b'\r6', b'\xab', b'\xb7'], [b'\xa5', b'\x01', b'\x95', b'\x08', b'\x04', b'\r2', b'\r/', b'\r2', b'\xab', b'\xac'], [b'\xa5', b'\x01', b'\x95', b'\x08', b'\x05', b'\r2', b'\r2', b'\r0', b'\xab', b'\xae'], [b'\xa5', b'\x01', b'\x95', b'\x08', b'\x06', b'\r1', b'\r2', b'\r0', b'\xab', b'\xae']]
[... cut ...]
cell_16_voltage                 3.377           V
2021-06-02 14:40:20,691:DEBUG:helpers:key_wanted@22: key_wanted: No filter and key cell_17_voltage not excluded by excl_filter None so key wanted
cell_17_voltage                 3.378           V
2021-06-02 14:40:20,691:DEBUG:helpers:key_wanted@22: key_wanted: No filter and key cell_18_voltage not excluded by excl_filter None so key wanted
cell_18_voltage                 3.376           V
2021-06-02 14:40:20,692:DEBUG:__init__:main@434: Not daemon, so not looping

```

SirkoVZ, 2021-06-03:
My tests show no issues any more, thank you very much :-)
A hopefully last question:
Currently I use a shell script to run several commands and send the result using mqtt:
```
mpp-solar -p /dev/ttyUSB1 --porttype daly -b 9600 -P DALY40 -o json_mqtt -q localhost --mqtttopic /SmartHome/DalyBMS/mosStatus/tele -c mosStatus
mpp-solar -p /dev/ttyUSB1 --porttype daly -b 9600 -P DALY40 -o json_mqtt -q localhost --mqtttopic /SmartHome/DalyBMS/Status/tele -c SOC

```
Is it possible to use only on command? Something like
 -c mosStatus,SOC
It would be ok, when the results of both commands are send to one and the same topic, e.g. /SmartHome/DalyBMS/UnifiedStatus/tele

jblance, 2021-06-03:
That should work

closed 2021-06-04 by jblance