jblance/mpp-solar #115: IndexError: list index out of range - 5048GK

duncanngenda, 2021-05-19:
Hello,

I have a 5048GK inverter and I am getting the following error when I run QPIGS. 
I also have an 8048 PIP Max and this does not have the same issue. 

Any one come across this before? Happy to provide more information.

root@pi002:/opt/juametrics# /usr/local/bin/mpp-solar -p /dev/hidraw0 -c QPIRI -I
INFO:root:Solar Device Command Utility, version: 0.7.49, recent changes: add DALY cellVoltages
INFO:root:Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI30"
INFO:device:set_port: Using hidrawio for communications
INFO:root:Looping 1 commands
INFO:root:Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.hidrawio.HIDRawIO object at 0xffff9d938a90>, protocol: <mppsolar.protocols.pi30.pi30 object at 0xffff9d938af0> for command: QPIRI, tag: QPIRI, outputs: screen
INFO:device:run_command: Running command QPIRI
INFO:AbstractProtocol:get_full_command: Using protocol b'PI30' with 36 commands
INFO:device:run_command: full command b'QPIRI\xf8T\r' for command QPIRI
INFO:AbstractProtocol:decode: response passed to decode: b'(230.0 21.7 230.0 50.0 21.7 5000 5000 48.0 47.0 46.5 57.6 57.6 5 30 080 0 1 2 1 01 0 0 52.0 0 1\x03$\r'
INFO:AbstractProtocol:decode: Processing response of type DEFAULT
INFO:AbstractProtocol:decode: Processing DEFAULT type responses
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.8/dist-packages/mppsolar/__init__.py", line 385, in main
    results = _device.run_command(command=_command)
  File "/usr/local/lib/python3.8/dist-packages/mppsolar/devices/device.py", line 320, in run_command
    decoded_response = self._protocol.decode(raw_response, command)
  File "/usr/local/lib/python3.8/dist-packages/mppsolar/protocols/abstractprotocol.py", line 232, in decode
    msgs[key] = [resp_format[2][int(result)], ""]
IndexError: list index out of range

jblance, 2021-05-19:
It looks to me to be a battery type issue - your response has a `5` for battery type (which implies the 6th item) there are only 3 defined in protocol PI30
Can you run the `QPI` command and post the result (check that it is still protocol PI30)
What are the battery types in your documentation? What is your battery type?

Regards

duncanngenda, 2021-05-21:
Hi @jblance 

Am running Weco batteries and there is an option to select Weco.  On the new GKs you have several Lithium batteries brands to select from - Weco is one of them along side Pylon, Solatora etc. See option 5 on p17 for full list of battery types I can see: http://www.mppsolar.com/manual/PIP-MAX/PIP-MAX%208KW-manual-20200923.pdf)

QPI returns P130

INFO:root:Solar Device Command Utility, version: 0.7.49, recent changes: add DALY cellVoltages
INFO:root:Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI30"
INFO:device:set_port: Using hidrawio for communications
INFO:root:Looping 1 commands
INFO:root:Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.hidrawio.HIDRawIO object at 0xffffb5643b50>, protocol: <mppsolar.protocols.pi30.pi30 object at 0xffffb5643a60> for command: QPI, tag: QPI, outputs: screen
INFO:device:run_command: Running command QPI
INFO:AbstractProtocol:get_full_command: Using protocol b'PI30' with 36 commands
INFO:device:run_command: full command b'QPI\xbe\xac\r' for command QPI
INFO:AbstractProtocol:decode: response passed to decode: b'(PI30\x9a\x0b\r'
INFO:AbstractProtocol:decode: Processing response of type DEFAULT
INFO:AbstractProtocol:decode: Processing DEFAULT type responses
INFO:device:run_command: Decoded response {'raw_response': ['(PI30\x9a\x0b\r', ''], '_command': 'QPI', '_command_description': 'Protocol ID inquiry', 'Protocol ID': ['PI30', '']}
INFO:root:attempting to create output processor: screen
INFO:screen:output: Using output processor: screen
Command: QPI - Protocol ID inquiry
------------------------------------------------------------
Parameter                     	Value           Unit
protocol_id                   	PI30

duncanngenda, 2021-05-21:
Thank you for your time @jblance  Appreciate

jblance, 2021-05-27:
List from Option 5 P17
* AGM (default)
* User-Defined 
* Pylontech
* WECO 
* Soltaro 
* LIb-protocol compatible
* 3rd party Lithium

jblance, 2021-05-27:
have added a change for this, be interested in a full -D run of QPIRI with the latest code to ensure the result is as expected

duncanngenda, 2021-05-28:
@jblance thanks for this. Have you published the code to the main branch?

It’s set to Weco

Happy to upgrade my current version and test. 

Thanks.

jblance, 2021-05-30:
yeah, it has been added to master

duncanngenda, 2021-05-30:
Thanks @jblance 

Given that a run. Index error sorted. See outputs below. 

You will notice, there are some undefined values. I need to investigate what these relate to.

[QPIRI 115 .txt](https://github.com/jblance/mpp-solar/files/6566077/QPIRI.115.txt)

closed 2021-06-02 by jblance