jblance/mpp-solar #117: Implement a SETTER type for PI17 protocol

VictorBucha, 2021-06-01:
I have tried various approaches to implement setter type for my 5kW hybrid, which uses PI17 protocol but failed

jblance, 2021-06-02:
0.7.61 has a first attempt, try with eg
`mpp-solar -P PI17 -c PEA `
let me know how it goes and post the debug output if it fails

VictorBucha, 2021-06-02:
Output is like this:
root@raspberrypi:/usr/local/lib/python3.7/dist-packages/mppsolar/protocols# mpp-solar -P PI17 -D -c PEA
INFO:root:Solar Device Command Utility, version: 0.7.49, recent changes: add DALY cellVoltages
INFO:root:Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=None)" using protocol "PI17"
DEBUG:root:device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
DEBUG:device:mppsolar __init__ args ()
DEBUG:device:mppsolar __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB0', 'protocol': 'PI17', 'baud': 2400, 'porttype': None, 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
DEBUG:device:get_port_type: port matches ttyusb
INFO:device:set_port: Using serialio for communications
DEBUG:device:set_protocol: Protocol PI17
DEBUG:device:mppsolar __init__ name unnamed, port <mppsolar.io.serialio.SerialIO object at 0x76628170>, protocol <mppsolar.protocols.pi17.pi17 object at 0x76628310>
DEBUG:root:Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x76628150>, 'PEA', 'PEA', 'screen', None, None)]
INFO:root:Looping 1 commands
INFO:root:Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.serialio.SerialIO object at 0x76628170>, protocol: <mppsolar.protocols.pi17.pi17 object at 0x76628310> for command: PEA, tag: PEA, outputs: screen
INFO:device:run_command: Running command PEA
INFO:pi17:Using protocol b'PI17' with 12 commands
DEBUG:AbstractProtocol:Processing command 'PEA'
DEBUG:AbstractProtocol:Regex commands _command: MCHGC
INFO:AbstractProtocol:No command_defn found for PEA
INFO:device:run_command: full command None for command PEA
ERROR:device:full_command not found for PEA in protocol b'PI17'
DEBUG:root:results: {'ERROR': ["full_command not found for PEA in protocol b'PI17'", '']}
INFO:root:attempting to create output processor: screen
DEBUG:screen:processor.screen __init__ kwargs {}
DEBUG:root:Using output filter: None
INFO:screen:output: Using output processor: screen
DEBUG:screen:output: kwargs {'data': {'ERROR': ["full_command not found for PEA in protocol b'PI17'", '']}, 'tag': 'PEA', 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None, 'mqtt_topic': 'mpp-solar', 'filter': None, 'excl_filter': None, 'keep_case': False}
Parameter                       Value           Unit
DEBUG:helpers:key_wanted: No filter and key error not excluded by excl_filter None so key wanted
error                           full_command not found for PEA in protocol b'PI17'
DEBUG:root:Not daemon, so not looping
root@raspberrypi:/usr/local/lib/python3.7/dist-packages/mppsolar/protocols#

VictorBucha, 2021-06-02:
Here it goes 

root@raspberrypi:/usr/local/lib/python3.7/dist-packages/mppsolar/protocols# mpp-solar -p /dev/hidraw0 -P PI17 -c PEA
ERROR:device:full_command not found for PEA in protocol b'PI17'
Parameter                       Value           Unit
error                           full_command not found for PEA in protocol b'PI17'
root@raspberrypi:/usr/local/lib/python3.7/dist-packages/mppsolar/protocols#

jblance, 2021-06-02:
Can you add `-I` to the command line and post all the output

VictorBucha, 2021-06-03:
root@raspberrypi:/usr/local/lib/python3.7/dist-packages/mppsolar/protocols# pip3 install mpp-solar
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: mpp-solar in /src/mpp-solar (0.7.62)
Requirement already satisfied: pyserial in /usr/local/lib/python3.7/dist-packages (from mpp-solar) (3.5)
Requirement already satisfied: paho-mqtt in /usr/local/lib/python3.7/dist-packages (from mpp-solar) (1.5.1)
root@raspberrypi:/usr/local/lib/python3.7/dist-packages/mppsolar/protocols# mpp-solar -p /dev/hidraw0 -P PI17 -I -c PEA
2021-06-03 11:01:45,292:INFO:__init__:main@235: Solar Device Command Utility, version: 0.7.62, dalyio add wiat to read loop
2021-06-03 11:01:45,292:INFO:__init__:main@339: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI17"
2021-06-03 11:01:45,308:INFO:device:set_port@142: Using hidrawio for communications
2021-06-03 11:01:45,351:INFO:__init__:main@394: Looping 1 commands
2021-06-03 11:01:45,351:INFO:__init__:main@405: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.hidrawio.HIDRawIO object at 0x75f61ab0>, protocol: <mppsolar.protocols.pi17.pi17 object at 0x75f61b10> for command: PEA, tag: PEA, outputs: screen
2021-06-03 11:01:45,352:INFO:device:run_command@270: Running command PEA
2021-06-03 11:01:45,352:INFO:pi17:get_full_command@281: Using protocol b'PI17' with 11 commands
2021-06-03 11:01:45,353:INFO:device:run_command@296: full command b'^S004PEA\r' for command PEA
2021-06-03 11:01:46,530:INFO:abstractprotocol:decode@137: response passed to decode: b'^0\x1b\xe3\r'
2021-06-03 11:01:46,531:INFO:abstractprotocol:decode@181: Processing response of type DEFAULT
2021-06-03 11:01:46,531:INFO:abstractprotocol:decode@188: Processing DEFAULT type responses
2021-06-03 11:01:46,532:INFO:device:run_command@334: Decoded response {'raw_response': ['^0\x1bã\r', ''], '_command': 'PEA', '_command_description': 'Set the enabled state of an Inverter setting', 'WARNING0': ['Command PEA was rejected', '']}
2021-06-03 11:01:46,533:INFO:__init__:get_outputs@26: attempting to create output processor: screen
2021-06-03 11:01:46,546:INFO:screen:output@19: Using output processor: screen
Command: PEA - Set the enabled state of an Inverter setting
------------------------------------------------------------
Parameter                       Value           Unit
warning0                        Command PEA was rejected

VictorBucha, 2021-06-03:
With -D:
root@raspberrypi:/usr/local/lib/python3.7/dist-packages/mppsolar/protocols# mpp-solar -p /dev/hidraw0 -P PI17 -D -c PEA
2021-06-03 11:04:22,068:INFO:__init__:main@235: Solar Device Command Utility, version: 0.7.62, dalyio add wiat to read loop
2021-06-03 11:04:22,068:INFO:__init__:main@339: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI17"
2021-06-03 11:04:22,072:DEBUG:__init__:main@342: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2021-06-03 11:04:22,073:DEBUG:device:__init__@31: __init__ args ()
2021-06-03 11:04:22,073:DEBUG:device:__init__@32: __init__ kwargs {'name': 'unnamed', 'port': '/dev/hidraw0', 'protocol': 'PI17', 'baud': 2400, 'porttype': None, 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
2021-06-03 11:04:22,073:DEBUG:device:get_port_type@59: port matches hidraw
2021-06-03 11:04:22,074:INFO:device:set_port@142: Using hidrawio for communications
2021-06-03 11:04:22,076:DEBUG:device:set_protocol@96: Protocol PI17
2021-06-03 11:04:22,090:DEBUG:device:__init__@36: __init__ name unnamed, port <mppsolar.io.hidrawio.HIDRawIO object at 0x75f15930>, protocol <mppsolar.protocols.pi17.pi17 object at 0x75f159b0>
2021-06-03 11:04:22,090:DEBUG:__init__:main@389: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x75f47c10>, 'PEA', 'PEA', 'screen', None, None)]
2021-06-03 11:04:22,091:INFO:__init__:main@394: Looping 1 commands
2021-06-03 11:04:22,091:INFO:__init__:main@405: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.hidrawio.HIDRawIO object at 0x75f15930>, protocol: <mppsolar.protocols.pi17.pi17 object at 0x75f159b0> for command: PEA, tag: PEA, outputs: screen
2021-06-03 11:04:22,091:INFO:device:run_command@270: Running command PEA
2021-06-03 11:04:22,091:INFO:pi17:get_full_command@281: Using protocol b'PI17' with 11 commands
2021-06-03 11:04:22,092:DEBUG:abstractprotocol:get_command_defn@42: Processing command 'PEA'
2021-06-03 11:04:22,092:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: PE
2021-06-03 11:04:22,093:DEBUG:abstractprotocol:get_command_defn@53: Matched: PEA to: PE value: A
2021-06-03 11:04:22,093:DEBUG:pi17:get_full_command@299: _pre_cmd: b'^S004PEA'
2021-06-03 11:04:22,093:DEBUG:protocol_helpers:crcPI@233: Calculating CRC for b'^S004PEA'
2021-06-03 11:04:22,093:DEBUG:protocol_helpers:crcPI@283: Generated CRC 0x56 0x08 0x5608
2021-06-03 11:04:22,094:DEBUG:pi17:get_full_command@306: full command: b'^S004PEA\r'
2021-06-03 11:04:22,094:INFO:device:run_command@296: full command b'^S004PEA\r' for command PEA
2021-06-03 11:04:22,094:DEBUG:abstractprotocol:get_command_defn@42: Processing command 'PEA'
2021-06-03 11:04:22,094:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: PE
2021-06-03 11:04:22,094:DEBUG:abstractprotocol:get_command_defn@53: Matched: PEA to: PE value: A
2021-06-03 11:04:22,163:DEBUG:hidrawio:send_and_receive@29: length of to_send: 9
2021-06-03 11:04:22,164:DEBUG:hidrawio:send_and_receive@40: 2 chunk send
2021-06-03 11:04:23,270:DEBUG:hidrawio:send_and_receive@69: usb response was: b'^1\x0b\xc2\r'
2021-06-03 11:04:23,272:DEBUG:device:run_command@315: Send and Receive Response b'^1\x0b\xc2\r'
2021-06-03 11:04:23,272:INFO:abstractprotocol:decode@137: response passed to decode: b'^1\x0b\xc2\r'
2021-06-03 11:04:23,273:DEBUG:abstractprotocol:get_command_defn@42: Processing command 'PEA'
2021-06-03 11:04:23,273:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: PE
2021-06-03 11:04:23,274:DEBUG:abstractprotocol:get_command_defn@53: Matched: PEA to: PE value: A
2021-06-03 11:04:23,274:DEBUG:abstractprotocol:decode@174: trimmed and split responses: [b'']
2021-06-03 11:04:23,275:INFO:abstractprotocol:decode@181: Processing response of type DEFAULT
2021-06-03 11:04:23,275:INFO:abstractprotocol:decode@188: Processing DEFAULT type responses
2021-06-03 11:04:23,276:INFO:device:run_command@334: Decoded response {'raw_response': ['^1\x0bÂ\r', ''], '_command': 'PEA', '_command_description': 'Set the enabled state of an Inverter setting', 'PE': ['', '']}
2021-06-03 11:04:23,277:DEBUG:__init__:main@408: results: {'raw_response': ['^1\x0bÂ\r', ''], '_command': 'PEA', '_command_description': 'Set the enabled state of an Inverter setting', 'PE': ['', '']}
2021-06-03 11:04:23,277:INFO:__init__:get_outputs@26: attempting to create output processor: screen
2021-06-03 11:04:23,283:DEBUG:screen:__init__@16: processor.screen __init__ kwargs {}
2021-06-03 11:04:23,283:DEBUG:__init__:main@414: Using output filter: None
2021-06-03 11:04:23,284:INFO:screen:output@19: Using output processor: screen
2021-06-03 11:04:23,284:DEBUG:screen:output@20: kwargs {'data': {'raw_response': ['^1\x0bÂ\r', ''], '_command': 'PEA', '_command_description': 'Set the enabled state of an Inverter setting', 'PE': ['', '']}, 'tag': 'PEA', 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None, 'mqtt_topic': 'mpp-solar', 'filter': None, 'excl_filter': None, 'keep_case': False}
Command: PEA - Set the enabled state of an Inverter setting
------------------------------------------------------------
Parameter                       Value           Unit
2021-06-03 11:04:23,285:DEBUG:helpers:key_wanted@22: key_wanted: No filter and key pe not excluded by excl_filter None so key wanted
pe
2021-06-03 11:04:23,286:DEBUG:__init__:main@434: Not daemon, so not looping

jblance, 2021-06-03:
Ok, so it worked but the response processing isn't working correctly. I'll have a look and fix it as soon as I get a chance

VictorBucha, 2021-06-14:
Is there any uprate with response processing?

jblance, 2021-06-14:
Sorry - yes I commited a fix previously
I've bumped the version number, but now the latest code should work for PEA success responses

jblance, 2021-06-15:
[pi17.py.txt](https://github.com/jblance/mpp-solar/files/6659253/pi17.py.txt)
from lonertic on diysolarforum

jblance, 2021-06-15:
added a bunch more commands (from lonertic's update) v0.7.64 or later
Now have the below:
```
$ mpp-solar -P PI17 -c
Command: command help - List available commands for protocol PI17
------------------------------------------------------------
Parameter                       Value           Unit
PI                              Device Protocol Version inquiry -- queries the device protocol version      
ID                              Device Serial Number inquiry -- queries the device serial number            
VFW                             Device CPU version inquiry -- queries the CPU version       
VFW2                            Device CPU 2 version inquiry -- queries the CPU 2 version           
MD                              Device Model inquiry -- queries the device model            
PIRI                            Device rated information -- queries rated information       
GS                              Device rated information -- queries rated information       
PS                              Device Power Status -- queries power status         
MOD                             Device working mode inquiry -- queries the device working mode      
WS                              Warning status inquiry -- queries any active warnings flags from the Inverter       
FLAG                            Query enable/disable flag status -- queries enable/disable flag status from the Inverter            
T                               Query current time -- queries current time from the Inverter        
ET                              Query total generated energy -- queries total generated energy from the Inverter            
BATS                            Query battery setting -- queries battery setting            
LON                             Set enable/disable machine supply power to the loads -- examples: LON1 (0: disable, 1: enable)      
PA                              Mute buzzer beep -- examples: PA1 (1: enable, 0: disable)           
PB                              Mute buzzer beep in standby mode -- examples: PB1 (1: enable, 0: disable)           
PC                              Mute buzzer beep only on battery discharged status -- examples: PC1 (1: enable, 0: disable)         
PD                              Generator as AC input -- examples: PD1 (1: enable, 0: disable)      
PE                              Wide AC input range -- examples: PE1 (1: enable, 0: disable)        
PF                              N/G relay function -- examples: PF1 (1: enable, 0: disable)         
DAT                             Set date time -- examples: DAT190518224530 (1: enable, 0: disable)          
LST                             Set LCD sleep wait time -- examples: LSTnn (nn: 00, 01, 02, 10, 20 for selection, unit : 30second.00 means LCD always light)        
EDA                             Enable/disable solar charge battery -- examples: EDA1 (1: enable, 0: disable)       
EDB                             Enable/disable AC charge battery -- examples: EDB1 (1: enable, 0: disable)          
EDC                             Enable/disable feed power to utility -- examples: EDC1 (1: enable, 0: disable)      
EDD                             Enable/disable battery discharge to loads when solar input normal -- examples: EDD1 (1: enable, 0: disable)         
EDE                             Enable/disable battery discharge to loads when solar input loss -- examples: EDE1 (1: enable, 0: disable)           
EDF                             Enable/disable battery discharge to feed power to utility when solar input normal -- examples: EDF1 (1: enable, 0: disable)         
EDG                             Enable/disable battery discharge to feed power to utility whensolar input loss -- examples: EDG1 (1: enable, 0: disable)            
BT                              Set battery type -- examples: EDG1 (1: Li-Fe, 0: Ordinary)          
ACCT                            Set AC charge time range -- examples: ACCT2200-0259 (Sets time range from 22:00 to 02:59. End minute is inclusive)
```

closed 2021-06-15 by jblance

VictorBucha, 2021-06-16:
Thanks a lot!!!

eMindFfM, 2024-01-03:
Hi, do you have a running example of the ACCT setter? I'm getting always a rejected messages

`(mppsolar) pi@Pi5Test:~ $ mpp-solar -p /dev/hidraw0 -P PI17 -c ACCT2200-0259
Command: ACCT2200-0259 - Set AC charge time range
--------------------------------------------------------------------------------
Parameter Value                 Unit
warning0  Command ACCT2200-0259 was rejected
--------------------------------------------------------------------------------
`