jblance/mpp-solar #124: sudo pip3 install bluepy

AnKloss, 2021-06-19:
Hello
for JKBMS i need the bluepy, right ? if i do it i get error
>>>>>
ubuntu@ubuntu:~$ sudo pip3 install bluepy
Collecting bluepy
  Using cached bluepy-1.3.0.tar.gz (217 kB)
Building wheels for collected packages: bluepy
  Building wheel for bluepy (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-13yu9fwp/bluepy/setup.py'"'"'; __file__='"'"'/tmp/pip-install-13yu9fwp/bluepy/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-qceed1pk
       cwd: /tmp/pip-install-13yu9fwp/bluepy/
  Complete output (11 lines):
  running bdist_wheel
  running build
  running build_py
  Working dir is /tmp/pip-install-13yu9fwp/bluepy
  execute make -C ./bluepy clean
  execute make -C bluepy -j1
  Failed to compile bluepy-helper. Exiting install.
  Command was 'make -C bluepy -j1' in /tmp/pip-install-13yu9fwp/bluepy
  Return code was 2

how to do it ?

mfg
Andreas

jblance, 2021-06-19:
Have you installed the dev libs?

sudo apt-get install libglib2.0

AnKloss, 2021-06-19:
no,  okay i do it now, thanks

AnKloss, 2021-06-19:
Successfully built bluepy
Installing collected packages: bluepy
Successfully installed bluepy-1.3.0

okay many thanks

closed 2021-06-19 by AnKloss