jblance/mpp-solar #128: PI30 QPIRI throws an error in 0.7.63

VictorBucha, 2021-07-03:
`mpp-solar -p /dev/hidraw0 -P PI30 -D -c QPIRI
2021-07-03 16:56:37,841:INFO:__init__:main@235: Solar Device Command Utility, version: 0.7.67, move jk04 to new processing, add lookup functionality
2021-07-03 16:56:37,841:INFO:__init__:main@339: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI30"
2021-07-03 16:56:37,844:DEBUG:__init__:main@342: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2021-07-03 16:56:37,844:DEBUG:device:__init__@31: __init__ args ()
2021-07-03 16:56:37,844:DEBUG:device:__init__@32: __init__ kwargs {'name': 'unnamed', 'port': '/dev/hidraw0', 'protocol': 'PI30', 'baud': 2400, 'porttype': None, 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
2021-07-03 16:56:37,844:DEBUG:device:get_port_type@59: port matches hidraw
2021-07-03 16:56:37,845:INFO:device:set_port@142: Using hidrawio for communications
2021-07-03 16:56:37,846:DEBUG:device:set_protocol@96: Protocol PI30
2021-07-03 16:56:37,857:DEBUG:device:__init__@36: __init__ name unnamed, port <mppsolar.io.hidrawio.HIDRawIO object at 0xb659c330>, protocol <mppsolar.protocols.pi30.pi30 object at 0xb659c390>
2021-07-03 16:56:37,857:DEBUG:__init__:main@389: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0xb659c190>, 'QPIRI', 'QPIRI', 'screen', None, None)]
2021-07-03 16:56:37,857:INFO:__init__:main@394: Looping 1 commands
2021-07-03 16:56:37,858:INFO:__init__:main@405: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.hidrawio.HIDRawIO object at 0xb659c330>, protocol: <mppsolar.protocols.pi30.pi30 object at 0xb659c390> for command: QPIRI, tag: QPIRI, outputs: screen
2021-07-03 16:56:37,858:INFO:device:run_command@270: Running command QPIRI
2021-07-03 16:56:37,858:INFO:abstractprotocol:get_full_command@30: Using protocol b'PI30' with 36 commands
2021-07-03 16:56:37,858:DEBUG:protocol_helpers:crcPI@299: Calculating CRC for b'QPIRI'
2021-07-03 16:56:37,858:DEBUG:protocol_helpers:crcPI@349: Generated CRC 0xf8 0x54 0xf854
2021-07-03 16:56:37,858:DEBUG:abstractprotocol:get_full_command@37: full command: b'QPIRI\xf8T\r'
2021-07-03 16:56:37,858:INFO:device:run_command@296: full command b'QPIRI\xf8T\r' for command QPIRI
2021-07-03 16:56:37,859:DEBUG:abstractprotocol:get_command_defn@41: Processing command 'QPIRI'
2021-07-03 16:56:37,859:DEBUG:abstractprotocol:get_command_defn@43: Found command QPIRI in protocol b'PI30'
2021-07-03 16:56:37,922:DEBUG:hidrawio:send_and_receive@29: length of to_send: 8
2021-07-03 16:56:37,922:DEBUG:hidrawio:send_and_receive@36: 1 chunk send
2021-07-03 16:56:40,627:DEBUG:hidrawio:send_and_receive@69: usb response was: b'(230.0 34.7 230.0 50.0 34.7 8000 8000 48.0 48.0 42.0 54.0 52.5 2 010 030 1 2 2 9 01 0 0 50.0 0 1 480 0 070\xd9`\r'
2021-07-03 16:56:40,628:DEBUG:device:run_command@315: Send and Receive Response b'(230.0 34.7 230.0 50.0 34.7 8000 8000 48.0 48.0 42.0 54.0 52.5 2 010 030 1 2 2 9 01 0 0 50.0 0 1 480 0 070\xd9`\r'
2021-07-03 16:56:40,628:INFO:abstractprotocol:decode@174: response passed to decode: b'(230.0 34.7 230.0 50.0 34.7 8000 8000 48.0 48.0 42.0 54.0 52.5 2 010 030 1 2 2 9 01 0 0 50.0 0 1 480 0 070\xd9`\r'
2021-07-03 16:56:40,628:DEBUG:protocol_helpers:crcPI@299: Calculating CRC for b'(230.0 34.7 230.0 50.0 34.7 8000 8000 48.0 48.0 42.0 54.0 52.5 2 010 030 1 2 2 9 01 0 0 50.0 0 1 480 0 070'
2021-07-03 16:56:40,629:DEBUG:protocol_helpers:crcPI@349: Generated CRC 0xd9 0x60 0xd960
2021-07-03 16:56:40,629:DEBUG:pi30:check_response_valid@860: CRCs match
2021-07-03 16:56:40,629:DEBUG:abstractprotocol:get_command_defn@41: Processing command 'QPIRI'
2021-07-03 16:56:40,629:DEBUG:abstractprotocol:get_command_defn@43: Found command QPIRI in protocol b'PI30'
2021-07-03 16:56:40,629:INFO:abstractprotocol:decode@214: Processing response of type SEQUENTIAL
2021-07-03 16:56:40,629:DEBUG:abstractprotocol:decode@218: trimmed and split responses: [b'230.0', b'34.7', b'230.0', b'50.0', b'34.7', b'8000', b'8000', b'48.0', b'48.0', b'42.0', b'54.0', b'52.5', b'2', b'010', b'030', b'1', b'2', b'2', b'9', b'01', b'0', b'0', b'50.0', b'0', b'1', b'480', b'0', b'070']
2021-07-03 16:56:40,630:DEBUG:abstractprotocol:decode@362: Processing SEQUENTIAL type responses
2021-07-03 16:56:40,630:DEBUG:abstractprotocol:process_response@85: Processing data_type: float for data_name: AC Input Voltage, raw_value b'230.0'
2021-07-03 16:56:40,630:DEBUG:abstractprotocol:process_response@153: Processing format string float(raw_value)
2021-07-03 16:56:40,630:DEBUG:abstractprotocol:decode@362: Processing SEQUENTIAL type responses
2021-07-03 16:56:40,630:DEBUG:abstractprotocol:process_response@85: Processing data_type: float for data_name: AC Input Current, raw_value b'34.7'
2021-07-03 16:56:40,630:DEBUG:abstractprotocol:process_response@153: Processing format string float(raw_value)
2021-07-03 16:56:40,630:DEBUG:abstractprotocol:decode@362: Processing SEQUENTIAL type responses
2021-07-03 16:56:40,631:DEBUG:abstractprotocol:process_response@85: Processing data_type: float for data_name: AC Output Voltage, raw_value b'230.0'
2021-07-03 16:56:40,631:DEBUG:abstractprotocol:process_response@153: Processing format string float(raw_value)
2021-07-03 16:56:40,631:DEBUG:abstractprotocol:decode@362: Processing SEQUENTIAL type responses
2021-07-03 16:56:40,631:DEBUG:abstractprotocol:process_response@85: Processing data_type: float for data_name: AC Output Frequency, raw_value b'50.0'
2021-07-03 16:56:40,631:DEBUG:abstractprotocol:process_response@153: Processing format string float(raw_value)
2021-07-03 16:56:40,631:DEBUG:abstractprotocol:decode@362: Processing SEQUENTIAL type responses
2021-07-03 16:56:40,631:DEBUG:abstractprotocol:process_response@85: Processing data_type: float for data_name: AC Output Current, raw_value b'34.7'
2021-07-03 16:56:40,632:DEBUG:abstractprotocol:process_response@153: Processing format string float(raw_value)
2021-07-03 16:56:40,632:DEBUG:abstractprotocol:decode@362: Processing SEQUENTIAL type responses
2021-07-03 16:56:40,632:DEBUG:abstractprotocol:process_response@85: Processing data_type: int for data_name: AC Output Apparent Power, raw_value b'8000'
2021-07-03 16:56:40,632:DEBUG:abstractprotocol:process_response@153: Processing format string int(raw_value)
2021-07-03 16:56:40,632:DEBUG:abstractprotocol:decode@362: Processing SEQUENTIAL type responses
2021-07-03 16:56:40,632:DEBUG:abstractprotocol:process_response@85: Processing data_type: int for data_name: AC Output Active Power, raw_value b'8000'
2021-07-03 16:56:40,633:DEBUG:abstractprotocol:process_response@153: Processing format string int(raw_value)
2021-07-03 16:56:40,633:DEBUG:abstractprotocol:decode@362: Processing SEQUENTIAL type responses
2021-07-03 16:56:40,633:DEBUG:abstractprotocol:process_response@85: Processing data_type: float for data_name: Battery Voltage, raw_value b'48.0'
2021-07-03 16:56:40,633:DEBUG:abstractprotocol:process_response@153: Processing format string float(raw_value)
2021-07-03 16:56:40,633:DEBUG:abstractprotocol:decode@362: Processing SEQUENTIAL type responses
2021-07-03 16:56:40,633:DEBUG:abstractprotocol:process_response@85: Processing data_type: float for data_name: Battery Recharge Voltage, raw_value b'48.0'
2021-07-03 16:56:40,633:DEBUG:abstractprotocol:process_response@153: Processing format string float(raw_value)
2021-07-03 16:56:40,634:DEBUG:abstractprotocol:decode@362: Processing SEQUENTIAL type responses
2021-07-03 16:56:40,634:DEBUG:abstractprotocol:process_response@85: Processing data_type: float for data_name: Battery Under Voltage, raw_value b'42.0'
2021-07-03 16:56:40,634:DEBUG:abstractprotocol:process_response@153: Processing format string float(raw_value)
2021-07-03 16:56:40,634:DEBUG:abstractprotocol:decode@362: Processing SEQUENTIAL type responses
2021-07-03 16:56:40,634:DEBUG:abstractprotocol:process_response@85: Processing data_type: float for data_name: Battery Bulk Charge Voltage, raw_value b'54.0'
2021-07-03 16:56:40,634:DEBUG:abstractprotocol:process_response@153: Processing format string float(raw_value)
2021-07-03 16:56:40,634:DEBUG:abstractprotocol:decode@362: Processing SEQUENTIAL type responses
2021-07-03 16:56:40,635:DEBUG:abstractprotocol:process_response@85: Processing data_type: float for data_name: Battery Float Charge Voltage, raw_value b'52.5'
2021-07-03 16:56:40,635:DEBUG:abstractprotocol:process_response@153: Processing format string float(raw_value)
[... cut ...]
  File "/usr/local/lib/python3.7/dist-packages/mppsolar/devices/device.py", line 333, in run_command
    decoded_response = self._protocol.decode(raw_response, command)
  File "/usr/local/lib/python3.7/dist-packages/mppsolar/protocols/abstractprotocol.py", line 417, in decode
    frame_number=frame_number,
  File "/usr/local/lib/python3.7/dist-packages/mppsolar/protocols/abstractprotocol.py", line 164, in process_response
    if "{" in data_name:
TypeError: argument of type 'int' is not iterable
`

VictorBucha, 2021-07-03:
works like a charm on 0.7.49 version.

jblance, 2021-07-04:
thanks, fixed the error in 0.7.71

there are extra results
```
unknown_value_in_response_25    b'480'              
unknown_value_in_response_26    b'0'                
unknown_value_in_response_27    b'070'  
```
What inverter do you have?
Is it using PI30?
Do you know what the values respond to?

VictorBucha, 2021-07-05:
I have access to 5 kw hybrid, but recently bought pip max MKSII - 8000w off grid one from mpp solar. 
Actually,  my guess is that unknown_value_in_response_25    b'480'      is the 
Maximum charging time at CV stage, min;   
unknown_value_in_response_26    b'0'       is the operation logic. 
But what is    the last one, I have no idea     
unknown_value_in_response_27    b'070'  

By the way, on some strange reason the fields 
Unknown_float and Unknown_flags? In QPGS return PV voltage and amps, respectively. Both are integers. Unfortunately I have only one string connected, so I cannot check whether it relates to a certain input or not.
I realize that this is not documented, but

jblance, 2021-07-06:
Do you get `PI30` from a `QPI` command?
I think some of the other protocol variants have the PV voltage and amps already in the definition

VictorBucha, 2021-07-06:
of course :-)
root@raspberrypi:/home/pi# mpp-solar -p /dev/hidraw0 -P PI30 -c QPI
Command: QPI - Protocol ID inquiry
------------------------------------------------------------
Parameter                       Value           Unit
protocol_id                     PI30
root@raspberrypi:/home/pi#

As I said, I have one with PI17 in my parent's house and the other with PI30 in mine. 
As I have only one string to test PV inputs - I have played today with both of them: unknown_float and unknown_flags? is the PV2 voltage and amps respectively. And those that were already present - were for PV1 input. so you can change naming accordingly.

VictorBucha, 2021-07-07:
Hi, 
PYPI's latest version is 0.7.67, not 71

jblance, 2021-07-08:
Yes, I dont update the pypi version for every development change
I post an update to pypi soon

jblance, 2021-07-22:
pypi has release 0.7.71 now

closed 2021-08-05 by jblance