jblance/mpp-solar #129: telegraf[1690]: 2021-07-12T06:41:42Z E! [inputs.mqtt_consumer] Error in plugin: metric parse error: expected tag at 1:4: "120"

AnKloss, 2021-07-12:
Hello 
i get a inputs.mqtt_consumer] Error in plugin: metric parse error: in Telegraf

[[inputs.mqtt_consumer]]
  servers = ["tcp://127.0.0.1:1883"]
  topics = [
    "#",
  ]
  data_format = "influx"

Any info how to fix it ?

mfg
Andreas Kloss

AnKloss, 2021-07-12:
QPGS1/status/pv_input_current/value 2
QPGS1/status/pv_input_current/unit A
QPGS1/status/battery_discharge_current/value 0
QPGS1/status/battery_discharge_current/unit A
QPGS1/status/unknown_float/value 188.8
QPGS1/status/unknown_flags?/value 02
^Cubuntu@ubuntu:~sudo systemctl status telegraf
● telegraf.service - The plugin-driven server agent for reporting metrics into InfluxDB
     Loaded: loaded (/lib/systemd/system/telegraf.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2021-07-12 06:40:34 UTC; 1min 16s ago
       Docs: https://github.com/influxdata/telegraf
   Main PID: 1690 (telegraf)
      Tasks: 10 (limit: 1825)
     CGroup: /system.slice/telegraf.service
             └─1690 /usr/bin/telegraf -config /etc/telegraf/telegraf.conf -config-directory /etc/telegraf/telegraf.d

Jul 12 06:41:42 ubuntu telegraf[1690]: 2021-07-12T06:41:42Z E! [inputs.mqtt_consumer] Error in plugin: metric parse error: expected tag at 1:4: "120"
Jul 12 06:41:42 ubuntu telegraf[1690]: 2021-07-12T06:41:42Z E! [inputs.mqtt_consumer] Error in plugin: metric parse error: expected tag at 1:2: "A"
Jul 12 06:41:42 ubuntu telegraf[1690]: 2021-07-12T06:41:42Z E! [inputs.mqtt_consumer] Error in plugin: metric parse error: expected tag at 1:3: "10"
Jul 12 06:41:42 ubuntu telegraf[1690]: 2021-07-12T06:41:42Z E! [inputs.mqtt_consumer] Error in plugin: metric parse error: expected tag at 1:2: "A"
Jul 12 06:41:42 ubuntu telegraf[1690]: 2021-07-12T06:41:42Z E! [inputs.mqtt_consumer] Error in plugin: metric parse error: expected tag at 1:2: "2"
Jul 12 06:41:42 ubuntu telegraf[1690]: 2021-07-12T06:41:42Z E! [inputs.mqtt_consumer] Error in plugin: metric parse error: expected tag at 1:2: "A"
Jul 12 06:41:42 ubuntu telegraf[1690]: 2021-07-12T06:41:42Z E! [inputs.mqtt_consumer] Error in plugin: metric parse error: expected tag at 1:2: "0"
Jul 12 06:41:42 ubuntu telegraf[1690]: 2021-07-12T06:41:42Z E! [inputs.mqtt_consumer] Error in plugin: metric parse error: expected tag at 1:2: "A"
Jul 12 06:41:42 ubuntu telegraf[1690]: 2021-07-12T06:41:42Z E! [inputs.mqtt_consumer] Error in plugin: metric parse error: expected tag at 1:6: "188.8"
Jul 12 06:41:42 ubuntu telegraf[1690]: 2021-07-12T06:41:42Z E! [inputs.mqtt_consumer] Error in plugin: metric parse error: expected tag at 1:3: "02"
ubuntu@ubuntu:~$ systemctl --user status mpp-solar
● mpp-solar.service - MPP Solar Service
     Loaded: loaded (/etc/xdg/systemd/user/mpp-solar.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2021-07-12 06:41:11 UTC; 49s ago
   Main PID: 1768 (python3)
     CGroup: /user.slice/user-1000.slice/user@1000.service/mpp-solar.service
             └─1768 /usr/bin/python3 /usr/local/bin/mpp-solar -C /etc/mpp-solar/mpp-solar.conf --daemon

Jul 12 06:41:56 ubuntu python3[1768]: output_mode                           single machine
Jul 12 06:41:56 ubuntu python3[1768]: charger_source_priority               Solar + Utility
Jul 12 06:41:56 ubuntu python3[1768]: max_charger_current                   80                     A
Jul 12 06:41:56 ubuntu python3[1768]: max_charger_range                     120                    A
Jul 12 06:41:56 ubuntu python3[1768]: max_ac_charger_current                10                     A
Jul 12 06:41:56 ubuntu python3[1768]: pv_input_current                      2                      A
Jul 12 06:41:56 ubuntu python3[1768]: battery_discharge_current             0                      A
Jul 12 06:41:56 ubuntu python3[1768]: unknown_float                         181.1
Jul 12 06:41:56 ubuntu python3[1768]: unknown_flags?                        02
Jul 12 06:41:56 ubuntu python3[1768]: Sleeping for 10 sec

jblance, 2021-07-12:
Can you post the mppsolar conf file
Also can you post the output of  `mosquitto_sub -t # -v`

AnKloss, 2021-07-12:
mppsolar conf

[SETUP]
# Number of seconds to pause between command execution loop
# 0 is no pause, greater than 60 will cause service restarts
pause=10
mqtt_broker=localhost

# This example would work on 2x PIP4048 installed in parallel
[Inverter_1]
protocol=PI30
port=/dev/ttyUSB0
porttype=serial
baud=2400
command=QPGS0
tag=QPGS0
outputs=mqtt,screen

[Inverter_2]
protocol=PI30
port=/dev/ttyUSB0
baud=2400
command=QPGS1
tag=QPGS1
outputs=mqtt,screen

AnKloss, 2021-07-12:
mosquitto_sub -t # -v

QPGS0/status/grid_voltage/value 236.3
QPGS0/status/grid_voltage/unit V
QPGS0/status/grid_frequency/value 49.97
QPGS0/status/grid_frequency/unit Hz
QPGS0/status/ac_output_voltage/value 229.9
QPGS0/status/ac_output_voltage/unit V
QPGS0/status/ac_output_frequency/value 49.92
QPGS0/status/ac_output_frequency/unit Hz
QPGS0/status/ac_output_apparent_power/value 850
QPGS0/status/ac_output_apparent_power/unit VA
QPGS0/status/ac_output_active_power/value 772
QPGS0/status/ac_output_active_power/unit W
QPGS0/status/load_percentage/value 10
QPGS0/status/load_percentage/unit %
QPGS0/status/battery_voltage/value 53.8
QPGS0/status/battery_voltage/unit V
QPGS0/status/battery_charging_current/value 0
QPGS0/status/battery_charging_current/unit A
QPGS0/status/battery_capacity/value 76
QPGS0/status/battery_capacity/unit %
QPGS0/status/pv_input_voltage/value 162.7
QPGS0/status/pv_input_voltage/unit V
QPGS0/status/total_charging_current/value 0
QPGS0/status/total_charging_current/unit A
QPGS0/status/total_ac_output_apparent_power/value 850
QPGS0/status/total_ac_output_apparent_power/unit VA
QPGS0/status/total_output_active_power/value 772
QPGS0/status/total_output_active_power/unit W
QPGS0/status/total_ac_output_percentage/value 9
QPGS0/status/total_ac_output_percentage/unit %
QPGS0/status/is_scc_ok/value 1
QPGS0/status/is_scc_ok/unit bool
QPGS0/status/is_ac_charging/value 0
QPGS0/status/is_ac_charging/unit bool
QPGS0/status/is_scc_charging/value 1
QPGS0/status/is_scc_charging/unit bool
QPGS0/status/is_battery_over_voltage/value 0
QPGS0/status/is_battery_over_voltage/unit bool
QPGS0/status/is_battery_under_voltage/value 0
QPGS0/status/is_battery_under_voltage/unit bool
QPGS0/status/is_line_lost/value 0
QPGS0/status/is_line_lost/unit bool
QPGS0/status/is_load_on/value 1
QPGS0/status/is_load_on/unit bool
QPGS0/status/is_configuration_changed/value 0
QPGS0/status/is_configuration_changed/unit bool
QPGS0/status/output_mode/value single machine
QPGS0/status/charger_source_priority/value Solar + Utility
QPGS0/status/max_charger_current/value 80
QPGS0/status/max_charger_current/unit A
QPGS0/status/max_charger_range/value 120
QPGS0/status/max_charger_range/unit A
QPGS0/status/max_ac_charger_current/value 10
QPGS0/status/max_ac_charger_current/unit A
QPGS0/status/pv_input_current/value 2
QPGS0/status/pv_input_current/unit A
QPGS0/status/battery_discharge_current/value 0
QPGS0/status/battery_discharge_current/unit A
QPGS0/status/unknown_float/value 178.8
QPGS0/status/unknown_flags?/value 02

jblance, 2021-07-12:
I don't know if the `mqtt` output works with telegraf. 

I used the `influx2_mqtt` output, can you try that (change in mpp-solar.conf)

AnKloss, 2021-07-12:
outputs=mqtt,screen
change to 
outputs=influx2_mqtt
right ?

jblance, 2021-07-12:
Yes.

jblance, 2021-07-12:
You can run mppsolar on the command line while passing the config file in with `mpp-solar -C /etc/mpp-solar/mpp-solar.conf` (make sure to use the path to your config file and stop the service first)
This will make testing easier

AnKloss, 2021-07-12:
looks like is working
mfg
Andreas Kloss

AnKloss, 2021-07-12:
in Telegraf i get this error
ul 12 07:21:08 ubuntu telegraf[1692]: time="2021-07-12T07:21:08Z" level=error msg="failed to open. Ignored. open /etc/telegraf/.cache/snowflake/ocsp_response_cache.json: permission denied\n" func="gosnowflak.....and so on

Folder i have make, as is not exsiting before

mfg
Andreas Kloss

jblance, 2021-07-12:
https://github.com/influxdata/telegraf/issues/9407 ? 

I don't use telegraf myself anymore but it looks like a telegraf specific error

closed 2021-07-12 by AnKloss