jblance/mpp-solar #130: Can not get all Info's out from mpp 8048max

AnKloss, 2021-07-14:
Hello,
After change output to mpp-solar, i get 36 command out from mpp controller, but missing all info's about  
- PV1 and 2 charging Power (watt),
- PV1 and 2 input current, 
- Total generation Wh, 
- Today generation Wh,

mpp-solsr is  version 0.7.67 , as can see in cron.out file

How can i get this info out ?

cron.out file >>>>
2021-07-14 02:29:01,239:INFO:__init__:main@235: Solar Device Command Utility, version: 0.7.67, move jk04 to new processing, ad>2021-07-14 02:29:01,240:INFO:__init__:main@338: Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=N>2021-07-14 02:29:01,244:INFO:device:set_port@160: Using serialio for communications
2021-07-14 02:29:01,274:INFO:__init__:main@394: Looping 1 commands
2021-07-14 02:29:01,275:INFO:__init__:main@404: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.>2021-07-14 02:29:01,275:INFO:device:run_command@270: Running command QPIGS
2021-07-14 02:29:01,275:INFO:abstractprotocol:get_full_command@30: Using protocol b'PI30' with 36 commands
2021-07-14 02:29:01,275:INFO:device:run_command@296: full command b'QPIGS\xb7\xa9\r' for command QPIGS
2021-07-14 02:29:02,841:INFO:abstractprotocol:decode@174: response passed to decode: b'(242.7 50.0 230.1 50.0 0828 0753 010 41>2021-07-14 02:29:02,844:INFO:abstractprotocol:decode@214: Processing response of type DEFAULT
2021-07-14 02:29:02,845:INFO:abstractprotocol:decode@225: Processing DEFAULT type responses
2021-07-14 02:29:02,846:INFO:device:run_command@334: Decoded response {'raw_response': ['(242.7 50.0 230.1 50.0 0828 0753 010 >2021-07-14 02:29:02,847:INFO:__init__:get_outputs@26: attempting to create output processor: screen
2021-07-14 02:29:02,850:INFO:screen:output@19: Using output processor: screen
Command: QPIGS - General Status Parameters inquiry

outout from mosquitto >>>
mpp-solar mpp-solar,command=QPGS1 parallel_instance_number="Not valid"
mpp-solar mpp-solar,command=QPGS1 serial_number="xxxxxxxxxxxxxx" (sorry i not show this)
mpp-solar mpp-solar,command=QPGS1 work_mode="Battery Mode"
mpp-solar mpp-solar,command=QPGS1 fault_code="No fault"
mpp-solar mpp-solar,command=QPGS1 grid_voltage=242.4
mpp-solar mpp-solar,command=QPGS1 grid_frequency=50.05
mpp-solar mpp-solar,command=QPGS1 ac_output_voltage=230.2
mpp-solar mpp-solar,command=QPGS1 ac_output_frequency=50.0
mpp-solar mpp-solar,command=QPGS1 ac_output_apparent_power=414
mpp-solar mpp-solar,command=QPGS1 ac_output_active_power=387
mpp-solar mpp-solar,command=QPGS1 load_percentage=5
mpp-solar mpp-solar,command=QPGS1 battery_voltage=53.0
mpp-solar mpp-solar,command=QPGS1 battery_charging_current=18
mpp-solar mpp-solar,command=QPGS1 battery_capacity=68
mpp-solar mpp-solar,command=QPGS1 pv_input_voltage=171.3  (<< is PV1 volt)
mpp-solar mpp-solar,command=QPGS1 total_charging_current=18
mpp-solar mpp-solar,command=QPGS1 total_ac_output_apparent_power=414
mpp-solar mpp-solar,command=QPGS1 total_output_active_power=387
mpp-solar mpp-solar,command=QPGS1 total_ac_output_percentage=4
mpp-solar mpp-solar,command=QPGS1 is_scc_ok=1
mpp-solar mpp-solar,command=QPGS1 is_ac_charging=0
mpp-solar mpp-solar,command=QPGS1 is_scc_charging=1
mpp-solar mpp-solar,command=QPGS1 is_battery_over_voltage=0
mpp-solar mpp-solar,command=QPGS1 is_battery_under_voltage=0
mpp-solar mpp-solar,command=QPGS1 is_line_lost=0
mpp-solar mpp-solar,command=QPGS1 is_load_on=1
mpp-solar mpp-solar,command=QPGS1 is_configuration_changed=0
mpp-solar mpp-solar,command=QPGS1 output_mode="single machine"
mpp-solar mpp-solar,command=QPGS1 charger_source_priority="Solar + Utility"
mpp-solar mpp-solar,command=QPGS1 max_charger_current=80
mpp-solar mpp-solar,command=QPGS1 max_charger_range=120
mpp-solar mpp-solar,command=QPGS1 max_ac_charger_current=10
mpp-solar mpp-solar,command=QPGS1 pv_input_current=4
mpp-solar mpp-solar,command=QPGS1 battery_discharge_current=0
mpp-solar mpp-solar,command=QPGS1 unknown_float=179.8  (<<- is PV2 volt)
mpp-solar mpp-solar,command=QPGS1 unknown_flags?="04"

AnKloss, 2021-07-14:
[image]
 screen looking now, i get happy
mfg
Andreas

jblance, 2021-07-14:
Sorry I dont understand - is there something still not working?

AnKloss, 2021-07-14:
Hello
i still missing this values  >>
PV1 and 2 charging Power (watt),
PV1 and 2 input current,
Total generation Wh,
Today generation Wh,
<<<

not show up at all.
How i can get this ?

mfg
Andreas

AnKloss, 2021-07-14:
[image]

AnKloss, 2021-07-14:
[image]

AnKloss, 2021-07-14:
This info show up on the wifi app

jblance, 2021-07-14:
Can you run `QPIGS` and `QPGS0` commands with `-D` flag and post all the
output.
(The 8048max uses pi30 but is a different configuration so the current
protocol definition is not correct)

AnKloss, 2021-07-14:
okay will do,
can you send my a sample of this commands please , as  i am not a programmer.

mfg
Andreas

AnKloss, 2021-07-14:
i use this command
sudo mpp-solar -q localhost -c QPIGS -D > /home/ubuntu/cron1.out 2>&1

output >>>
2021-07-14 05:49:50,311:INFO:__init__:main@235: Solar Device Command Utility, version: 0.7.67, move jk04 to new processing, add lookup functionality
2021-07-14 05:49:50,312:INFO:__init__:main@338: Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=None)" using protocol "PI30"
2021-07-14 05:49:50,319:DEBUG:__init__:main@342: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2021-07-14 05:49:50,319:DEBUG:device:__init__@31: __init__ args ()
2021-07-14 05:49:50,320:DEBUG:device:__init__@32: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB0', 'protocol': 'PI30', 'baud': 2400, 'porttype': None, 'mqtt_broker': 'localhost', '>2021-07-14 05:49:50,320:DEBUG:device:get_port_type@86: port matches ttyusb
2021-07-14 05:49:50,321:INFO:device:set_port@160: Using serialio for communications
2021-07-14 05:49:50,342:DEBUG:device:set_protocol@96: Protocol PI30
2021-07-14 05:49:50,367:DEBUG:device:__init__@36: __init__ name unnamed, port <mppsolar.io.serialio.SerialIO object at 0xb68ced30>, protocol <mppsolar.protocols.pi30.pi30 object at 0xb68cecb8>2021-07-14 05:49:50,368:DEBUG:__init__:main@389: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0xb68cece8>, 'QPIGS', 'QPIGS', 'screen', None, None)]
2021-07-14 05:49:50,368:INFO:__init__:main@394: Looping 1 commands
2021-07-14 05:49:50,369:INFO:__init__:main@404: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.serialio.SerialIO object at 0xb68ced30>, protocol: <mppsolar.p>2021-07-14 05:49:50,369:INFO:device:run_command@270: Running command QPIGS
2021-07-14 05:49:50,369:INFO:abstractprotocol:get_full_command@30: Using protocol b'PI30' with 36 commands
2021-07-14 05:49:50,370:DEBUG:protocol_helpers:crcPI@299: Calculating CRC for b'QPIGS'
2021-07-14 05:49:50,370:DEBUG:protocol_helpers:crcPI@349: Generated CRC 0xb7 0xa9 0xb7a9
2021-07-14 05:49:50,371:DEBUG:abstractprotocol:get_full_command@37: full command: b'QPIGS\xb7\xa9\r'
2021-07-14 05:49:50,371:INFO:device:run_command@296: full command b'QPIGS\xb7\xa9\r' for command QPIGS
2021-07-14 05:49:50,371:DEBUG:abstractprotocol:get_command_defn@41: Processing command 'QPIGS'
2021-07-14 05:49:50,371:DEBUG:abstractprotocol:get_command_defn@43: Found command QPIGS in protocol b'PI30'
2021-07-14 05:49:50,372:DEBUG:serialio:send_and_receive@19: port /dev/ttyUSB0, baudrate 2400
2021-07-14 05:49:50,377:DEBUG:serialio:send_and_receive@22: Executing command via serialio...
2021-07-14 05:49:50,870:DEBUG:serialio:send_and_receive@30: serial response was: b'(238.1 50.0 230.1 50.0 0874 0787 010 416 53.00 002 068 0052 02.7 174.6 00.00 00000 00010110 00 00 00482 010s>2021-07-14 05:49:51,961:DEBUG:device:run_command@315: Send and Receive Response b'(238.1 50.0 230.1 50.0 0874 0787 010 416 53.00 002 068 0052 02.7 174.6 00.00 00000 00010110 00 00 00482 010s\>2021-07-14 05:49:51,962:INFO:abstractprotocol:decode@174: response passed to decode: b'(238.1 50.0 230.1 50.0 0874 0787 010 416 53.00 002 068 0052 02.7 174.6 00.00 00000 00010110 00 00 00482 >2021-07-14 05:49:51,962:DEBUG:protocol_helpers:crcPI@299: Calculating CRC for b'(238.1 50.0 230.1 50.0 0874 0787 010 416 53.00 002 068 0052 02.7 174.6 00.00 00000 00010110 00 00 00482 010'
2021-07-14 05:49:51,964:DEBUG:protocol_helpers:crcPI@349: Generated CRC 0x73 0xfa 0x73fa
2021-07-14 05:49:51,964:DEBUG:pi30:check_response_valid@860: CRCs match
2021-07-14 05:49:51,965:DEBUG:abstractprotocol:get_command_defn@41: Processing command 'QPIGS'
2021-07-14 05:49:51,965:DEBUG:abstractprotocol:get_command_defn@43: Found command QPIGS in protocol b'PI30'
2021-07-14 05:49:51,965:INFO:abstractprotocol:decode@214: Processing response of type DEFAULT
2021-07-14 05:49:51,966:DEBUG:abstractprotocol:decode@218: trimmed and split responses: [b'238.1', b'50.0', b'230.1', b'50.0', b'0874', b'0787', b'010', b'416', b'53.00', b'002', b'068', b'00>2021-07-14 05:49:51,966:INFO:abstractprotocol:decode@225: Processing DEFAULT type responses
2021-07-14 05:49:51,967:INFO:device:run_command@334: Decoded response {'raw_response': ['(238.1 50.0 230.1 50.0 0874 0787 010 416 53.00 002 068 0052 02.7 174.6 00.00 00000 00010110 00 00 0048>2021-07-14 05:49:51,967:DEBUG:__init__:main@408: results: {'raw_response': ['(238.1 50.0 230.1 50.0 0874 0787 010 416 53.00 002 068 0052 02.7 174.6 00.00 00000 00010110 00 00 00482 010sú\r', >2021-07-14 05:49:51,967:INFO:__init__:get_outputs@26: attempting to create output processor: screen
2021-07-14 05:49:51,971:DEBUG:screen:__init__@16: processor.screen __init__ kwargs {}
2021-07-14 05:49:51,972:DEBUG:__init__:main@414: Using output filter: None
2021-07-14 05:49:51,972:INFO:screen:output@19: Using output processor: screen
2021-07-14 05:49:51,973:DEBUG:screen:output@20: kwargs {'data': {'raw_response': ['(238.1 50.0 230.1 50.0 0874 0787 010 416 53.00 002 068 0052 02.7 174.6 00.00 00000 00010110 00 00 00482 010s>2021-07-14 05:49:51,973:DEBUG:helpers:key_wanted@21: key_wanted: No filter and key ac_input_voltage not excluded by excl_filter None so key wanted
2021-07-14 05:49:51,974:DEBUG:helpers:key_wanted@21: key_wanted: No filter and key ac_input_frequency not excluded by excl_filter None so key wanted
2021-07-14 05:49:51,974:DEBUG:helpers:key_wanted@21: key_wanted: No filter and key ac_output_voltage not excluded by excl_filter None so key wanted
[... cut ...]
inverter_heat_sink_temperature  52              °C
pv_input_current_for_battery    2.7             A
pv_input_voltage                174.6           V
battery_voltage_from_scc        0.0             V
battery_discharge_current       0               A
is_sbu_priority_version_added   0               bool
is_configuration_changed        0               bool
is_scc_firmware_updated         0               bool
is_load_on                      1               bool
is_battery_voltage_to_steady_while_charging     0               bool

AnKloss, 2021-07-14:
is_charging_on                  1               bool
is_scc_charging_on              1               bool
is_ac_charging_on               0               bool
rsv1                            0               A
rsv2                            0               A
pv_input_power                  482             W
is_charging_to_float            0               bool
is_switched_on                  1               bool
is_reserved                     0               bool

AnKloss, 2021-07-14:
command 
mpp-solar -q localhost -c QPGS0 -D > /home/ubuntu/cron2.out 2>&1

output >>>
  GNU nano 4.8                                                                                           cron2.out                                                                                                      2021-07-14 05:57:59,654:INFO:__init__:main@235: Solar Device Command Utility, version: 0.7.67, move jk04 to new processing, add lookup functionality
2021-07-14 05:57:59,654:INFO:__init__:main@338: Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=None)" using protocol "PI30"
2021-07-14 05:57:59,662:DEBUG:__init__:main@342: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2021-07-14 05:57:59,662:DEBUG:device:__init__@31: __init__ args ()
2021-07-14 05:57:59,663:DEBUG:device:__init__@32: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB0', 'protocol': 'PI30', 'baud': 2400, 'porttype': None, 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_>2021-07-14 05:57:59,663:DEBUG:device:get_port_type@86: port matches ttyusb
2021-07-14 05:57:59,664:INFO:device:set_port@160: Using serialio for communications
2021-07-14 05:57:59,678:DEBUG:device:set_protocol@96: Protocol PI30
2021-07-14 05:57:59,703:DEBUG:device:__init__@36: __init__ name unnamed, port <mppsolar.io.serialio.SerialIO object at 0xb6871f88>, protocol <mppsolar.protocols.pi30.pi30 object at 0xb6871f70>
2021-07-14 05:57:59,703:DEBUG:__init__:main@389: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0xb6871fe8>, 'QPGS0', 'QPGS0', 'screen', None, None)]
2021-07-14 05:57:59,703:INFO:__init__:main@394: Looping 1 commands
2021-07-14 05:57:59,703:INFO:__init__:main@404: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.serialio.SerialIO object at 0xb6871f88>, protocol: <mppsolar.protocols.pi30.pi30 objec>2021-07-14 05:57:59,704:INFO:device:run_command@270: Running command QPGS0
2021-07-14 05:57:59,704:INFO:abstractprotocol:get_full_command@30: Using protocol b'PI30' with 36 commands
2021-07-14 05:57:59,704:DEBUG:protocol_helpers:crcPI@299: Calculating CRC for b'QPGS0'
2021-07-14 05:57:59,704:DEBUG:protocol_helpers:crcPI@349: Generated CRC 0x3f 0xda 0x3fda
2021-07-14 05:57:59,705:DEBUG:abstractprotocol:get_full_command@37: full command: b'QPGS0?\xda\r'
2021-07-14 05:57:59,705:INFO:device:run_command@296: full command b'QPGS0?\xda\r' for command QPGS0
2021-07-14 05:57:59,705:DEBUG:abstractprotocol:get_command_defn@41: Processing command 'QPGS0'
2021-07-14 05:57:59,705:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: F
2021-07-14 05:57:59,706:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: MCHGC
2021-07-14 05:57:59,707:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: MNCHGC
2021-07-14 05:57:59,708:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: MUCHGC
2021-07-14 05:57:59,709:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PBCV
2021-07-14 05:57:59,709:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PBDV
2021-07-14 05:57:59,710:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PBFT
2021-07-14 05:57:59,711:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PBT
2021-07-14 05:57:59,711:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PCP
2021-07-14 05:57:59,712:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PCVV
2021-07-14 05:57:59,713:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PE
2021-07-14 05:57:59,714:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PD
2021-07-14 05:57:59,715:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PGR
2021-07-14 05:57:59,715:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: POP
2021-07-14 05:57:59,716:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: POPM
2021-07-14 05:57:59,717:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PPCP
2021-07-14 05:57:59,718:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PPVOKC
2021-07-14 05:57:59,719:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PSDV
2021-07-14 05:57:59,719:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PSPB
2021-07-14 05:57:59,720:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: QPGS
2021-07-14 05:57:59,721:DEBUG:abstractprotocol:get_command_defn@51: Matched: QPGS0 to: QPGS value: 0
2021-07-14 05:57:59,721:DEBUG:serialio:send_and_receive@19: port /dev/ttyUSB0, baudrate 2400
2021-07-14 05:57:59,729:DEBUG:serialio:send_and_receive@22: Executing command via serialio...
2021-07-14 05:58:00,361:DEBUG:serialio:send_and_receive@30: serial response was: b'(0 92932105105335 B 00 237.3 50.03 229.9 50.06 0459 0427 005 53.1 007 069 176.6 007 00459 00427 005 10100010 0 2 080 120 010 02 000 >2021-07-14 05:58:01,433:DEBUG:device:run_command@315: Send and Receive Response b'(0 92932105105335 B 00 237.3 50.03 229.9 50.06 0459 0427 005 53.1 007 069 176.6 007 00459 00427 005 10100010 0 2 080 120 010 02 000 1>2021-07-14 05:58:01,433:INFO:abstractprotocol:decode@174: response passed to decode: b'(0 92932105105335 B 00 237.3 50.03 229.9 50.06 0459 0427 005 53.1 007 069 176.6 007 00459 00427 005 10100010 0 2 080 120 010 02 >2021-07-14 05:58:01,433:DEBUG:protocol_helpers:crcPI@299: Calculating CRC for b'(0 92932105105335 B [... cut ...]
is_ac_charging                  0               bool
is_scc_charging                 1               bool
is_battery_over_voltage         0               bool
is_battery_under_voltage        0               bool
is_line_lost                    0               bool
is_load_on                      1               bool
is_configuration_changed        0               bool
output_mode                     single machine
charger_source_priority         Solar + Utility
max_charger_current             80              A

AnKloss, 2021-07-15:
Hello good day,
is this the right output you looking for ?

mfg
Andreas

AnKloss, 2021-07-15:
Hello,
Can you help my to fix this ?
mfg
Andreas

jblance, 2021-07-15:
Yep, it might take a couple of days before I can resolve the issue though.

AnKloss, 2021-07-15:
okay may thanks.
i will ask to mpp service , is there any change between 5048 and 8048 
Do you need any spec. info from mpp ?
mfg
Andreas

jblance, 2021-07-15:
If you can get the protocol document that would make a big difference

jblance, 2021-07-15:
So QPGS0 gives 4 extra values as shown at the end, do these match anything?
```
Command: QPGS0 - Parallel Information inquiry
------------------------------------------------------------
Parameter                       Value           Unit	Raw response
parallel_instance_number        Not valid		0
serial_number                   92932105105335		92932105105335
work_mode                       Battery Mode		B
fault_code                      No fault		00
grid_voltage                    237.3           V	237.3 
grid_frequency                  50.03           Hz	50.03
ac_output_voltage               229.9           V	229.9
ac_output_frequency             50.06           Hz	50.06
ac_output_apparent_power        459             VA	0459  
ac_output_active_power          427             W	0427
load_percentage                 5               %	005 
battery_voltage                 53.1            V	53.1 
battery_charging_current        7               A	007  
battery_capacity                69              %	069
pv_input_voltage                176.6           V	176.6  
total_charging_current          7               A	007
total_ac_output_apparent_power  459             VA	00459
total_output_active_power       427             W	00427
total_ac_output_percentage      5               %	005
is_scc_ok                       1               bool	10100010
is_ac_charging                  0               bool
is_scc_charging                 1               bool
is_battery_over_voltage         0               bool
is_battery_under_voltage        0               bool
is_line_lost                    0               bool
is_load_on                      1               bool
is_configuration_changed        0               bool
output_mode                     single machine		0
charger_source_priority         Solar + Utility		2
max_charger_current             80              A	080 
```
extra results:
```
							120 
							010
							02
							000
```

Can you try:
`mpp-solar -P PI41 -c QP2GS0 -D` and post the output (make sure the service / cron isnt processing at the same time)

AnKloss, 2021-07-15:
ubuntu@ubuntu:~$ mpp-solar -P PI41 -c QP2GS0 -D
2021-07-15 23:13:54,650:INFO:__init__:main@235: Solar Device Command Utility, version: 0.7.67, move jk04 to new processing, add lookup functionality
2021-07-15 23:13:54,650:INFO:__init__:main@338: Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=None)" using protocol "PI41"
2021-07-15 23:13:54,660:DEBUG:__init__:main@342: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2021-07-15 23:13:54,661:DEBUG:device:__init__@31: __init__ args ()
2021-07-15 23:13:54,661:DEBUG:device:__init__@32: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB0', 'protocol': 'PI41', 'baud': 2400, 'porttype': None, 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
2021-07-15 23:13:54,662:DEBUG:device:get_port_type@86: port matches ttyusb
2021-07-15 23:13:54,662:INFO:device:set_port@160: Using serialio for communications
2021-07-15 23:13:54,683:DEBUG:device:set_protocol@96: Protocol PI41
2021-07-15 23:13:54,712:DEBUG:device:__init__@36: __init__ name unnamed, port <mppsolar.io.serialio.SerialIO object at 0xb68eff88>, protocol <mppsolar.protocols.pi41.pi41 object at 0xb68eff70>
2021-07-15 23:13:54,712:DEBUG:__init__:main@389: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0xb68effe8>, 'QP2GS0', 'QP2GS0', 'screen', None, None)]
2021-07-15 23:13:54,713:INFO:__init__:main@394: Looping 1 commands
2021-07-15 23:13:54,713:INFO:__init__:main@404: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.serialio.SerialIO object at 0xb68eff88>, protocol: <mppsolar.protocols.pi41.pi41 object at 0xb68eff70> for command: QP2GS0, tag: QP2GS0, outputs: screen
2021-07-15 23:13:54,713:INFO:device:run_command@270: Running command QP2GS0
2021-07-15 23:13:54,713:INFO:abstractprotocol:get_full_command@30: Using protocol b'PI41' with 38 commands
2021-07-15 23:13:54,714:DEBUG:protocol_helpers:crcPI@299: Calculating CRC for b'QP2GS0'
2021-07-15 23:13:54,714:DEBUG:protocol_helpers:crcPI@349: Generated CRC 0x14 0x05 0x1405
2021-07-15 23:13:54,714:DEBUG:abstractprotocol:get_full_command@37: full command: b'QP2GS0\x14\x05\r'
2021-07-15 23:13:54,714:INFO:device:run_command@296: full command b'QP2GS0\x14\x05\r' for command QP2GS0
2021-07-15 23:13:54,715:DEBUG:abstractprotocol:get_command_defn@41: Processing command 'QP2GS0'
2021-07-15 23:13:54,715:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: F
2021-07-15 23:13:54,716:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: MCHGC
2021-07-15 23:13:54,718:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: MNCHGC
2021-07-15 23:13:54,719:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: MUCHGC
2021-07-15 23:13:54,720:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PBCV
2021-07-15 23:13:54,722:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PBDV
2021-07-15 23:13:54,723:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PBFT
2021-07-15 23:13:54,725:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PBT
2021-07-15 23:13:54,726:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PCP
2021-07-15 23:13:54,727:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PCVV
2021-07-15 23:13:54,729:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PE
2021-07-15 23:13:54,730:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PD
2021-07-15 23:13:54,731:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PGR
2021-07-15 23:13:54,732:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: POP
2021-07-15 23:13:54,733:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: POPM
2021-07-15 23:13:54,735:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PPCP
2021-07-15 23:13:54,736:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PPVOKC
2021-07-15 23:13:54,737:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PSDV
2021-07-15 23:13:54,738:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PSPB
2021-07-15 23:13:54,739:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: QPGS
2021-07-15 23:13:54,741:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: QP2GS
2021-07-15 23:13:54,742:DEBUG:abstractprotocol:get_command_defn@51: Matched: QP2GS0 to: QP2GS value: 0
2021-07-15 23:13:54,742:DEBUG:serialio:send_and_receive@19: port /dev/ttyUSB0, baudrate 2400
2021-07-15 23:13:55,397:DEBUG:serialio:send_and_receive@22: Executing command via serialio...
2021-07-15 23:13:55,499:DEBUG:serialio:send_and_receive@30: serial response was: b'(NAKss\r'
2021-07-15 23:13:55,500:DEBUG:device:run_command@315: Send and Receive Response b'(NAKss\r'
2021-07-15 23:13:55,500:INFO:abstractprotocol:decode@174: response passed to decode: b'(NAKss\r'
2021-07-15 23:13:55,500:INFO:abstractprotocol:decode@178: NAK
2021-07-15 23:13:55,500:INFO:device:run_command@334: Decoded response {'ERROR': ['NAK', '']}
2021-07-15 23:13:55,501:DEBUG:__init__:main@408: results: {'ERROR': ['NAK', '']}
2021-07-15 23:13:55,501:INFO:__init__:get_outputs@26: attempting to create output processor: screen
2021-07-15 23:13:55,507:DEBUG:screen:__init__@16: processor.screen __init__ kwargs {}
2021-07-15 23:13:55,508:DEBUG:__init__:main@414: Using output filter: None
[... cut ...]
2021-07-15 23:13:55,509:DEBUG:screen:output@20: kwargs {'data': {'ERROR': ['NAK', '']}, 'tag': 'QP2GS0', 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None, 'mqtt_topic': 'mpp-solar', 'filter': None, 'excl_filter': None, 'keep_case': False}
Parameter                       Value           Unit
2021-07-15 23:13:55,509:DEBUG:helpers:key_wanted@21: key_wanted: No filter and key error not excluded by excl_filter None so key wanted
error                           NAK
2021-07-15 23:13:55,509:DEBUG:__init__:main@434: Not daemon, so not looping
ubuntu@ubuntu:~$

AnKloss, 2021-07-15:
cron is off

jblance, 2021-07-15:
ok, so that command isnt supported
how about
`mpp-solar -P PI41 -c QPIGS2 -D`

AnKloss, 2021-07-15:
ubuntu@ubuntu:~$ mpp-solar -P PI41 -c QPIGS2 -D
2021-07-15 23:49:19,813:INFO:__init__:main@235: Solar Device Command Utility, version: 0.7.67, move jk04 to new processing, add lookup functionality
2021-07-15 23:49:19,813:INFO:__init__:main@338: Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=None)" using protocol "PI41"
2021-07-15 23:49:19,817:DEBUG:__init__:main@342: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2021-07-15 23:49:19,818:DEBUG:device:__init__@31: __init__ args ()
2021-07-15 23:49:19,818:DEBUG:device:__init__@32: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB0', 'protocol': 'PI41', 'baud': 2400, 'porttype': None, 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
2021-07-15 23:49:19,818:DEBUG:device:get_port_type@86: port matches ttyusb
2021-07-15 23:49:19,818:INFO:device:set_port@160: Using serialio for communications
2021-07-15 23:49:19,830:DEBUG:device:set_protocol@96: Protocol PI41
2021-07-15 23:49:19,851:DEBUG:device:__init__@36: __init__ name unnamed, port <mppsolar.io.serialio.SerialIO object at 0xb6865f88>, protocol <mppsolar.protocols.pi41.pi41 object at 0xb6865f70>
2021-07-15 23:49:19,851:DEBUG:__init__:main@389: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0xb6865fe8>, 'QPIGS2', 'QPIGS2', 'screen', None, None)]
2021-07-15 23:49:19,852:INFO:__init__:main@394: Looping 1 commands
2021-07-15 23:49:19,852:INFO:__init__:main@404: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.serialio.SerialIO object at 0xb6865f88>, protocol: <mppsolar.protocols.pi41.pi41 object at 0xb6865f70> for command: QPIGS2, tag: QPIGS2, outputs: screen
2021-07-15 23:49:19,852:INFO:device:run_command@270: Running command QPIGS2
2021-07-15 23:49:19,852:INFO:abstractprotocol:get_full_command@30: Using protocol b'PI41' with 38 commands
2021-07-15 23:49:19,853:DEBUG:protocol_helpers:crcPI@299: Calculating CRC for b'QPIGS2'
2021-07-15 23:49:19,853:DEBUG:protocol_helpers:crcPI@349: Generated CRC 0x68 0x2d 0x682d
2021-07-15 23:49:19,853:DEBUG:abstractprotocol:get_full_command@37: full command: b'QPIGS2h-\r'
2021-07-15 23:49:19,853:INFO:device:run_command@296: full command b'QPIGS2h-\r' for command QPIGS2
2021-07-15 23:49:19,853:DEBUG:abstractprotocol:get_command_defn@41: Processing command 'QPIGS2'
2021-07-15 23:49:19,853:DEBUG:abstractprotocol:get_command_defn@43: Found command QPIGS2 in protocol b'PI41'
2021-07-15 23:49:19,854:DEBUG:serialio:send_and_receive@19: port /dev/ttyUSB0, baudrate 2400
2021-07-15 23:49:19,858:DEBUG:serialio:send_and_receive@22: Executing command via serialio...
2021-07-15 23:49:19,986:DEBUG:serialio:send_and_receive@30: serial response was: b'(01.0 100.5 00105 /=\r'
2021-07-15 23:49:21,081:DEBUG:device:run_command@315: Send and Receive Response b'(01.0 100.5 00105 /=\r'
2021-07-15 23:49:21,081:INFO:abstractprotocol:decode@174: response passed to decode: b'(01.0 100.5 00105 /=\r'
2021-07-15 23:49:21,082:DEBUG:protocol_helpers:crcPI@299: Calculating CRC for b'(01.0 100.5 00105 '
2021-07-15 23:49:21,083:DEBUG:protocol_helpers:crcPI@349: Generated CRC 0x2f 0x3d 0x2f3d
2021-07-15 23:49:21,083:DEBUG:pi30:check_response_valid@860: CRCs match
2021-07-15 23:49:21,083:DEBUG:abstractprotocol:get_command_defn@41: Processing command 'QPIGS2'
2021-07-15 23:49:21,084:DEBUG:abstractprotocol:get_command_defn@43: Found command QPIGS2 in protocol b'PI41'
2021-07-15 23:49:21,084:INFO:abstractprotocol:decode@214: Processing response of type DEFAULT
2021-07-15 23:49:21,085:DEBUG:abstractprotocol:decode@218: trimmed and split responses: [b'01.0', b'100.5', b'00105', b'']
2021-07-15 23:49:21,085:INFO:abstractprotocol:decode@225: Processing DEFAULT type responses
2021-07-15 23:49:21,085:DEBUG:abstractprotocol:decode@250: Error resolving  as float
2021-07-15 23:49:21,086:INFO:device:run_command@334: Decoded response {'raw_response': ['(01.0 100.5 00105 /=\r', ''], '_command': 'QPIGS2', '_command_description': 'General Status Parameters inquiry', 'L2 AC Input Voltage': [1.0, 'V'], 'L2 AC Input Frequency': [100.5, 'Hz'], 'L2 AC Output Voltage': [105.0, 'V'], 'L2 AC Output Frequency': ['', 'Hz']}
2021-07-15 23:49:21,086:DEBUG:__init__:main@408: results: {'raw_response': ['(01.0 100.5 00105 /=\r', ''], '_command': 'QPIGS2', '_command_description': 'General Status Parameters inquiry', 'L2 AC Input Voltage': [1.0, 'V'], 'L2 AC Input Frequency': [100.5, 'Hz'], 'L2 AC Output Voltage': [105.0, 'V'], 'L2 AC Output Frequency': ['', 'Hz']}
2021-07-15 23:49:21,087:INFO:__init__:get_outputs@26: attempting to create output processor: screen
2021-07-15 23:49:21,091:DEBUG:screen:__init__@16: processor.screen __init__ kwargs {}
2021-07-15 23:49:21,092:DEBUG:__init__:main@414: Using output filter: None
2021-07-15 23:49:21,093:INFO:screen:output@19: Using output processor: screen
2021-07-15 23:49:21,093:DEBUG:screen:output@20: kwargs {'data': {'raw_response': ['(01.0 100.5 00105 /=\r', ''], '_command': 'QPIGS2', '_command_description': 'General Status Parameters inquiry', 'L2 AC Input Voltage': [1.0, 'V'], 'L2 AC Input Frequency': [100.5, 'Hz'], 'L2 AC Output Voltage': [105.0, 'V'], 'L2 AC Output Frequency': ['', 'Hz']}, 'tag': 'QPIGS2', 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None, 'mqtt_topic': 'mpp-solar', 'filter': None, 'excl_filter': None, 'keep_case': False}
Command: QPIGS2 - General Status Parameters inquiry
[... cut ...]
l2_ac_input_frequency           100.5           Hz
2021-07-15 23:49:21,094:DEBUG:helpers:key_wanted@21: key_wanted: No filter and key l2_ac_output_voltage not excluded by excl_filter None so key wanted
l2_ac_output_voltage            105.0           V
2021-07-15 23:49:21,095:DEBUG:helpers:key_wanted@21: key_wanted: No filter and key l2_ac_output_frequency not excluded by excl_filter None so key wanted
l2_ac_output_frequency                          Hz
2021-07-15 23:49:21,095:DEBUG:__init__:main@434: Not daemon, so not looping
ubuntu@ubuntu:~$

jblance, 2021-07-15:
so some response there (though clearly not for the details as displayed)
can you do
`mpp-solar -P PI41 -c QPIGS2 -D` and `mpp-solar -c QPGS0 -D` with a screenshot of the app as close as possible (want to see if the extra/new values correspond to items in the app)

AnKloss, 2021-07-15:
ubuntu@ubuntu:~$ mpp-solar -c QPGS0 -D
2021-07-15 23:58:34,930:INFO:__init__:main@235: Solar Device Command Utility, version: 0.7.67, move jk04 to new processing, add lookup functionality
2021-07-15 23:58:34,931:INFO:__init__:main@338: Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=None)" using protocol "PI30"
2021-07-15 23:58:34,935:DEBUG:__init__:main@342: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2021-07-15 23:58:34,936:DEBUG:device:__init__@31: __init__ args ()
2021-07-15 23:58:34,936:DEBUG:device:__init__@32: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB0', 'protocol': 'PI30', 'baud': 2400, 'porttype': None, 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
2021-07-15 23:58:34,937:DEBUG:device:get_port_type@86: port matches ttyusb
2021-07-15 23:58:34,937:INFO:device:set_port@160: Using serialio for communications
2021-07-15 23:58:34,950:DEBUG:device:set_protocol@96: Protocol PI30
2021-07-15 23:58:34,970:DEBUG:device:__init__@36: __init__ name unnamed, port <mppsolar.io.serialio.SerialIO object at 0xb688df70>, protocol <mppsolar.protocols.pi30.pi30 object at 0xb688de38>
2021-07-15 23:58:34,971:DEBUG:__init__:main@389: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0xb688dfd0>, 'QPGS0', 'QPGS0', 'screen', None, None)]
2021-07-15 23:58:34,971:INFO:__init__:main@394: Looping 1 commands
2021-07-15 23:58:34,972:INFO:__init__:main@404: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.serialio.SerialIO object at 0xb688df70>, protocol: <mppsolar.protocols.pi30.pi30 object at 0xb688de38> for command: QPGS0, tag: QPGS0, outputs: screen
2021-07-15 23:58:34,972:INFO:device:run_command@270: Running command QPGS0
2021-07-15 23:58:34,972:INFO:abstractprotocol:get_full_command@30: Using protocol b'PI30' with 36 commands
2021-07-15 23:58:34,972:DEBUG:protocol_helpers:crcPI@299: Calculating CRC for b'QPGS0'
2021-07-15 23:58:34,973:DEBUG:protocol_helpers:crcPI@349: Generated CRC 0x3f 0xda 0x3fda
2021-07-15 23:58:34,973:DEBUG:abstractprotocol:get_full_command@37: full command: b'QPGS0?\xda\r'
2021-07-15 23:58:34,973:INFO:device:run_command@296: full command b'QPGS0?\xda\r' for command QPGS0
2021-07-15 23:58:34,973:DEBUG:abstractprotocol:get_command_defn@41: Processing command 'QPGS0'
2021-07-15 23:58:34,973:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: F
2021-07-15 23:58:34,974:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: MCHGC
2021-07-15 23:58:34,975:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: MNCHGC
2021-07-15 23:58:34,976:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: MUCHGC
2021-07-15 23:58:34,976:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PBCV
2021-07-15 23:58:34,977:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PBDV
2021-07-15 23:58:34,978:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PBFT
2021-07-15 23:58:34,979:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PBT
2021-07-15 23:58:34,980:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PCP
2021-07-15 23:58:34,981:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PCVV
2021-07-15 23:58:34,982:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PE
2021-07-15 23:58:34,982:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PD
2021-07-15 23:58:34,983:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PGR
2021-07-15 23:58:34,984:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: POP
2021-07-15 23:58:34,985:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: POPM
2021-07-15 23:58:34,985:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PPCP
2021-07-15 23:58:34,987:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PPVOKC
2021-07-15 23:58:34,988:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PSDV
2021-07-15 23:58:34,989:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: PSPB
2021-07-15 23:58:34,990:DEBUG:abstractprotocol:get_command_defn@47: Regex commands _command: QPGS
2021-07-15 23:58:34,991:DEBUG:abstractprotocol:get_command_defn@51: Matched: QPGS0 to: QPGS value: 0
2021-07-15 23:58:34,991:DEBUG:serialio:send_and_receive@19: port /dev/ttyUSB0, baudrate 2400
2021-07-15 23:58:35,838:DEBUG:serialio:send_and_receive@22: Executing command via serialio...
2021-07-15 23:58:36,471:DEBUG:serialio:send_and_receive@30: serial response was: b'(0 92932105105335 B 00 221.6 50.04 230.0 50.01 1150 1120 014 52.2 000 060 092.2 000 01150 01120 014 10100010 0 2 080 120 010 00 018 090.9 01I\xe7\r'
2021-07-15 23:58:37,561:DEBUG:device:run_command@315: Send and Receive Response b'(0 92932105105335 B 00 221.6 50.04 230.0 50.01 1150 1120 014 52.2 000 060 092.2 000 01150 01120 014 10100010 0 2 080 120 010 00 018 090.9 01I\xe7\r'
2021-07-15 23:58:37,562:INFO:abstractprotocol:decode@174: response passed to decode: b'(0 92932105105335 B 00 221.6 50.04 230.0 50.01 1150 1120 014 52.2 000 060 092.2 000 01150 01120 014 10100010 0 2 080 120 010 00 018 090.9 01I\xe7\r'
2021-07-15 23:58:37,562:DEBUG:protocol_helpers:crcPI@299: Calculating CRC for b'(0 92932105105335 B 00 221.6 50.04 230.0 50.01 1150 1120 014 52.2 000 060 092.2 000 01150 01120 014 10100010 0 2 080 120 010 00 018 090. [... cut ...]
battery_discharge_current       18              A
2021-07-15 23:58:37,590:DEBUG:helpers:key_wanted@21: key_wanted: No filter and key unknown_float not excluded by excl_filter None so key wanted
unknown_float                   90.9
2021-07-15 23:58:37,590:DEBUG:helpers:key_wanted@21: key_wanted: No filter and key unknown_flags? not excluded by excl_filter None so key wanted
unknown_flags?                  01
2021-07-15 23:58:37,591:DEBUG:__init__:main@434: Not daemon, so not looping
ubuntu@ubuntu:~$

AnKloss, 2021-07-16:
[image]

jblance, 2021-07-16:
sorry I ment the wifi app, ie
[image]

AnKloss, 2021-07-16:
is the wifi app

jblance, 2021-07-16:
The screenshot you posted is the terminal screen.
I want the data from mppsolar and the wifi app at the same time to compare results

AnKloss, 2021-07-16:
PV1 and 2 input watt  (Current * Volt = watt) not Wh
For PV1 i get current,  for PV 2 i not get it , but in the History is show

AnKloss, 2021-07-16:
[image]

jblance, 2021-07-16:
I don't understand what you mean.

AnKloss, 2021-07-16:
wifi app do not delivery the results on time, is a delay from 2-3 min.
i can connect watchpower and see what i can get, but then is in this time no data to mppsolar. What you think ?

jblance, 2021-07-16:
Just do mppsolar commands then watchpower. It should be close enough time wise

AnKloss, 2021-07-16:
[image]

AnKloss, 2021-07-16:
get update info all 2 sec

AnKloss, 2021-07-16:
sorry all 5 sec

AnKloss, 2021-07-20:
Hello,
any Idea how to get missing data ?

mfg
Andreas

jblance, 2021-07-20:
Did you have any luck getting the protocol document? 

It's possible that the 8 extra results are what is needed, but we need to figure out what it is. Hence the request for the output of the command and the screen shot of watchpower straight after

Otherwise there could be another command that we haven't tried, so would need to packet capture the watchpower comms

AnKloss, 2021-07-20:
i ask mpp 2 time , still ignoring , bat i will call today by phone.
Do you know same hardware , wish can by put in between watchpower connect (DB9 or usb) to the 8048 cable ?
May i buy a other Pi and can use it as in/out for recording, if have same software to do it.
mfg
Andreas

AnKloss, 2021-07-20:
just call mpp, he tell my will send the protocol today, let looking what we get.
mfg
Andreas

jblance, 2021-07-20:
Wow, good job. That will help a lot

AnKloss, 2021-07-20:
[MAX Communication Protocol20210217.pdf](https://github.com/jblance/mpp-solar/files/6844911/MAX.Communication.Protocol20210217.pdf)

AnKloss, 2021-07-20:
Protocol max 8048 get it just now

AnKloss, 2021-07-20:
Thanks
As i see mpp add more commands (QPIGS2 (2.10) and change same structure  (example Max AC change current from PP to PPP )

AnKloss, 2021-07-26:
Good morning,
i see you add pi30max protocol in the mpp project , many thanks.
As i am not a programmer, how can i install this to the mppsolar program ?

mfg
Andreas

AnKloss, 2021-07-26:
Hello
I make a version (venv) and is change tp 0.7.71 .
But can not use Protocol PI30MAX
>>>>
(mppsolar) ubuntu@ubuntu:~/venv/mppsolar/bin$ mpp-solar --p PI30MAX
usage: mpp-solar [-h] [-n NAME] [-p PORT] [--porttype PORTTYPE] [-P {PI00,PI16,PI17,PI18,PI30,PI30REVO,PI41,VED,DALY,DALY40}] [-T TAG] [-b BAUD] [-o [OUTPUT]] [--keepcase] [--filter FILTER]
                 [--exclfilter EXCLFILTER] [-q MQTTBROKER] [--mqttport MQTTPORT] [--mqtttopic MQTTTOPIC] [--mqttuser MQTTUSER] [--mqttpass MQTTPASS] [-c [COMMAND]] [-C [CONFIGFILE]]
                 [--daemon] [--getstatus] [--getsettings] [-v] [-D] [-I]
mpp-solar: error: ambiguous option: --p could match --port, --porttype, --protocol
<<<<
How to do ?

mfg
Andreas

AnKloss, 2021-07-26:
(mppsolar) ubuntu@ubuntu:~$ mpp-solar -p /dev/ttyUSB0 --porttype serial -P PI30MAX -T QPGS0 -b 2400 -o influx2_mqtt
Traceback (most recent call last):
  File "/home/ubuntu/venv/mppsolar/bin/mpp-solar", line 33, in <module>
    sys.exit(load_entry_point('mpp-solar', 'console_scripts', 'mpp-solar')())
  File "/home/ubuntu/venv/mppsolar/src/mpp-solar/mppsolar/__init__.py", line 345, in main
    device = device_class(
  File "/home/ubuntu/venv/mppsolar/src/mpp-solar/mppsolar/devices/mppsolar.py", line 7, in __init__
    super().__init__(*args, **kwargs)
  File "/home/ubuntu/venv/mppsolar/src/mpp-solar/mppsolar/devices/device.py", line 35, in __init__
    self.set_protocol(**kwargs)
  File "/home/ubuntu/venv/mppsolar/src/mpp-solar/mppsolar/devices/device.py", line 120, in set_protocol
    self._protocol = self._protocol_class(
  File "/home/ubuntu/venv/mppsolar/src/mpp-solar/mppsolar/protocols/pi30max.py", line 394, in __init__
    self.COMMANDS = COMMANDS
NameError: name 'COMMANDS' is not defined
(mppsolar) ubuntu@ubuntu:~$

AnKloss, 2021-07-26:
version is 0.7.72

jblance, 2021-07-26:
Hi

PI30MAX is still a work in progress, 0.7.73 should function - but its not yet complete
Also for testing, best to drop `-o influx2_mqtt` from the command so the results can be seen on screen (until it is whown to work)

jblance, 2021-07-26:
> mpp-solar -p /dev/ttyUSB0 --porttype serial -P PI30MAX -T QPGS0 -b 2400

this doesnt run the QPGS0 command either - you need a `-c QPGS0` as well

AnKloss, 2021-07-26:
change pip to pip3
pip3 install -e "git+https://github.com/jblance/mpp-solar.git#egg=mpp-solar"
no errors,
(mppsolar) ubuntu@ubuntu:~/venv/mppsolar/bin$ mpp-solar -p /dev/ttyUSB0 --porttype serial -P PI30MAX -T QPGS0 -c QPGS0 -b 2400
Command: QPGS0 - Parallel Information inquiry
------------------------------------------------------------
Parameter                       Value           Unit
parallel_instance_number        Not valid
serial_number                   92932105105335
work_mode                       Battery Mode
fault_code                      No fault
grid_voltage                    248.3           V
grid_frequency                  49.94           Hz
ac_output_voltage               230.1           V
ac_output_frequency             49.92           Hz
ac_output_apparent_power        1196            VA
ac_output_active_power          1163            W
load_percentage                 14              %
battery_voltage                 54.2            V
battery_charging_current        64              A
battery_capacity                80              %
pv1_input_voltage               244.0           V
total_charging_current          64              A
total_ac_output_apparent_power  1196            VA
total_output_active_power       1163            W
total_ac_output_percentage      14              %
is_scc_ok                       1               bool
is_ac_charging                  0               bool
is_scc_charging                 1               bool
is_battery_over_voltage         0               bool
is_battery_under_voltage        0               bool
is_line_lost                    0               bool
is_load_on                      1               bool
is_configuration_changed        0               bool
output_mode                     single machine
charger_source_priority         Solar + Utility
max_charger_current             80              A
max_charger_range               120             A
max_ac_charger_current          20              A
pv_input_current                9               A
battery_discharge_current       0               A
unknown_float                   292.5
unknown_flags?                  08

AnKloss, 2021-07-26:
output is same as with PI30, 
version test 0.7.73

jblance, 2021-07-26:
yes, well that command isnt updated :-)
try `-c QPIGS`
and `-c QPIGS2`

AnKloss, 2021-07-26:
(mppsolar) ubuntu@ubuntu:~/venv/mppsolar/bin$ mpp-solar -p /dev/ttyUSB0 --porttype serial -P PI30MAX -T QPGS0 -c QPIGS2 -b 2400
Command: QPIGS2 - General Status Parameters inquiry 2
------------------------------------------------------------
Parameter                       Value           Unit
pv2_input_current               8.4             A
pv2_input_voltage               317.1           V
pv2_charging_power              2690            W

AnKloss, 2021-07-26:
PV2 working

AnKloss, 2021-07-26:
(mppsolar) ubuntu@ubuntu:~/venv/mppsolar/bin$ mpp-solar -p /dev/ttyUSB0 --porttype serial -P PI30MAX -T QPGS0 -c QPIGS -b 2400
Command: QPIGS - General Status Parameters inquiry
------------------------------------------------------------
Parameter                       Value           Unit
ac_input_voltage                250.9           V
ac_input_frequency              49.9            Hz
ac_output_voltage               229.7           V
ac_output_frequency             49.9            Hz
ac_output_apparent_power        782             VA
ac_output_active_power          697             W
ac_output_load                  9               %
bus_voltage                     447             V
battery_voltage                 54.4            V
battery_charging_current        80              A
battery_capacity                82              %
inverter_heat_sink_temperature  62              °C
pv1_input_current               8.2             A
pv1_input_voltage               308.6           V
battery_voltage_from_scc        0.0             V
battery_discharge_current       0               A
is_sbu_priority_version_added   0               bool
is_configuration_changed        0               bool
is_scc_firmware_updated         0               bool
is_load_on                      1               bool
is_battery_voltage_to_steady_while_charging     0               bool
is_charging_on                  1               bool
is_scc_charging_on              1               bool
is_ac_charging_on               0               bool
battery_voltage_offset_for_fans_on      0               10mV
eeprom_version                  0
pv1_charging_power              2551            W
is_charging_to_float            0               bool
is_switched_on                  1               bool
is_dustproof_installed          0               bool

jblance, 2021-07-26:
There are a bunch of commands that (if you have time) can be tested, you can see the list with
`mpp-solar -p /dev/ttyUSB0 -P PI30MAX -c`

What would be really useful is if you can grab the raw output of each command (so i can use it for tests) to do this add `-o raw` to the command line. e.g.
`mpp-solar -p /dev/ttyUSB0 -P PI30MAX -c QPI -o raw`
```
Command: QPI - Protocol ID inquiry
------------------------------------------------------------
raw_response                    '(PI30\x9a\x0b\r'
```

AnKloss, 2021-07-26:
yes i have time , as work form home...
(mppsolar) ubuntu@ubuntu:~/venv/mppsolar/bin$ mpp-solar -p /dev/ttyUSB0 -P PI30MAX -c QPI -o raw
Command: QPI - Protocol ID inquiry
------------------------------------------------------------
raw_response                    '(PI30\x9a\x0b\r'

AnKloss, 2021-07-26:
all commands in the pi30max  protocol ?

jblance, 2021-07-26:
Yes please, I think so will not work properly

There are still more to do, including the ones that give total power per hour, day etc...

AnKloss, 2021-07-26:
(mppsolar) ubuntu@ubuntu:~/venv/mppsolar/bin$ mpp-solar -p /dev/ttyUSB0 -P PI30MAX -c QID -o raw
2021-07-26 05:13:20,054:WARNING:serialio:send_and_receive@33: Serial read error: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

[148 more replies]

closed 2021-08-05 by jblance