jblance/mpp-solar #132: Question

Oneguice, 2021-07-20:
Hello. I am new to raspberry pi and mpp solar. I am having a hard time loading the mpp software on the Sd card. I have a raspberry 4 but am clueless on installing the software. Could I pay someone to provide instructions or load a sd card for me?

jblance, 2021-07-20:
Start by getting the pi to work normally
https://www.raspberrypi.org/documentation/installation/installing-images/

Then once you log in to the pi you can install mppsolar as per https://github.com/jblance/mpp-solar/blob/master/docs/installation.md

Eg `sudo pip install mpp-solar`

Oneguice, 2021-07-20:
Thank you for your email.

I am getting an error message. Python setup.py egg info failed.

jblance, 2021-07-20:
Can you post the full output starting with the command you ran

Oneguice, 2021-07-20:
pi@raspberrypi:~ $ sudo pip install -e "git+https://github.com/jblance/mpp-solar.git#egg=mpp-solar"
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Obtaining mpp-solar from git+https://github.com/jblance/mpp-solar.git#egg=mpp-solar
  Cloning https://github.com/jblance/mpp-solar.git to ./src/mpp-solar
  Installing build dependencies ... done
mpp-solar requires Python '>=3.6' but the running Python is 2.7.16
pi@raspberrypi:~ $ sudo pip install mpp-solar
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting mpp-solar
  Using cached https://files.pythonhosted.org/packages/c2/36/fad21f11cdadabbde9ff6a031975ed96a56bb8f8c64120885afe7664e50b/mpp-solar-0.4.2.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-L7Nh7G/mpp-solar/setup.py", line 18, in <module>
        with open(path.join(here, 'pypi_README.md'), encoding='utf-8') as f:
      File "/usr/lib/python2.7/codecs.py", line 898, in open
        file = __builtin__.open(filename, mode, buffering)
    IOError: [Errno 2] No such file or directory: '/tmp/pip-install-L7Nh7G/mpp-solar/pypi_README.md'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-L7Nh7G/mpp-solar/
pi@raspberrypi:~ $ python3 -m venv ~/venv/mppsolar
pi@raspberrypi:~ $ mkdir ~/venv
mkdir: cannot create directory \u2018/home/pi/venv\u2019: File exists
pi@raspberrypi:~ $ source venv/mppsolar/bin/activate
(mppsolar) pi@raspberrypi:~ $ pip install -e "git+https://github.com/jblance/mpp-solar.git#egg=mpp-solar"
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Obtaining mpp-solar from git+https://github.com/jblance/mpp-solar.git#egg=mpp-solar
  Cloning https://github.com/jblance/mpp-solar.git to ./venv/mppsolar/src/mpp-solar
  Installing build dependencies ... done
Collecting pyserial (from mpp-solar)
  Downloading https://files.pythonhosted.org/packages/07/bc/587a445451b253b285629263eb51c2d8e9bcea4fc97826266d186f96f558/pyserial-3.5-py2.py3-none-any.whl (90kB)
    100% |\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588| 92kB 1.8MB/s 
Collecting paho-mqtt (from mpp-solar)
  Downloading https://www.piwheels.org/simple/paho-mqtt/paho_mqtt-1.5.1-py3-none-any.whl (74kB)
    100% |\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588| 81kB 319kB/s 
Installing collected packages: pyserial, paho-mqtt, mpp-solar
  Running setup.py develop for mpp-solar
Successfully installed mpp-solar paho-mqtt-1.5.1 pyserial-3.5
(mppsolar) pi@raspberrypi:~ $ sudo pip install mpp-solar
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting mpp-solar
  Using cached https://files.pythonhosted.org/packages/c2/36/fad21f11cdadabbde9ff6a031975ed96a56bb8f8c64120885afe7664e50b/mpp-solar-0.4.2.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-KYns0t/mpp-solar/setup.py", line 18, in <module>
        with open(path.join(here, 'pypi_README.md'), encoding='utf-8') as f:
      File "/usr/lib/python2.7/codecs.py", line 898, in open
        file = __builtin__.open(filename, mode, buffering)
    IOError: [Errno 2] No such file or directory: '/tmp/pip-install-KYns0t/mpp-solar/pypi_README.md'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-KYns0t/mpp-solar/
(mppsolar) pi@raspberrypi:~ $ Eg sudo pip install mpp-solar
bash: Eg: command not found
(mppsolar) pi@raspberrypi:~ $ sudo pip install mpp-solar
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting mpp-solar
  Using cached https://files.pythonhosted.org/packages/c2/36/fad21f11cdadabbde9ff6a031975ed96a56bb8f8c64120885afe7664e50b/mpp-solar-0.4.2.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-_fZ0BU/mpp-solar/setup.py", line 18, in <module>
        with open(path.join(here, 'pypi_README.md'), encoding='utf-8') as f:
      File "/usr/lib/python2.7/codecs.py", line 898, in open
        file = __builtin__.open(filename, mode, buffering)
    IOError: [Errno 2] No such file or directory: '/tmp/pip-install-_fZ0BU/mpp-solar/pypi_README.md'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-_fZ0BU/mpp-solar/
(mppsolar) pi@raspberrypi:~ $ sudo pip install mpp-solar
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting mpp-solar
  Using cached https://files.pythonhosted.org/packages/c2/36/fad21f11cdadabbde9ff6a031975ed96a56bb8f8c64120885afe7664e50b/mpp-solar-0.4.2.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-Q8gAD2/mpp-solar/setup.py", line 18, in <module>
[... cut ...]
(mppsolar) pi@raspberrypi:~ $ Eg sudo pip install mpp-solar
bash: Eg: command not found
(mppsolar) pi@raspberrypi:~ $ sudo pip install -e "git+https://github.com/jblance/mpp-solar.git#egg=mpp-solar"
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Obtaining mpp-solar from git+https://github.com/jblance/mpp-solar.git#egg=mpp-solar
  Updating ./src/mpp-solar clone
  Installing build dependencies ... done
mpp-solar requires Python '>=3.6' but the running Python is 2.7.16
(mppsolar) pi@raspberrypi:~ $

jblance, 2021-07-20:
> File "", line 1, in
> (mppsolar) ***@***.***:~ $ pip install -e "git+
> (mppsolar) ***@***.***:~ $ sudo pip install mpp-solar
> (mppsolar) ***@***.***:~ $ Eg sudo pip install mpp-solar
> (mppsolar) ***@***.***:~ $ sudo pip install -e "git+

You need to use python3 not python2.7, try with pip3 instead of pip
`sudo pip3 install mpp-solar`

Oneguice, 2021-07-20:
done, than i typed in sudo pip install mpp-solar  once completed do I open the software.  how can u pay you?

jblance, 2021-07-20:
> The software is command line only,
> `mpp-solar -h` will list all the options.
>
> It is intended to allow you to get info and send it to either an
> automation tool (I use home assistant) or to something that can graph or
> display the data.

Also the software is free and open source so no need to pay for it.

Oneguice, 2021-07-20:
Cool. Thank you. Looks like I need to purchase more equipment to view my
data.

jblance, 2021-07-20:
Shouldn't need to.

You should be able to see on screen at the minimum.

Then once that works You can decide what you'd like to achieve.

jblance, 2021-07-20:
Here's an example of what I have

Oneguice, 2021-07-21:
-h pulled up all the options but I do not see any commands to view inverter graph or gui

jblance, 2021-07-21:
It's not a gui application, it's just a way to get the info out. Most hardware comes with dedicated software to display the status, you might be better with that?

closed 2021-08-05 by jblance