jblance/mpp-solar #136: jkbms - json parse errors

hoschult, 2021-08-07:
Hi all,

thanks for providing such a great tool!

I have some parsing errors with my latest jkbms installation:

```
(mppsolar) pi@raspi-battery2:~ $ /home/pi/timeout_bash.sh /home/pi/venv/mppsolar/bin/jkbms -p xxx -P JK02 -n "DIY_Powerwall" -o json
{"_command": "getCellData", "_command_description": "BLE Cell Data inquiry", "header": "55aaeb90", "record_type": "02", "record_counter": 49, "voltage_cell01": 3.799, "voltage_cell02": 3.78, "voltage_cell03": 3.809, "voltage_cell04": 3.776, "voltage_cell05": 3.802, "voltage_cell06": 3.798, "voltage_cell07": 3.788, "voltage_cell08": 3.809, "voltage_cell09": 3.788, "voltage_cell10": 3.792, "voltage_cell11": 3.81, "voltage_cell12": 3.779, "voltage_cell13": 3.797, "voltage_cell14": 3.8, "voltage_cell15": 0.0, "voltage_cell16": 0.0, "voltage_cell17": 0.0, "voltage_cell18": 0.0, "voltage_cell19": 0.0, "voltage_cell20": 0.0, "voltage_cell21": 0.0, "voltage_cell22": 0.0, "voltage_cell23": 0.0, "voltage_cell24": 0.0, "average_cell_voltage": 3.794, "delta_cell_voltage": 0.035, "current_balancer": 3.337, "resistance_cell01": 0.125, "resistance_cell02": 0.117, "resistance_cell03": 0.113, "resistance_cell04": 0.115, "resistance_cell05": 0.113, "resistance_cell06": 0.105, "resistance_cell07": 0.104, "resistance_cell08": 0.107, "resistance_cell09": 0.108, "resistance_cell10": 0.103, "resistance_cell11": 0.101, "resistance_cell12": 0.128, "resistance_cell13": 0.133, "resistance_cell14": 0.104, "resistance_cell15": 0.0, "resistance_cell16": 0.0, "resistance_cell17": 0.0, "resistance_cell18": 0.0, "resistance_cell19": 0.0, "resistance_cell20": 0.0, "resistance_cell21": 0.0, "resistance_cell22": 0.0, "resistance_cell23": 0.0, "resistance_cell24": 0.0, "battery_voltage": 53.12, "battery_power": 6.030095577651839e-40, "charge_current": NaN, "battery_t1": 30.7, "battery_t2": 19.8, "mos_temp": 26.9, "percent_remain": 85, "capacity_remain": 141.399, "nominal_capacity": 166.0, "cycle_count": "13000000", "capacity_cycle": 3273.233, "time": "105D23H6M2S", "current_charge": 0.004, "current_discharge": 11.264}
```

This output is parsed in a node red flow where I get the following syntax exception:

`"Unexpected token N in JSON at position 1545"`

Seems to me that it might be connected with a encoding issue:

`"battery_power": 6.030095577651839e-40`

or

`"charge_current": NaN`

What do you think? What can I do to not run in this error?

Version is:

`Solar Device Command Utility, version: 0.7.76`

Thanks,
Holger

jblance, 2021-08-07:
It is likely the NaN result. Can you do a full debug run and post the output

hoschult, 2021-08-07:
yes, of course. Here we go.

`2021-08-07 22:27:09,335:INFO:__init__:main@236: Solar Device Command Utility, version: 0.7.76, remove brackets in ved protocol
2021-08-07 22:27:09,336:INFO:__init__:main@340: Creating device "DIY_Powerwall" (type: "jkbms") on port "3C:A5:51:84:7B:4B (porttype=None)" using protocol "JK02"
2021-08-07 22:27:09,342:DEBUG:__init__:main@343: device_class <class 'mppsolar.devices.jkbms.jkbms'>
2021-08-07 22:27:09,343:DEBUG:device:__init__@31: __init__ args ()
2021-08-07 22:27:09,343:DEBUG:device:__init__@32: __init__ kwargs {'name': 'DIY_Powerwall', 'port': '3C:A5:51:84:7B:4B', 'protocol': 'JK02', 'baud': 2400, 'porttype': None, 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
2021-08-07 22:27:09,344:DEBUG:device:get_port_type@71: port matches jkble ':'
2021-08-07 22:27:09,345:INFO:device:set_port@154: Using jkbleio for communications
2021-08-07 22:27:09,407:DEBUG:device:set_protocol@96: Protocol JK02
2021-08-07 22:27:09,424:DEBUG:device:__init__@36: __init__ name DIY_Powerwall, port <mppsolar.io.jkbleio.JkBleIO object at 0x75f30cf0>, protocol <mppsolar.protocols.jk02.jk02 object at 0x75f30bd0>
2021-08-07 22:27:09,425:DEBUG:__init__:main@390: Commands [(<mppsolar.devices.jkbms.jkbms object at 0x75f30cd0>, '', '', 'json', None, None)]
2021-08-07 22:27:09,426:INFO:__init__:main@395: Looping 1 commands
2021-08-07 22:27:09,427:INFO:__init__:main@406: Getting results from device: jkbms device - name: DIY_Powerwall, port: <mppsolar.io.jkbleio.JkBleIO object at 0x75f30cf0>, protocol: <mppsolar.protocols.jk02.jk02 object at 0x75f30bd0> for command: , tag: , outputs: json
2021-08-07 22:27:09,427:INFO:device:run_command@270: Running command 
2021-08-07 22:27:09,428:INFO:jkabstractprotocol:get_full_command@70: Using protocol b'JK02' with 2 commands
2021-08-07 22:27:09,429:DEBUG:jkabstractprotocol:get_command_defn@95: get_command_defn for: getCellData
2021-08-07 22:27:09,430:DEBUG:abstractprotocol:get_command_defn@42: Processing command 'getCellData'
2021-08-07 22:27:09,431:DEBUG:abstractprotocol:get_command_defn@44: Found command getCellData in protocol b'JK02'
2021-08-07 22:27:09,432:DEBUG:jkabstractprotocol:get_full_command@74: self._command = getCellData, self._command_defn = {'name': 'getCellData', 'command_code': '96', 'record_type': '2', 'description': 'BLE Cell Data inquiry', 'help': ' -- queries the ble device for the cell data', 'type': 'QUERY', 'response_type': 'POSITIONAL', 'response': [['Hex2Str', 4, 'Header', ''], ['Hex2Str', 1, 'Record_Type', ''], ['Hex2Int', 1, 'Record_Counter', ''], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell01', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell02', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell03', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell04', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell05', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell06', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell07', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell08', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell09', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell10', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell11', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell12', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell13', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell14', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell15', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell16', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell17', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell18', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell19', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell20', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell21', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell22', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell23', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell24', 'V'], ['discard', 4, 'discard1', ''], ['LittleHex2Short:r/1000', 2, 'Average_Cell_Voltage', 'V'], ['LittleHex2Short:r/1000', 2, 'Delta_Cell_Voltage', 'V'], ['LittleHex2Short:r/1000', 2, 'Current_Balancer', ''], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell01', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell02', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell03', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell04', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell05', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell06', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell07', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell08', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell09', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell10', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell11', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell12', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell13', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell14', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell15', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell16', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell17', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell18', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell19', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell20', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell21', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell22', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell23', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell24', [... cut ...] cell22": 0.0, "resistance_cell23": 0.0, "resistance_cell24": 0.0, "battery_voltage": 52.953, "battery_power": 5.062078598496256e-40, "charge_current": NaN, "battery_t1": 31.4, "battery_t2": 19.9, "mos_temp": 26.9, "percent_remain": 83, "capacity_remain": 138.149, "nominal_capacity": 166.0, "cycle_count": "13000000", "capacity_cycle": 3276.483, "time": "105D23H30M54S", "current_charge": 0.004, "current_discharge": 8.192}
2021-08-07 22:27:17,682:DEBUG:__init__:main@435: Not daemon, so not looping`

jblance, 2021-08-08:
So looking at the results there are 2 that are obviously wrong
```
battery_power                   5.062078598496256e-40   W   
charge_current                  nan             A   
```
None of my test responses have values here, so I'm a bit lost as to what the correct answers could be
Do you have an appropriate answer for these 2 for when the debug above was taken?

jblance, 2021-08-08:
note these protocols have been reverse engineered - i.e. guessed from logs and results, so it is not surprising that the could be errors. 

To correct things I need the debug log (so i can grab the data) and the correct answer (as best as is possible)

Cheers

hoschult, 2021-08-08:
Hi,

ok, got it. Thinking about a procedure how to get the debug values and the real values from the app at the same time..... Came to the conclusion that it might be hard due to the fact that I cannot login via jkbms utility and app at the same time. Is there some other reverse engineering strategy how I can obtain the values at the same time?

I did the gathering with a slight delay, hopefully it helps anyhow.

Here is the debug and a screenshot out of the app.

Thanks,
Holger

`(mppsolar) pi@raspi-battery2:~ $ /home/pi/timeout_bash.sh /home/pi/venv/mppsolar/bin/jkbms -p 3C:A5:51:84:7B:4B -P JK02 -n "DIY_Powerwall" -D -o json
2021-08-08 06:51:43,635:INFO:__init__:main@236: Solar Device Command Utility, version: 0.7.76, remove brackets in ved protocol
2021-08-08 06:51:43,635:INFO:__init__:main@340: Creating device "DIY_Powerwall" (type: "jkbms") on port "3C:A5:51:84:7B:4B (porttype=None)" using protocol "JK02"
2021-08-08 06:51:43,641:DEBUG:__init__:main@343: device_class <class 'mppsolar.devices.jkbms.jkbms'>
2021-08-08 06:51:43,641:DEBUG:device:__init__@31: __init__ args ()
2021-08-08 06:51:43,642:DEBUG:device:__init__@32: __init__ kwargs {'name': 'DIY_Powerwall', 'port': '3C:A5:51:84:7B:4B', 'protocol': 'JK02', 'baud': 2400, 'porttype': None, 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
2021-08-08 06:51:43,642:DEBUG:device:get_port_type@71: port matches jkble ':'
2021-08-08 06:51:43,643:INFO:device:set_port@154: Using jkbleio for communications
2021-08-08 06:51:43,704:DEBUG:device:set_protocol@96: Protocol JK02
2021-08-08 06:51:43,721:DEBUG:device:__init__@36: __init__ name DIY_Powerwall, port <mppsolar.io.jkbleio.JkBleIO object at 0x75f98ff0>, protocol <mppsolar.protocols.jk02.jk02 object at 0x75f98c90>
2021-08-08 06:51:43,721:DEBUG:__init__:main@390: Commands [(<mppsolar.devices.jkbms.jkbms object at 0x764d8830>, '', '', 'json', None, None)]
2021-08-08 06:51:43,722:INFO:__init__:main@395: Looping 1 commands
2021-08-08 06:51:43,722:INFO:__init__:main@406: Getting results from device: jkbms device - name: DIY_Powerwall, port: <mppsolar.io.jkbleio.JkBleIO object at 0x75f98ff0>, protocol: <mppsolar.protocols.jk02.jk02 object at 0x75f98c90> for command: , tag: , outputs: json
2021-08-08 06:51:43,723:INFO:device:run_command@270: Running command 
2021-08-08 06:51:43,723:INFO:jkabstractprotocol:get_full_command@70: Using protocol b'JK02' with 2 commands
2021-08-08 06:51:43,724:DEBUG:jkabstractprotocol:get_command_defn@95: get_command_defn for: getCellData
2021-08-08 06:51:43,724:DEBUG:abstractprotocol:get_command_defn@42: Processing command 'getCellData'
2021-08-08 06:51:43,725:DEBUG:abstractprotocol:get_command_defn@44: Found command getCellData in protocol b'JK02'
2021-08-08 06:51:43,726:DEBUG:jkabstractprotocol:get_full_command@74: self._command = getCellData, self._command_defn = {'name': 'getCellData', 'command_code': '96', 'record_type': '2', 'description': 'BLE Cell Data inquiry', 'help': ' -- queries the ble device for the cell data', 'type': 'QUERY', 'response_type': 'POSITIONAL', 'response': [['Hex2Str', 4, 'Header', ''], ['Hex2Str', 1, 'Record_Type', ''], ['Hex2Int', 1, 'Record_Counter', ''], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell01', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell02', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell03', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell04', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell05', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell06', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell07', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell08', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell09', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell10', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell11', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell12', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell13', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell14', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell15', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell16', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell17', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell18', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell19', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell20', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell21', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell22', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell23', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell24', 'V'], ['discard', 4, 'discard1', ''], ['LittleHex2Short:r/1000', 2, 'Average_Cell_Voltage', 'V'], ['LittleHex2Short:r/1000', 2, 'Delta_Cell_Voltage', 'V'], ['LittleHex2Short:r/1000', 2, 'Current_Balancer', ''], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell01', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell02', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell03', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell04', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell05', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell06', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell07', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell08', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell09', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell10', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell11', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell12', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell13', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Res [... cut ...] e_cell23": 0.0, "resistance_cell24": 0.0, "battery_voltage": 51.054, "battery_power": 1.5252573264789904e-40, "charge_current": NaN, "battery_t1": 26.0, "battery_t2": 20.1, "mos_temp": 26.5, "percent_remain": 58, "capacity_remain": 96.308, "nominal_capacity": 166.0, "cycle_count": "13000000", "capacity_cycle": 3318.324, "time": "106D7H51M15S", "current_charge": 0.004, "current_discharge": 2.304}
2021-08-08 06:51:52,092:DEBUG:__init__:main@435: Not daemon, so not looping`

[image]
[image]
[image]

hoschult, 2021-08-08:
[image]
[image]

hoschult, 2021-08-08:
some thoughts: In the first screenshot balancer current is positive, in the next screenshot balancer current is negative... That might explain differences?

jblance, 2021-08-16:
the time differences are some of the differences
have added some changes. still some unresolved issues.
another set of -I debugs and screenshots would help - espescially with battery power values

closed 2024-05-07 by jblance