jblance/mpp-solar #137: jkbms - balancer_current wrong?

hoschult, 2021-08-07:
Hi,

I'm using jkbms, version 0.7.76 and I get suspicious balancer_currents. In example, jkbms utility show me the following value:

`"current_balancer": 3.337`

However, I'm using a hardware (JK-B2A24S15P) which is specified with a maximum balancer current of 2A. In bluetooth app jkbms values seems to be ok (never go beyond 2A)

What can be wrong here?

Thanks,
Holger

jblance, 2021-08-07:
Can you a full debug run and post the output

hoschult, 2021-08-07:
here we go:

`(mppsolar) pi@raspi-battery2:~ $ /home/pi/venv/mppsolar/bin/jkbms -p 3C:A5:51:84:7B:4B -P JK02 -n "DIY_Powerwall" -D -o json
2021-08-07 22:32:55,906:INFO:__init__:main@236: Solar Device Command Utility, version: 0.7.76, remove brackets in ved protocol
2021-08-07 22:32:55,907:INFO:__init__:main@340: Creating device "DIY_Powerwall" (type: "jkbms") on port "3C:A5:51:84:7B:4B (porttype=None)" using protocol "JK02"
2021-08-07 22:32:55,912:DEBUG:__init__:main@343: device_class <class 'mppsolar.devices.jkbms.jkbms'>
2021-08-07 22:32:55,912:DEBUG:device:__init__@31: __init__ args ()
2021-08-07 22:32:55,913:DEBUG:device:__init__@32: __init__ kwargs {'name': 'DIY_Powerwall', 'port': '3C:A5:51:84:7B:4B', 'protocol': 'JK02', 'baud': 2400, 'porttype': None, 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
2021-08-07 22:32:55,913:DEBUG:device:get_port_type@71: port matches jkble ':'
2021-08-07 22:32:55,913:INFO:device:set_port@154: Using jkbleio for communications
2021-08-07 22:32:55,975:DEBUG:device:set_protocol@96: Protocol JK02
2021-08-07 22:32:55,991:DEBUG:device:__init__@36: __init__ name DIY_Powerwall, port <mppsolar.io.jkbleio.JkBleIO object at 0x75fb5c90>, protocol <mppsolar.protocols.jk02.jk02 object at 0x75fb5c30>
2021-08-07 22:32:55,991:DEBUG:__init__:main@390: Commands [(<mppsolar.devices.jkbms.jkbms object at 0x75fb5d10>, '', '', 'json', None, None)]
2021-08-07 22:32:55,992:INFO:__init__:main@395: Looping 1 commands
2021-08-07 22:32:55,992:INFO:__init__:main@406: Getting results from device: jkbms device - name: DIY_Powerwall, port: <mppsolar.io.jkbleio.JkBleIO object at 0x75fb5c90>, protocol: <mppsolar.protocols.jk02.jk02 object at 0x75fb5c30> for command: , tag: , outputs: json
2021-08-07 22:32:55,992:INFO:device:run_command@270: Running command 
2021-08-07 22:32:55,993:INFO:jkabstractprotocol:get_full_command@70: Using protocol b'JK02' with 2 commands
2021-08-07 22:32:55,993:DEBUG:jkabstractprotocol:get_command_defn@95: get_command_defn for: getCellData
2021-08-07 22:32:55,993:DEBUG:abstractprotocol:get_command_defn@42: Processing command 'getCellData'
2021-08-07 22:32:55,993:DEBUG:abstractprotocol:get_command_defn@44: Found command getCellData in protocol b'JK02'
2021-08-07 22:32:55,994:DEBUG:jkabstractprotocol:get_full_command@74: self._command = getCellData, self._command_defn = {'name': 'getCellData', 'command_code': '96', 'record_type': '2', 'description': 'BLE Cell Data inquiry', 'help': ' -- queries the ble device for the cell data', 'type': 'QUERY', 'response_type': 'POSITIONAL', 'response': [['Hex2Str', 4, 'Header', ''], ['Hex2Str', 1, 'Record_Type', ''], ['Hex2Int', 1, 'Record_Counter', ''], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell01', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell02', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell03', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell04', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell05', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell06', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell07', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell08', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell09', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell10', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell11', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell12', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell13', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell14', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell15', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell16', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell17', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell18', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell19', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell20', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell21', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell22', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell23', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell24', 'V'], ['discard', 4, 'discard1', ''], ['LittleHex2Short:r/1000', 2, 'Average_Cell_Voltage', 'V'], ['LittleHex2Short:r/1000', 2, 'Delta_Cell_Voltage', 'V'], ['LittleHex2Short:r/1000', 2, 'Current_Balancer', ''], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell01', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell02', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell03', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell04', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell05', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell06', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell07', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell08', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell09', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell10', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell11', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell12', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell13', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell14', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell15', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell16', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell17', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell18', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell19', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell20', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell21', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell22', 'Ohm'],  [... cut ...] e_cell22": 0.0, "resistance_cell23": 0.0, "resistance_cell24": 0.0, "battery_voltage": 52.85, "battery_power": 5.526230688834565e-40, "charge_current": NaN, "battery_t1": 31.6, "battery_t2": 19.8, "mos_temp": 26.9, "percent_remain": 82, "capacity_remain": 137.371, "nominal_capacity": 166.0, "cycle_count": "13000000", "capacity_cycle": 3277.261, "time": "105D23H36M38S", "current_charge": 0.004, "current_discharge": 8.96}
2021-08-07 22:33:04,293:DEBUG:__init__:main@435: Not daemon, so not looping`

sebmueller, 2022-09-01:
Hi,

at my Setup (B2A24S20P) I got also Values above 2 Volts for Balance_Current

If I compare the Values with the Values from my Inverter, it looks like the total charge/discharge current. So maybe the name is wrong, it is not the actual balance current, it is the Total InOut  Current.

[image]