jblance/mpp-solar #138: "new" jkbms Protocol changes..

Scotty-89, 2021-08-09:
hey,

currently i am working on to read out my JKBMS  HW Version 6.X SW Version 6.1.3S (JK-b2a24S15P) over an EPS32 via Bluetooth ;)
it looks not so bad the Basics are working only the responses what i got from the Balancer are not fitting to the "known" types yet..

maybe for some else this could be helpful .. or can help me to find out some more out of it ;) 

this is now how it looks like ..:

["Hex2Str", 4, "Header", ""],
            ["Hex2Str", 1, "Record_Type", ""],
            ["Hex2Int", 1, "Record_Counter", ""],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell01", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell02", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell03", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell04", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell05", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell06", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell07", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell08", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell09", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell10", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell11", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell12", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell13", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell14", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell15", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell16", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell17", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell18", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell19", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell20", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell21", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell22", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell23", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell24", "V"],
            ["discard", 4, "Unknown", ""],
            ["LittleHex2Short:r/1000", 2, "Average_Cell_Voltage", "V"],
            ["LittleHex2Short:r/1000", 2, "Delta_Cell_Voltage", "V"],
            ["discard", 2, "Unknown", ""],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell01", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell02", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell03", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell04", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell05", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell06", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell07", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell08", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell09", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell10", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell11", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell12", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell13", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell14", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell15", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell16", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell17", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell18", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell19", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell20", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell21", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell22", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell23", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell24", "Ohm"],
            ["discard", 6, "Unknown", ""],
            ["LittleHex2UInt:r/1000", 4, "Battery_Voltage", "V"],
            ["LittleHex2Float", 4, "Battery_Power", "W"],
            ["LittleHex2Float", 4, "Charge_Current", "A"], 
            ["discard", 6, "Unknown", ""],
            ["LittleHex2Short:r/10", 2, "Battery_T1", "°C"],
            ["LittleHex2Short:r/10", 2, "Battery_T2", "°C"],
            ["LittleHex2Short:r/10", 2, "MOS_Temp", "°C"],
			["discard", 2, "Unknown", ""],
			["LittleHex2Short:r/1000", 2, "Balancer Current", "A"], 
			["Hex2Int", 1, "Balancer State", "0 = off; 1 = ON positiv current; ON negativ current"], 
			["Hex2Int", 1, "Percent_Remain", ""],
            ["LittleHex2UInt:r/1000", 4, "Capacity_Remain", ""],  # Unknown6+7
            ["LittleHex2UInt:r/1000", 4, "Nominal_Capacity", ""],  # Unknown8+9
            ["Hex2Str", 4, "Cycle_Count", ""],
			["discard", 6, "Unknown", ""],	
            ["LittleHex2UInt:r/1000", 4, "Capacity_Cycle", ""],  # Unknown10+11
            ["uptime", 3, "Time", ""],
            ["discard", 1, "Unknown", ""],
			["Hex2Int", 1, "Charge State", "0 = off; 1 = ON"], 
			["Hex2Int", 1, "Discharge State", "0 = off; 1 = ON"],
			
            ...
			...
			...

jblance, 2021-08-11:
Im not sure what you are asking - or what you are showing in the image

one of the things i did when i was trying to reverse engineer is to fake the BMS and send messages to the app to confirm that changes delivered the expected answer - that was a huge PITA though but was the only way i found to be certain

ulinieto, 2021-08-20:
Hello, i had a jkbms 16s working with your mpp-solar. Ununfortunaly i broken it and i've bought a new one but with harware version 8.x, 

The new one doesnt work, are ther any way to make it work again??

jblance, 2021-08-20:
What does the debug output show?

ulinieto, 2021-08-20:
Unable to connect to bluetooth

jblance, 2021-08-20:
Do you have the correct mac address?
Have you restarted bluetooth?
Is anything else connected to the bms?
Does the app connect correctly?

ulinieto, 2021-08-20:
Do you have the correct mac address? YES
Have you restarted bluetooth? YES
Is anything else connected to the bms? NO
Does the app connect correctly? YES

This what a get at the terminal:

ago 20 23:00:42 raspberrypi systemd[1]: jkbms.service: Service RestartSec=5s expired, scheduling restart.
ago 20 23:00:42 raspberrypi systemd[1]: jkbms.service: Scheduled restart job, restart counter is at 20.
ago 20 23:00:42 raspberrypi systemd[1]: Stopped jkbms is a script for jkbms.
ago 20 23:00:42 raspberrypi systemd[1]: Started jkbms is a script for jkbms.
ago 20 23:02:44 raspberrypi jkbms.py[1636]: Cannot connect to wtfsection with mac c8:47:8c:f3:73:32 - exceeded 3 attempts
ago 20 23:02:44 raspberrypi jkbms.py[1636]: Failed to connect to wtfsection c8:47:8c:f3:73:32
ago 20 23:02:44 raspberrypi systemd[1]: jkbms.service: Succeeded.
ago 20 23:02:49 raspberrypi systemd[1]: jkbms.service: Service RestartSec=5s expired, scheduling restart.
ago 20 23:02:49 raspberrypi systemd[1]: jkbms.service: Scheduled restart job, restart counter is at 21.
ago 20 23:02:49 raspberrypi systemd[1]: Stopped jkbms is a script for jkbms.
ago 20 23:02:49 raspberrypi systemd[1]: Started jkbms is a script for jkbms.
ago 20 23:03:50 raspberrypi jkbms.py[1645]: Traceback (most recent call last):
ago 20 23:03:50 raspberrypi jkbms.py[1645]:   File "/usr/bin/jkbms.py", line 469, in <module>
ago 20 23:03:50 raspberrypi jkbms.py[1645]:     jk.getBLEData()
ago 20 23:03:50 raspberrypi jkbms.py[1645]:   File "/usr/bin/jkbms.py", line 328, in getBLEData
ago 20 23:03:50 raspberrypi jkbms.py[1645]:     characteristicRead = serviceNotify.getCharacteristics(characteristicReadUuid)[0]
ago 20 23:03:50 raspberrypi jkbms.py[1645]: IndexError: list index out of range
ago 20 23:03:50 raspberrypi systemd[1]: jkbms.service: Main process exited, code=exited, status=1/FAILURE
ago 20 23:03:50 raspberrypi systemd[1]: jkbms.service: Failed with result 'exit-code'.
ago 20 23:03:56 raspberrypi systemd[1]: jkbms.service: Service RestartSec=5s expired, scheduling restart.
ago 20 23:03:56 raspberrypi systemd[1]: jkbms.service: Scheduled restart job, restart counter is at 22.
ago 20 23:03:56 raspberrypi systemd[1]: Stopped jkbms is a script for jkbms.
ago 20 23:03:56 raspberrypi systemd[1]: Started jkbms is a script for jkbms.
ago 20 23:03:59 raspberrypi jkbms.py[1653]: Cannot connect to wtfsection with mac c8:47:8c:f3:73:32 - exceeded 3 attempts
ago 20 23:03:59 raspberrypi jkbms.py[1653]: Failed to connect to wtfsection c8:47:8c:f3:73:32

jblance, 2021-08-20:
Try using the command line first. What version are you running?

ulinieto, 2021-08-20:
Where can i see what version i'm running?

jblance, 2021-08-20:
jkbms -v

ulinieto, 2021-08-20:
Solar Device Command Utility, version: 0.7.54, recent changes: add jk rs232/485 protocol

jblance, 2021-08-20:
Ok, so fairly new, try

`sudo hcitool lescan`

ulinieto, 2021-08-20:
I've done the scan and the rasp detects without problem the new jkbms. Thanks for your help but may be I'll try another BMS

PR #221 (2022-05-09): Make JK02 v8 compatible
I assume this change works fine with JK-BMS software/hardware version >= v6. I would be happy if somebody using a JK-BMS v3 could verify this isn't a breaking change.

Fixes: #202
Fixes: #170
Fixes: #138
closed 2022-05-09 by jblance