jblance/mpp-solar #139: Feature Request - Daly BMS with Sinowealth chip

brendanjerwin, 2021-08-14:
I seem to have "lucked out" and gotten a Daly BMS with a sinowealth chip. After struggling to get mpp-solar to work I searched around for another implementation to see if I had a hardware issue or not. I found this: https://github.com/dreadnought/python-daly-bms/issues/1 and that tool _does_ work for me if I tell it to work in sinowealth mode. (So no hardware issue.)

My goal, though, is to get the readings into mqtt for home-assistant and mpp-solar is the best for that!

It'd be great if you could extend mpp-solar with yet-another-protocol for Daly...

hacksics, 2021-08-15:
I think the reason this is not working is that your BMS for 12S or above battery pack. The Sinowealth feature assumes the battery pack is less than 10s. 

"""
List from BMStool PC / Sinowealth
1 = Cell 1 Voltage
...
09 = Cell 9 Voltage
0A = Cell 10 Voltage
0B = Total Voltage
0C = External Temperature 1
0D = External Temperature 2
0E = IC Temperature 1
0F = IC Temperature 2
10 = CADC Current (4 byte)
11 = Full Charge Capacity (4 byte)
12 = Remaining Capacity (4 byte)
13 = RSOC
14 = Cycle Count
15 = Pack Status
16 = Battery Status
17 = Pack Config
18 = Manufacture Access
"""

To help out, I increased the cell count to 16 and ran the command with more debug 

(python3venv) 342-MC-C02YF1VVJG5H:python3venv buddikag$ daly-bms-cli  -d /dev/cu.usbserial-1440 --sinowealth --cell-voltages --verbose
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x01\x02'), 0a0102
DEBUG    [daly_sinowealth.py:91] 0e7dbf (3)
DEBUG    [daly_sinowealth.py:102] Response data for cell 1 is 3709
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x02\x02'), 0a0202
DEBUG    [daly_sinowealth.py:91] 0e7d85 (3)
DEBUG    [daly_sinowealth.py:102] Response data for cell 2 is 3709
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x03\x02'), 0a0302
DEBUG    [daly_sinowealth.py:91] 0e7d93 (3)
DEBUG    [daly_sinowealth.py:102] Response data for cell 3 is 3709
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x04\x02'), 0a0402
DEBUG    [daly_sinowealth.py:91] 0e7ef8 (3)
DEBUG    [daly_sinowealth.py:102] Response data for cell 4 is 3710
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x05\x02'), 0a0502
DEBUG    [daly_sinowealth.py:91] 0e7fe9 (3)
DEBUG    [daly_sinowealth.py:102] Response data for cell 5 is 3711
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x06\x02'), 0a0602
DEBUG    [daly_sinowealth.py:91] 0e7ddd (3)
DEBUG    [daly_sinowealth.py:102] Response data for cell 6 is 3709
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x07\x02'), 0a0702
DEBUG    [daly_sinowealth.py:91] 0e7ec2 (3)
DEBUG    [daly_sinowealth.py:102] Response data for cell 7 is 3710
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x08\x02'), 0a0802
DEBUG    [daly_sinowealth.py:91] 0e7d19 (3)
DEBUG    [daly_sinowealth.py:102] Response data for cell 8 is 3709
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\t\x02'), 0a0902
DEBUG    [daly_sinowealth.py:91] 0e80f2 (3)
DEBUG    [daly_sinowealth.py:102] Response data for cell 9 is 3712
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\n\x02'), 0a0a02
DEBUG    [daly_sinowealth.py:91] 0e7e3c (3)
DEBUG    [daly_sinowealth.py:102] Response data for cell 10 is 3710
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x0b\x02'), 0a0b02
DEBUG    [daly_sinowealth.py:91] 0e7e2a (3)
DEBUG    [daly_sinowealth.py:102] Response data for cell 11 is 3710
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x0c\x02'), 0a0c02
DEBUG    [daly_sinowealth.py:91] 0e80bc (3)
DEBUG    [daly_sinowealth.py:102] Response data for cell 12 is 3712
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\r\x02'), 0a0d02
DEBUG    [daly_sinowealth.py:91] 0e7f59 (3)
DEBUG    [daly_sinowealth.py:102] Response data for cell 13 is 3711
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x0e\x02'), 0a0e02
DEBUG    [daly_sinowealth.py:91] 0e7f63 (3)
DEBUG    [daly_sinowealth.py:102] Response data for cell 14 is 3711
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x0f\x02'), 0a0f02
DEBUG    [daly_sinowealth.py:91] 0e7f75 (3)
DEBUG    [daly_sinowealth.py:102] Response data for cell 15 is 3711
DEBUG    [daly_sinowealth.py:61] message: bytearray(b'\n\x10\x04'), 0a1004
DEBUG    [daly_sinowealth.py:91] 0e7f0000a6 (5)
DEBUG    [daly_sinowealth.py:102] Response data for cell 16 is 243204096

It looks like 0B is cell voltage of 11th cell now.

hacksics, 2021-08-15:
Also I connected the BMS Tool v0.2 with the BMS to see tool only supports reading up to 10 cells. Even there total voltage is also wrong.

hacksics, 2021-08-15:
[btsnoop_hci.log](https://github.com/jblance/mpp-solar/files/6987559/btsnoop_hci.log)

Here is the bluetooth snoop file I generated since bluetooth application seems to show all information accurately. I don't know how to read this.