jblance/mpp-solar #140: Run command Received Status_todo

irmajavi, 2021-08-17:
Hi I run this command:
rock64@rock64:~$ mpp-solar -P PI16 -c QPIGS -p /dev/ttyUSB0
2021-08-16 20:08:10,989:WARNING:abstractprotocol:process_response@161: Failed to eval format bytes.decode(raw_value), error: descriptor 'decode' requires a 'bytes' object but received a 'str'
Command: QPIGS - General status query
------------------------------------------------------------
Parameter                       Value           Unit
grid_voltage                    0               V
output_power                    0               W
grid_frequency                  0               Hz
output_current                  0               A
ac_output_voltage_r             0               V
ac_output_power_r               0               W
ac_output_frequency             0               Hz
ac_output_current_r             0               A
output_load_percent             0               %
pbus_voltage                    0               V
sbus_voltage                    0               V
positive_battery_voltage        0               V
negative_battery_voltage        0               V
battery_capacity                0               %
pv1_input_power                 0               W
pv2_input_power                 0               W
pv3_input_power                 0               W
pv1_input_voltage               0               V
pv2_input_voltage               0               V
pv3_input_voltage               0               V
max_temperature                 0               °C
status_todo                     bytes.decode(raw_value)
rock64@rock64:~$ rock64@rock64:~$ mpp-solar -P PI16 -c QPIGS -p /dev/ttyUSB0
ac_output_frequency             0               Hz
ac_output_current_r             0               A
output_load_percent             0               %
pbus_voltage                    0               V
sbus_voltage                    0               V
positive_battery_voltage        0               V
negative_battery_voltage        0               V
battery_capacity                0               %
pv1_input_power                 0               W
pv2_input_power                 0               W
pv3_input_power                 0               W
pv1_input_voltage               0               V
pv2_input_voltage               0               V
pv3_input_voltage               0               V
max_temperature                 0               °C
status_todo                     bytes.decode(raw_value)

and when run this command:

rock64@rock64:~$ mpp-solar -p /dev/hidraw0 -c QPIGS
Parameter                       Value           Unit
error                           USB open error: [Errno 2] No such file or directory: '/dev/hidraw0'

I tried every possible combination without success. My inverter is connect via usb cable. Any suggestions?

jblance, 2021-08-17:
You need to determine where the inverter connection is in the Linux environment, try `dmesg` just after plugging the inverter cable into the Linux device to see where it gets connected

For mppsolar issues add `-I` to the command to get more info and post the results. The QPI or PI command is a good start to make sure you are using the correct protocol

Post the full output as it will help diagnose what is wrong (and if you are PI16 then it is probably some bugs as that protocol hasn't had much testing)

irmajavi, 2021-08-17:
rock64@rock64:~$ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.190-1233-rockchip-ayufan-gd3f1be0ed310 (root@runner-f725ff63-project-5943294-concurrent-0) (gcc version 7.4.0 (Ubuntu/Linaro 7.4.0-1ubuntu1~18.04.1) ) #1 SMP Wed Aug 28 08:59:34 UTC 2019
[    0.000000] Boot CPU: AArch64 Processor [410fd034]
[    0.000000] earlycon: Early serial console at MMIO32 0xff130000 (options '')
[    0.000000] bootconsole [uart0] enabled
[    0.000000] On node 0 totalpages: 261632
[    0.000000]   DMA zone: 4088 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 261632 pages, LIFO batch:31
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] PERCPU: Embedded 21 pages/cpu @ffffffc03ff3e000 s46248 r8192 d31576 u86016
[    0.000000] pcpu-alloc: s46248 r8192 d31576 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: enabling workaround for ARM erratum 845719
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 257544
[    0.000000] Kernel command line: earlycon=uart8250,mmio32,0xff130000 swiotlb=1 kpti=0 no_console_suspend=1 usbcore.autosuspend=-1 rw panic=10 init=/sbin/init coherent_pool=1M ethaddr=42:3d:94:c1:0b:07 eth1addr=42:3d:94:c1:0b:27 serial=4baaed680c44ae2 cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory swapaccount=1 root=LABEL=linux-root rootwait rootfstype=ext4
[    0.000000] PID hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes)
[    0.000000] software IO TLB: mapped [mem 0x3fdf6000-0x3fe36000] (0MB)
[    0.000000] Memory: 999232K/1046528K available (12478K kernel code, 1760K rwdata, 4552K rodata, 1216K init, 1962K bss, 47296K reserved, 0K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   modules : 0xffffff8000000000 - 0xffffff8008000000   (   128 MB)
                   vmalloc : 0xffffff8008000000 - 0xffffffbdbfff0000   (   246 GB)
                     .init : 0xffffff8009130000 - 0xffffff8009260000   (  1216 KB)
                     .text : 0xffffff8008080000 - 0xffffff8008cb0000   ( 12480 KB)
                   .rodata : 0xffffff8008cb0000 - 0xffffff8009130000   (  4608 KB)
                     .data : 0xffffff8009260000 - 0xffffff8009418008   (  1761 KB)
                   vmemmap : 0xffffffbdc0000000 - 0xffffffbfc0000000   (     8 GB maximum)
                             0xffffffbdc0008000 - 0xffffffbdc1000000   (    15 MB actual)
                   fixed   : 0xffffffbffe7fb000 - 0xffffffbffec00000   (  4116 KB)
                   PCI I/O : 0xffffffbffee00000 - 0xffffffbfffe00000   (    16 MB)
                   memory  : 0xffffffc000200000 - 0xffffffc040000000   (  1022 MB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  Build-time adjustment of leaf fanout to 64.
[    0.000000]  RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=4.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=64, nr_cpu_ids=4
[    0.000000] NR_IRQS:64 nr_irqs:64 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] rockchip_mmc_get_phase: invalid clk rate
[    0.000000] rockchip_mmc_get_phase: invalid clk rate
[    0.000000] rockchip_mmc_get_phase: invalid clk rate
[    0.000000] rockchip_mmc_get_phase: invalid clk rate
[    0.000000] rockchip_mmc_get_phase: invalid clk rate
[    0.000000] rockchip_mmc_get_phase: invalid clk rate
[    0.000000] rockchip_mmc_get_phase: invalid clk rate
[    0.000000] rockchip_mmc_get_phase: invalid clk rate
[    0.000000] Architected cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000008] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.001863] Console: colour dummy device 80x25
[    0.002318] console [tty0] enabled
[    0.002644] bootconsole [uart0] disabled
[    0.003026] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=96000)
[    0.003064] pid_max: default: 32768 minimum: 301
[    0.003209] Security Framework initialized
[    0.003231] Yama: becoming mindful.
[    0.003264] AppArmor: AppArmor disabled by boot time parameter
[    0.003341] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes)
[    0.003370] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes)
[    0.004211] Initializing cgroup subsys io
[... 717 lines cut ...]
[   12.089459] Adding 125820k swap on /dev/zram3.  Priority:5 extents:1 across:125820k SS
[   15.953803] [drm:dw_hdmi_rockchip_set_property] *ERROR* failed to set rockchip hdmi connector property
[   15.954787] [drm:dw_hdmi_rockchip_set_property] *ERROR* failed to set rockchip hdmi connector property
[   15.955633] [drm:dw_hdmi_rockchip_set_property] *ERROR* failed to set rockchip hdmi connector property
[... cut ...]
[ 5760.935812] ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
[ 5760.935926] ch341 4-1:1.0: device disconnected
[ 6156.977979] usb 3-1: new full-speed USB device number 3 using ohci-platform
[ 6157.182375] usb 3-1: New USB device found, idVendor=1a86, idProduct=7523
[ 6157.182392] usb 3-1: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[ 6157.182401] usb 3-1: Product: USB Serial
[ 6157.184843] ch341 3-1:1.0: ch341-uart converter detected
[ 6157.208817] usb 3-1: ch341-uart converter now attached to ttyUSB0

irmajavi, 2021-08-17:
rock64@rock64:~$ mpp-solar -P PI16 -c QPI -p /dev/ttyUSB0 -I
2021-08-16 21:32:19,270:INFO:__init__:main@236: Solar Device Command Utility, version: 0.7.76, remove brackets in ved protocol
2021-08-16 21:32:19,271:INFO:__init__:main@340: Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=None)" using protocol "PI16"
2021-08-16 21:32:19,277:INFO:device:set_port@160: Using serialio for communications
2021-08-16 21:32:19,347:INFO:__init__:main@395: Looping 1 commands
2021-08-16 21:32:19,347:INFO:__init__:main@406: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.serialio.SerialIO object at 0xf6430db0>, protocol: <mppsolar.protocols.pi16.pi16 object at 0xf6430d30> for command: QPI, tag: QPI, outputs: screen
2021-08-16 21:32:19,348:INFO:device:run_command@270: Running command QPI
2021-08-16 21:32:19,348:INFO:pi16:get_full_command@180: Using protocol b'PI16' with 6 commands
2021-08-16 21:32:19,349:INFO:device:run_command@296: full command b'QPI\r' for command QPI
2021-08-16 21:32:20,497:INFO:abstractprotocol:decode@175: response passed to decode: b''
2021-08-16 21:32:20,498:INFO:abstractprotocol:decode@215: Processing response of type SEQUENTIAL
2021-08-16 21:32:20,499:INFO:device:run_command@334: Decoded response {'raw_response': ['', ''], '_command': 'QPI', '_command_description': 'Device Protocol Version inquiry', 'Protocol Version': ['', '']}
2021-08-16 21:32:20,500:INFO:__init__:get_outputs@26: attempting to create output processor: screen
2021-08-16 21:32:20,509:INFO:screen:output@19: Using output processor: screen
Command: QPI - Device Protocol Version inquiry
------------------------------------------------------------
Parameter                       Value           Unit
protocol_version

jblance, 2021-08-17:
So it is connected on ttyUSB0 

What inverter are you using?

irmajavi, 2021-08-17:
its from srne

jblance, 2021-08-17:
QPI doesnt work for PI16 - try PI

jblance, 2021-08-17:
> srne

What model? Do you know that it supports PI16 protocol?

irmajavi, 2021-08-17:
rock64@rock64:~$ mpp-solar -P PI16 -c PI -p /dev/ttyUSB0 -I
2021-08-16 21:33:16,525:INFO:__init__:main@236: Solar Device Command Utility, version: 0.7.76, remove brackets in ved protocol
2021-08-16 21:33:16,525:INFO:__init__:main@340: Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=None)" using protocol "PI16"
2021-08-16 21:33:16,531:INFO:device:set_port@160: Using serialio for communications
2021-08-16 21:33:16,601:INFO:__init__:main@395: Looping 1 commands
2021-08-16 21:33:16,602:INFO:__init__:main@406: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.serialio.SerialIO object at 0xf69d4db0>, protocol: <mppsolar.protocols.pi16.pi16 object at 0xf69d4d30> for command: PI, tag: PI, outputs: screen
2021-08-16 21:33:16,602:INFO:device:run_command@270: Running command PI
2021-08-16 21:33:16,603:INFO:pi16:get_full_command@180: Using protocol b'PI16' with 6 commands
2021-08-16 21:33:16,604:INFO:abstractprotocol:get_command_defn@57: No command_defn found for PI
2021-08-16 21:33:16,604:INFO:device:run_command@296: full command b'PI\r' for command PI
2021-08-16 21:33:16,605:INFO:abstractprotocol:get_command_defn@57: No command_defn found for PI
2021-08-16 21:33:17,753:INFO:abstractprotocol:decode@175: response passed to decode: b''
2021-08-16 21:33:17,754:INFO:abstractprotocol:get_command_defn@57: No command_defn found for PI
2021-08-16 21:33:17,755:INFO:device:run_command@334: Decoded response {'raw_response': ['', ''], '_command': 'PI', 'WARNING': ["No definition for command PI in protocol b'PI16'", ''], 'response': ['', '']}
2021-08-16 21:33:17,755:INFO:__init__:get_outputs@26: attempting to create output processor: screen
2021-08-16 21:33:17,764:INFO:screen:output@19: Using output processor: screen
Command: PI - No description found
------------------------------------------------------------
Parameter                       Value           Unit
warning                         No definition for command PI in protocol b'PI16'
response

irmajavi, 2021-08-17:
I tried all protocols but not response. Which one you suggest?

jblance, 2021-08-17:
PI30 is the most common in use from mpp-solar activity

irmajavi, 2021-08-17:
SRNE Model HF4835U80-145

irmajavi, 2021-08-17:
rock64@rock64:~$ mpp-solar -P PI30 -c QPI -p /dev/ttyUSB0 -I
2021-08-16 21:40:20,039:INFO:__init__:main@236: Solar Device Command Utility, version: 0.7.76, remove brackets in ved protocol
2021-08-16 21:40:20,040:INFO:__init__:main@340: Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=None)" using protocol "PI30"
2021-08-16 21:40:20,045:INFO:device:set_port@160: Using serialio for communications
2021-08-16 21:40:20,117:INFO:__init__:main@395: Looping 1 commands
2021-08-16 21:40:20,118:INFO:__init__:main@406: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.serialio.SerialIO object at 0xf6a89db0>, protocol: <mppsolar.protocols.pi30.pi30 object at 0xf6a89d30> for command: QPI, tag: QPI, outputs: screen
2021-08-16 21:40:20,118:INFO:device:run_command@270: Running command QPI
2021-08-16 21:40:20,119:INFO:abstractprotocol:get_full_command@31: Using protocol b'PI30' with 36 commands
2021-08-16 21:40:20,119:INFO:device:run_command@296: full command b'QPI\xbe\xac\r' for command QPI
2021-08-16 21:40:21,266:INFO:abstractprotocol:decode@175: response passed to decode: b''
2021-08-16 21:40:21,267:INFO:abstractprotocol:decode@179: Response to short
2021-08-16 21:40:21,268:INFO:device:run_command@334: Decoded response {'ERROR': ['Response to short', '']}
2021-08-16 21:40:21,269:INFO:__init__:get_outputs@26: attempting to create output processor: screen
2021-08-16 21:40:21,277:INFO:screen:output@19: Using output processor: screen
Parameter                       Value           Unit
error                           Response to short

jblance, 2021-08-17:
doesnt look like it uses the protocols that are supported by mpp-solar so far

irmajavi, 2021-08-17:
Oh really, I bought another inverter a few months ago and I use this program mpp-solar with it because is so cool and I thought it might work with this one too.

jblance, 2021-08-17:
well it probably _can_ work, we just need to find the protocol and implement it if it is not currently supported

jblance, 2021-08-17:
It claims 'Supply of a variety of user-friendly communication modules, such as RS485(GPRS, WiFi, Bluetooth), CAN, USB etc.,
and suitable for computer, mobile phones, Internet monitoring as well as remote operations'
Can you ask the supplier for the communication protocol definition?

irmajavi, 2021-08-17:
Yes I can do that. If I get a response I will post it here.

jblance, 2021-08-17:
you could try something like:
```python
import serial
ser = serial.Serial('/dev/ttyUSB0', 19200, timeout=1)
print(ser)

ser.write(':010310010001EA\r\n')
print(repr(ser.read(1000)))  # Read 1000 bytes, or wait for timeout
```

and see what response you get

irmajavi, 2021-08-17:
which program should I use to write and execute the code?

jblance, 2021-08-17:
just use an editor to put it into a file like `test.py` and then run it
with `python3 test.py`

irmajavi, 2021-08-17:
rock64@rock64:~/Documents$ python3 test.py
Serial<id=0xf70716f0, open=True>(port='/dev/ttyUSB0', baudrate=19200, bytesize=8, parity='N', stopbits=1, timeout=1, xonxoff=False, rtscts=False, dsrdtr=False)
Traceback (most recent call last):
  File "test.py", line 5, in <module>
    ser.write(':010310010001EA\r\n')
  File "/usr/local/lib/python3.6/dist-packages/pyserial-3.5-py3.6.egg/serial/serialposix.py", line 616, in write
    d = to_bytes(data)
  File "/usr/local/lib/python3.6/dist-packages/pyserial-3.5-py3.6.egg/serial/serialutil.py", line 65, in to_bytes
    raise TypeError('unicode strings are not supported, please encode to bytes: {!r}'.format(seq))
TypeError: unicode strings are not supported, please encode to bytes: ':010310010001EA\r\n'

irmajavi, 2021-08-17:
I get a response from SRNE, they send me this file:

[逆控一体机modbus寄存器定义-客户版v1.3 - 中性（English）.pdf](https://github.com/jblance/mpp-solar/files/7000275/modbus.-.v1.3.-.English.pdf)

jblance, 2021-08-19:
ok, well that helps
the code doesnt have any modbus protocols at the moment and i dont have any modbus devices myself, so if you are interested in this in mpp-solar it will be a lot of testing on your part and will likely take a while. Are you up for that?

irmajavi, 2021-08-20:
Yeah I can do that and not just me, other people can benefit from this.

pwalsh110, 2021-12-14:
> 

I also have this HF4835U80-145 unit.  Was communication with this device ever solved?  So far I'm not having any luck with the USB port or my generic FTDI serial adapter in the RS485 port, but I have no documentation to go off.

pwalsh110, 2021-12-15:
> I write my own python script and is working fine. It need improvement but it work for my needs. It is a script that send that data from the inverter to an mqtt server. This type of inverter use Modbus for communicate that is why mpp-solar is not working. If you have programming skills you can modified for your needs.

Thank you that would be a very helpful start could please provide a link to your code?

irmajavi, 2021-12-17:
Write me an email javier.matos@gmail.com and as soon as I can I will send it to you.

jblance, 2023-01-31:
will see if @irmajavi is ok to share their code and if so and it can be added I will create an enhancement to track it

closed 2023-01-31 by jblance