jblance/mpp-solar #141: MQTT_Influx_Grafana  > empty grafana

jeca32, 2021-08-17:
I followed your procedure but I must make several errors because it does not work.

I manage to extract the information from the BMS but the database remains empty
can someone help me?

pi@raspberrypi:~ $ jkbms -C /etc/jkbms/jkbms.conf
Parameter                     Value           Unit
header                         55aaeb90          
record_type                   02                
record_counter                 61                
voltage_cell01                 3.289           V  
voltage_cell02                 3.291           V  
voltage_cell03                 3.291           V  
voltage_cell04                 3.291           V  
voltage_cell05                 3.291           V  
voltage_cell06                 3.291           V  
voltage_cell07                 3.288           V  
voltage_cell08                 3.282           V  
voltage_cell09                 3.288           V  
voltage_cell10                 3.291           V  
voltage_cell11                 3.289           V  
voltage_cell12                 3.286           V  
voltage_cell13                 3.288           V  
voltage_cell14                 3.288           V  
voltage_cell15                 3.289           V  
voltage_cell16                 3.283           V  
voltage_cell17                 0.0             V  
voltage_cell18                 0.0             V  
voltage_cell19                 0.0             V  
voltage_cell20                 0.0             V  
voltage_cell21                 0.0             V  
voltage_cell22                 0.0             V  
voltage_cell23                 0.0             V  
voltage_cell24                 0.0             V  
average_cell_voltage           3.289           V  
delta_cell_voltage             0.01           V  
current_balancer               1.793              
resistance_cell01             0.12           Ohm
resistance_cell02             0.122           Ohm
resistance_cell03             0.124           Ohm
resistance_cell04             0.121           Ohm
resistance_cell05             0.123           Ohm
resistance_cell06             0.122           Ohm
resistance_cell07             0.121           Ohm
resistance_cell08             0.12           Ohm
resistance_cell09             0.119           Ohm
resistance_cell10             0.119           Ohm
resistance_cell11             0.12           Ohm
resistance_cell12             0.119           Ohm
resistance_cell13             0.118           Ohm
resistance_cell14             0.12           Ohm
resistance_cell15             0.12           Ohm
resistance_cell16             0.118           Ohm
resistance_cell17             0.0             Ohm
resistance_cell18             0.0             Ohm
resistance_cell19             0.0             Ohm
resistance_cell20             0.0             Ohm
resistance_cell21             0.0             Ohm
resistance_cell22             0.0             Ohm
resistance_cell23             0.0             Ohm
resistance_cell24             0.0             Ohm
battery_voltage               52.624         V  
battery_power                 0.0             W  
charge_current                 0.0             A  
battery_t1                     24.1           °C  
battery_t2                     23.7           °C  
mos_temp                       25.4           °C  
percent_remain                 69                
capacity_remain               216.019            
nominal_capacity               310.0              
cycle_count                   00000000          
capacity_cycle                 2.531              
time                           14D8H57M10S        
current_charge                 0.004              
current_discharge             0.0                
pi@raspberrypi:~ $

pi@raspberrypi:~ $ mosquitto_sub -h localhost -v -t "#"
CellData/status/header/value 55aaeb90
CellData/status/record_type/value 02
CellData/status/record_counter/value 61
CellData/status/voltage_cell01/value 3.289
....

pi@raspberrypi:~ $ sudo systemctl status telegraf
● telegraf.service - The plugin-driven server agent for reporting metrics into InfluxDB
   Loaded: loaded (/lib/systemd/system/telegraf.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2021-08-15 18:52:45 CEST; 2 days ago
     Docs: https://github.com/influxdata/telegraf
 Main PID: 522 (telegraf)
    Tasks: 12 (limit: 3720)
   CGroup: /system.slice/telegraf.service
           └─522 /usr/bin/telegraf -config /etc/telegraf/telegraf.conf -config-directory /etc/telegraf/telegraf.d

août 17 23:27:37 raspberrypi telegraf[522]: 2021-08-17T21:27:37Z E! [inputs.mqtt_consumer] Error in plugin: metric parse error: expected tag at 1:5: "25.4"
août 17 23:27:37 raspberrypi telegraf[522]: 2021-08-17T21:27:37Z E! [inputs.mqtt_consumer] Error in plugin: metric parse error: expected tag at 1:4: "°C"
août 17 23:27:37 raspberrypi telegraf[522]: 2021-08-17T21:27:37Z E! [inputs.mqtt_consumer] Error in plugin: metric parse error: expected tag at 1:3: "69"
août 17 23:27:37 raspberrypi telegraf[522]: 2021-08-17T21:27:37Z E! [inputs.mqtt_consumer] Error in plugin: metric parse error: expected tag at 1:8: "216.019"
août 17 23:27:37 raspberrypi telegraf[522]: 2021-08-17T21:27:37Z E! [inputs.mqtt_consumer] Error in plugin: metric parse error: expected tag at 1:6: "310.0"
août 17 23:27:37 raspberrypi telegraf[522]: 2021-08-17T21:27:37Z E! [inputs.mqtt_consumer] Error in plugin: metric parse error: expected tag at 1:9: "00000000"
août 17 23:27:37 raspberrypi telegraf[522]: 2021-08-17T21:27:37Z E! [inputs.mqtt_consumer] Error in plugin: metric parse error: expected tag at 1:6: "2.531"
août 17 23:27:37 raspberrypi telegraf[522]: 2021-08-17T21:27:37Z E! [inputs.mqtt_consumer] Error in plugin: metric parse error: expected tag at 1:12: "14D8H57M10S"
août 17 23:27:37 raspberrypi telegraf[522]: 2021-08-17T21:27:37Z E! [inputs.mqtt_consumer] Error in plugin: metric parse error: expected tag at 1:6: "0.004"
août 17 23:27:37 raspberrypi telegraf[522]: 2021-08-17T21:27:37Z E! [inputs.mqtt_consumer] Error in plugin: metric parse error: expected tag at 1:4: "0.0"

pi@raspberrypi:~ $ influx
Connected to http://localhost:8086 version 1.8.9
InfluxDB shell version: 1.8.9
> use mppsolar
Using database mppsolar
> show measurements
name: measurements
name
----
cpu
disk
diskio
kernel
mem
processes
swap
system
>

my configuration:

2021-05-07-raspios-buster-armhf-full
Python 3.9.1

JK-B1A24S1 (x16 Lifepo4)
raspberry pi4
Voltronic InfiniSolar E 5.5KW

----------------------------------
/etc/jkbms/jkbms.conf
[SETUP]
pause=5
mqtt_broker=localhost

[Bmsjeca]
type=jkbms
protocol=JK02
port=3C:A5:51:87:DD:68
command=getCellData
tag=CellData
outputs=mqtt,screen
----------------------------
/etc/systemd/user/jkbms.service
[Unit]
Description=JKBMS Service
[Service]
ExecStart=/usr/bin/python3 /usr/local/bin/jkbms -C /etc/jkbms/jkbms.conf --daemon
Environment=PYTHONUNBUFFERED=1
Restart=always
WatchdogSec=300
Type=notify

[Install]
WantedBy=default.target

sanwizard, 2021-08-17:
I am using a Heltec 200amp BMS, which has a tiny RS485 and CAN connector.
What cables do you recommend to go from RS485 to USB on the Pi?

jblance, 2021-08-17:
> `/etc/jkbms/jkbms.conf`
> 
> ```
> [SETUP]
> pause=5
> mqtt_broker=localhost
> 
> [Bmsjeca]
> type=jkbms
> protocol=JK02
> port=3C:A5:51:87:DD:68
> command=getCellData
> tag=CellData
> outputs=mqtt,screen
> ```

try with 
`outputs=influx2_mqtt,screen`

jeca32, 2021-08-17:
> > `/etc/jkbms/jkbms.conf`
> > ```
> > [SETUP]
> > pause=5
> > mqtt_broker=localhost
> > 
> > [Bmsjeca]
> > type=jkbms
> > protocol=JK02
> > port=3C:A5:51:87:DD:68
> > command=getCellData
> > tag=CellData
> > outputs=mqtt,screen
> > ```
> 
> try with
> `outputs=influx2_mqtt,screen`

magic thank you 

pi@raspberrypi:~ $ influx
Connected to http://localhost:8086 version 1.8.9
InfluxDB shell version: 1.8.9
> use mppsolar
Using database mppsolar
> show measurements
name: measurements
name
----
cpu
disk
diskio
jkbms
kernel
mem
processes
swap
system
> select * from jkbms
name: jkbms
time                average_cell_voltage battery_power battery_t1 battery_t2 .....

jeca32, 2021-08-17:
> I am using a Heltec 200amp BMS, which has a tiny RS485 and CAN connector. What cables do you recommend to go from RS485 to USB on the Pi?
> […](#)

I don't know, currently I use bluetooth

closed 2021-08-17 by jeca32

jblance, 2021-08-17:
> > I am using a Heltec 200amp BMS, which has a tiny RS485 and CAN connector. What cables do you recommend to go from RS485 to USB on the Pi?
> > […](#)
> 
> I don't know, currently I use bluetooth

There quite a few on Aliexpress - find one that matches the connector. I use the one that the manufacturer of my BMS supplied. It is quite a simple device, so any should work