jblance/mpp-solar #142: MQTT_Influx_Grafana > InfluxDB Error: Bad Gateway

jeca32, 2021-08-21:
I cannot add the database in grafana and the influxdb service only starts with the user pi. 

I run these commands but the problem is still 

sudo rm -rf /var/lib/influxdb/
sudo mkdir /var/lib/influxdb
sudo chown -R 1000:1000 /var/lib/influxdb

Does anyone have the solution ?
[image]
[image]

jblance, 2021-08-22:
This seems to be an influx problem - I'm no expert with influxdb (in fact I
dont use it)

Perhaps ask in an influx specific forum?

jeca32, 2021-08-22:
I nevertheless followed your MQTT_Influx_Grafana tutorial. 
I continue to look for the solution if someone has an idea or a procedure to find the fault, I am taker

DMIINC, 2021-08-23:
If it is of any help, in one instance I was not able to overcome this issue. No matter what I tried it did not work. For what ever reason Grafana wants to be on the same machine as influxDb and its database. As jblance suggested try influx forums for more help. Post a solution if you find one. I am interested to know too. I have since moved to Home Assistant.

jblance, 2021-08-28:
Great news