jblance/mpp-solar #146: jkbms.service > irregular data

jeca32, 2021-08-28:
My installation works but the database is not filled regularly, 

[image]

as if the BMS was saturated with requests from the server.
it also emits random beeps without recording new values.

is there a method to limit data logging to only 12 values: 

delta_cell_voltage
battery_voltage
battery_power
charge_current
battery_t1
battery_t2
capacity_remain
cycle_count
capacity_cycle
time
current_charge
current_discharge

here is my config 
----------------------------------------------
/etc/jkbms/jkbms.conf
[SETUP]
pause=0
mqtt_broker=localhost

[Bmsjeca]
type=jkbms
protocol=JK02
port=3C:A5:51:87:DD:68
command=getCellData
tag=CellData
outputs=influx2_mqtt,screen
----------------------------------------------

/etc/systemd/user/jkbms.service
[Unit]
Description=JKBMS Service

[Service]
ExecStart=/usr/bin/python3 /usr/local/bin/jkbms -C /etc/jkbms/jkbms.conf --daemon
Environment=PYTHONUNBUFFERED=1
Restart=always
WatchdogSec=300
Type=notify

[Install]
WantedBy=default.target
----------------------------------------------

jeca32, 2021-08-28:
with
pause = 10

the data is regular but the bms beeps every 10 seconds

with the application on my phone there is one request per second but without the BEEP 

Is there a solution to limit the values and remove the noise? 

[image]

jblance, 2021-08-28:
I think it beeps on connect and the code reconnects each message - I'm looking at changing this but need to do some refactoring first.

jeca32, 2021-08-29:
yes there is indeed a beep on connection and another for disconnection after providing the result of the command.
while waiting for a probable evolution of the script, I will set the "pause = 10" and I will muffle this beep beep noise.

closed 2021-08-29 by jeca32