jblance/mpp-solar #149: JKBMS and RS485

trixing, 2021-08-30:
I have the following BMS, working via Bluetooth, but occassionally it crashes the Bluetooth stack of the RPI 3 and I need to reboot. I've got myself an RS485 cable (from ICGOGO) and trying to get it to work.

If I use the JK232 protocol I get the following response (edited):
- full command bytearray(b'\xdd\xa5\x03\x00\xff\xfdw') for command getBalancerData
- jk232:get_responses@149: get_responses: responses [b'\x11', b'-', b'~', b'\x01', b'\x01D', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'']

For RS485 protocol:
- full command bytearray(b'U\xaa\x01\xff\x00\x00\xff') for command getBalancerData
- response passed to decode: b'U\xa5\x01\xff\x01'

Both are obviously too short and it fails processing. Interesting I do get responses, but different ones. Any clue how to debug this further? I tried a short python script to reproduce, which gives me consistently the same results:
"""
import serial
ser = serial.Serial('/dev/ttyUSB1', 9600, timeout=1)
#cmd = bytearray(b'\xdd\xa5\x03\x00\xff\xfdw')
cmd = bytearray(b'\x55\xAA\x01\xFF\x00\x00\xFF')
ser.write(cmd)
print(ser.read(100))
ser.close()
"""

Command: getInfo - BLE Device Information inquiry
------------------------------------------------------------
Parameter                       Value           Unit
header                          55aaeb90            
record_type                     03                  
record_counter                  184                 
device_model                    JK-B1A24S1          
hardware_version                5P6.X               
software_version                6.1.0S              
device_name                     JK-B1A24S15P        
device_passcode                 xxxx                
manufacturing_date              210106              
serial_number                   2012281234          
user_data                       Input Userdata      
settings_passcode?              xxxxxx

jblance, 2021-09-05:
> I have the following BMS, working via Bluetooth, but occassionally it crashes the Bluetooth stack of the RPI 3 and I need to reboot.

This can be fixed by restarting the bluetooth service (which is a hack, but works for me)

I've not had any success with the RS485 connections - what was the jk232 response passed to decode?

Maybe try with `\n` added to the end of the cmd in your script?

trixing, 2021-09-06:
Yeah, unfortunately restarting Bluetooth service stops working after a
while due to a kernel NULL pointer reference in create_le_conn_complete.

This patch would address it (
https://www.spinics.net/lists/linux-bluetooth/msg70764.html) . Still not
merged after the issue got discovered ~2 years ago... The only fix at that
point is to reboot.

Well, today I get no responses anymore from the dongle. I'm starting to
wonder if my cheap dongle is actually broken or if I need a termination
resistance after all.

trixing, 2021-09-09:
Never mind, I've mixed up A/B on the RS485 and what I was seeing is the reflection on the line. Duh. Works now.

closed 2021-09-09 by trixing