jblance/mpp-solar #150: PI30MAX protocol

borisgit86, 2021-09-05:
Hi,
i try to use commands QEY, QEM, QED, QLY, QLM, QLD the inverter return NAK.
QET and QLT works.

Do you check and help me?

jblance, 2021-09-05:
Please post the command line and output
At a guess, are you supplying the year etc, ie QEY2021 etc?

borisgit86, 2021-09-05:
this is the command line:
pi@raspberrypi:~ $ sudo mpp-solar -p /dev/hidraw0 -P PI30MAX -c QEY
Parameter                     	Value           Unit
error                         	NAK

jblance, 2021-09-06:
> this is the command line:
> Parameter Value Unit
> error NAK

you need to supply the year for QEY, eg
sudo mpp-solar -p /dev/hidraw0 -P PI30MAX -c QEY2021

borisgit86, 2021-09-06:
I try the command line as in your example...
pi@raspberrypi:~ $ sudo mpp-solar -p /dev/hidraw0 -P PI30MAX -c QEY2021
Parameter                     	Value           Unit
error                         	NAK            	    

but it don't work...

jblance, 2021-09-06:
> I try the command line as in your example...
> Parameter Value Unit
> error NAK
>
> but it don't work...

Can younpost the debug output of both that command and the QET that works

borisgit86, 2021-09-06:
this is the command QEY that don't work:

pi@raspberrypi:~ $ sudo mpp-solar -p /dev/hidraw0 -P PI30MAX -c QEY2021 -D
2021-09-07 00:32:02,902:INFO:__init__:main@236: Solar Device Command Utility, version: 0.7.78, jk02 changes - add first attempt at setter command
2021-09-07 00:32:02,903:INFO:__init__:main@340: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI30MAX"
2021-09-07 00:32:02,906:DEBUG:__init__:main@343: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2021-09-07 00:32:02,906:DEBUG:device:__init__@31: __init__ args ()
2021-09-07 00:32:02,906:DEBUG:device:__init__@32: __init__ kwargs {'name': 'unnamed', 'port': '/dev/hidraw0', 'protocol': 'PI30MAX', 'baud': 2400, 'porttype': None, 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
2021-09-07 00:32:02,906:DEBUG:device:get_port_type@59: port matches hidraw
2021-09-07 00:32:02,907:INFO:device:set_port@142: Using hidrawio for communications
2021-09-07 00:32:02,909:DEBUG:device:set_protocol@96: Protocol PI30MAX
2021-09-07 00:32:02,961:DEBUG:device:__init__@36: __init__ name unnamed, port <mppsolar.io.hidrawio.HIDRawIO object at 0xb6657ff0>, protocol <mppsolar.protocols.pi30max.pi30max object at 0xb6657c90>
2021-09-07 00:32:02,962:DEBUG:__init__:main@390: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0xb6657fb0>, 'QEY2021', 'QEY2021', 'screen', None, None)]
2021-09-07 00:32:02,962:INFO:__init__:main@395: Looping 1 commands
2021-09-07 00:32:02,962:INFO:__init__:main@406: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.hidrawio.HIDRawIO object at 0xb6657ff0>, protocol: <mppsolar.protocols.pi30max.pi30max object at 0xb6657c90> for command: QEY2021, tag: QEY2021, outputs: screen
2021-09-07 00:32:02,963:INFO:device:run_command@270: Running command QEY2021
2021-09-07 00:32:02,963:INFO:abstractprotocol:get_full_command@32: Using protocol b'PI30MAX' with 30 commands
2021-09-07 00:32:02,963:DEBUG:protocol_helpers:crcPI@313: Calculating CRC for b'QEY2021'
2021-09-07 00:32:02,963:DEBUG:protocol_helpers:crcPI@363: Generated CRC 0x87 0x08 0x8708
2021-09-07 00:32:02,964:DEBUG:abstractprotocol:get_full_command@39: full command: b'QEY2021\x87\x08\r'
2021-09-07 00:32:02,964:INFO:device:run_command@296: full command b'QEY2021\x87\x08\r' for command QEY2021
2021-09-07 00:32:02,964:DEBUG:abstractprotocol:get_command_defn@43: Processing command 'QEY2021'
2021-09-07 00:32:02,964:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: QPGS
2021-09-07 00:32:02,965:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: QEY
2021-09-07 00:32:02,966:DEBUG:abstractprotocol:get_command_defn@54: Matched: QEY2021 to: QEY value: 2021
2021-09-07 00:32:02,967:DEBUG:hidrawio:send_and_receive@29: length of to_send: 10
2021-09-07 00:32:02,967:DEBUG:hidrawio:send_and_receive@40: 2 chunk send
2021-09-07 00:32:04,073:DEBUG:hidrawio:send_and_receive@69: usb response was: b'(NAKss\r'
2021-09-07 00:32:04,075:DEBUG:device:run_command@315: Send and Receive Response b'(NAKss\r'
2021-09-07 00:32:04,076:INFO:abstractprotocol:decode@176: response passed to decode: b'(NAKss\r'
2021-09-07 00:32:04,076:INFO:abstractprotocol:decode@180: NAK
2021-09-07 00:32:04,077:INFO:device:run_command@334: Decoded response {'ERROR': ['NAK', '']}
2021-09-07 00:32:04,077:DEBUG:__init__:main@409: results: {'ERROR': ['NAK', '']}
2021-09-07 00:32:04,077:INFO:__init__:get_outputs@26: attempting to create output processor: screen
2021-09-07 00:32:04,082:DEBUG:screen:__init__@16: processor.screen __init__ kwargs {}
2021-09-07 00:32:04,083:DEBUG:__init__:main@415: Using output filter: None
2021-09-07 00:32:04,083:INFO:screen:output@19: Using output processor: screen
2021-09-07 00:32:04,083:DEBUG:screen:output@20: kwargs {'data': {'ERROR': ['NAK', '']}, 'tag': 'QEY2021', 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None, 'mqtt_topic': 'mpp-solar', 'filter': None, 'excl_filter': None, 'keep_case': False}
Parameter                     	Value           Unit
error                         	NAK            	    
2021-09-07 00:32:04,084:DEBUG:__init__:main@435: Not daemon, so not looping

This is the command QET that it works perfectly:

pi@raspberrypi:~ $ sudo mpp-solar -p /dev/hidraw0 -P PI30MAX -c QET -D
2021-09-07 00:33:21,258:INFO:__init__:main@236: Solar Device Command Utility, version: 0.7.78, jk02 changes - add first attempt at setter command
2021-09-07 00:33:21,259:INFO:__init__:main@340: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI30MAX"
2021-09-07 00:33:21,263:DEBUG:__init__:main@343: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2021-09-07 00:33:21,263:DEBUG:device:__init__@31: __init__ args ()
2021-09-07 00:33:21,263:DEBUG:device:__init__@32: __init__ kwargs {'name': 'unnamed', 'port': '/dev/hidraw0', 'protocol': 'PI30MAX', 'baud': 2400, 'porttype': None, 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
2021-09-07 00:33:21,263:DEBUG:device:get_port_type@59: port matches hidraw
2021-09-07 00:33:21,263:INFO:device:set_port@142: Using hidrawio for communications
2021-09-07 00:33:21,266:DEBUG:device:set_protocol@96: Protocol PI30MAX
[... cut ...]
2021-09-07 00:33:22,318:DEBUG:__init__:main@435: Not daemon, so not looping

I have implemented PI30MAX in my Home Assistant Platform trough MQTT and crontab (refresh values every minutes).
I was unable to set it as service, i tried many times but nothing. 
I read all previous issues on systemd and i tried a lot of suggest solution but nothing.
In this moment i need to use the commands in this issue to implement the value in my energy platform, after if you want can help me to configure it as service.
I would be really happy.
Thank you very much for all!!!

jblance, 2021-09-07:
the only thing I see is that it is breaking the command into 2 to send - though Im sure this worked for someone
can you post the output of a `QT` command
what inverter model do you have?

borisgit86, 2021-09-08:
pi@raspberrypi:~ $ sudo mpp-solar -p /dev/hidraw0 -P PI30MAX -c QT
Command: QT - Device Time Inquiry
------------------------------------------------------------
Parameter                     	Value           Unit
device_time                   	20210908105752 	    

The data is set correctly.
I test QED, QEM, QLD, QLM supplying they by timestamp, for example QED20210908 and works perfectly.
QEY and QLY don't work.

Another command that don't work is QPIGS2, this command is most important because it returns the values of 2nd Panel String and them are different from those of QPGS0.

pi@raspberrypi:~ $ sudo mpp-solar -p /dev/hidraw0 -P PI30MAX -c QPIGS2
Parameter                     	Value           Unit
error                         	NAK           

Inverter Model is "Voltronic / Axpert 7200MAX 48V"

jblance, 2021-09-08:
I wonder if that inverter has slightly different commands?

jblance, 2021-09-08:
Does QPG2S0 work?

borisgit86, 2021-09-08:
I think that there aren't different command,  anyway QPGS0 and QPIGS works.

borisgit86, 2021-09-09:
hi, 
do you have a news on PI30MAX protocol?

jblance, 2021-09-09:
Did you try  QPG2S0 ?

Not sure what to do next to be honest - the commands have been said to
work. I have a 8048MAX that uses this command set that will be installed
soon, so will be able to test more then

borisgit86, 2021-09-09:
Ok, thanks for all...
keep me updated...

VictorBucha, 2021-09-30:
It seems that QEY sometimes produces output after one year of operation of inverter, this might depend on firmware. Mine MAX8048 reverts with a Wh number (comissioned this year), but the other 3P hybrid declares 0 for 2021 (also comissioned in 2021).

jblance, 2021-09-30:
I have just commissioned my 8048MAX, QEY and QPIGS2 both return data (though I dont have PV2 connected yet)

```
i@batteryshed:~ $ mpp-solar -p /dev/ttyUSB1 -P PI30MAX -c QPIGS2
Command: QPIGS2 - General Status Parameters inquiry 2
------------------------------------------------------------
Parameter                     	Value           Unit
pv2_input_current             	0.0            	A   
pv2_input_voltage             	0.0            	V   
pv2_charging_power            	0              	W   
unknown_value_in_response_3   	               	    

pi@batteryshed:~ $ mpp-solar -p /dev/ttyUSB1 -P PI30MAX -c QEY2021
Command: QEY2021 - Yearly PV Generated Energy Inquiry
------------------------------------------------------------
Parameter                     	Value           Unit
pv_generated_energy_for_year  	21700          	Wh  
year                          	2021
```

Not really sure how to troubleshoot your setup further - maybe compare firmware versions of the inverters? I do have a newer inverter to install but that wont happen until I build another solar array

audio-catalyst, 2021-10-16:
i am running into the samen issues a boris above.
I also run a voltronic 8000max, running firmware 00046.04

it is communicating :

sudo mpp-solar -p /dev/hidraw0 --getstatus
Command: Q1 - Q1 query
------------------------------------------------------------
Parameter                       Value           Unit
ac_input_voltage                0.0             V
ac_input_frequency              0.0             Hz
ac_output_voltage               230.0           V
ac_output_frequency             50.0            Hz
ac_output_apparent_power        0               VA
ac_output_active_power          0               W
ac_output_load                  0               %
bus_voltage                     370             V
battery_voltage                 0.0             V
battery_charging_current        2               A
battery_capacity                0               %
inverter_heat_sink_temperature  27              °C
pv_input_current_for_battery    0.2             A
pv_input_voltage                266.3           V
battery_voltage_from_scc        0.0             V
battery_discharge_current       0               A
is_sbu_priority_version_added   0               bool
is_configuration_changed        0               bool
is_scc_firmware_updated         0               bool
is_load_on                      1               bool
is_battery_voltage_to_steady_while_charging     0               bool
is_charging_on                  0               bool
is_scc_charging_on              0               bool
is_ac_charging_on               0               bool
rsv1                            0               A
rsv2                            0               A
pv_input_power                  73              W
is_charging_to_float            0               bool
is_switched_on                  1               bool
is_reserved                     0               bool
time_until_the_end_of_absorb_charging   1               sec
time_until_the_end_of_float_charging    20030           sec
scc_flag                        SCC is powered and communicating
allowscconflag                  00
chargeaveragecurrent            00
scc_pwm_temperature             24              °C
inverter_temperature            27              °C
battery_temperature             20              °C
transformer_temperature         21              °C
gpio13                          2
fan_lock_status                 Not locked
not_used                        000
fan_pwm_speed                   30              Percent
scc_charge_power                72              W
parallel_warning??              0000
sync_frequency                  0.0
inverter_charge_status          nocharging
unknown_value_in_response_17    0
unknown_value_in_response_18    060
unknown_value_in_response_19    030
unknown_value_in_response_20    060
unknown_value_in_response_21    030
unknown_value_in_response_22    58.40
unknown_value_in_response_23    000
unknown_value_in_response_24    120
unknown_value_in_response_25    0
unknown_value_in_response_26    0000

pi@serial-test-raspberry:~ $ sudo mpp-solar -p /dev/hidraw0 -P PI30MAX
Command: QPI - Protocol ID inquiry
------------------------------------------------------------
Parameter                       Value           Unit
protocol_id                     PI30
pi@serial-test-raspberry:~ $ sudo mpp-solar -p /dev/hidraw0 -P PI30MAX -c QT
Command: QT - Device Time Inquiry
------------------------------------------------------------
Parameter                       Value           Unit
device_time                     20220106092550
pi@serial-test-raspberry:~ $ sudo mpp-solar -p /dev/hidraw0 -P PI30MAX -c QPG2S0
Parameter                       Value           Unit
error                           NAK
pi@serial-test-raspberry:~ $ sudo mpp-solar -p /dev/hidraw0 -P PI30MAX -c QPQPIGS
Parameter                       Value           Unit
error                           NAK
pi@serial-test-raspberry:~ $ sudo mpp-solar -p /dev/hidraw0 -P PI30MAX -c QEY2021
Parameter                       Value           Unit
error                           NAK
pi@serial-test-raspberry:~ $ sudo mpp-solar -p /dev/hidraw0 -P PI30MAX -c QPIGS2
Parameter                       Value           Unit
error                           NAK

but as show above , a lot of NAK

gasparekk, 2022-05-22:
I make a usb capture for model MKS2-8000, if it is helpful.[https://drive.google.com/file/d/1IV4FOw8sAXEPSO6bvCOgRsrGEvZqVQnw/view?usp=sharing](url)

frederickjh, 2022-07-25:
I hope this issue is the right place to give this information as it is titled PI30MAX protocol. I have a PIP 8048MAX with the LED light bar. A number of the commands in the PI30MAX protocol seem to be missing, while others that are the same commands as in the PI30 protocol are only in the PI30 protocol so I need to switch back and forth between the two protocols. 

PE and PD are two that are only in the PI30 protocol but are also listed in the PI30MAX documentation, that I have found but there could be more.

All the commands starting PLED are missing from the PI30MAX protocol. The documentation lists the following: PLEDE, PLEDS, PLEDM, PLEDB, PLEDT, and PLEDC.

The interesting thing is that when I run `mpp-solar -c QPI` to do a Protocol ID inquiry it returns PI30 not PI30MAX. However the QLED command will not work with the PI30 protocol.

I figured this out as I would like to disable the LED bar at night to get rid of light pollution,along with turning off the backlight. I was able to find commands to do the later but without the PLEDE command I cannot do the former.

Thanks for any help you can give!

jblance, 2022-10-11:
Yes I haven't implemented all the commands for the pi30max yet. (Tbh I've
been a bit slack on updates lately as I have a bunch of work on)

Pi30max isn't a real protocol version but just a name I used as there are
different versions of pi30 (yay for standards)

The later versions of mppsolar should allow you to run an unknown command

Can you run the led disable command with debug and post the output

frederickjh, 2022-10-13:
Below are the debug outputs from running the PLEDE command for three protocols. The protocol documentation ([PI30MAX.Communication.Protocol20210217.pdf](https://github.com/jblance/mpp-solar/blob/master/docs/protocols/PI30MAX.Communication.Protocol20210217.pdf)) says this:

> 3.32 PLEDE<n><cr>: Enable/disable LED function
Computer: PLEDE<n><cr>
UPS: (ACK<cr> if UPS accepts this command, otherwise, responds (NAK<cr>
n: 0 means disable; 1 means enable

I wanted to disable so I used a 0 (zero) after the command *PLEDE0*.

<details><summary>CLICK ME to see collapsed text of DEBUG  for PLEDE for protocol PI30</summary>
<p>

```bash
mpp-solar -D -p /dev/mppsolar/direct/0 -P PI30 -b 2400 -c PLEDE0                                                   [Wed Oct 12 11:01:08 PM]
2022-10-12 23:06:08,826:INFO:__init__:main@218: Solar Device Command Utility, version: 0.12.5, tweak jkbleio for getInfo and to return first 300 bytes
2022-10-12 23:06:08,826:DEBUG:__init__:main@232: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f9a2d2caca0>, _isConnected=False)
2022-10-12 23:06:08,827:DEBUG:__init__:main@234: udp port 5555
2022-10-12 23:06:08,827:DEBUG:__init__:main@237: Using Mongo None with mppsolar
2022-10-12 23:06:08,827:INFO:__init__:main@337: Creating device "unnamed" (type: "mppsolar") on port "/dev/mppsolar/direct/0 (porttype=None)" using protocol "PI30"
2022-10-12 23:06:08,828:DEBUG:__init__:main@341: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2022-10-12 23:06:08,828:DEBUG:device:__init__@33: __init__ args ()
2022-10-12 23:06:08,828:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': '/dev/mppsolar/direct/0', 'protocol': 'PI30', 'baud': 2400, 'porttype': None, 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f9a2d2caca0>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2022-10-12 23:06:08,828:DEBUG:__init__:get_port_type@41: port matches mppsolar
2022-10-12 23:06:08,828:INFO:__init__:get_port@89: Using hidrawio for communications
2022-10-12 23:06:08,829:DEBUG:__init__:get_protocol@12: Protocol PI30
2022-10-12 23:06:08,830:DEBUG:device:__init__@38: __init__ name unnamed, port <mppsolar.inout.hidrawio.HIDRawIO object at 0x7f9a2d247f70>, protocol <mppsolar.protocols.pi30.pi30 object at 0x7f9a2d247f40>
2022-10-12 23:06:08,830:DEBUG:__init__:main@390: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x7f9a2d247fa0>, 'PLEDE0', 'PLEDE0', 'screen', None, None)]
2022-10-12 23:06:08,830:INFO:__init__:main@395: Looping 1 commands
2022-10-12 23:06:08,830:INFO:__init__:main@405: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.hidrawio.HIDRawIO object at 0x7f9a2d247f70>, protocol: <mppsolar.protocols.pi30.pi30 object at 0x7f9a2d247f40> for command: PLEDE0, tag: PLEDE0, outputs: screen
2022-10-12 23:06:08,830:INFO:device:run_command@274: Running command PLEDE0
2022-10-12 23:06:08,830:INFO:abstractprotocol:get_full_command@50: Using protocol b'PI30' with 40 commands
2022-10-12 23:06:08,830:DEBUG:protocol_helpers:crcPI@313: Calculating CRC for b'PLEDE0'
2022-10-12 23:06:08,830:DEBUG:protocol_helpers:crcPI@363: Generated CRC 0x3d 0x09 0x3d09
2022-10-12 23:06:08,830:DEBUG:abstractprotocol:get_full_command@56: full command: b'PLEDE0=\t\r'
2022-10-12 23:06:08,830:INFO:device:run_command@303: full command b'PLEDE0=\t\r' for command PLEDE0
2022-10-12 23:06:08,830:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'PLEDE0'
2022-10-12 23:06:08,830:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: F
2022-10-12 23:06:08,830:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: MCHGC
2022-10-12 23:06:08,831:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: MNCHGC
2022-10-12 23:06:08,831:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: MUCHGC
2022-10-12 23:06:08,831:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PBCV
2022-10-12 23:06:08,831:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PBDV
2022-10-12 23:06:08,831:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PBFT
2022-10-12 23:06:08,831:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PBT
2022-10-12 23:06:08,832:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PCP
2022-10-12 23:06:08,832:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PCVV
2022-10-12 23:06:08,832:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PE
2022-10-12 23:06:08,832:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PD
2022-10-12 23:06:08,832:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PGR
2022-10-12 23:06:08,832:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: POP
2022-10-12 23:06:08,832:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: POPLG
2022-10-12 23:06:08,833:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: POPM
2022-10-12 23:06:08,833:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PPCP
2022-10-12 23:06:08,833:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PPVOKC
[... cut ...]
2022-10-12 23:06:55,554:DEBUG:screen:output@20: kwargs {'data': {'ERROR': ['NAK', '']}, 'tag': 'PLEDE0', 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7efcfd0d8ca0>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar', 'filter': None, 'excl_filter': None, 'keep_case': False}
Parameter                     	Value           Unit
error                         	NAK            	    
2022-10-12 23:06:55,554:DEBUG:__init__:main@438: Not daemon, so not looping
```

</p>
</details>

frederickjh, 2022-10-13:
```bash
mpp-solar -v                                                                                                       [Wed Oct 12 11:21:08 PM]
Solar Device Command Utility, version: 0.12.5, tweak jkbleio for getInfo and to return first 300 bytes
```

eliafino, 2022-10-19:
Hi, also my  Axpert MAX II 10kW return NAK to QPIGS2. Maybe have only 1 MPPT?

jblance, 2022-10-19:
i think qpigs2 is broken if you have a direct usb connection

i need to figure out how to fix (and test)

do you have 2 strings of solar? if so qpigs2 is what is needed for the data
for the second string

eliafino, 2022-10-20:
> i think qpigs2 is broken if you have a direct usb connection i need to figure out how to fix (and test) do you have 2 strings of solar? if so qpigs2 is what is needed for the data for the second string
> […](#)

I can try with rs232 connection. I don't have the second string yet, I'm getting ready to have

eliafino, 2022-11-07:
Finally solved!

QPIGS2h

frederickjh, 2022-12-04:
> Yes I haven't implemented all the commands for the pi30max yet. (Tbh I've been a bit slack on updates lately as I have a bunch of work on) Pi30max isn't a real protocol version but just a name I [...]

@jblance is there documentation anywhere on how to do development work for this project?  I am unsure how to build  mpp-solar locally to test. If I knew that I might take a crack at putting pull request together. I want to be able to test any changes  I make before submitting a pull request.

jblance, 2022-12-05:
Not really any documentation
closest is [here](https://github.com/jblance/mpp-solar/blob/master/docs/installation.md#venv-install---recommended-if-testing-new-features--release) but youll want to clone the repo locally, make changes and install in dev mode from the local source

jblance, 2022-12-06:
made some changes to the git version (v0.14.3+) that should include pi30 commands into pi30max and pi30mst
just need to add commands that are missing - if you let me know which ones are your priority i can try to get them in

frederickjh, 2022-12-09:
Thanks @jblance for offering to try and implement the missing commands.

PLEDE along with PLEDS, PLEDM, PLEDB, PLEDT, and PLEDC, as I find when you disable the LEDs it many times loses the settings for them and goes back to the defaults. I want to be able to automate turning off the LEDs at night and on in the day and restore the settings for the LEDs that I want.

Otherwise the commands I am looking for seem to be implemented.

jblance, 2022-12-10:
v 0.14.7 has first go at these commands (untested)
let me know how they go

frederickjh, 2022-12-10:
Thanks I will test them.

frederickjh, 2022-12-10:
All the new commands are returning an error and a  NAK response.

```
$ /usr/local/bin/mpp-solar -p /dev/mppsolar/direct/0 -P PI30MAX -b 2400 -c PLEDE1

Parameter                     	Value           Unit
error                         	NAK            	    

$ /usr/local/bin/mpp-solar -p /dev/mppsolar/direct/0 -P PI30MAX -b 2400 -c PLEDE0 

Parameter                     	Value           Unit
error                         	NAK            	    

$ /usr/local/bin/mpp-solar -p /dev/mppsolar/direct/0 -P PI30MAX -b 2400 -c PLEDS0

Parameter                     	Value           Unit
error                         	NAK            	    

$ /usr/local/bin/mpp-solar -p /dev/mppsolar/direct/0 -P PI30MAX -b 2400 -c PLEDS1

Parameter                     	Value           Unit
error                         	NAK            	    

$ /usr/local/bin/mpp-solar -p /dev/mppsolar/direct/0 -P PI30MAX -b 2400 -c PLEDS2

Parameter                     	Value           Unit
error                         	NAK            	    

$ /usr/local/bin/mpp-solar -p /dev/mppsolar/direct/0 -P PI30MAX -b 2400 -c PLEDM0

Parameter                     	Value           Unit
error                         	NAK            	    

$ /usr/local/bin/mpp-solar -p /dev/mppsolar/direct/0 -P PI30MAX -b 2400 -c PLEDM2

Parameter                     	Value           Unit
error                         	NAK            	    

$ /usr/local/bin/mpp-solar -p /dev/mppsolar/direct/0 -P PI30MAX -b 2400 -c PLEDM3

Parameter                     	Value           Unit
error                         	NAK            	    

$ /usr/local/bin/mpp-solar -p /dev/mppsolar/direct/0 -P PI30MAX -b 2400 -c PLEDB1

Parameter                     	Value           Unit
error                         	NAK            	    

$ /usr/local/bin/mpp-solar -p /dev/mppsolar/direct/0 -P PI30MAX -b 2400 -c PLEDB5

Parameter                     	Value           Unit
error                         	NAK            	    

$ /usr/local/bin/mpp-solar -p /dev/mppsolar/direct/0 -P PI30MAX -b 2400 -c PLEDB9  

Parameter                     	Value           Unit
error                         	NAK            	    

$ /usr/local/bin/mpp-solar -p /dev/mppsolar/direct/0 -P PI30MAX -b 2400 -c PLEDT2

Parameter                     	Value           Unit
error                         	NAK            	    

$ /usr/local/bin/mpp-solar -p /dev/mppsolar/direct/0 -P PI30MAX -b 2400 -c PLEDT3

Parameter                     	Value           Unit
error                         	NAK            	    

$ /usr/local/bin/mpp-solar -p /dev/mppsolar/direct/0 -P PI30MAX -b 2400 -c QLED

Command: QLED - LED Status Parameters Inquiry
------------------------------------------------------------
Parameter                     	Value           Unit
led_enabled                   	Enabled        	    
led_speed                     	Medium         	    
led_effect                    	Solid          	    
led_brightness                	5              	    
led_number_of_colors          	2              	    
rgb                           	255000147255064000	    

$ /usr/local/bin/mpp-solar -p /dev/mppsolar/direct/0 -P PI30MAX -b 2400 -c PLEDC1255000147

Parameter                     	Value           Unit
error                         	NAK            	    

$ /usr/local/bin/mpp-solar -p /dev/mppsolar/direct/0 -P PI30MAX -b 2400 -c PLEDC3255064000 

Parameter                     	Value           Unit
error                         	NAK            	    

$ /usr/local/bin/mpp-solar -v
Solar Device Command Utility, version: 0.14.7

$ /usr/local/bin/mpp-solar -p /dev/mppsolar/direct/0 -P PI30MAX -b 2400 --getstatus

Processing DEFAULT type responses
Command: QPIGS - General Status Parameters inquiry
------------------------------------------------------------
Parameter                     	Value           Unit
ac_input_voltage              	0.0            	V   
ac_input_frequency            	0.0            	Hz  
ac_output_voltage             	229.9          	V   
ac_output_frequency           	60.0           	Hz  
ac_output_apparent_power      	230            	VA  
ac_output_active_power        	207            	W   
ac_output_load                	2              	%   
bus_voltage                   	384            	V   
battery_voltage               	52.3           	V   
battery_charging_current      	0              	A   
battery_capacity              	81             	%   
inverter_heat_sink_temperature	33             	°C  
pv1_input_current             	0.0            	A   
pv1_input_voltage             	0.0            	V   
battery_voltage_from_scc      	0.0            	V   
battery_discharge_current     	4              	A   
is_sbu_priority_version_added 	0              	bool
is_configuration_changed      	0              	bool
is_scc_firmware_updated       	0              	bool
is_load_on                    	1              	bool
is_battery_voltage_to_steady_while_charging	0              	bool
is_charging_on                	0              	bool
is_scc_charging_on            	0              	bool
is_ac_charging_on             	0              	bool
battery_voltage_offset_for_fans_on	0              	10mV
eeprom_version                	0              	    
pv1_charging_power            	0              	W   
is_charging_to_float          	0              	bool
is_switched_on                	1              	bool
is_dustproof_installed        	0              	bool
error                         	NAK  
```

jblance, 2022-12-10:
Can you run one with debug on

frederickjh, 2022-12-10:
Sorry, I should have thought of that.

<details><summary>CLICK ME to see collapsed debug text</summary>
<p>

```bash
$ /usr/local/bin/mpp-solar -p /dev/mppsolar/direct/0 -P PI30MAX -b 2400 --debug -c PLEDE1                                  4018ms  [Fri Dec  9 09:19:37 PM]
2022-12-09 21:34:12,265:INFO:__init__:main@218: Solar Device Command Utility, version: 0.14.7
2022-12-09 21:34:12,265:DEBUG:__init__:main@238: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f438983ae80>, _isConnected=False)
2022-12-09 21:34:12,265:DEBUG:__init__:main@240: udp port 5555
2022-12-09 21:34:12,265:DEBUG:__init__:main@242: Using Postgres None
2022-12-09 21:34:12,265:DEBUG:__init__:main@245: Using Mongo None with mppsolar
2022-12-09 21:34:12,265:INFO:__init__:main@347: Creating device "unnamed" (type: "mppsolar") on port "/dev/mppsolar/direct/0 (porttype=None)" using protocol "PI30MAX"
2022-12-09 21:34:12,266:DEBUG:__init__:main@351: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2022-12-09 21:34:12,266:DEBUG:device:__init__@33: __init__ args ()
2022-12-09 21:34:12,266:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': '/dev/mppsolar/direct/0', 'protocol': 'PI30MAX', 'baud': 2400, 'porttype': None, 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f438983ae80>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2022-12-09 21:34:12,266:DEBUG:__init__:get_port_type@42: port matches mppsolar
2022-12-09 21:34:12,267:INFO:__init__:get_port@92: Using hidrawio for communications
2022-12-09 21:34:12,267:DEBUG:__init__:get_protocol@13: Protocol PI30MAX
2022-12-09 21:34:12,269:INFO:pi30max:__init__@967: Using protocol b'PI30MAX' with 61 commands
2022-12-09 21:34:12,269:DEBUG:device:__init__@38: __init__ name unnamed, port <mppsolar.inout.hidrawio.HIDRawIO object at 0x7f43890b7eb0>, protocol PI30 protocol handler for LV6048MAX and similar inverters
2022-12-09 21:34:12,269:DEBUG:__init__:main@399: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x7f43890b7f40>, 'PLEDE1', 'PLEDE1', 'screen', None, None)]
2022-12-09 21:34:12,269:INFO:__init__:main@404: Looping 1 commands
2022-12-09 21:34:12,269:INFO:__init__:main@414: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.hidrawio.HIDRawIO object at 0x7f43890b7eb0>, protocol: PI30 protocol handler for LV6048MAX and similar inverters for command: PLEDE1, tag: PLEDE1, outputs: screen
2022-12-09 21:34:12,269:INFO:device:run_command@274: Running command PLEDE1
2022-12-09 21:34:12,269:INFO:abstractprotocol:get_full_command@50: Using protocol b'PI30MAX' with 61 commands
2022-12-09 21:34:12,269:DEBUG:protocol_helpers:crcPI@313: Calculating CRC for b'PLEDE1'
2022-12-09 21:34:12,269:DEBUG:protocol_helpers:crcPI@363: Generated CRC 0x2d 0x29 0x2d29
2022-12-09 21:34:12,269:DEBUG:abstractprotocol:get_full_command@56: full command: b'PLEDE1-)\r'
2022-12-09 21:34:12,269:INFO:device:run_command@303: full command b'PLEDE1-)\r' for command PLEDE1
2022-12-09 21:34:12,269:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'PLEDE1'
2022-12-09 21:34:12,269:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: F
2022-12-09 21:34:12,269:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: MCHGC
2022-12-09 21:34:12,269:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: MNCHGC
2022-12-09 21:34:12,269:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: MUCHGC
2022-12-09 21:34:12,270:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PBCV
2022-12-09 21:34:12,270:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PBDV
2022-12-09 21:34:12,270:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PBFT
2022-12-09 21:34:12,270:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PBT
2022-12-09 21:34:12,270:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PCP
2022-12-09 21:34:12,270:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PCVV
2022-12-09 21:34:12,270:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PE
2022-12-09 21:34:12,270:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PD
2022-12-09 21:34:12,271:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PGR
2022-12-09 21:34:12,271:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: POP
2022-12-09 21:34:12,271:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: POPLG
2022-12-09 21:34:12,271:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: POPM
2022-12-09 21:34:12,271:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PPCP
2022-12-09 21:34:12,271:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PPVOKC
2022-12-09 21:34:12,271:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PSDV
2022-12-09 21:34:12,271:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PSPB
2022-12-09 21:34:12,272:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PBATCD
2022-12-09 21:34:12,272:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: DAT
[... cut ...]
2022-12-09 21:34:13,443:DEBUG:screen:output@20: kwargs {'data': {'ERROR': ['NAK', '']}, 'tag': 'PLEDE1', 'name': 'unnamed', 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f438983ae80>, _isConnected=False), 'udp_port': '5555', 'postgres_url': None, 'mongo_url': None, 'mongo_db': 'mppsolar', 'filter': None, 'excl_filter': None, 'keep_case': False}
Parameter                     	Value           Unit
error                         	NAK            	    
2022-12-09 21:34:13,443:DEBUG:__init__:main@449: Not daemon, so not looping
```

</p>
</details>

jblance, 2022-12-10:
> 2022-12-09 ***@***.***: Solar Device Command Utility, version: 0.14.7
> 2022-12-09 ***@***.***: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f438983ae80>, _isConnected=False)
> 2022-12-09 ***@***.***: udp port 5555
> 2022-12-09 ***@***.***: Using Postgres None
> 2022-12-09 ***@***.***: Using Mongo None with mppsolar
> 2022-12-09 ***@***.***: Creating device "unnamed" (type: "mppsolar") on port "/dev/mppsolar/direct/0 (porttype=None)" using protocol "PI30MAX"
> 2022-12-09 ***@***.***: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
> 2022-12-09 ***@***.***: __init__ args ()
> 2022-12-09 ***@***.***: __init__ kwargs {'name': 'unnamed', 'port': '/dev/mppsolar/direct/0', 'protocol': 'PI30MAX', 'baud': 2400, 'porttype': None, 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f438983ae80>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
> 2022-12-09 ***@***.***: port matches mppsolar
> 2022-12-09 ***@***.***: Using hidrawio for communications
> 2022-12-09 ***@***.***: Protocol PI30MAX
> 2022-12-09 ***@***.***: Using protocol b'PI30MAX' with 61 commands
> 2022-12-09 ***@***.***: __init__ name unnamed, port <mppsolar.inout.hidrawio.HIDRawIO object at 0x7f43890b7eb0>, protocol PI30 protocol handler for LV6048MAX and similar inverters
> 2022-12-09 ***@***.***: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x7f43890b7f40>, 'PLEDE1', 'PLEDE1', 'screen', None, None)]
> 2022-12-09 ***@***.***: Looping 1 commands
> [... cut ...]
> 2022-12-09 ***@***.***: kwargs {'data': {'ERROR': ['NAK', '']}, 'tag': 'PLEDE1', 'name': 'unnamed', 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f438983ae80>, _isConnected=False), 'udp_port': '5555', 'postgres_url': None, 'mongo_url': None, 'mongo_db': 'mppsolar', 'filter': None, 'excl_filter': None, 'keep_case': False}
> 2022-12-09 ***@***.***: Not daemon, so not looping

Try with `--porttype hidfull`

frederickjh, 2022-12-10:
<details><summary>CLICK ME to see collapsed debug text</summary>
<p>

```bash
$ /usr/local/bin/mpp-solar -p /dev/mppsolar/direct/0 -P PI30MAX -b 2400 --debug --porttype hidfull -c PLEDE1               1259ms  [Fri Dec  9 09:57:47 PM]
2022-12-09 21:57:57,857:INFO:__init__:main@218: Solar Device Command Utility, version: 0.14.7
2022-12-09 21:57:57,857:DEBUG:__init__:main@238: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f64e6cf0e80>, _isConnected=False)
2022-12-09 21:57:57,857:DEBUG:__init__:main@240: udp port 5555
2022-12-09 21:57:57,857:DEBUG:__init__:main@242: Using Postgres None
2022-12-09 21:57:57,857:DEBUG:__init__:main@245: Using Mongo None with mppsolar
2022-12-09 21:57:57,857:INFO:__init__:main@347: Creating device "unnamed" (type: "mppsolar") on port "/dev/mppsolar/direct/0 (porttype=hidfull)" using protocol "PI30MAX"
2022-12-09 21:57:57,859:DEBUG:__init__:main@351: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2022-12-09 21:57:57,859:DEBUG:device:__init__@33: __init__ args ()
2022-12-09 21:57:57,859:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': '/dev/mppsolar/direct/0', 'protocol': 'PI30MAX', 'baud': 2400, 'porttype': 'hidfull', 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f64e6cf0e80>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2022-12-09 21:57:57,859:INFO:__init__:get_port@81: Port overide - using port 'hidfull'
2022-12-09 21:57:57,859:DEBUG:__init__:get_port_type@45: port matches hidfull
2022-12-09 21:57:57,859:INFO:__init__:get_port@97: Using hidfullio for communications
2022-12-09 21:57:57,859:DEBUG:__init__:get_protocol@13: Protocol PI30MAX
2022-12-09 21:57:57,861:INFO:pi30max:__init__@967: Using protocol b'PI30MAX' with 61 commands
2022-12-09 21:57:57,861:DEBUG:device:__init__@38: __init__ name unnamed, port <mppsolar.inout.hidfullio.HIDFullIO object at 0x7f64e656deb0>, protocol PI30 protocol handler for LV6048MAX and similar inverters
2022-12-09 21:57:57,861:DEBUG:__init__:main@399: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x7f64e656df40>, 'PLEDE1', 'PLEDE1', 'screen', None, None)]
2022-12-09 21:57:57,861:INFO:__init__:main@404: Looping 1 commands
2022-12-09 21:57:57,861:INFO:__init__:main@414: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.hidfullio.HIDFullIO object at 0x7f64e656deb0>, protocol: PI30 protocol handler for LV6048MAX and similar inverters for command: PLEDE1, tag: PLEDE1, outputs: screen
2022-12-09 21:57:57,861:INFO:device:run_command@274: Running command PLEDE1
2022-12-09 21:57:57,861:INFO:abstractprotocol:get_full_command@50: Using protocol b'PI30MAX' with 61 commands
2022-12-09 21:57:57,861:DEBUG:protocol_helpers:crcPI@313: Calculating CRC for b'PLEDE1'
2022-12-09 21:57:57,861:DEBUG:protocol_helpers:crcPI@363: Generated CRC 0x2d 0x29 0x2d29
2022-12-09 21:57:57,861:DEBUG:abstractprotocol:get_full_command@56: full command: b'PLEDE1-)\r'
2022-12-09 21:57:57,861:INFO:device:run_command@303: full command b'PLEDE1-)\r' for command PLEDE1
2022-12-09 21:57:57,861:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'PLEDE1'
2022-12-09 21:57:57,861:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: F
2022-12-09 21:57:57,862:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: MCHGC
2022-12-09 21:57:57,862:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: MNCHGC
2022-12-09 21:57:57,862:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: MUCHGC
2022-12-09 21:57:57,862:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PBCV
2022-12-09 21:57:57,862:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PBDV
2022-12-09 21:57:57,862:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PBFT
2022-12-09 21:57:57,862:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PBT
2022-12-09 21:57:57,862:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PCP
2022-12-09 21:57:57,862:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PCVV
2022-12-09 21:57:57,863:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PE
2022-12-09 21:57:57,863:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PD
2022-12-09 21:57:57,863:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PGR
2022-12-09 21:57:57,863:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: POP
2022-12-09 21:57:57,863:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: POPLG
2022-12-09 21:57:57,863:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: POPM
2022-12-09 21:57:57,863:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PPCP
2022-12-09 21:57:57,863:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PPVOKC
2022-12-09 21:57:57,863:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PSDV
2022-12-09 21:57:57,864:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PSPB
2022-12-09 21:57:57,864:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PBATCD
[... cut ...]
2022-12-09 21:58:13,611:DEBUG:screen:output@20: kwargs {'data': {'ERROR': ['Response to short', '']}, 'tag': 'PLEDE1', 'name': 'unnamed', 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f64e6cf0e80>, _isConnected=False), 'udp_port': '5555', 'postgres_url': None, 'mongo_url': None, 'mongo_db': 'mppsolar', 'filter': None, 'excl_filter': None, 'keep_case': False}
Parameter                     	Value           Unit
error                         	Response to short	    
2022-12-09 21:58:13,611:DEBUG:__init__:main@449: Not daemon, so not looping
```

</p>
</details>

jblance, 2022-12-10:
> 2022-12-09 ***@***.***: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f64e6cf0e80>, _isConnected=False)
> 2022-12-09 ***@***.***: Creating device "unnamed" (type: "mppsolar") on port "/dev/mppsolar/direct/0 (porttype=hidfull)" using protocol "PI30MAX"
> 2022-12-09 ***@***.***: __init__ kwargs {'name': 'unnamed', 'port': '/dev/mppsolar/direct/0', 'protocol': 'PI30MAX', 'baud': 2400, 'porttype': 'hidfull', 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f64e6cf0e80>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
> 2022-12-09 ***@***.***: Port overide - using port 'hidfull'
> 2022-12-09 ***@***.***: port matches hidfull
> 2022-12-09 ***@***.***: Using hidfullio for communications
> 2022-12-09 ***@***.***: __init__ name unnamed, port <mppsolar.inout.hidfullio.HIDFullIO object at 0x7f64e656deb0>, protocol PI30 protocol handler for LV6048MAX and similar inverters
> 2022-12-09 ***@***.***: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x7f64e656df40>, 'PLEDE1', 'PLEDE1', 'screen', None, None)]
> 2022-12-09 ***@***.***: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.hidfullio.HIDFullIO object at 0x7f64e656deb0>, protocol: PI30 protocol handler for LV6048MAX and similar inverters for command: PLEDE1, tag: PLEDE1, outputs: screen
> 2022-12-09 ***@***.***: 1 chunk send
> 2022-12-09 ***@***.***: USB read error: [Errno 11] Resource temporarily unavailable
> 2022-12-09 ***@***.***: Response to short
> [... cut ...]
> 2022-12-09 ***@***.***: kwargs {'data': {'ERROR': ['Response to short', '']}, 'tag': 'PLEDE1', 'name': 'unnamed', 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f64e6cf0e80>, _isConnected=False), 'udp_port': '5555', 'postgres_url': None, 'mongo_url': None, 'mongo_db': 'mppsolar', 'filter': None, 'excl_filter': None, 'keep_case': False}

Hmm. Something not right there. I'll need to do some more digging

frederickjh, 2022-12-10:
Hold off on digging a bit. I notice earlier today that the inverter is accepting the commands for the LED bar but it is not functioning as the settings are. I think I need to turn it off by completely disconnecting everything. We are off-grid and it is night time here, so I will wait until tomorrow morning to reset it. I think it has done this once before.

frederickjh, 2022-12-10:
@jblance  I was able to reset the inverter this morning by powering it off by disconnecting all power sources. The LED bar is working correctly now, but unfortunately running the `PLEDE1` still returns the same output that I posted above.

frederickjh, 2023-02-03:
Hi @jblance!

The OP wrote:

> i try to use commands QEY, QEM, QED, QLY, QLM, QLD the inverter return NAK.
> QET and QLT works.

I just tested these commands on my MPPSolar PIP 8048MAX with the following results:

The following work and return results:
QET, QEM, QLT, QLM

The following do not work. Debug messages below:
 <details><summary>CLICK ME to see collapsed debug for the QEY command</summary>
<p>

```bash
$ /usr/local/bin/mpp-solar -p /dev/mppsolarusb/direct/0 -P PI30MAX -b 2400 -c QEY2023 --debug                                            
2023-02-02 23:09:02,961:INFO:__init__:main@218: Solar Device Command Utility, version: 0.14.7
2023-02-02 23:09:02,961:DEBUG:__init__:main@238: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7fcc8e38eeb0>, _isConnected=False)
2023-02-02 23:09:02,961:DEBUG:__init__:main@240: udp port 5555
2023-02-02 23:09:02,961:DEBUG:__init__:main@242: Using Postgres None
2023-02-02 23:09:02,961:DEBUG:__init__:main@245: Using Mongo None with mppsolar
2023-02-02 23:09:02,961:INFO:__init__:main@347: Creating device "unnamed" (type: "mppsolar") on port "/dev/mppsolarusb/direct/0 (porttype=None)" using protocol "PI30MAX"
2023-02-02 23:09:02,962:DEBUG:__init__:main@351: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2023-02-02 23:09:02,962:DEBUG:device:__init__@33: __init__ args ()
2023-02-02 23:09:02,962:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': '/dev/mppsolarusb/direct/0', 'protocol': 'PI30MAX', 'baud': 2400, 'porttype': None, 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7fcc8e38eeb0>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2023-02-02 23:09:02,962:DEBUG:__init__:get_port_type@42: port matches mppsolar
2023-02-02 23:09:02,962:INFO:__init__:get_port@92: Using hidrawio for communications
2023-02-02 23:09:02,963:DEBUG:__init__:get_protocol@13: Protocol PI30MAX
2023-02-02 23:09:02,965:INFO:pi30max:__init__@967: Using protocol b'PI30MAX' with 61 commands
2023-02-02 23:09:02,965:DEBUG:device:__init__@38: __init__ name unnamed, port <mppsolar.inout.hidrawio.HIDRawIO object at 0x7fcc8dc0cee0>, protocol PI30 protocol handler for LV6048MAX and similar inverters
2023-02-02 23:09:02,965:DEBUG:__init__:main@399: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x7fcc8dc0cf70>, 'QEY2023', 'QEY2023', 'screen', None, None)]
2023-02-02 23:09:02,965:INFO:__init__:main@404: Looping 1 commands
2023-02-02 23:09:02,966:INFO:__init__:main@414: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.hidrawio.HIDRawIO object at 0x7fcc8dc0cee0>, protocol: PI30 protocol handler for LV6048MAX and similar inverters for command: QEY2023, tag: QEY2023, outputs: screen
2023-02-02 23:09:02,966:INFO:device:run_command@274: Running command QEY2023
2023-02-02 23:09:02,966:INFO:abstractprotocol:get_full_command@50: Using protocol b'PI30MAX' with 61 commands
2023-02-02 23:09:02,966:DEBUG:protocol_helpers:crcPI@313: Calculating CRC for b'QEY2023'
2023-02-02 23:09:02,966:DEBUG:protocol_helpers:crcPI@363: Generated CRC 0xa7 0x4a 0xa74a
2023-02-02 23:09:02,966:DEBUG:abstractprotocol:get_full_command@56: full command: b'QEY2023\xa7J\r'
2023-02-02 23:09:02,966:INFO:device:run_command@303: full command b'QEY2023\xa7J\r' for command QEY2023
2023-02-02 23:09:02,966:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'QEY2023'
2023-02-02 23:09:02,966:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: F
2023-02-02 23:09:02,966:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: MCHGC
2023-02-02 23:09:02,966:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: MNCHGC
2023-02-02 23:09:02,967:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: MUCHGC
2023-02-02 23:09:02,967:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PBCV
2023-02-02 23:09:02,967:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PBDV
2023-02-02 23:09:02,967:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PBFT
2023-02-02 23:09:02,967:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PBT
2023-02-02 23:09:02,967:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PCP
2023-02-02 23:09:02,968:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PCVV
2023-02-02 23:09:02,968:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PE
2023-02-02 23:09:02,968:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PD
2023-02-02 23:09:02,968:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PGR
2023-02-02 23:09:02,968:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: POP
2023-02-02 23:09:02,968:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: POPLG
2023-02-02 23:09:02,968:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: POPM
2023-02-02 23:09:02,968:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PPCP
2023-02-02 23:09:02,969:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PPVOKC
2023-02-02 23:09:02,969:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PSDV
[... cut ...]
Parameter                     	Value           Unit
error                         	NAK            	    
2023-02-02 23:19:33,200:DEBUG:__init__:main@449: Not daemon, so not looping
```

</p>
</details>

You had asked one other time what commands would be important to me to implement. I realized that the **QED** and **QLD** daily total would be very important to me as these could be queried at timed intervals to see the input or output during a period by subtracting the previous total.
Thanks again for all your work on this project! 
Frederick

Saentist, 2023-02-03:
@frederickjh what return your inverter with this command `QPI#QMN#QGMN`
https://github.com/jblance/mpp-solar/wiki/Protocol-Recommendations

frederickjh, 2023-02-03:
@Saentist 
```bash
$ /usr/local/bin/mpp-solar -p /dev/mppsolarusb/direct/0 -P PI30MAX -b 2400 -c QPI#QMN#QGMN                                           
Processing DEFAULT type responses
Command: QPI - Protocol ID inquiry
------------------------------------------------------------
Parameter                     	Value           Unit
protocol_id                   	PI30           	    
Command: QMN - Model Name Inquiry
------------------------------------------------------------
Parameter                     	Value           Unit
model_name                    	MKS2-8000      	    
Command: QGMN - No description found
------------------------------------------------------------
Parameter                     	Value           Unit
warning                       	No definition for command QGMN in protocol b'PI30MAX'	    
                              	(044È®
```

``` bash
$ /usr/local/bin/mpp-solar -p /dev/mppsolarusb/direct/0 -P PI30 -b 2400 -c QPI#QMN#QGMN 
Processing DEFAULT type responses
Command: QPI - Protocol ID inquiry
------------------------------------------------------------
Parameter                     	Value           Unit
protocol_id                   	PI30           	    
Command: QMN - No description found
------------------------------------------------------------
Parameter                     	Value           Unit
  rning                       	No definition for command QMN in protocol b'PI30'	    
Command: QGMN - No description found2-8000²
------------------------------------------------------------
Parameter                     	Value           Unit
warning                       	No definition for command QGMN in protocol b'PI30'	    
                              	(044È®
```

Saentist, 2023-02-04:
@frederickjh 
PI30-044:MKS2-8000
@jblance 
[some docs](http://ftps.voltronic.com.tw/E/Easunpower-93CB312922AC474787C6/RS232%20通訊協議(communication%20protocol)/) about inverter protocol, xlsx files ;)

frederickjh, 2023-02-04:
> @frederickjh
> PI30-044:MKS2-8000

@Saentist not sure what I should do with that information. I already saw that information in the wiki you linked. I am guessing that it the original manufactures model number.

Saentist, 2023-02-04:
@frederickjh 
Information is important because each device support different command set.
Just simple firmware update can change supported commands aka can add or remove commands on exact same model.