jblance/mpp-solar #152: Trying to extend list of setter commands in PI 17 but

VictorBucha, 2021-09-16:
I'm trying to implement a ACCB setter command, which can save kilowatts per day in summer but keep battery charged in winter but receiving a mistake. So I'm inserting in the line 600 of PI17.py file the following code:
    "ACCB": {
        "name": "ACCB",
        "description": "AC Charger  keep  battery voltage",
        "help": " -- examples: ACCB1,450 (1: enable, 0: disable),(400-600) mV",
        "type": "SETTER",
        "response": [
            ["ack", "Command execution", {"NAK": "Failed", "ACK": "Successful"}],
        ],
        "test_responses": [
            b"^1\x0b\xc2\r",
            b"^0\x1b\xe3\r",
        ],
        "regex": "ACCB([01]),([400-600])$",
    },
but get the result:
root@uman1:/usr/local/lib/python3.7/dist-packages/mppsolar/protocols# mpp-solar -p /dev/hidraw0 -b 9600 -P PI17 -c ACCB0,490
2021-09-16 15:40:50,187:ERROR:device:run_command@299: full_command not found for ACCB0 in protocol b'PI17'
Parameter                       Value           Unit
error                           full_command not found for ACCB0 in protocol b'PI17'
2021-09-16 15:40:50,190:ERROR:device:run_command@299: full_command not found for 490 in protocol b'PI17'
Parameter                       Value           Unit
error                           full_command not found for 490 in protocol b'PI17'
What am I missing?

jblance, 2021-09-16:
> 2021-09-16 ***@***.***: full_command not

The comma is used as a command separator so the code is seeing 2 commands.
I'd try something like this

"regex": "ACCB[01]([400-600])$"

VictorBucha, 2021-09-17:
Changed regex as you advised.
Below is the result
root@uman1:/usr/local/lib/python3.7/dist-packages/mppsolar/protocols# mpp-solar -p /dev/hidraw0 -b 9600 -P PI17 -c ACCB0490
2021-09-17 10:40:36,782:ERROR:device:run_command@299: full_command not found for ACCB0490 in protocol b'PI17'
Parameter                       Value           Unit
error                           full_command not found for ACCB0490 in protocol b'PI17'

jblance, 2021-09-17:
> 2021-09-17 ***@***.***: full_command not

Can you post full debug

VictorBucha, 2021-09-17:
root@uman1:/usr/local/lib/python3.7/dist-packages/mppsolar/protocols# mpp-solar -p /dev/hidraw0 -b 9600 -P PI17 -D -c ACCB0490
2021-09-17 13:45:01,915:INFO:__init__:main@236: Solar Device Command Utility, version: 0.7.78, jk02 changes - add first attempt at setter command
2021-09-17 13:45:01,916:INFO:__init__:main@340: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI17"
2021-09-17 13:45:01,920:DEBUG:__init__:main@343: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2021-09-17 13:45:01,920:DEBUG:device:__init__@31: __init__ args ()
2021-09-17 13:45:01,920:DEBUG:device:__init__@32: __init__ kwargs {'name': 'unnamed', 'port': '/dev/hidraw0', 'protocol': 'PI17', 'baud': 9600, 'porttype': None, 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
2021-09-17 13:45:01,921:DEBUG:device:get_port_type@59: port matches hidraw
2021-09-17 13:45:01,921:INFO:device:set_port@142: Using hidrawio for communications
2021-09-17 13:45:01,924:DEBUG:device:set_protocol@96: Protocol PI17
2021-09-17 13:45:01,953:DEBUG:device:__init__@36: __init__ name unnamed, port <mppsolar.io.hidrawio.HIDRawIO object at 0x766361b0>, protocol <mppsolar.protocols.pi17.pi17 object at 0x76636330>
2021-09-17 13:45:01,953:DEBUG:__init__:main@390: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x76636310>, 'ACCB0490', 'ACCB0490', 'screen', None, None)]
2021-09-17 13:45:01,953:INFO:__init__:main@395: Looping 1 commands
2021-09-17 13:45:01,954:INFO:__init__:main@406: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.hidrawio.HIDRawIO object at 0x766361b0>, protocol: <mppsolar.protocols.pi17.pi17 object at 0x76636330> for command: ACCB0490, tag: ACCB0490, outputs: screen
2021-09-17 13:45:01,954:INFO:device:run_command@270: Running command ACCB0490
2021-09-17 13:45:01,954:INFO:pi17:get_full_command@646: Using protocol b'PI17' with 33 commands
2021-09-17 13:45:01,954:DEBUG:abstractprotocol:get_command_defn@43: Processing command 'ACCB0490'
2021-09-17 13:45:01,955:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: LON
2021-09-17 13:45:01,956:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: PA
2021-09-17 13:45:01,957:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: PB
2021-09-17 13:45:01,957:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: PC
2021-09-17 13:45:01,958:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: PD
2021-09-17 13:45:01,959:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: PE
2021-09-17 13:45:01,959:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: PF
2021-09-17 13:45:01,960:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: DAT
2021-09-17 13:45:01,963:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: LST
2021-09-17 13:45:01,963:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: EDA
2021-09-17 13:45:01,964:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: EDB
2021-09-17 13:45:01,964:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: EDC
2021-09-17 13:45:01,965:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: EDD
2021-09-17 13:45:01,966:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: EDE
2021-09-17 13:45:01,966:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: EDF
2021-09-17 13:45:01,967:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: EDG
2021-09-17 13:45:01,968:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: BT
2021-09-17 13:45:01,968:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: ACCT
2021-09-17 13:45:01,970:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: ACCB
2021-09-17 13:45:01,971:INFO:abstractprotocol:get_command_defn@58: No command_defn found for ACCB0490
2021-09-17 13:45:01,972:INFO:device:run_command@296: full command None for command ACCB0490
2021-09-17 13:45:01,972:ERROR:device:run_command@299: full_command not found for ACCB0490 in protocol b'PI17'
2021-09-17 13:45:01,972:DEBUG:__init__:main@409: results: {'ERROR': ["full_command not found for ACCB0490 in protocol b'PI17'", '']}
2021-09-17 13:45:01,972:INFO:__init__:get_outputs@26: attempting to create output processor: screen
2021-09-17 13:45:01,975:DEBUG:screen:__init__@16: processor.screen __init__ kwargs {}
2021-09-17 13:45:01,975:DEBUG:__init__:main@415: Using output filter: None
2021-09-17 13:45:01,976:INFO:screen:output@19: Using output processor: screen
2021-09-17 13:45:01,976:DEBUG:screen:output@20: kwargs {'data': {'ERROR': ["full_command not found for ACCB0490 in protocol b'PI17'", '']}, 'tag': 'ACCB0490', 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None, 'mqtt_topic': 'mpp-solar', 'filter': None, 'excl_filter': None, 'keep_case': False}
Parameter                       Value           Unit
error                           full_command not found for ACCB0490 in protocol b'PI17'
2021-09-17 13:45:01,976:DEBUG:__init__:main@435: Not daemon, so not looping

jblance, 2021-09-17:
> 2021-09-17 ***@***.***: Solar Device Command
> 2021-09-17 ***@***.***: Creating device "unnamed"
> 2021-09-17 ***@***.***: device_class <class
> 2021-09-17 ***@***.***: port matches hidraw
> 2021-09-17 ***@***.***: Using hidrawio for
> 2021-09-17 ***@***.***: Protocol PI17
> 2021-09-17 ***@***.***: Commands
> 2021-09-17 ***@***.***: Looping 1 commands
> 2021-09-17 ***@***.***: Getting results from
> 2021-09-17 ***@***.***: Running command
> 2021-09-17 ***@***.***: Using protocol
> 2021-09-17 ***@***.***: Regex
> 2021-09-17 ***@***.***: No
> 2021-09-17 ***@***.***: full command None
> 2021-09-17 ***@***.***: attempting to create
> 2021-09-17 ***@***.***: Using output filter: None
> 2021-09-17 ***@***.***: Not daemon, so not looping

Hmm, and the changed part of the pi17 protocol?

VictorBucha, 2021-09-17:
found the proper variant for regex:
    "ACCB": {
        "name": "ACCB",
        "description": "AC Charger  keep  battery voltage",
        "help": " -- examples: ACCB1450 (1: enable, 0: disable),(400-600) mV",
        "type": "SETTER",
        "response": [
            ["ack", "Command execution", {"NAK": "Failed", "ACK": "Successful"}],
        ],
        "test_responses": [
            b"^1\x0b\xc2\r",
            b"^0\x1b\xe3\r",
        ], "regex": "ACCB(0[4-6][0-9][0-9])|(1[4-6][0-9][0-9])$",
    },

VictorBucha, 2021-09-17:
and the output:
mpp-solar -p /dev/hidraw0 -b 9600 -P PI17 -D -c ACCB0500
2021-09-17 15:27:17,860:INFO:__init__:main@236: Solar Device Command Utility, version: 0.7.78, jk02 changes - add first attempt at setter command
2021-09-17 15:27:17,861:INFO:__init__:main@340: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI17"
2021-09-17 15:27:17,865:DEBUG:__init__:main@343: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2021-09-17 15:27:17,865:DEBUG:device:__init__@31: __init__ args ()
2021-09-17 15:27:17,865:DEBUG:device:__init__@32: __init__ kwargs {'name': 'unnamed', 'port': '/dev/hidraw0', 'protocol': 'PI17', 'baud': 9600, 'porttype': None, 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
2021-09-17 15:27:17,866:DEBUG:device:get_port_type@59: port matches hidraw
2021-09-17 15:27:17,866:INFO:device:set_port@142: Using hidrawio for communications
2021-09-17 15:27:17,870:DEBUG:device:set_protocol@96: Protocol PI17
2021-09-17 15:27:17,899:DEBUG:device:__init__@36: __init__ name unnamed, port <mppsolar.io.hidrawio.HIDRawIO object at 0x765e91b0>, protocol <mppsolar.protocols.pi17.pi17 object at 0x765e9330>
2021-09-17 15:27:17,900:DEBUG:__init__:main@390: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x765e9310>, 'ACCB0500', 'ACCB0500', 'screen', None, None)]
2021-09-17 15:27:17,900:INFO:__init__:main@395: Looping 1 commands
2021-09-17 15:27:17,900:INFO:__init__:main@406: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.hidrawio.HIDRawIO object at 0x765e91b0>, protocol: <mppsolar.protocols.pi17.pi17 object at 0x765e9330> for command: ACCB0500, tag: ACCB0500, outputs: screen
2021-09-17 15:27:17,901:INFO:device:run_command@270: Running command ACCB0500
2021-09-17 15:27:17,901:INFO:pi17:get_full_command@645: Using protocol b'PI17' with 33 commands
2021-09-17 15:27:17,901:DEBUG:abstractprotocol:get_command_defn@43: Processing command 'ACCB0500'
2021-09-17 15:27:17,902:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: LON
2021-09-17 15:27:17,903:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: PA
2021-09-17 15:27:17,903:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: PB
2021-09-17 15:27:17,904:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: PC
2021-09-17 15:27:17,905:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: PD
2021-09-17 15:27:17,905:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: PE
2021-09-17 15:27:17,906:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: PF
2021-09-17 15:27:17,907:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: DAT
2021-09-17 15:27:17,909:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: LST
2021-09-17 15:27:17,910:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: EDA
2021-09-17 15:27:17,911:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: EDB
2021-09-17 15:27:17,912:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: EDC
2021-09-17 15:27:17,912:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: EDD
2021-09-17 15:27:17,913:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: EDE
2021-09-17 15:27:17,914:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: EDF
2021-09-17 15:27:17,915:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: EDG
2021-09-17 15:27:17,915:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: BT
2021-09-17 15:27:17,916:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: ACCT
2021-09-17 15:27:17,918:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: ACCB
2021-09-17 15:27:17,919:DEBUG:abstractprotocol:get_command_defn@54: Matched: ACCB0500 to: ACCB value: 0500
2021-09-17 15:27:17,920:DEBUG:pi17:get_full_command@662: _pre_cmd: b'^S009ACCB0500'
2021-09-17 15:27:17,920:DEBUG:protocol_helpers:crcPI@313: Calculating CRC for b'^S009ACCB0500'
2021-09-17 15:27:17,920:DEBUG:protocol_helpers:crcPI@363: Generated CRC 0x37 0xd4 0x37d4
2021-09-17 15:27:17,921:DEBUG:pi17:get_full_command@669: full command: b'^S009ACCB0500\r'
2021-09-17 15:27:17,921:INFO:device:run_command@296: full command b'^S009ACCB0500\r' for command ACCB0500
2021-09-17 15:27:17,921:DEBUG:abstractprotocol:get_command_defn@43: Processing command 'ACCB0500'
2021-09-17 15:27:17,921:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: LON
2021-09-17 15:27:17,921:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: PA
2021-09-17 15:27:17,922:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: PB
2021-09-17 15:27:17,922:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: PC
2021-09-17 15:27:17,922:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: PD
2021-09-17 15:27:17,922:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: PE
2021-09-17 15:27:17,922:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: PF
2021-09-17 15:27:17,923:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: DAT
2021-09-17 15:27:17,923:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: LST
2021-09-17 15:27:17,923:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: EDA
[... cut ...]  '']}, 'tag': 'ACCB0500', 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None, 'mqtt_topic': 'mpp-solar', 'filter': None, 'excl_filter': None, 'keep_case': False}
Command: ACCB0500 - AC Charger  keep  battery voltage
------------------------------------------------------------
Parameter                       Value           Unit
warning0                        Command ACCB0500 was rejected
2021-09-17 15:27:19,117:DEBUG:__init__:main@435: Not daemon, so not looping

VictorBucha, 2021-09-17:
I think the problem is in the fact that the prefix for a full command is calculated incorrectly: in the above case it is S009 instead of S016. I have made a couple of other setter instances with simpler regexses but still the same result.
root@uman1:/usr/local/lib/python3.7/dist-packages/mppsolar/protocols# mpp-solar -p /dev/hidraw0 -b 9600 -P PI17 -I -c MCHGV05760569
2021-09-17 22:14:19,417:INFO:__init__:main@236: Solar Device Command Utility, version: 0.7.78, jk02 changes - add first attempt at setter command
2021-09-17 22:14:19,418:INFO:__init__:main@340: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI17"
2021-09-17 22:14:19,422:INFO:device:set_port@142: Using hidrawio for communications
2021-09-17 22:14:19,454:INFO:__init__:main@395: Looping 1 commands
2021-09-17 22:14:19,455:INFO:__init__:main@406: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.hidrawio.HIDRawIO object at 0x76624170>, protocol: <mppsolar.protocols.pi17.pi17 object at 0x766242f0> for command: MCHGV05760569, tag: MCHGV05760569, outputs: screen
2021-09-17 22:14:19,456:INFO:device:run_command@270: Running command MCHGV05760569
2021-09-17 22:14:19,457:INFO:pi17:get_full_command@692: Using protocol b'PI17' with 35 commands
2021-09-17 22:14:19,467:INFO:device:run_command@296: full command b'^S014MCHGV05760569\r' for command MCHGV05760569
2021-09-17 22:14:20,998:INFO:abstractprotocol:decode@176: response passed to decode: b'^0\x1b\xe3\r'
2021-09-17 22:14:20,999:INFO:abstractprotocol:decode@216: Processing response of type DEFAULT
2021-09-17 22:14:21,000:INFO:abstractprotocol:decode@227: Processing DEFAULT type responses
2021-09-17 22:14:21,000:INFO:device:run_command@334: Decoded response {'raw_response': ['^0\x1bã\r', ''], '_command': 'MCHGV05760569', '_command_description': 'Set battery charge voltages', 'WARNING0': ['Command MCHGV05760569 was rejected', '']}
2021-09-17 22:14:21,001:INFO:__init__:get_outputs@26: attempting to create output processor: screen
2021-09-17 22:14:21,006:INFO:screen:output@19: Using output processor: screen
Command: MCHGV05760569 - Set battery charge voltages
------------------------------------------------------------
Parameter                       Value           Unit
warning0                        Command MCHGV05760569 was rejected

and the code

    "MCHGV": {
        "name": "MCHGV",
        "description": "Set battery charge voltages",
        "help": " -- examples: MCHGV05760566 (CV voltage in mV xxxx, Float voltage xxxx in mV)",
        "type": "SETTER",
        "response": [
            ["ack", "Command execution", {"NAK": "Failed", "ACK": "Successful"}],
        ],
        "test_responses": [
            b"^1\x0b\xc2\r",
            b"^0\x1b\xe3\r",
        ],
        "regex": "MCHGV(05\d\d05\d\d)$",
    },

VictorBucha, 2021-09-17:
but this piece of code works like a charm:
    "MUCHGC": {
        "name": "MUCHGC",
        "description": "Set battery charge current",
        "help": " -- examples: MUCHGC060 (Current in mA xxxx)",
        "type": "SETTER",
        "response": [
            ["ack", "Command execution", {"NAK": "Failed", "ACK": "Successful"}],
        ],
        "test_responses": [
            b"^1\x0b\xc2\r",
            b"^0\x1b\xe3\r",
        ],
        "regex": "MUCHGC([012]\d\d\d)$",
    },
and reply:
mpp-solar -p /dev/hidraw0 -b 9600 -P PI17 -I -c MUCHGC0600
2021-09-17 22:44:55,957:INFO:__init__:main@236: Solar Device Command Utility, version: 0.7.78, jk02 changes - add first attempt at setter command
2021-09-17 22:44:55,957:INFO:__init__:main@340: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI17"
2021-09-17 22:44:55,961:INFO:device:set_port@142: Using hidrawio for communications
2021-09-17 22:44:55,993:INFO:__init__:main@395: Looping 1 commands
2021-09-17 22:44:55,993:INFO:__init__:main@406: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.hidrawio.HIDRawIO object at 0x765991b0>, protocol: <mppsolar.protocols.pi17.pi17 object at 0x76599330> for command: MUCHGC0600, tag: MUCHGC0600, outputs: screen
2021-09-17 22:44:55,993:INFO:device:run_command@270: Running command MUCHGC0600
2021-09-17 22:44:55,993:INFO:pi17:get_full_command@705: Using protocol b'PI17' with 36 commands
2021-09-17 22:44:56,006:INFO:device:run_command@296: full command b'^S011MUCHGC0600\r' for command MUCHGC0600
2021-09-17 22:44:57,175:INFO:abstractprotocol:decode@176: response passed to decode: b'^1\x0b\xc2\r'
2021-09-17 22:44:57,176:INFO:abstractprotocol:decode@216: Processing response of type DEFAULT
2021-09-17 22:44:57,177:INFO:abstractprotocol:decode@227: Processing DEFAULT type responses
2021-09-17 22:44:57,178:INFO:device:run_command@334: Decoded response {'raw_response': ['^1\x0bÂ\r', ''], '_command': 'MUCHGC0600', '_command_description': 'Set battery charge current', 'MUCHGC': ['ACK', '']}
2021-09-17 22:44:57,178:INFO:__init__:get_outputs@26: attempting to create output processor: screen
2021-09-17 22:44:57,184:INFO:screen:output@19: Using output processor: screen
Command: MUCHGC0600 - Set battery charge current
------------------------------------------------------------
Parameter                       Value           Unit
muchgc                          ACK

VictorBucha, 2021-09-17:
The problem occurs, when multiple parameters are required in a row. Maybe a solution is not to calculate the prefix for these but to include it into the command

jblance, 2021-09-17:
What's an example?
But yes it is possible to do it that way with some small changes - I tried
to keep things as generic as possible,  but that does not always work

VictorBucha, 2021-09-17:
> What's an example? But yes it is possible to do it that way with some small changes - I tried to keep things as generic as possible, but that does not always work
> […](#)

Example is two posts up.  With muchgs command.
What should I comment to enter prefixes manually?

jblance, 2021-09-17:
Do you mean MCHGV05760569?
Arent the values wrong? 0576 is 5.76V

VictorBucha, 2021-09-18:
yes they are wrong, but prefix is incorrectly calculated even with correct ones. The code:
    "MCHGV": {
        "name": "MCHGV",
        "description": "Set battery charge voltages",
        "help": " -- examples: MCHGV57605660 (CV voltage in mV xxxx, Float voltage xxxx in mV)",
        "type": "SETTER",
        "response": [
            ["ack", "Command execution", {"NAK": "Failed", "ACK": "Successful"}],
        ],
        "test_responses": [
            b"^1\x0b\xc2\r",
            b"^0\x1b\xe3\r",
        ],
        "regex": "MCHGV(5\d\d\d5\d\d\d)$",
    },
and the answer:

root@uman1:/usr/local/lib/python3.7/dist-packages/mppsolar/protocols# mpp-solar -p /dev/hidraw0 -b 9600 -P PI17 -I -c MCHGV57605690
2021-09-18 08:18:37,211:INFO:__init__:main@236: Solar Device Command Utility, version: 0.7.78, jk02 changes - add first attempt at setter command
2021-09-18 08:18:37,212:INFO:__init__:main@340: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI17"
2021-09-18 08:18:37,218:INFO:device:set_port@142: Using hidrawio for communications
2021-09-18 08:18:37,258:INFO:__init__:main@395: Looping 1 commands
2021-09-18 08:18:37,259:INFO:__init__:main@406: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.hidrawio.HIDRawIO object at 0x766401b0>, protocol: <mppsolar.protocols.pi17.pi17 object at 0x76640330> for command: MCHGV57605690, tag: MCHGV57605690, outputs: screen
2021-09-18 08:18:37,260:INFO:device:run_command@270: Running command MCHGV57605690
2021-09-18 08:18:37,260:INFO:pi17:get_full_command@705: Using protocol b'PI17' with 36 commands
2021-09-18 08:18:37,270:INFO:device:run_command@296: full command b'^S014MCHGV57605690\r' for command MCHGV57605690
2021-09-18 08:18:38,799:INFO:abstractprotocol:decode@176: response passed to decode: b'^0\x1b\xe3\r'
2021-09-18 08:18:38,801:INFO:abstractprotocol:decode@216: Processing response of type DEFAULT
2021-09-18 08:18:38,801:INFO:abstractprotocol:decode@227: Processing DEFAULT type responses
2021-09-18 08:18:38,802:INFO:device:run_command@334: Decoded response {'raw_response': ['^0\x1bã\r', ''], '_command': 'MCHGV57605690', '_command_description': 'Set battery charge voltages', 'WARNING0': ['Command MCHGV57605690 was rejected', '']}
2021-09-18 08:18:38,802:INFO:__init__:get_outputs@26: attempting to create output processor: screen
2021-09-18 08:18:38,807:INFO:screen:output@19: Using output processor: screen
Command: MCHGV57605690 - Set battery charge voltages
------------------------------------------------------------
Parameter                       Value           Unit
warning0                        Command MCHGV57605690 was rejected

You see, prefix is S014 and should be S015!

jblance, 2021-09-18:
> 2021-09-18 ***@***.***: Solar Device Command
> 2021-09-18 ***@***.***: Creating device "unnamed"
> 2021-09-18 ***@***.***: Using hidrawio for
> 2021-09-18 ***@***.***: Looping 1 commands
> 2021-09-18 ***@***.***: Getting results from
> 2021-09-18 ***@***.***: Running command
> 2021-09-18 ***@***.***: Using protocol
> 2021-09-18 ***@***.***: full command
> 2021-09-18 ***@***.***: response passed
> 2021-09-18 ***@***.***: Processing
> 2021-09-18 ***@***.***: Decoded response
> 2021-09-18 ***@***.***: attempting to create

It should? I count 14 (though i was wrong with mV conversion before so ....

If the Calc is wrong for prefix then best would be to fix it. To force a
prefix would need a bit of code changes

VictorBucha, 2021-09-18:
definitely! 
[image]
If I could fix the calc- I would have done so already, but I dont know how.
Paste the prefix in each of setter command takes some time but I can do that. If you tell me what to change in a full command

jblance, 2021-09-18:
Think the problem is that it needs the comma as part of the command - and
currently the code splits -c on comma.

Hmm, best hacky fix for you would be to remove the split on comma. Better
would be to allow an override, something like `--nosplit`

jblance, 2021-09-18:
Hacky fix for you would be line 319 in __init__.py change the comma in the split command to another character eg ";" this would allow you to correctly format the command (you'd need to fix the regex again thou)

VictorBucha, 2021-09-18:
> Hacky fix for you would be line 319 in **init**.py change the comma in the split command to another character eg ";" this would allow you to correctly format the command (you'd need to fix the regex again thou)

no, it would not. Changed line 319 for ";" and changed regex as follows (and rebooted): 
    "MCHGV": {
        "name": "MCHGV",
        "description": "Set battery charge voltages",
        "help": " -- examples: MCHGV57605660 (CV voltage in mV xxxx, Float voltage xxxx in mV)",
        "type": "SETTER",
        "response": [
            ["ack", "Command execution", {"NAK": "Failed", "ACK": "Successful"}],
        ],
        "test_responses": [
            b"^1\x0b\xc2\r",
            b"^0\x1b\xe3\r",
        ],
        "regex": "MCHGV(5\d\d\d\,5\d\d\d)$",
    },
the reply is 
root@uman1:/usr/local/lib/python3.7/dist-packages/mppsolar/protocols# mpp-solar -p /dev/hidraw0 -b 9600 -P PI17 -I -c MCHGV5760,5690
2021-09-18 10:18:26,115:INFO:__init__:main@236: Solar Device Command Utility, version: 0.7.78, jk02 changes - add first attempt at setter command
2021-09-18 10:18:26,115:INFO:__init__:main@340: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI17"
2021-09-18 10:18:26,120:INFO:device:set_port@142: Using hidrawio for communications
2021-09-18 10:18:26,167:INFO:__init__:main@395: Looping 2 commands
2021-09-18 10:18:26,170:INFO:__init__:main@406: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.hidrawio.HIDRawIO object at 0x765b61b0>, protocol: <mppsolar.protocols.pi17.pi17 object at 0x765b6330> for command: MCHGV5760, tag: MCHGV5760, outputs: screen
2021-09-18 10:18:26,171:INFO:device:run_command@270: Running command MCHGV5760
2021-09-18 10:18:26,171:INFO:pi17:get_full_command@705: Using protocol b'PI17' with 36 commands
2021-09-18 10:18:26,187:INFO:abstractprotocol:get_command_defn@58: No command_defn found for MCHGV5760
2021-09-18 10:18:26,199:INFO:device:run_command@296: full command None for command MCHGV5760
2021-09-18 10:18:26,200:ERROR:device:run_command@299: full_command not found for MCHGV5760 in protocol b'PI17'
2021-09-18 10:18:26,200:INFO:__init__:get_outputs@26: attempting to create output processor: screen
2021-09-18 10:18:26,204:INFO:screen:output@19: Using output processor: screen
Parameter                       Value           Unit
error                           full_command not found for MCHGV5760 in protocol b'PI17'
2021-09-18 10:18:26,206:INFO:__init__:main@406: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.hidrawio.HIDRawIO object at 0x765b61b0>, protocol: <mppsolar.protocols.pi17.pi17 object at 0x765b6330> for command: 5690, tag: 5690, outputs: screen
2021-09-18 10:18:26,206:INFO:device:run_command@270: Running command 5690
2021-09-18 10:18:26,206:INFO:pi17:get_full_command@705: Using protocol b'PI17' with 36 commands
2021-09-18 10:18:26,207:INFO:abstractprotocol:get_command_defn@58: No command_defn found for 5690
2021-09-18 10:18:26,207:INFO:device:run_command@296: full command None for command 5690
2021-09-18 10:18:26,208:ERROR:device:run_command@299: full_command not found for 5690 in protocol b'PI17'
2021-09-18 10:18:26,208:INFO:__init__:get_outputs@26: attempting to create output processor: screen
2021-09-18 10:18:26,209:INFO:screen:output@19: Using output processor: screen
Parameter                       Value           Unit
error                           full_command not found for 5690 in protocol b'PI17'

jblance, 2021-09-18:
> 2021-09-18 ***@***.***: Looping 2 commands
> 2021-09-18 ***@***.***: No
> 2021-09-18 ***@***.***: full command None
> 2021-09-18 ***@***.***: full_command not
> 2021-09-18 ***@***.***: Running command 5690

Sorry also at about line 381

jblance, 2021-09-18:
(I'm replying on my phone so apologies for inaccuracies) 

In addition to fixing the other split line, I think your initial numbers were correct 0560 is 56.0 so you'll need to go back to the earlier values

VictorBucha, 2021-09-18:
Done for MCHGV with changing comma to semicolumn in 319 and 381 and reverting to previous regex.
But ACCB still gives wrong calculation of prefix. 
 "regex": "ACCB([01],[456]\d\d)$"
answer:
root@uman1:/usr/local/lib/python3.7/dist-packages/mppsolar/protocols# mpp-solar -p /dev/hidraw0 -b 9600 -P PI17 -I -c ACCB0,490
2021-09-18 13:48:21,268:INFO:__init__:main@236: Solar Device Command Utility, version: 0.7.78, jk02 changes - add first attempt at setter command
2021-09-18 13:48:21,269:INFO:__init__:main@340: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI17"
2021-09-18 13:48:21,273:INFO:device:set_port@142: Using hidrawio for communications
2021-09-18 13:48:21,304:INFO:__init__:main@395: Looping 1 commands
2021-09-18 13:48:21,304:INFO:__init__:main@406: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.hidrawio.HIDRawIO object at 0x7663c170>, protocol: <mppsolar.protocols.pi17.pi17 object at 0x7663c2f0> for command: ACCB0,490, tag: ACCB0,490, outputs: screen
2021-09-18 13:48:21,304:INFO:device:run_command@270: Running command ACCB0,490
2021-09-18 13:48:21,304:INFO:pi17:get_full_command@706: Using protocol b'PI17' with 36 commands
2021-09-18 13:48:21,316:INFO:device:run_command@296: full command b'^S010ACCB0,490\r' for command ACCB0,490
2021-09-18 13:48:22,486:INFO:abstractprotocol:decode@176: response passed to decode: b'^0\x1b\xe3\r'
2021-09-18 13:48:22,488:INFO:abstractprotocol:decode@216: Processing response of type DEFAULT
2021-09-18 13:48:22,489:INFO:abstractprotocol:decode@227: Processing DEFAULT type responses
2021-09-18 13:48:22,489:INFO:device:run_command@334: Decoded response {'raw_response': ['^0\x1bã\r', ''], '_command': 'ACCB0,490', '_command_description': 'AC Charger  keep  battery voltage', 'WARNING0': ['Command ACCB0,490 was rejected', '']}
2021-09-18 13:48:22,490:INFO:__init__:get_outputs@26: attempting to create output processor: screen
2021-09-18 13:48:22,497:INFO:screen:output@19: Using output processor: screen
Command: ACCB0,490 - AC Charger  keep  battery voltage
------------------------------------------------------------
Parameter                       Value           Unit
warning0                        Command ACCB0,490 was rejected

VictorBucha, 2021-09-18:
root@uman1:/opt/Temp/inverter# mpp-solar -p /dev/hidraw0 -b 9600 -P PI17 -c ACCB1,0510
Command: ACCB1,0510 - AC Charger  keep  battery voltage
------------------------------------------------------------
Parameter                       Value           Unit
accb                            ACK

Sent a pull request ( for the first time in my life)

closed 2021-09-18 by VictorBucha