jblance/mpp-solar #154: Change get_full_command logic to implement EY, EM, ED,EH commands in PI17

VictorBucha, 2021-09-21:
I'm trying to implement command to get a daily generated energy by my 3P hybrid inverter. PI17 does not use CRC  but for the above commands they use "the sum of character string". provided we take the command with prefix as a=  "P010EDyyyymmdd". This sum is calculated as  
b=sum(ord(ch) for ch in a) 

and full command will be "P010EDyyyymmdd"+str(b)
So the decimal sum for get the monthly value for this September will be 675 for command P012EM202109.   
Can you please guide me further?

jblance, 2021-09-22:
I might be missing something, but something like:
```python
        # calculate the CRC
        crc = sum(ord(ch) for ch in _pre_cmd)
        full_command = _pre_cmd + _crc + bytes([13])  
        log.debug(f"full command: {full_command}")
        return full_command
```

VictorBucha, 2021-09-22:
@jblance, assuming the bytes([13]) is command plus arguments, like YYYYMM, the full command should be ordered like this
full_command = _pre_cmd +  bytes([13]) + _crc

also some attributes should be changed to differentiate between simple QUERY and query for energy generated.
If my understanding is correct - I will proceed with changes

jblance, 2021-09-22:
Bytes([13]) is newline and should be the last byte of the sequence

VictorBucha, 2021-09-22:
Tried to incorporate the code into protocol.
[pi17.zip](https://github.com/jblance/mpp-solar/files/7212145/pi17.zip)

Stuck here.
root@raspberrypi:/home/pi# mpp-solar -p test -P PI17 -D -c EH202105081250
2021-09-22 16:48:14,433:INFO:__init__:main@236: Solar Device Command Utility, version: 0.7.76, remove brackets in ved protocol
2021-09-22 16:48:14,434:INFO:__init__:main@340: Creating device "unnamed" (type: "mppsolar") on port "test (porttype=None)" using protocol "PI17"
2021-09-22 16:48:14,436:DEBUG:__init__:main@343: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2021-09-22 16:48:14,437:DEBUG:device:__init__@31: __init__ args ()
2021-09-22 16:48:14,437:DEBUG:device:__init__@32: __init__ kwargs {'name': 'unnamed', 'port': 'test', 'protocol': 'PI17', 'baud': 2400, 'porttype': None, 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
2021-09-22 16:48:14,437:DEBUG:device:get_port_type@51: port matches test
2021-09-22 16:48:14,437:INFO:device:set_port@136: Using testio for communications
2021-09-22 16:48:14,450:DEBUG:device:set_protocol@96: Protocol PI17
2021-09-22 16:48:14,490:DEBUG:device:__init__@36: __init__ name unnamed, port <mppsolar.io.testio.TestIO object at 0xb6520590>, protocol <mppsolar.protocols.pi17.pi17 object at 0xb6520db0>
2021-09-22 16:48:14,490:DEBUG:__init__:main@390: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0xb65203f0>, 'EH202105081250', 'EH202105081250', 'screen', None, None)]
2021-09-22 16:48:14,490:INFO:__init__:main@395: Looping 1 commands
2021-09-22 16:48:14,491:INFO:__init__:main@406: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.testio.TestIO object at 0xb6520590>, protocol: <mppsolar.protocols.pi17.pi17 object at 0xb6520db0> for command: EH202105081250, tag: EH202105081250, outputs: screen
2021-09-22 16:48:14,491:INFO:device:run_command@270: Running command EH202105081250
2021-09-22 16:48:14,491:INFO:pi17:get_full_command@793: Using protocol b'PI17' with 42 commands
2021-09-22 16:48:14,492:DEBUG:abstractprotocol:get_command_defn@42: Processing command 'EH202105081250'
2021-09-22 16:48:14,492:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: EY
2021-09-22 16:48:14,493:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: EM
2021-09-22 16:48:14,493:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: ED
2021-09-22 16:48:14,494:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: EH
2021-09-22 16:48:14,495:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: LON
2021-09-22 16:48:14,495:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: PA
2021-09-22 16:48:14,496:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: PB
2021-09-22 16:48:14,496:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: PC
2021-09-22 16:48:14,497:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: PD
2021-09-22 16:48:14,497:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: PE
2021-09-22 16:48:14,498:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: PF
2021-09-22 16:48:14,498:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: DAT
2021-09-22 16:48:14,499:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: LST
2021-09-22 16:48:14,500:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: EDA
2021-09-22 16:48:14,501:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: EDB
2021-09-22 16:48:14,501:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: EDC
2021-09-22 16:48:14,502:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: EDD
2021-09-22 16:48:14,502:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: EDE
2021-09-22 16:48:14,503:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: EDF
2021-09-22 16:48:14,503:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: EDG
2021-09-22 16:48:14,504:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: BT
2021-09-22 16:48:14,504:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: MCHGV
2021-09-22 16:48:14,505:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: ACCT
2021-09-22 16:48:14,506:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: ACCB
2021-09-22 16:48:14,507:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: MUCHGC
2021-09-22 16:48:14,507:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: SEP
2021-09-22 16:48:14,508:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: BDCM
2021-09-22 16:48:14,509:INFO:abstractprotocol:get_command_defn@57: No command_defn found for EH202105081250
2021-09-22 16:48:14,509:INFO:device:run_command@296: full command None for command EH202105081250
2021-09-22 16:48:14,509:ERROR:device:run_command@299: full_command not found for EH202105081250 in protocol b'PI17'
2021-09-22 16:48:14,509:DEBUG:__init__:main@409: results: {'ERROR': ["full_command not found for EH202105081250 in protocol b'PI17'", '']}
2021-09-22 16:48:14,510:INFO:__init__:get_outputs@26: attempting to create output processor: screen
2021-09-22 16:48:14,512:DEBUG:screen:__init__@16: processor.screen __init__ kwargs {}
2021-09-22 16:48:14,512:DEBUG:__init__:main@415: Using output filter: None
[... cut ...]
2021-09-22 16:48:14,512:DEBUG:screen:output@20: kwargs {'data': {'ERROR': ["full_command not found for EH202105081250 in protocol b'PI17'", '']}, 'tag': 'EH202105081250', 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None, 'mqtt_topic': 'mpp-solar', 'filter': None, 'excl_filter': None, 'keep_case': False}
Parameter                       Value           Unit
error                           full_command not found for EH202105081250 in protocol b'PI17'
2021-09-22 16:48:14,513:DEBUG:__init__:main@435: Not daemon, so not looping

VictorBucha, 2021-09-22:
So presently I am here, trying to get a "nnn is a decimal number, and it is low 8 bits of its hexadecimal type" as it is said in Chinese PI17 protocol
file is below
[pi17.zip](https://github.com/jblance/mpp-solar/files/7212380/pi17.zip)

The answer to it is as follows:

root@raspberrypi:/usr/local/lib/python3.7/dist-packages/mppsolar/protocols# mpp-solar -p test -P PI17 -D -c ED20210805
2021-09-22 17:20:37,752:INFO:__init__:main@236: Solar Device Command Utility, version: 0.7.76, remove brackets in ved protocol
2021-09-22 17:20:37,752:INFO:__init__:main@340: Creating device "unnamed" (type: "mppsolar") on port "test (porttype=None)" using protocol "PI17"
2021-09-22 17:20:37,755:DEBUG:__init__:main@343: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2021-09-22 17:20:37,755:DEBUG:device:__init__@31: __init__ args ()
2021-09-22 17:20:37,755:DEBUG:device:__init__@32: __init__ kwargs {'name': 'unnamed', 'port': 'test', 'protocol': 'PI17', 'baud': 2400, 'porttype': None, 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
2021-09-22 17:20:37,755:DEBUG:device:get_port_type@51: port matches test
2021-09-22 17:20:37,755:INFO:device:set_port@136: Using testio for communications
2021-09-22 17:20:37,768:DEBUG:device:set_protocol@96: Protocol PI17
2021-09-22 17:20:37,797:DEBUG:device:__init__@36: __init__ name unnamed, port <mppsolar.io.testio.TestIO object at 0xb650c590>, protocol <mppsolar.protocols.pi17.pi17 object at 0xb650cdb0>
2021-09-22 17:20:37,797:DEBUG:__init__:main@390: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0xb650c3f0>, 'ED20210805', 'ED20210805', 'screen', None, None)]
2021-09-22 17:20:37,797:INFO:__init__:main@395: Looping 1 commands
2021-09-22 17:20:37,798:INFO:__init__:main@406: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.testio.TestIO object at 0xb650c590>, protocol: <mppsolar.protocols.pi17.pi17 object at 0xb650cdb0> for command: ED20210805, tag: ED20210805, outputs: screen
2021-09-22 17:20:37,798:INFO:device:run_command@270: Running command ED20210805
2021-09-22 17:20:37,798:INFO:pi17:get_full_command@793: Using protocol b'PI17' with 42 commands
2021-09-22 17:20:37,798:DEBUG:abstractprotocol:get_command_defn@42: Processing command 'ED20210805'
2021-09-22 17:20:37,798:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: EY
2021-09-22 17:20:37,799:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: EM
2021-09-22 17:20:37,799:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: ED
2021-09-22 17:20:37,800:DEBUG:abstractprotocol:get_command_defn@53: Matched: ED20210805 to: ED value: 20210805
2021-09-22 17:20:37,800:DEBUG:pi17:get_full_command@820: _pre_cmd: b'^P011ED20210805'
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 10, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.7/dist-packages/mppsolar/__init__.py", line 408, in main
    results = _device.run_command(command=_command)
  File "/usr/local/lib/python3.7/dist-packages/mppsolar/devices/device.py", line 295, in run_command
    full_command = self._protocol.get_full_command(command)
  File "/usr/local/lib/python3.7/dist-packages/mppsolar/protocols/pi17.py", line 822, in get_full_command
    _crc = bytes([crc_low])
NameError: name 'crc_low' is not defined

And I have no idea how to implement it...

VictorBucha, 2021-09-22:
This is where I have come to a full frustration: for some reason prefix is distorted, and moreover, the result of conversion to bytes makes the real inverter crazy.   (yes, I'm testing that on a real device as well) I have added a couple of debugging points to see what is going on.
Here is the file.
[pi17.zip](https://github.com/jblance/mpp-solar/files/7213573/pi17.zip)
root@raspberrypi:/usr/local/lib/python3.7/dist-packages/mppsolar/protocols# mpp-solar -p test -P PI17 -D -c EY2020
2021-09-22 21:22:03,134:INFO:__init__:main@236: Solar Device Command Utility, version: 0.7.76, remove brackets in ved protocol
2021-09-22 21:22:03,135:INFO:__init__:main@340: Creating device "unnamed" (type: "mppsolar") on port "test (porttype=None)" using protocol "PI17"
2021-09-22 21:22:03,137:DEBUG:__init__:main@343: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2021-09-22 21:22:03,137:DEBUG:device:__init__@31: __init__ args ()
2021-09-22 21:22:03,137:DEBUG:device:__init__@32: __init__ kwargs {'name': 'unnamed', 'port': 'test', 'protocol': 'PI17', 'baud': 2400, 'porttype': None, 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
2021-09-22 21:22:03,137:DEBUG:device:get_port_type@51: port matches test
2021-09-22 21:22:03,138:INFO:device:set_port@136: Using testio for communications
2021-09-22 21:22:03,150:DEBUG:device:set_protocol@96: Protocol PI17
2021-09-22 21:22:03,166:DEBUG:device:__init__@36: __init__ name unnamed, port <mppsolar.io.testio.TestIO object at 0xb65145b0>, protocol <mppsolar.protocols.pi17.pi17 object at 0xb6514dd0>
2021-09-22 21:22:03,167:DEBUG:__init__:main@390: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0xb6514410>, 'EY2020', 'EY2020', 'screen', None, None)]
2021-09-22 21:22:03,167:INFO:__init__:main@395: Looping 1 commands
2021-09-22 21:22:03,167:INFO:__init__:main@406: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.testio.TestIO object at 0xb65145b0>, protocol: <mppsolar.protocols.pi17.pi17 object at 0xb6514dd0> for command: EY2020, tag: EY2020, outputs: screen
2021-09-22 21:22:03,167:INFO:device:run_command@270: Running command EY2020
2021-09-22 21:22:03,167:INFO:pi17:get_full_command@793: Using protocol b'PI17' with 42 commands
2021-09-22 21:22:03,167:DEBUG:abstractprotocol:get_command_defn@42: Processing command 'EY2020'
2021-09-22 21:22:03,167:DEBUG:abstractprotocol:get_command_defn@48: Regex commands _command: EY
2021-09-22 21:22:03,168:DEBUG:abstractprotocol:get_command_defn@53: Matched: EY2020 to: EY value: 2020
2021-09-22 21:22:03,168:DEBUG:pi17:get_full_command@820: _prefix: ^P007
2021-09-22 21:22:03,168:DEBUG:pi17:get_full_command@822: _pre_cmd: b'^P007EY2020'
2021-09-22 21:22:03,168:DEBUG:pi17:get_full_command@824: _numb: 679
2021-09-22 21:22:03,169:DEBUG:pi17:get_full_command@826: full command: b'^P007EY2020\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x [... cut ...] nergy': [6591, 'kWh']}, 'tag': 'EY2020', 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None, 'mqtt_topic': 'mpp-solar', 'filter': None, 'excl_filter': None, 'keep_case': False}
Command: EY2020 - Query generated energy of year
------------------------------------------------------------
Parameter                       Value           Unit
generated_energy                6591            kWh
2021-09-22 21:22:03,174:DEBUG:__init__:main@435: Not daemon, so not looping

VictorBucha, 2021-09-27:
@jblance, It seems that I have managed to write the code but failed to understand the concept of the "sum of characters string, which  is a decimal number, and it is low 8 bits of its hexadecimal type".
[pi17.zip](https://github.com/jblance/mpp-solar/files/7238030/pi17.zip)
And below is the response from a real inverter:

root@uman1:/home/pi# mpp-solar -p /dev/hidraw0 -b 9600 -P PI17 -D -c EY2021
2021-09-27 19:47:44,565:INFO:__init__:main@236: Solar Device Command Utility, version: 0.7.78, jk                                                              02 changes - add first attempt at setter command
2021-09-27 19:47:44,566:INFO:__init__:main@340: Creating device "unnamed" (type: "mppsolar") on p                                                              ort "/dev/hidraw0 (porttype=None)" using protocol "PI17"
2021-09-27 19:47:44,570:DEBUG:__init__:main@343: device_class <class 'mppsolar.devices.mppsolar.m                                                              ppsolar'>
2021-09-27 19:47:44,571:DEBUG:device:__init__@31: __init__ args ()
2021-09-27 19:47:44,571:DEBUG:device:__init__@32: __init__ kwargs {'name': 'unnamed', 'port': '/d                                                              ev/hidraw0', 'protocol': 'PI17', 'baud': 9600, 'porttype': None, 'mqtt_broker': 'localhost', 'mqt                                                              t_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
2021-09-27 19:47:44,572:DEBUG:device:get_port_type@59: port matches hidraw
2021-09-27 19:47:44,572:INFO:device:set_port@142: Using hidrawio for communications
2021-09-27 19:47:44,575:DEBUG:device:set_protocol@96: Protocol PI17
2021-09-27 19:47:44,606:DEBUG:device:__init__@36: __init__ name unnamed, port <mppsolar.io.hidraw                                                              io.HIDRawIO object at 0x7663f150>, protocol <mppsolar.protocols.pi17.pi17 object at 0x7663f2d0>
2021-09-27 19:47:44,607:DEBUG:__init__:main@390: Commands [(<mppsolar.devices.mppsolar.mppsolar o                                                              bject at 0x7663f2b0>, 'EY2021', 'EY2021', 'screen', None, None)]
2021-09-27 19:47:44,607:INFO:__init__:main@395: Looping 1 commands
2021-09-27 19:47:44,607:INFO:__init__:main@406: Getting results from device: mppsolar device - na                                                              me: unnamed, port: <mppsolar.io.hidrawio.HIDRawIO object at 0x7663f150>, protocol: <mppsolar.prot                                                              ocols.pi17.pi17 object at 0x7663f2d0> for command: EY2021, tag: EY2021, outputs: screen
2021-09-27 19:47:44,608:INFO:device:run_command@270: Running command EY2021
2021-09-27 19:47:44,608:INFO:pi17:get_full_command@792: Using protocol b'PI17' with 42 commands
2021-09-27 19:47:44,608:DEBUG:abstractprotocol:get_command_defn@43: Processing command 'EY2021'
2021-09-27 19:47:44,609:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: EY
2021-09-27 19:47:44,610:DEBUG:abstractprotocol:get_command_defn@54: Matched: EY2021 to: EY value:                                                               2021
2021-09-27 19:47:44,610:DEBUG:pi17:get_full_command@820: _prefix: ^P010
2021-09-27 19:47:44,610:DEBUG:pi17:get_full_command@823: _numb: 674
2021-09-27 19:47:44,611:DEBUG:pi17:get_full_command@825: _pre_cmd: ^P010EY2021674
2021-09-27 19:47:44,611:DEBUG:pi17:get_full_command@827: full command: b'^P010EY2021674\r'
2021-09-27 19:47:44,611:INFO:device:run_command@296: full command b'^P010EY2021674\r' for command EY2021
2021-09-27 19:47:44,611:DEBUG:abstractprotocol:get_command_defn@43: Processing command 'EY2021'
2021-09-27 19:47:44,612:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: EY
2021-09-27 19:47:44,612:DEBUG:abstractprotocol:get_command_defn@54: Matched: EY2021 to: EY value:                                                               2021
2021-09-27 19:47:44,677:DEBUG:hidrawio:send_and_receive@29: length of to_send: 15
2021-09-27 19:47:44,678:DEBUG:hidrawio:send_and_receive@47: multiple chunk send
2021-09-27 19:47:44,678:DEBUG:hidrawio:send_and_receive@50: send: b'^P010EY2', to_send: b'021674\                                                              r'
2021-09-27 19:47:45,031:DEBUG:hidrawio:send_and_receive@47: multiple chunk send
2021-09-27 19:47:45,031:DEBUG:hidrawio:send_and_receive@50: send: b'021674\r', to_send: b''
2021-09-27 19:47:45,786:DEBUG:hidrawio:send_and_receive@69: usb response was: b'^0\x1b\xe3\r'
2021-09-27 19:47:45,786:DEBUG:device:run_command@315: Send and Receive Response b'^0\x1b\xe3\r'
2021-09-27 19:47:45,787:INFO:abstractprotocol:decode@176: response passed to decode: b'^0\x1b\xe3                                                              \r'
2021-09-27 19:47:45,787:DEBUG:abstractprotocol:get_command_defn@43: Processing command 'EY2021'
2021-09-27 19:47:45,788:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: EY
2021-09-27 19:47:45,788:DEBUG:abstractprotocol:get_command_defn@54: Matched: EY2021 to: EY value:                                                               2021
2021-09-27 19:47:45,789:INFO:abstractprotocol:decode@216: Processing response of type DEFAULT
2021-09-27 19:47:45,789:DEBUG:abstractprotocol:decode@220: trimmed and split responses: ['NAK']
2021-09-27 19:47:45,790:INFO:abstractprotocol:decode@227: Processing DEFAULT type responses
[... cut ...] None, 'mqtt_pass': None, 'mqtt_topic': 'mpp-solar', 'filter': None, 'excl_filter'                                                              : None, 'keep_case': False}
Command: EY2021 - Query generated energy of year
------------------------------------------------------------
Parameter                       Value           Unit
warning0                        Command EY2021 was rejected
2021-09-27 19:47:45,796:DEBUG:__init__:main@435: Not daemon, so not looping

Any idea how to get that nnn?

jblance, 2021-09-30:
you could try `_numb = sum(bytearray(intermedstr,'utf-8')) & 255` at line 822 in your code

VictorBucha, 2021-09-30:
@jblance this did some miracle, but in this instance ED20210926 _numb is returned as a double digit, not three digits:
root@uman1:/home/pi# mpp-solar -p /dev/hidraw0 -b 9600 -P PI17 -D -c ED20210926
2021-09-30 10:54:58,735:INFO:__init__:main@236: Solar Device Command Utility, version: 0.7.78, jk02 changes - add first attempt at setter command
2021-09-30 10:54:58,736:INFO:__init__:main@340: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI17"
2021-09-30 10:54:58,740:DEBUG:__init__:main@343: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2021-09-30 10:54:58,740:DEBUG:device:__init__@31: __init__ args ()
2021-09-30 10:54:58,740:DEBUG:device:__init__@32: __init__ kwargs {'name': 'unnamed', 'port': '/dev/hidraw0', 'protocol': 'PI17', 'baud': 9600, 'porttype': None, 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
2021-09-30 10:54:58,741:DEBUG:device:get_port_type@59: port matches hidraw
2021-09-30 10:54:58,741:INFO:device:set_port@142: Using hidrawio for communications
2021-09-30 10:54:58,744:DEBUG:device:set_protocol@96: Protocol PI17
2021-09-30 10:54:58,773:DEBUG:device:__init__@36: __init__ name unnamed, port <mppsolar.io.hidrawio.HIDRawIO object at 0x765f51b0>, protocol <mppsolar.protocols.pi17.pi17 object at 0x765f5330>
2021-09-30 10:54:58,773:DEBUG:__init__:main@390: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x765f5310>, 'ED20210926', 'ED20210926', 'screen', None, None)]
2021-09-30 10:54:58,774:INFO:__init__:main@395: Looping 1 commands
2021-09-30 10:54:58,774:INFO:__init__:main@406: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.hidrawio.HIDRawIO object at 0x765f51b0>, protocol: <mppsolar.protocols.pi17.pi17 object at 0x765f5330> for command: ED20210926, tag: ED20210926, outputs: screen
2021-09-30 10:54:58,774:INFO:device:run_command@270: Running command ED20210926
2021-09-30 10:54:58,775:INFO:pi17:get_full_command@792: Using protocol b'PI17' with 42 commands
2021-09-30 10:54:58,775:DEBUG:abstractprotocol:get_command_defn@43: Processing command 'ED20210926'
2021-09-30 10:54:58,775:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: EY
2021-09-30 10:54:58,776:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: EM
2021-09-30 10:54:58,777:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: ED
2021-09-30 10:54:58,778:DEBUG:abstractprotocol:get_command_defn@54: Matched: ED20210926 to: ED value: 20210926
2021-09-30 10:54:58,778:DEBUG:pi17:get_full_command@820: _prefix: ^P014
2021-09-30 10:54:58,778:DEBUG:pi17:get_full_command@823: _numb: 98
2021-09-30 10:54:58,779:DEBUG:pi17:get_full_command@825: _pre_cmd: ^P014ED2021092698
2021-09-30 10:54:58,779:DEBUG:pi17:get_full_command@827: full command: b'^P014ED2021092698\r'
2021-09-30 10:54:58,779:INFO:device:run_command@296: full command b'^P014ED2021092698\r' for command ED20210926
2021-09-30 10:54:58,779:DEBUG:abstractprotocol:get_command_defn@43: Processing command 'ED20210926'
2021-09-30 10:54:58,779:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: EY
2021-09-30 10:54:58,780:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: EM
2021-09-30 10:54:58,780:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: ED
2021-09-30 10:54:58,780:DEBUG:abstractprotocol:get_command_defn@54: Matched: ED20210926 to: ED value: 20210926
2021-09-30 10:54:58,847:DEBUG:hidrawio:send_and_receive@29: length of to_send: 18
2021-09-30 10:54:58,848:DEBUG:hidrawio:send_and_receive@47: multiple chunk send
2021-09-30 10:54:58,848:DEBUG:hidrawio:send_and_receive@50: send: b'^P014ED2', to_send: b'021092698\r'
2021-09-30 10:54:59,200:DEBUG:hidrawio:send_and_receive@47: multiple chunk send
2021-09-30 10:54:59,201:DEBUG:hidrawio:send_and_receive@50: send: b'02109269', to_send: b'8\r'
2021-09-30 10:54:59,554:DEBUG:hidrawio:send_and_receive@47: multiple chunk send
2021-09-30 10:54:59,555:DEBUG:hidrawio:send_and_receive@50: send: b'8\r', to_send: b''
2021-09-30 10:55:00,308:DEBUG:hidrawio:send_and_receive@69: usb response was: b'^0\x1b\xe3\r'
2021-09-30 10:55:00,309:DEBUG:device:run_command@315: Send and Receive Response b'^0\x1b\xe3\r'
2021-09-30 10:55:00,309:INFO:abstractprotocol:decode@176: response passed to decode: b'^0\x1b\xe3\r'
2021-09-30 10:55:00,310:DEBUG:abstractprotocol:get_command_defn@43: Processing command 'ED20210926'
2021-09-30 10:55:00,310:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: EY
2021-09-30 10:55:00,310:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: EM
2021-09-30 10:55:00,311:DEBUG:abstractprotocol:get_command_defn@49: Regex commands _command: ED
2021-09-30 10:55:00,311:DEBUG:abstractprotocol:get_command_defn@54: Matched: ED20210926 to: ED value: 20210926
2021-09-30 10:55:00,311:INFO:abstractprotocol:decode@216: Processing response of type DEFAULT
2021-09-30 10:55:00,312:DEBUG:abstractprotocol:decode@220: trimmed and split responses: ['NAK']
2021-09-30 10:55:00,312:INFO:abstractprotocol:decode@227: Processing DEFAULT type responses
2021-09-30 10:55:00,312:INFO:device:run_command@334: Decoded response {'raw_response': ['^0\x1bã\r', ''], '_command': 'ED20210926', '_command_description': 'Query generated energy of day', 'WARNING0': ['Command ED20210926 was rejected', '']}
[... cut ...] ']}, 'tag': 'ED20210926', 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None, 'mqtt_topic': 'mpp-solar', 'filter': None, 'excl_filter': None, 'keep_case': False}
Command: ED20210926 - Query generated energy of day
------------------------------------------------------------
Parameter                       Value           Unit
warning0                        Command ED20210926 was rejected
2021-09-30 10:55:00,317:DEBUG:__init__:main@435: Not daemon, so not looping

jblance, 2021-09-30:
> 2021-09-30 ***@***.***: Solar Device Command
> 2021-09-30 ***@***.***: Creating device "unnamed"
> 2021-09-30 ***@***.***: device_class <class
> 2021-09-30 ***@***.***: port matches hidraw
> 2021-09-30 ***@***.***: Using hidrawio for
> 2021-09-30 ***@***.***: Protocol PI17
> 2021-09-30 ***@***.***: Commands
> 2021-09-30 ***@***.***: Looping 1 commands
> 2021-09-30 ***@***.***: Getting results from
> 2021-09-30 ***@***.***: Running command
> 2021-09-30 ***@***.***: Using protocol
> 2021-09-30 ***@***.***: Regex
> 2021-09-30 ***@***.***: _prefix: ^P014
> 2021-09-30 ***@***.***: _numb: 98
> 2021-09-30 ***@***.***: full command
> 2021-09-30 ***@***.***: length of
> 2021-09-30 ***@***.***: multiple
> 2021-09-30 ***@***.***: send: b'8\r',
> 2021-09-30 ***@***.***: usb response
> 2021-09-30 ***@***.***: Send and Receive
> 2021-09-30 ***@***.***: response passed
> 2021-09-30 ***@***.***: Processing
> 2021-09-30 ***@***.***: trimmed and
> 2021-09-30 ***@***.***: Decoded response
> 2021-09-30 ***@***.***: attempting to create
> 2021-09-30 ***@***.***: Using output filter: None
> 2021-09-30 ***@***.***: Not daemon, so not looping

Try using a format string to pad it to 3 digits

jblance, 2021-09-30:
As example
```
_numb = 90
print(f"{_numb:03d}")
```

VictorBucha, 2021-09-30:
Bingo! Will commit changes right now!
Thank you!

closed 2021-09-30 by VictorBucha