jblance/mpp-solar #155: PIP4048ms not connecting

sa1g, 2021-09-26:
Hi, I've followed your guide, but I got no response from the inverter (output below).
I'm using PIP4048ms, RaspberryPi 2B+, [this cable (FT232RL chip)
](https://www.amazon.it/gp/product/B07MYQPYKR/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&psc=1).

`pi@raspberrypi:~ $ mpp-solar -p /dev/ttyUSB0 -c QPI -D
2021-09-26 09:19:43,182:INFO:__init__:main@236: Solar Device Command Utility, version: 0.7.76, remove brackets in ved protocol
2021-09-26 09:19:43,187:INFO:__init__:main@340: Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=None)" using protocol "PI30"
2021-09-26 09:19:43,216:DEBUG:__init__:main@343: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2021-09-26 09:19:43,221:DEBUG:device:__init__@31: __init__ args ()
2021-09-26 09:19:43,225:DEBUG:device:__init__@32: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB0', 'protocol': 'PI30', 'baud': 2400, 'porttype': None, 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
2021-09-26 09:19:43,231:DEBUG:device:get_port_type@86: port matches ttyusb
2021-09-26 09:19:43,235:INFO:device:set_port@160: Using serialio for communications
2021-09-26 09:19:43,326:DEBUG:device:set_protocol@96: Protocol PI30
2021-09-26 09:19:43,511:DEBUG:device:__init__@36: __init__ name unnamed, port <mppsolar.io.serialio.SerialIO object at 0xb688d2b0>, protocol <mppsolar.protocols.pi30.pi30 object at 0xb688d2d0>
2021-09-26 09:19:43,516:DEBUG:__init__:main@390: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0xb688d410>, 'QPI', 'QPI', 'screen', None, None)]
2021-09-26 09:19:43,520:INFO:__init__:main@395: Looping 1 commands
2021-09-26 09:19:43,524:INFO:__init__:main@406: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.serialio.SerialIO object at 0xb688d2b0>, protocol: <mppsolar.protocols.pi30.pi30 object at 0xb688d2d0> for command: QPI, tag: QPI, outputs: screen
2021-09-26 09:19:43,528:INFO:device:run_command@270: Running command QPI
2021-09-26 09:19:43,533:INFO:abstractprotocol:get_full_command@31: Using protocol b'PI30' with 36 commands
2021-09-26 09:19:43,536:DEBUG:protocol_helpers:crcPI@299: Calculating CRC for b'QPI'
2021-09-26 09:19:43,539:DEBUG:protocol_helpers:crcPI@349: Generated CRC 0xbe 0xac 0xbeac
2021-09-26 09:19:43,542:DEBUG:abstractprotocol:get_full_command@38: full command: b'QPI\xbe\xac\r'
2021-09-26 09:19:43,545:INFO:device:run_command@296: full command b'QPI\xbe\xac\r' for command QPI
2021-09-26 09:19:43,548:DEBUG:abstractprotocol:get_command_defn@42: Processing command 'QPI'
2021-09-26 09:19:43,551:DEBUG:abstractprotocol:get_command_defn@44: Found command QPI in protocol b'PI30'
2021-09-26 09:19:43,554:DEBUG:serialio:send_and_receive@19: port /dev/ttyUSB0, baudrate 2400
2021-09-26 09:19:43,563:DEBUG:serialio:send_and_receive@22: Executing command via serialio...
2021-09-26 09:19:44,671:DEBUG:serialio:send_and_receive@30: serial response was: b''
2021-09-26 09:19:44,679:DEBUG:device:run_command@315: Send and Receive Response b''
2021-09-26 09:19:44,684:INFO:abstractprotocol:decode@175: response passed to decode: b''
2021-09-26 09:19:44,688:INFO:abstractprotocol:decode@179: Response to short
2021-09-26 09:19:44,693:INFO:device:run_command@334: Decoded response {'ERROR': ['Response to short', '']}
2021-09-26 09:19:44,698:DEBUG:__init__:main@409: results: {'ERROR': ['Response to short', '']}
2021-09-26 09:19:44,703:INFO:__init__:get_outputs@26: attempting to create output processor: screen
2021-09-26 09:19:44,729:DEBUG:screen:__init__@16: processor.screen __init__ kwargs {}
2021-09-26 09:19:44,733:DEBUG:__init__:main@415: Using output filter: None
2021-09-26 09:19:44,738:INFO:screen:output@19: Using output processor: screen
2021-09-26 09:19:44,742:DEBUG:screen:output@20: kwargs {'data': {'ERROR': ['Response to short', '']}, 'tag': 'QPI', 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None, 'mqtt_topic': 'mpp-solar', 'filter': None, 'excl_filter': None, 'keep_case': False}
Parameter                     	Value           Unit
error                         	Response to short	    
2021-09-26 09:19:44,752:DEBUG:__init__:main@435: Not daemon, so not looping
`
I also tested [this](https://github.com/jblance/mpp-solar/issues/140#issuecomment-899939001) code of yours, the result is:
`pi@raspberrypi:~ $ python3 test.py 
Serial<id=0xb6962770, open=True>(port='/dev/ttyUSB0', baudrate=19200, bytesize=8, parity='N', stopbits=1, timeout=1, xonxoff=False, rtscts=False, dsrdtr=False)
Traceback (most recent call last):
  File "test.py", line 5, in <module>
    ser.write(':010310010001EA\r\n')
  File "/usr/local/lib/python3.7/dist-packages/serial/serialposix.py", line 616, in write
    d = to_bytes(data)
  File "/usr/local/lib/python3.7/dist-packages/serial/serialutil.py", line 65, in to_bytes
    raise TypeError('unicode strings are not supported, please encode to bytes: {!r}'.format(seq))
TypeError: unicode strings are not supported, please encode to bytes: ':010310010001EA\r\n'
`
Do you have any suggestion?

jblance, 2021-09-26:
Are you sure it's connected to usb0?
Try dmesg and lsusb to see what devices are connected

sa1g, 2021-09-26:
I'm pretty sure it's connected to /dev/ttyUSB0 : 
dmesg:
`[63170.397891] usb 1-1.5: new full-speed USB device number 7 using dwc_otg
[63170.554288] usb 1-1.5: New USB device found, idVendor=0403, idProduct=6001, bcdDevice= 6.00
[63170.554330] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[63170.554351] usb 1-1.5: Product: FT232R USB UART
[63170.554371] usb 1-1.5: Manufacturer: FTDI
[63170.554391] usb 1-1.5: SerialNumber: A50285BI
[63170.576156] ftdi_sio 1-1.5:1.0: FTDI USB Serial Device converter detected
[63170.576545] usb 1-1.5: Detected FT232RL
[63170.581000] usb 1-1.5: FTDI USB Serial Device converter now attached to ttyUSB0
`

`pi@raspberrypi:~ $ lsusb
Bus 001 Device 007: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 001 Device 004: ID 0b05:17ba ASUSTek Computer, Inc. N10 Nano 802.11n Network Adapter [Realtek RTL8192CU]
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`

sa1g, 2021-09-26:
Okay maybe I found the reason why it doesn't work: the cable is wrong ("the RX-DX and ground layout isn't right for this inverter" - form the forum, next phrase). In a forum a guy sent me this schematic about the connection from the inverter's RJ45 to a DB9. I'll make one by myself and search a DB9-USB compatible converter, hopefully it'll work. 
[image]
.

VictorBucha, 2021-09-26:
--porttype derial

sa1g, 2021-09-26:
> just add this:
> --porttype serial
> cable is ok.

Still getting the same output:
`pi@raspberrypi:~ $ mpp-solar -p /dev/ttyUSB0 -c QPI --porttype serial -D
2021-09-26 19:41:10,433:INFO:__init__:main@236: Solar Device Command Utility, version: 0.7.76, remove brackets in ved protocol
2021-09-26 19:41:10,438:INFO:__init__:main@340: Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=serial)" using protocol "PI30"
2021-09-26 19:41:10,468:DEBUG:__init__:main@343: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2021-09-26 19:41:10,473:DEBUG:device:__init__@31: __init__ args ()
2021-09-26 19:41:10,478:DEBUG:device:__init__@32: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB0', 'protocol': 'PI30', 'baud': 2400, 'porttype': 'serial', 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
2021-09-26 19:41:10,483:INFO:device:set_port@130: Port overide - using port 'serial'
2021-09-26 19:41:10,488:DEBUG:device:get_port_type@83: port matches serial
2021-09-26 19:41:10,493:INFO:device:set_port@160: Using serialio for communications
2021-09-26 19:41:10,593:DEBUG:device:set_protocol@96: Protocol PI30
2021-09-26 19:41:10,777:DEBUG:device:__init__@36: __init__ name unnamed, port <mppsolar.io.serialio.SerialIO object at 0xb67a4470>, protocol <mppsolar.protocols.pi30.pi30 object at 0xb67a44d0>
2021-09-26 19:41:10,782:DEBUG:__init__:main@390: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0xb67a42d0>, 'QPI', 'QPI', 'screen', None, None)]
2021-09-26 19:41:10,787:INFO:__init__:main@395: Looping 1 commands
2021-09-26 19:41:10,792:INFO:__init__:main@406: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.serialio.SerialIO object at 0xb67a4470>, protocol: <mppsolar.protocols.pi30.pi30 object at 0xb67a44d0> for command: QPI, tag: QPI, outputs: screen
2021-09-26 19:41:10,797:INFO:device:run_command@270: Running command QPI
2021-09-26 19:41:10,802:INFO:abstractprotocol:get_full_command@31: Using protocol b'PI30' with 36 commands
2021-09-26 19:41:10,808:DEBUG:protocol_helpers:crcPI@299: Calculating CRC for b'QPI'
2021-09-26 19:41:10,813:DEBUG:protocol_helpers:crcPI@349: Generated CRC 0xbe 0xac 0xbeac
2021-09-26 19:41:10,818:DEBUG:abstractprotocol:get_full_command@38: full command: b'QPI\xbe\xac\r'
2021-09-26 19:41:10,823:INFO:device:run_command@296: full command b'QPI\xbe\xac\r' for command QPI
2021-09-26 19:41:10,828:DEBUG:abstractprotocol:get_command_defn@42: Processing command 'QPI'
2021-09-26 19:41:10,833:DEBUG:abstractprotocol:get_command_defn@44: Found command QPI in protocol b'PI30'
2021-09-26 19:41:10,838:DEBUG:serialio:send_and_receive@19: port /dev/ttyUSB0, baudrate 2400
2021-09-26 19:41:10,850:DEBUG:serialio:send_and_receive@22: Executing command via serialio...
2021-09-26 19:41:11,960:DEBUG:serialio:send_and_receive@30: serial response was: b''
2021-09-26 19:41:11,971:DEBUG:device:run_command@315: Send and Receive Response b''
2021-09-26 19:41:11,977:INFO:abstractprotocol:decode@175: response passed to decode: b''
2021-09-26 19:41:11,982:INFO:abstractprotocol:decode@179: Response to short
2021-09-26 19:41:11,987:INFO:device:run_command@334: Decoded response {'ERROR': ['Response to short', '']}
2021-09-26 19:41:11,992:DEBUG:__init__:main@409: results: {'ERROR': ['Response to short', '']}
2021-09-26 19:41:11,997:INFO:__init__:get_outputs@26: attempting to create output processor: screen
2021-09-26 19:41:12,023:DEBUG:screen:__init__@16: processor.screen __init__ kwargs {}
2021-09-26 19:41:12,027:DEBUG:__init__:main@415: Using output filter: None
2021-09-26 19:41:12,032:INFO:screen:output@19: Using output processor: screen
2021-09-26 19:41:12,037:DEBUG:screen:output@20: kwargs {'data': {'ERROR': ['Response to short', '']}, 'tag': 'QPI', 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None, 'mqtt_topic': 'mpp-solar', 'filter': None, 'excl_filter': None, 'keep_case': False}
Parameter                     	Value           Unit
error                         	Response to short	    
2021-09-26 19:41:12,048:DEBUG:__init__:main@435: Not daemon, so not looping
`
What should I do?

jblance, 2021-09-26:
> 2021-09-26 ***@***.***: Solar Device Command
> 2021-09-26 ***@***.***: device_class <class
> 2021-09-26 ***@***.***: port matches
> serial 2021-09-26 ***@***.***: Using serialio
> Protocol PI30 2021-09-26 ***@***.***: __init__
> 2021-09-26 ***@***.***: Generated CRC
> 2021-09-26 ***@***.***: Send and Receive
> 2021-09-26 ***@***.***: Decoded response
> None 2021-09-26 ***@***.***: Using output
> processor: screen 2021-09-26 ***@***.***: kwargs

Is anything else connected to the inverter?

Have you tried the USB connection (does your inverter have a USB socket? )

sa1g, 2021-09-26:
No, nothing else is connected (only 220V, photovoltaic panels and batteries). My inverter doesn't have a USB socket.

jblance, 2021-09-26:
Does the cable work with a windows machine and the supplied software
(watchpower)

sa1g, 2021-09-26:
Haven't tried yet. I'll let you know in the next few days :D (I have to find a Windows machine at home)

sa1g, 2021-09-28:
Tested, it doesn't connect to the inverter

jblance, 2021-09-28:
Can you try different cables and adapters?
Otherwise it may be an Inverter issue

sa1g, 2021-09-29:
Yes, I'll build the cable (as pic above) and use [this](https://www.amazon.it/Manhattan-Convertitore-Adattatore-USB-Seriale/dp/B0007OWNYA/ref=sr_1_6?__mk_it_IT=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dchild=1&keywords=USB+RS232&qid=1632661121&sr=8-6) adapter from DB9 to USB - a guy in the forum (the same who sent me the schematic) uses this and said it works perfectly, so I'll try with this. I'll tell you the results, thanks so much :)

jblance, 2021-09-30:
ok, I will close this issue. Raise another if you have problems after getting a connection to work

closed 2021-09-30 by jblance