jblance/mpp-solar #156: working with the LV-MK

kchiem, 2021-09-27:
Hello,

I recently got a 2024LV-MK and tried your program on it.  It mostly works on the PI30 protocol, except for at least one missing command that I'm interested in (POPLG), and I've noticed that the QPIRI command has a couple of additional responses that aren't handled in the protocols/pi30.py file, but they seem to be handled in pi30max.py.  However, pi30max.py has no setter commands defined for some reason.  So, this brings me to the question: How do I go about adding support for my inverter?  Do I:

1. add the POPLG command to pi30.py and copy the 2 additional responses defined in pi30max.py for the QPIRI command over to handle my inverter?  pi30max.py actually has 3 additional responses for the QPIRI command, but my inverter only has the first two--well, I assume the first additional response is correct, though I have no experience with it, but the second additional response is the POPLG setting, so I can confirm that one to be correct.

or

2. add the POPLG command to pi30max.py, and copy the rest of the setter commands from pi30.py to pi30max.py?

or ?

jblance, 2021-09-27:
>    1. add the POPLG command to pi30max.py, and copy the rest of the

First does your inverter give pi30 as a response to a QPI command?

What is the poplg command?

The pi30max protocol implementation is new but will get setters soon....

kchiem, 2021-09-27:
Yes, it does:

```
$ mpp-solar -p /dev/hidraw0 -c QPI
Command: QPI - Protocol ID inquiry
------------------------------------------------------------
Parameter                       Value           Unit
protocol_id                     PI30
```

POPLG is "Operation Logic".  It has 3 settings, Auto, Online, and ECO.  I found it in a protocol documentation you uploaded in this repository actually:

https://github.com/jblance/mpp-solar/blob/master/docs/protocols/PI30_PIP-GK_MK-Protocol.pdf

jblance, 2021-09-27:
Ok, sounds like I should add it to the main pi30 protocol then

kchiem, 2021-09-27:
I've done it on my end, want the pull request?

jblance, 2021-09-27:
Awesome, yes that would be great

kchiem, 2021-09-27:
#157

jblance, 2021-09-30:
merged

closed 2021-09-30 by jblance

saviothecnic, 2024-01-11:
Guys I have a Voltronic Power KING UNO 5Kw
Which in fact is from the MK family which stands for Master King
I chose the PI30 protocol but I don't see this data
What should I do for the operational logic?

Also another question, I have two KIngs connected in parallel to each other, how do I also view the second unit
connected in parallel?

Thank you

jblance, 2024-01-11:
i suggest opening a new issue
you will need to post what you have done (including debug runs) and the results
make sure that you are using the correct dev and have access to it (and nothing else is using it)

saviothecnic, 2024-01-12:
Sorry, maybe I spelled it wrong, I use Google Translate
I don't know how to program mine was a request if someone could add this function
I can only help test as a user by having this particular configuration, i.e. two King UNO inverters connected in parallel
should be PIP30 protocol or called PI35 MK

jblance, 2024-02-14:
I cant help without more information
1. how is the inverter connected (what is it connected to?) 
2. what port is the inverter mounted on
3. what commands have you tried / what was the result

jblance, 2024-02-14:
it depends, for some - yes
so the inverter is connected to a usb port on a raspberry pi?
what port does it show up as? (eg `/dev/ttyUSB0`)
what commands have you run / what result did you get?

saviothecnic, 2024-02-14:
it is on a virtualized machine
On x86 proxmox servers
I know debian as port sees the
/dev/ttyUSB0

Now I'm not at home and I can't test on the inverter
I'll try and let you know what I get
Intent you can explain to me how to do it and or what command to launch
to have the vision of all the inverts with a single cable
connected in parallel?

jblance, 2024-02-14:
first command is
`mppsolar -p /dev/ttyUSB -c QPI -D`