jblance/mpp-solar #161: Error in last development version

VictorBucha, 2021-10-01:
Yesterday decided to upgrade the library on one of my PIs and got this error:

root@raspberrypi:/home/pi# mpp-solar -p test -D -c QPIRI
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 6, in <module>
    from mppsolar import main
  File "/usr/local/lib/python3.7/dist-packages/mppsolar/__init__.py", line 5, in <module>
    from .lib.mqttbroker import MqttBroker
ModuleNotFoundError: No module named 'mppsolar.lib'
root@raspberrypi:/home/pi#

VictorBucha, 2021-10-01:
this is the installation:
root@raspberrypi:/home/pi/mpp-solar# python3 -m pip install -r requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting mpp-solar from git+https://github.com/jblance/mpp-solar.git#egg=mpp-solar (from -r requirements.txt (line 1))
  Cloning https://github.com/jblance/mpp-solar.git to /tmp/pip-install-x51l6k5r/mpp-solar
  Installing build dependencies ... done
  Running setup.py (path:/tmp/pip-install-x51l6k5r/mpp-solar/setup.py) egg_info for package mpp-solar produced metadata for project name mppsolar. Fix your #egg=mpp-solar fragments.
Requirement already satisfied: pyserial in /usr/lib/python3/dist-packages (from mppsolar->-r requirements.txt (line 1)) (3.4)
Collecting paho-mqtt (from mppsolar->-r requirements.txt (line 1))
  Downloading https://www.piwheels.org/simple/paho-mqtt/paho_mqtt-1.5.1-py3-none-any.whl (74kB)
    100% |████████████████████████████████| 81kB 655kB/s
Building wheels for collected packages: mppsolar, mppsolar
  Running setup.py bdist_wheel for mppsolar ... done
  Stored in directory: /tmp/pip-ephem-wheel-cache-b_lke9te/wheels/17/71/3f/901b5039ec87f91bd9e5086f73a7996dd10925178700861782
  Running setup.py bdist_wheel for mppsolar ... error
  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-x51l6k5r/mppsolar/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-6wsv_fv_ --python-tag cp37:
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/usr/lib/python3.7/tokenize.py", line 447, in open
      buffer = _builtin_open(filename, 'rb')
  FileNotFoundError: [Errno 2] No such file or directory: '/tmp/pip-install-x51l6k5r/mppsolar/setup.py'

  ----------------------------------------
  Failed building wheel for mppsolar
  Running setup.py clean for mppsolar
  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-x51l6k5r/mppsolar/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" clean --all:
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/usr/lib/python3.7/tokenize.py", line 447, in open
      buffer = _builtin_open(filename, 'rb')
  FileNotFoundError: [Errno 2] No such file or directory: '/tmp/pip-install-x51l6k5r/mppsolar/setup.py'

  ----------------------------------------
  Failed cleaning build dir for mppsolar
Successfully built mppsolar
Failed to build mppsolar
Installing collected packages: paho-mqtt, mppsolar
Successfully installed mppsolar-0.7.81 paho-mqtt-1.5.1

jblance, 2021-10-01:
Bummer, that looks like a bug I introduced, I don't know why it works on my system but I'll look at a fix as soon as I can

VictorBucha, 2021-10-01:
I just took files from my other Pi (lucky to have 2) - If needed  I can attach the whole bunch here

jblance, 2021-10-01:
I've committed the missing files, hopefully this resolves the issue

VictorBucha, 2021-10-03:
tested install - ok

closed 2021-10-03 by VictorBucha