jblance/mpp-solar #162: New DALY 8S 200A BMS does not work with MPP-Solar

cyborgxxl, 2021-10-05:
I have two DALY 8S 200A BMS. An older one and a brand new one. The old one works good with mpp-solar, the new one doesn't. It seems that some parameters have changed?

**Output from old one:**
```
sudo mpp-solar -p /dev/ttyUSB0 -b 9600 -P DALY40 -c SOC,cellVoltages,cellTemperatures
MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar')
Command: SOC - State of Charge
------------------------------------------------------------
Parameter                       Value           Unit
battery_bank_voltage            26.1            V
acquistion                      0.0             V
current                         5.5             A
soc                             35.0            %
Command: cellVoltages - Cell Voltages Information
------------------------------------------------------------
Parameter                       Value           Unit
cell_01_voltage                 3.264           V
cell_02_voltage                 3.273           V
cell_03_voltage                 3.272           V
cell_04_voltage                 3.273           V
cell_05_voltage                 3.272           V
cell_06_voltage                 3.273           V
cell_07_voltage                 3.273           V
cell_08_voltage                 3.27            V
cell_09_voltage                 3.273           V
Command: cellTemperatures - Cell Temperature Information
------------------------------------------------------------
Parameter                       Value           Unit
cell_01_temperature             17              °C
cell_02_temperature             -40             °C
cell_03_temperature             -40             °C
cell_04_temperature             -40             °C
cell_05_temperature             -40             °C
cell_06_temperature             -40             °C
cell_07_temperature             -40             °C
cell_08_temperature             -40             °C
cell_09_temperature             -40             °C
cell_10_temperature             -40             °C
cell_11_temperature             -40             °C
cell_12_temperature             -40             °C
cell_13_temperature             -40             °C
cell_14_temperature             -40             °C
```

**Output from new one:**
```
sudo mpp-solar -p /dev/ttyUSB1 -b 9600 -P DALY40 -c SOC,cellVoltages,cellTemperatures
MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar')
Command: SOC - State of Charge
------------------------------------------------------------
Parameter                       Value           Unit
battery_bank_voltage            27.0            V
acquistion                      0.0             V
current                         -1.9            A
soc                             100.0           %
Parameter                       Value           Unit
error                           DALY checksum did not match for response b'\xa5\x01\x95\x08\x01\r'
Parameter                       Value           Unit
error                           DALY checksum did not match for response b'\xa5\x01\x96\x08\x0170\r'
```

Any ideas how to fix that?

jblance, 2021-10-05:
can you run just the `cellVoltages` command with the debug flag added `-D`

cyborgxxl, 2021-10-14:
A small update... I tested daly-bms-cli with my new BMS and it looks like reading out the values ​​works here.

```
daly-bms-cli  -d /dev/ttyUSB0 --cell-voltages
{
  "1": 3.568,
  "2": 3.57,
  "3": 3.569,
  "4": 3.57,
  "5": 3.57,
  "6": 3.57,
  "7": 3.568,
  "8": 3.57
}
```

or

```
daly-bms-cli  -d /dev/ttyUSB0 --soc
{
  "total_voltage": 28.5,
  "current": 0.0,
  "soc_percent": 100.0
}
```

It would be great if you could update the query in your project jblance .

cyborgxxl, 2021-10-15:
In the meantime, I changed the USB port as a test. /dev/ttyUSB1 is now /dev/ttyUSB0. Sorry for the confusion. ;-)

jblance, 2021-10-15:
The debug just showed that it didn't get back much in the way of a response

Can you run it a couple of times and post the output

cyborgxxl, 2021-10-17:
I'll go crazy. For weeks only one BMS has been working and out of sheer desperation I wrote a sh logging script that records the output for ttyUSB0 and ttyUSB1. Both suddenly work fine. I do not understand that....

[mppt-test-ttyUSB0.txt](https://github.com/jblance/mpp-solar/files/7360822/mppt-test-ttyUSB0.txt)
[mppt-test-ttyUSB1.txt](https://github.com/jblance/mpp-solar/files/7360823/mppt-test-ttyUSB1.txt)

closed 2023-01-08 by jblance