jblance/mpp-solar #178: getting a json.decoder.JSONDecodeError

shanrahul, 2021-11-02:
I'm preparing a python script using the mpp-solar to catch several data from the inverter and write them down to influxdb. So [here](https://gist.github.com/shanrahul/aef4b15827529410591804803c2c9028) is my code. When i run the code for a day, sometimes i receive an error like this, 

`Traceback (most recent call last):
  File "/home/pi/getdata.py", line 145, in <module>
    write_influx()
  File "/home/pi/getdata.py", line 31, in write_influx
    data_dict_energy_generated = json.loads(payload_energy_generated)
  File "/usr/lib/python3.7/json/__init__.py", line 348, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.7/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.7/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)`

 How can i overcome this issue?

jblance, 2021-11-02:
no idea. i dont know what your script is doing or what data you are processing
suggest you add exception handling to your script and dump the json when a JSONDecodeError is caught

VictorBucha, 2021-11-10:
starting from version 0.7.82 (or maybe earlier)  a totally unnecessary line appeared in JSON output
root@raspberrypi:/opt/Temp/inverter# mpp-solar -p /dev/hidraw0 -P PI30MAX -c QPGS0 -o json
MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar')
{"_command": "QPGS0", "_command_description": "Parallel Information inquiry", "parallel_instance_number": "Not valid", "serial_number": "92932012101959", "work_mode": "Line Mode", "fault_code": "No fault", "grid_voltage": 228.6, "grid_frequency": 49.92, "ac_output_voltage": 228.6, "ac_output_frequency": 49.92, "ac_output_apparent_power": 525, "ac_output_active_power": 439, "load_percentage": 6, "battery_voltage": 56.7, "battery_charging_current": 0, "battery_capacity": 100, "pv1_input_voltage": 0.0, "total_charging_current": 0, "total_ac_output_apparent_power": 525, "total_output_active_power": 439, "total_ac_output_percentage": 5, "is_scc_ok": 1, "is_ac_charging": 1, "is_scc_charging": 1, "is_battery_over_voltage": 0, "is_battery_under_voltage": 0, "is_line_lost": 0, "is_load_on": 1, "is_configuration_changed": 0, "output_mode": "single machine", "charger_source_priority": "Solar first", "max_charger_current": 90, "max_charger_range": 120, "max_ac_charger_current": 60, "pv1_input_current": 0, "battery_discharge_current": 0, "pv2_input_voltage": 290.9, "pv2_input_current": 1}

Especially when I was parsing it and sending directly to Influx, not with instruments of mpp-solar package. I am not using mqtt at all. how can I get rid of this first line in -json responce?

jblance, 2021-11-10:
That was a bug, is it still there in the latest version?

VictorBucha, 2021-11-10:
had my MAX2 inverter offline - revealed only yesterday. will upgrade and revert

jblance, 2021-11-10:
Cool.

I have been refactoring (while my own system has been in the midst of
change) to allow the fixing of some issues that are design related, so it
is possible (or likely) that bugs have been introduced

Good luck 🙃

VictorBucha, 2021-11-10:
If newest release fail, can I have the latest pre refactoring version?

jblance, 2021-11-10:
The fix was in the middle of refactoring. Try first and see if it works

VictorBucha, 2021-11-10:
0.7.91 - has no first line

closed 2021-11-22 by jblance