jblance/mpp-solar #183: Parameter mqtttopic is ignored

SirkoVZ, 2021-11-18:
Hi, 

I updated mpp-solar from something 0.7x to 0.9.02 using
sudo pip install -e "git+https://github.com/jblance/mpp-solar.git#egg=mpp-solar"

and found that after that the --mqtttopic isn't used any more, example:

```
$ mpp-solar -p /dev/ttyUSB1 --porttype daly -b 9600 -P DALY40 -o json_mqtt -q screen --mqtttopic SmartHome -c mosStatus
{'topic': 'mpp-solar', 'payload': '{"charge_status": "charged", "charging_mos_tube_status": "01", "discharging_mos_tube_status": "01", "bms_life": 251, "residual_capacity": 8.74}'}

```
--mqtttopic SmartHome is ignored.

here is the debug and info output:
```
$ mpp-solar -p /dev/ttyUSB1 --porttype daly -b 9600 -P DALY40 -o json_mqtt -q screen --mqtttopic SmartHome -c mosStatus -D -I
2021-11-18 09:55:42,082:INFO:__init__:main@195: Solar Device Command Utility, version: 0.9.02,
2021-11-18 09:55:42,082:DEBUG:__init__:main@204: MqttBroker(name='screen', port=1883, username=None, password=None, results_topic='SmartHome', mqttc=<paho.mqtt.client.Client object at 0xb5f62db0>, _isConnected=False)
2021-11-18 09:55:42,082:INFO:__init__:main@300: Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB1 (porttype=daly)" using protocol "DALY40"
2021-11-18 09:55:42,092:DEBUG:__init__:main@303: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2021-11-18 09:55:42,092:DEBUG:device:__init__@33: __init__ args ()
2021-11-18 09:55:42,092:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB1', 'protocol': 'DALY40', 'baud': 9600, 'porttype': 'daly', 'mqtt_broker': MqttBroker(name='screen', port=1883, username=None, password=None, results_topic='SmartHome', mqttc=<paho.mqtt.client.Client object at 0xb5f62db0>, _isConnected=False)}
2021-11-18 09:55:42,092:INFO:__init__:get_port@77: Port overide - using port 'daly'
2021-11-18 09:55:42,092:DEBUG:__init__:get_port_type@56: port matches daly
2021-11-18 09:55:42,093:INFO:__init__:get_port@113: Using dalyserialio for communications
2021-11-18 09:55:42,101:DEBUG:__init__:get_protocol@12: Protocol DALY40
2021-11-18 09:55:42,120:DEBUG:device:__init__@38: __init__ name unnamed, port <mppsolar.io.dalyserialio.DalySerialIO object at 0xb5f2eed0>, protocol <mppsolar.protocols.daly40.daly40 object at 0xb5dc86d0>
2021-11-18 09:55:42,121:DEBUG:__init__:main@352: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0xb5fdc150>, 'mosStatus', 'mosStatus', 'json_mqtt', None, None)]
2021-11-18 09:55:42,121:INFO:__init__:main@357: Looping 1 commands
2021-11-18 09:55:42,121:INFO:__init__:main@368: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.dalyserialio.DalySerialIO object at 0xb5f2eed0>, protocol: <mppsolar.protocols.daly40.daly40 object at 0xb5dc86d0> for command: mosStatus, tag: mosStatus, outputs: json_mqtt
2021-11-18 09:55:42,121:INFO:device:run_command@271: Running command mosStatus
2021-11-18 09:55:42,121:INFO:daly:get_full_command@211: Using protocol b'DALY40' with 7 commands
2021-11-18 09:55:42,121:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'mosStatus'
2021-11-18 09:55:42,122:DEBUG:abstractprotocol:get_command_defn@62: Found command mosStatus in protocol b'DALY40'
2021-11-18 09:55:42,122:DEBUG:daly:get_full_command@228: full command: b'\xa5@\x93\x08\x00\x00\x00\x00\x00\x00\x00\x00\x80\n'
2021-11-18 09:55:42,122:INFO:device:run_command@298: full command b'\xa5@\x93\x08\x00\x00\x00\x00\x00\x00\x00\x00\x80\n' for command mosStatus
2021-11-18 09:55:42,122:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'mosStatus'
2021-11-18 09:55:42,122:DEBUG:abstractprotocol:get_command_defn@62: Found command mosStatus in protocol b'DALY40'
2021-11-18 09:55:42,122:DEBUG:dalyserialio:send_and_receive@19: port /dev/ttyUSB1, baudrate 9600
2021-11-18 09:55:42,127:DEBUG:dalyserialio:send_and_receive@22: Executing command via dalyserialio...
2021-11-18 09:55:42,378:DEBUG:dalyserialio:send_and_receive@32: bytes waiting 13
2021-11-18 09:55:42,628:DEBUG:dalyserialio:send_and_receive@32: bytes waiting 0
2021-11-18 09:55:42,629:DEBUG:dalyserialio:send_and_receive@38: serial response was: b'\xa5\x01\x93\x08\x01\x01\x01J\x00\x00"$\xd4'
2021-11-18 09:55:42,630:DEBUG:device:run_command@317: Send and Receive Response b'\xa5\x01\x93\x08\x01\x01\x01J\x00\x00"$\xd4'
2021-11-18 09:55:42,630:INFO:abstractprotocol:decode@196: response passed to decode: b'\xa5\x01\x93\x08\x01\x01\x01J\x00\x00"$\xd4'
2021-11-18 09:55:42,630:DEBUG:daly:check_response_valid@247: checking validity of b'\xa5\x01\x93\x08\x01\x01\x01J\x00\x00"$\xd4'
2021-11-18 09:55:42,630:DEBUG:daly:check_response_valid@260: DALY Checksum matches response 'b'\xa5\x01\x93\x08\x01\x01\x01J\x00\x00"$\xd4'' checksum:212
2021-11-18 09:55:42,630:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'mosStatus'
2021-11-18 09:55:42,631:DEBUG:abstractprotocol:get_command_defn@62: Found command mosStatus in protocol b'DALY40'
2021-11-18 09:55:42,631:INFO:abstractprotocol:decode@236: Processing response of type POSITIONAL
2021-11-18 09:55:42,631:DEBUG:abstractprotocol:decode@240: trimmed and split responses: [b'\xa5', b'\x01', b'\x93', b'\x08', b'\x01', b'\x01', b'\x01', b'J', b'\x00\x00"$', b'\xd4']
2021-11-18 09:55:42,631:DEBUG:abstractprotocol:decode@409: Processing POSITIONAL type responses
2021-11-18 09:55:42,631:DEBUG:abstractprotocol:decode@431: Got defn ['discard', 1, 'start flag', '']
2021-11-18 09:55:42,631:DEBUG:abstractprotocol:process_response@104: Processing data_type: discard for data_name: start flag, raw_value b'\xa5'
2021-11-18 09:55:42,631:DEBUG:abstractprotocol:process_response@111: Discarding start flag:b'\xa5'
2021-11-18 09:55:42,631:DEBUG:abstractprotocol:decode@409: Processing POSITIONAL type responses
2021-11-18 09:55:42,631:DEBUG:abstractprotocol:decode@431: Got defn ['discard', 1, 'module address', '']
2021-11-18 09:55:42,632:DEBUG:abstractprotocol:process_response@104: Processing data_type: discard for data_name: module address, raw_value b'\x01'
2021-11-18 09:55:42,632:DEBUG:abstractprotocol:process_response@111: Discarding module address:b'\x01'
2021-11-18 09:55:42,632:DEBUG:abstractprotocol:decode@409: Processing POSITIONAL type responses
2021-11-18 09:55:42,632:DEBUG:abstractprotocol:decode@431: Got defn ['discard', 1, 'command id', '']
2021-11-18 09:55:42,632:DEBUG:abstractprotocol:process_response@104: Processing data_type: discard for data_name: command id, raw_value b'\x93'
2021-11-18 09:55:42,632:DEBUG:abstractprotocol:process_response@111: Discarding command id:b'\x93'
2021-11-18 09:55:42,632:DEBUG:abstractprotocol:decode@409: Processing POSITIONAL type responses
2021-11-18 09:55:42,632:DEBUG:abstractprotocol:decode@431: Got defn ['discard', 1, 'data length', '']
2021-11-18 09:55:42,632:DEBUG:abstractprotocol:process_response@104: Processing data_type: discard for data_name: data length, raw_value b'\x08'
2021-11-18 09:55:42,633:DEBUG:abstractprotocol:process_response@111: Discarding data length:b'\x08'
2021-11-18 09:55:42,633:DEBUG:abstractprotocol:decode@409: Processing POSITIONAL type responses
2021-11-18 09:55:42,633:DEBUG:abstractprotocol:decode@431: Got defn ['hex_option', 1, 'Charge Status', ['stationary', 'charged', 'discharged']]
[... 13 lines cut ...]
2021-11-18 09:55:42,635:DEBUG:abstractprotocol:process_response@104: Processing data_type: Hex2Int for data_name: BMS Life, raw_value b'J'
2021-11-18 09:55:42,635:DEBUG:abstractprotocol:process_response@175: Processing format string Hex2Int(raw_value)
2021-11-18 09:55:42,635:DEBUG:protocol_helpers:Hex2Int@96: Hex b'J' decoded to 74
2021-11-18 09:55:42,635:DEBUG:abstractprotocol:decode@409: Processing POSITIONAL type responses
2021-11-18 09:55:42,635:DEBUG:abstractprotocol:decode@431: Got defn ['BigHex2Float:r/1000', 4, 'Residual Capacity', 'AH']
2021-11-18 09:55:42,635:DEBUG:abstractprotocol:process_response@102: Got template r/1000 for Residual Capacity b'\x00\x00"$'
2021-11-18 09:55:42,635:DEBUG:abstractprotocol:process_response@104: Processing data_type: BigHex2Float for data_name: Residual Capacity, raw_value b'\x00\x00"$'
2021-11-18 09:55:42,636:DEBUG:abstractprotocol:process_response@175: Processing format string BigHex2Float(raw_value)
2021-11-18 09:55:42,636:DEBUG:protocol_helpers:BigHex2Float@166: Hex b'\x00\x00"$' 4 byte decoded to 8740
2021-11-18 09:55:42,636:DEBUG:abstractprotocol:decode@409: Processing POSITIONAL type responses
2021-11-18 09:55:42,636:DEBUG:abstractprotocol:decode@431: Got defn ['discard', 1, 'checksum', '']
[... cut ...]
2021-11-18 09:55:42,650:DEBUG:mqtt:output@86: mqtt.output msgs [{'topic': 'mpp-solar', 'payload': '{"charge_status": "charged", "charging_mos_tube_status": "01", "discharging_mos_tube_status": "01", "bms_life": 74, "residual_capacity": 8.74}'}]
{'topic': 'mpp-solar', 'payload': '{"charge_status": "charged", "charging_mos_tube_status": "01", "discharging_mos_tube_status": "01", "bms_life": 74, "residual_capacity": 8.74}'}
2021-11-18 09:55:42,650:DEBUG:__init__:main@397: Not daemon, so not looping
```

SirkoVZ, 2021-11-19:
the reason is in:
https://github.com/jblance/mpp-solar/blob/458086f5cf8fec4aec4954ea3fc41ea75c329c07/mppsolar/outputs/json_mqtt.py#L23
_kwargs_ doesn't contain _mqtt_topic_
It contains _MqttBroker_ that contains _results_topic_, which holds the command line's _mqtttopic_ instead.
(It is the same for the other outputs, like influx_mqtt.py ...)

jblance, 2021-11-19:
Ok thanks. Will get that fixed soon

jblance, 2021-11-22:
have added a fix (attempt) to version 0.9.03

SirkoVZ, 2021-11-23:
tested, yes, looks good for json_mqtt now :-)

closed 2021-11-23 by SirkoVZ