jblance/mpp-solar #184: PIP 8048 Max - missing PV2 stats

dwtaylornz, 2021-11-28:
Hi John, Awesome tool! 

Everything connects fine on hidraw1 via protocol PI30...

My only issue is that I'm only getting the PV1 stats and i'm missing the stats for the 2nd MPPT tracker.
Is there a way i can help fine where these stats might be hiding? 

Many thanks.

jblance, 2021-11-28:
Try PI30MAX protocol

dwtaylornz, 2021-11-28:
Sorry, should this be in discussions for future? whoop. 

Tried this using PI30MAX but get NAK error, looks to be when its trying for QPIGS2? 

```
mpp-solar -p /dev/hidraw1 -P PI30MAX --getstatus
Command: QPIGS - General Status Parameters inquiry
------------------------------------------------------------
Parameter                       Value           Unit
ac_input_voltage                236.5           V   
ac_input_frequency              50.0            Hz  
ac_output_voltage               236.5           V   
ac_output_frequency             50.0            Hz  
ac_output_apparent_power        260             VA  
ac_output_active_power          225             W   
ac_output_load                  3               %   
bus_voltage                     325             V   
battery_voltage                 0.0             V   
battery_charging_current        0               A   
battery_capacity                0               %   
inverter_heat_sink_temperature  28              °C  
pv1_input_current               0.0             A   
pv1_input_voltage               0.0             V   
battery_voltage_from_scc        0.0             V   
battery_discharge_current       0               A   
is_sbu_priority_version_added   0               bool
is_configuration_changed        0               bool
is_scc_firmware_updated         0               bool
is_load_on                      1               bool
is_battery_voltage_to_steady_while_charging     0               bool
is_charging_on                  0               bool
is_scc_charging_on              0               bool
is_ac_charging_on               0               bool
battery_voltage_offset_for_fans_on      0               10mV
eeprom_version                  0                   
pv1_charging_power              0               W   
is_charging_to_float            0               bool
is_switched_on                  1               bool
is_dustproof_installed          0               bool
error                           NAK

mpp-solar -p /dev/hidraw1 -P PI30MAX -c QPIGS2 -I 
2021-11-28 09:22:38,330:INFO:__init__:main@195: Solar Device Command Utility, version: 0.9.03, add results_topic to json_mqtt output
2021-11-28 09:22:38,330:INFO:__init__:main@299: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw1 (porttype=None)" using protocol "PI30MAX"
2021-11-28 09:22:38,331:INFO:__init__:get_port@89: Using hidrawio for communications
2021-11-28 09:22:38,332:INFO:__init__:main@357: Looping 1 commands
2021-11-28 09:22:38,332:INFO:__init__:main@367: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.hidrawio.HIDRawIO object at 0x1484a30e1eb0>, protocol: <mppsolar.protocols.pi30max.pi30max object at 0x1484a30e1dc0> for command: QPIGS2, tag: QPIGS2, outputs: screen
2021-11-28 09:22:38,332:INFO:device:run_command@271: Running command QPIGS2
2021-11-28 09:22:38,332:INFO:abstractprotocol:get_full_command@50: Using protocol b'PI30MAX' with 30 commands
2021-11-28 09:22:38,332:INFO:device:run_command@298: full command b'QPIGS2h-\r' for command QPIGS2
2021-11-28 09:22:39,490:INFO:abstractprotocol:decode@196: response passed to decode: b'(NAKss\r'
2021-11-28 09:22:39,490:INFO:abstractprotocol:decode@200: NAK
2021-11-28 09:22:39,490:INFO:device:run_command@336: Decoded response {'ERROR': ['NAK', '']}
2021-11-28 09:22:39,490:INFO:__init__:get_outputs@37: attempting to create output processor: screen
2021-11-28 09:22:39,491:INFO:screen:output@19: Using output processor: screen
Parameter                       Value           Unit
error                           NAK               
```

jblance, 2021-11-28:
> 2021-11-28 ***@***.***: Solar Device Command Utility, version: 0.9.03, add results_topic to json_mqtt output
> 2021-11-28 ***@***.***: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw1 (porttype=None)" using protocol "PI30MAX"
> 2021-11-28 ***@***.***: Using hidrawio for communications
> 2021-11-28 ***@***.***: Looping 1 commands
> 2021-11-28 ***@***.***: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.hidrawio.HIDRawIO object at 0x1484a30e1eb0>, protocol: <mppsolar.protocols.pi30max.pi30max object at 0x1484a30e1dc0> for command: QPIGS2, tag: QPIGS2, outputs: screen
> 2021-11-28 ***@***.***: Running command QPIGS2
> 2021-11-28 ***@***.***: Using protocol b'PI30MAX' with 30 commands
> 2021-11-28 ***@***.***: full command b'QPIGS2h-\r' for command QPIGS2
> 2021-11-28 ***@***.***: response passed to decode: b'(NAKss\r'
> 2021-11-28 ***@***.***: NAK
> 2021-11-28 ***@***.***: Decoded response {'ERROR': ['NAK', '']}
> 2021-11-28 ***@***.***: attempting to create output processor: screen
> 2021-11-28 ***@***.***: Using output processor: screen

Weird - what does command `QMN` return?
What does command `QPGS0` return?

dwtaylornz, 2021-11-28:
Looked into the comms link in more detail, I'm seeing some intermittent EMI errors on the USB -> USB connection that i'm using. Wondering if there is some corruption going on? 

Results of commands ; 

```
# mpp-solar -p /dev/hidraw2 -P PI30MAX -c QMN
Command: QMN - Model Name Inquiry
------------------------------------------------------------
Parameter                       Value           Unit
model_name                      MKS2-8000           
# mpp-solar -p /dev/hidraw2 -P PI30MAX -c QPGS0
Command: QPGS0 - Parallel Information inquiry
------------------------------------------------------------
Parameter                       Value           Unit
parallel_instance_number        Not valid           
serial_number                   92932108104624      
work_mode                       Line Mode           
fault_code                      No fault            
grid_voltage                    234.5           V   
grid_frequency                  50.03           Hz  
ac_output_voltage               234.5           V   
ac_output_frequency             50.03           Hz  
ac_output_apparent_power        375             VA  
ac_output_active_power          355             W   
load_percentage                 4               %   
battery_voltage                 6.7             V   
battery_charging_current        1               A   
battery_capacity                0               %   
pv1_input_voltage               205.5           V   
total_charging_current          1               A   
total_ac_output_apparent_power  375             VA  
total_output_active_power       355             W   
total_ac_output_percentage      4               %   
is_scc_ok                       1               bool
is_ac_charging                  1               bool
is_scc_charging                 1               bool
is_battery_over_voltage         0               bool
is_battery_under_voltage        2               bool
is_line_lost                    0               bool
is_load_on                      1               bool
is_configuration_changed        0               bool
output_mode                     single machine      
charger_source_priority         Solar first         
max_charger_current             60              A   
max_charger_range               120             A   
max_ac_charger_current          10              A   
pv1_input_current               0               A   
battery_discharge_current       0               A   
pv2_input_voltage               127.4           V   
pv2_input_current               0               A   
```

jblance, 2021-11-29:
great news

closed 2021-11-29 by jblance