jblance/mpp-solar #194: Installing Ubuntu

erosenst, 2022-01-14:
Hello I want compile your pack instead pip install (like more .deb packs). How do that? There are not common INSTALL file..

~/src/mpp-solar$ make
flake8 setup.py mppsolar tests
tests/__init__.py:1:1: W391 blank line at end of file
makefile:2: recipe for target 'format' failed
make: *** [format] Error 1

Pip install fails too and I don't like to use it.

pip install mpp-solar
Collecting mpp-solar
  Downloading https://files.pythonhosted.org/packages/c2/36/fad21f11cdadabbde9ff6a031975ed96a56bb8f8c64120885afe7664e50b/mpp-solar-0.4.2.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-kFGbRL/mpp-solar/setup.py", line 18, in <module>
        with open(path.join(here, 'pypi_README.md'), encoding='utf-8') as f:
      File "/usr/lib/python2.7/codecs.py", line 898, in open
        file = __builtin__.open(filename, mode, buffering)
    IOError: [Errno 2] No such file or directory: '/tmp/pip-build-kFGbRL/mpp-solar/pypi_README.md'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-kFGbRL/mpp-solar/

btw. is VMII5K48 1:7 device supported? (Yangtze solar brand)

erosenst, 2022-01-14:
pip3 install mpp-solar seems to work..

Some info from Yangtze 5500w device
mpp-solar -p /dev/ttyUSB3 -c QVFW
Command: QVFW - Main CPU firmware version inquiry
------------------------------------------------------------
Parameter                       Value           Unit
main_cpu_firmware_version       VERFW:00072.40     

~# mpp-solar -p /dev/ttyUSB3 -P PI30 -c QPIRI
Command: QPIRI - Current Settings inquiry
------------------------------------------------------------
Parameter                       Value           Unit
ac_input_voltage                230.0           V   
ac_input_current                20.0            A   
ac_output_voltage               230.0           V   
ac_output_frequency             50.0            Hz  
ac_output_current               20.0            A   
ac_output_apparent_power        5500            VA  
ac_output_active_power          5500            W   
battery_voltage                 48.0            V   
battery_recharge_voltage        44.0            V   
battery_under_voltage           42.0            V   
battery_bulk_charge_voltage     56.0            V   
battery_float_charge_voltage    54.0            V   
battery_type                    User                
max_ac_charging_current         60              A   
max_charging_current            0               A   
input_voltage_range             UPS                 
output_source_priority          Utility first       
charger_source_priority         Solar first         
max_parallel_units              6               units
machine_type                    Off Grid            
topology                        transformerless     
output_mode                     single machine output       
battery_redischarge_voltage     52.0            V   
pv_ok_condition                 As long as one unit of inverters has connect PV, parallel system will consider PV OK        
pv_power_balance                PV input max power will be the sum of the max charged power and loads power      

Hmm, this is selled by 24A AC amps, full 5500W and 9 units paraller ok. Currently setup has no battery (yet).

jblance, 2022-01-15:
Yes python 2.7 is not supported anymore.

That seems to be working though. Settings can be changed so maybe it can be set to 24A?
Try command QPI to see what protocol it claims to use (there are some differences)

closed 2022-02-01 by jblance