jblance/mpp-solar #198: EG4, Jakiper Server Rack Battery connection

Ea1506, 2022-02-12:
I am looking for guidance on how to read the RS485 / RS232 comms information from those respective platforms.  I have only recently started learning these skills, and while I've managed to install the system to display data from 2 victron shunts successfully, I've run into a brick wall when trying to go beyond anything that someone else clearly documented.

Any guidance would be appreciated.

Ed

jblance, 2022-02-12:
This is probably belongs in the discussion area (as it's not really an issue)

These are not currently supported.  To add support requires figuring out how to communicate with and decode the responses of the devices.
2 ways of doing this are:
- find the protocol documentation 
- capture the commands when using the supplied software (eg with wireshark or similar)

I'm not familiar with either device but the Jakiper seems to have windows software so it should be possible to capture the communications 

Happy to add support for things but note it takes a bit of trial and error, especially since I don't have access to these devices

Ea1506, 2022-02-13:
My bad,  I will reach out to both and see if I can get protocols,  if not I
will setup a sniffer

Thanks.   Now, how do we get this moved to where it belongs?

jblance, 2022-02-17:
well, if you can get the docs then it becomes a feature request - will just leave it here and close it if the docs or info is unavailable

pchiquit, 2022-10-21:
I have been working on integrating the EG4 batteries with Victron equipment so I have made lots of progress on the EG4 protocol documentation. It is documented on this isssue: https://github.com/Louisvdw/dbus-serialbattery/pull/212 . Just a FYI if it helps

hasselk, 2023-08-16:
Jaikiper, SOK, BSLBATT, FreedomWon eTower all seem to use an PACE BMS if its a 100Ah Batery. Mostly Pace BMS P16S100A.

A working python implementation with home assistant autodiscovery by mqtt is [Tertiush/bmspace Repo](https://github.com/Tertiush/bmspace).

Im currently at modifying the code a bit so that balancing and cellwarnings and stuff is output as an switch too instead of an bitmask like it is donne now in the original code;).
Ive added the documentation i found to [pace_documentation](https://github.com/hasselk/bmspace/tree/main/pace_documentation)