jblance/mpp-solar #202: JK-B2A24S15P Error

Smoeller0, 2022-03-18:
Hello i Buy an new JK-B2A24S15P and have a problem reading the Data now. 
Before i had the jk only balancer without bms abd this worked fine.
Anyone any idea how to solve this Problem? 

jkbms -p C8:47:8C:F2:80:D9 -c getCellData -P JK02
Traceback (most recent call last):
  File "/usr/local/bin/jkbms", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/__init__.py", line 408, in main
    results = _device.run_command(command=_command)
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/devices/device.py", line 309, in run_command
    raw_response = self._port.send_and_receive(
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/io/jkbleio.py", line 34, in send_and_receive
    response = self.ble_get_data(full_command)
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/io/jkbleio.py", line 91, in ble_get_data
    characteristicRead = serviceNotify.getCharacteristics(characteristicReadUuid)[0]
IndexError: list index out of range

jblance, 2022-03-19:
Run the command with debug (add ``-D```) and post the results
Might be worth trying jk04 protocol as well

Smoeller0, 2022-03-19:
Hello 
I Try jk04 but the same result

root@raspberrypi:~# jkbms -p C8:47:8C:F2:80:D9 -P JK02 -D
2022-03-19 06:44:11,894:INFO:__init__:main@236: Solar Device Command Utility, version: 0.7.76, remove brackets in ved protocol
2022-03-19 06:44:11,894:INFO:__init__:main@339: Creating device "unnamed" (type: "jkbms") on port "C8:47:8C:F2:80:D9 (porttype=None)" using protocol "JK02"
2022-03-19 06:44:11,897:DEBUG:__init__:main@343: device_class <class 'mppsolar.devices.jkbms.jkbms'>
2022-03-19 06:44:11,897:DEBUG:device:__init__@31: __init__ args ()
2022-03-19 06:44:11,897:DEBUG:device:__init__@32: __init__ kwargs {'name': 'unnamed', 'port': 'C8:47:8C:F2:80:D9', 'protocol': 'JK02', 'baud': 2400, 'porttype': None, 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
2022-03-19 06:44:11,897:DEBUG:device:get_port_type@71: port matches jkble ':'
2022-03-19 06:44:11,897:INFO:device:set_port@154: Using jkbleio for communications
2022-03-19 06:44:11,939:DEBUG:device:set_protocol@96: Protocol JK02
2022-03-19 06:44:11,953:DEBUG:device:__init__@36: __init__ name unnamed, port <mppsolar.io.jkbleio.JkBleIO object at 0xb6542190>, protocol <mppsolar.protocols.jk02.jk02 object at 0xb6542208>
2022-03-19 06:44:11,953:DEBUG:__init__:main@390: Commands [(<mppsolar.devices.jkbms.jkbms object at 0xb6542178>, '', '', 'screen', None, None)]
2022-03-19 06:44:11,954:INFO:__init__:main@395: Looping 1 commands
2022-03-19 06:44:11,954:INFO:__init__:main@405: Getting results from device: jkbms device - name: unnamed, port: <mppsolar.io.jkbleio.JkBleIO object at 0xb6542190>, protocol: <mppsolar.protocols.jk02.jk02 object at 0xb6542208> for command: , tag: , outputs: screen
2022-03-19 06:44:11,954:INFO:device:run_command@270: Running command 
2022-03-19 06:44:11,954:INFO:jkabstractprotocol:get_full_command@70: Using protocol b'JK02' with 2 commands
2022-03-19 06:44:11,954:DEBUG:jkabstractprotocol:get_command_defn@95: get_command_defn for: getCellData
2022-03-19 06:44:11,954:DEBUG:abstractprotocol:get_command_defn@42: Processing command 'getCellData'
2022-03-19 06:44:11,954:DEBUG:abstractprotocol:get_command_defn@44: Found command getCellData in protocol b'JK02'
2022-03-19 06:44:11,955:DEBUG:jkabstractprotocol:get_full_command@74: self._command = getCellData, self._command_defn = {'name': 'getCellData', 'command_code': '96', 'record_type': '2', 'description': 'BLE Cell Data inquiry', 'help': ' -- queries the ble device for the cell data', 'type': 'QUERY', 'response_type': 'POSITIONAL', 'response': [['Hex2Str', 4, 'Header', ''], ['Hex2Str', 1, 'Record_Type', ''], ['Hex2Int', 1, 'Record_Counter', ''], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell01', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell02', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell03', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell04', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell05', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell06', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell07', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell08', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell09', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell10', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell11', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell12', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell13', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell14', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell15', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell16', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell17', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell18', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell19', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell20', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell21', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell22', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell23', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell24', 'V'], ['discard', 4, 'discard1', ''], ['LittleHex2Short:r/1000', 2, 'Average_Cell_Voltage', 'V'], ['LittleHex2Short:r/1000', 2, 'Delta_Cell_Voltage', 'V'], ['LittleHex2Short:r/1000', 2, 'Current_Balancer', ''], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell01', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell02', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell03', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell04', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell05', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell06', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell07', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell08', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell09', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell10', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell11', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell12', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell13', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell14', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell15', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell16', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell17', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell18', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell19', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell20', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell21', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell22', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell23', 'Ohm'], ['L [... cut ...]
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/devices/device.py", line 309, in run_command
    raw_response = self._port.send_and_receive(
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/io/jkbleio.py", line 34, in send_and_receive
    response = self.ble_get_data(full_command)
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/io/jkbleio.py", line 91, in ble_get_data
    characteristicRead = serviceNotify.getCharacteristics(characteristicReadUuid)[0]
IndexError: list index out of range
root@raspberrypi:~#

Smoeller0, 2022-03-19:
Hello
I Try the jk04 but the same result

jkbms -p C8:47:8C:F2:80:D9 -P JK02 -D
2022-03-19 06:44:11,894:INFO:__init__:main@236: Solar Device Command Utility, version: 0.7.76, remove brackets in ved protocol
2022-03-19 06:44:11,894:INFO:__init__:main@339: Creating device "unnamed" (type: "jkbms") on port "C8:47:8C:F2:80:D9 (porttype=None)" using protocol "JK02"
2022-03-19 06:44:11,897:DEBUG:__init__:main@343: device_class <class 'mppsolar.devices.jkbms.jkbms'>
2022-03-19 06:44:11,897:DEBUG:device:__init__@31: __init__ args ()
2022-03-19 06:44:11,897:DEBUG:device:__init__@32: __init__ kwargs {'name': 'unnamed', 'port': 'C8:47:8C:F2:80:D9', 'protocol': 'JK02', 'baud': 2400, 'porttype': None, 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
2022-03-19 06:44:11,897:DEBUG:device:get_port_type@71: port matches jkble ':'
2022-03-19 06:44:11,897:INFO:device:set_port@154: Using jkbleio for communications
2022-03-19 06:44:11,939:DEBUG:device:set_protocol@96: Protocol JK02
2022-03-19 06:44:11,953:DEBUG:device:__init__@36: __init__ name unnamed, port <mppsolar.io.jkbleio.JkBleIO object at 0xb6542190>, protocol <mppsolar.protocols.jk02.jk02 object at 0xb6542208>
2022-03-19 06:44:11,953:DEBUG:__init__:main@390: Commands [(<mppsolar.devices.jkbms.jkbms object at 0xb6542178>, '', '', 'screen', None, None)]
2022-03-19 06:44:11,954:INFO:__init__:main@395: Looping 1 commands
2022-03-19 06:44:11,954:INFO:__init__:main@405: Getting results from device: jkbms device - name: unnamed, port: <mppsolar.io.jkbleio.JkBleIO object at 0xb6542190>, protocol: <mppsolar.protocols.jk02.jk02 object at 0xb6542208> for command: , tag: , outputs: screen
2022-03-19 06:44:11,954:INFO:device:run_command@270: Running command 
2022-03-19 06:44:11,954:INFO:jkabstractprotocol:get_full_command@70: Using protocol b'JK02' with 2 commands
2022-03-19 06:44:11,954:DEBUG:jkabstractprotocol:get_command_defn@95: get_command_defn for: getCellData
2022-03-19 06:44:11,954:DEBUG:abstractprotocol:get_command_defn@42: Processing command 'getCellData'
2022-03-19 06:44:11,954:DEBUG:abstractprotocol:get_command_defn@44: Found command getCellData in protocol b'JK02'
2022-03-19 06:44:11,955:DEBUG:jkabstractprotocol:get_full_command@74: self._command = getCellData, self._command_defn = {'name': 'getCellData', 'command_code': '96', 'record_type': '2', 'description': 'BLE Cell Data inquiry', 'help': ' -- queries the ble device for the cell data', 'type': 'QUERY', 'response_type': 'POSITIONAL', 'response': [['Hex2Str', 4, 'Header', ''], ['Hex2Str', 1, 'Record_Type', ''], ['Hex2Int', 1, 'Record_Counter', ''], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell01', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell02', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell03', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell04', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell05', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell06', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell07', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell08', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell09', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell10', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell11', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell12', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell13', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell14', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell15', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell16', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell17', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell18', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell19', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell20', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell21', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell22', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell23', 'V'], ['LittleHex2Short:r/1000', 2, 'Voltage_Cell24', 'V'], ['discard', 4, 'discard1', ''], ['LittleHex2Short:r/1000', 2, 'Average_Cell_Voltage', 'V'], ['LittleHex2Short:r/1000', 2, 'Delta_Cell_Voltage', 'V'], ['LittleHex2Short:r/1000', 2, 'Current_Balancer', ''], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell01', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell02', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell03', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell04', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell05', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell06', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell07', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell08', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell09', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell10', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell11', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell12', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell13', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell14', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell15', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell16', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell17', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell18', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell19', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell20', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell21', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell22', 'Ohm'], ['LittleHex2Short:r/1000', 2, 'Resistance_Cell23', 'Ohm'], ['LittleHex2Short:r/ [... cut ...]
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/devices/device.py", line 309, in run_command
    raw_response = self._port.send_and_receive(
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/io/jkbleio.py", line 34, in send_and_receive
    response = self.ble_get_data(full_command)
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/io/jkbleio.py", line 91, in ble_get_data
    characteristicRead = serviceNotify.getCharacteristics(characteristicReadUuid)[0]
IndexError: list index out of range

jmck82, 2022-03-21:
Good luck, been trying to fix this for months.   See my topic

https://github.com/jblance/mpp-solar/issues/170

Smoeller0, 2022-03-22:
This is very bad. Is the Problem in rs485 too?
I Just Open my working Balancer and solder wires to the Bluetooth module to Read the Data and Try to understand...

jmck82, 2022-03-22:
Yup, I worked on it for I don't know how many hours, and got some of it to work, but it was spotty at best.    I honestly got it 100% because of this so when it become more of a chore than it was worth I just made my own with an ESP32 and ADS1115 with a couple of relays.   Got one of those 5a active balancers and haven't looked back.

Smoeller0, 2022-04-08:
I Record first Data from the BMS.

[85,170,235,144,2,8,89,14,4,14,5,14,156,15,151,15,36,14,148,15,3,14,38,14,254,13,68,14,40,14,160,15,88,15,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,255,63,0,0,162,14,162,1,12,9,154,0,179,0,232,0,186,0,205,0,211,0,181,0,178,0,197,0,15,1,250,0,245,0,196,0,186,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,219,204,0,0,0,0,0,0,0,0,0,0,48,248,48,248,207,0,0,0,40,248,2,99,62,156,0,0,64,156,0,0,22,0,0,0,139,238,13,0,100,0,192,2,161,7,34,0,1,1,172,6,0,0,0,0,0,0,0,0,0,0,0,0,1,0,1,0,0,0,250,3,0,0,0,0,65,255,64,64,156,0,0,0,88,170,253,255,0,0,0,1,0,2,0,0,182,43,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,128]

 Where Battery Voltage ist int 16 start at byte 7+8 [85,170,235,144,2,8,89,14,4,14,5,14
(14*255 )+89= 3,659V

PR #221 (2022-05-09): Make JK02 v8 compatible
I assume this change works fine with JK-BMS software/hardware version >= v6. I would be happy if somebody using a JK-BMS v3 could verify this isn't a breaking change.

Fixes: #202
Fixes: #170
Fixes: #138
closed 2022-05-09 by jblance