jblance/mpp-solar #203: JKBMS Charging and discharging with wrong values

miguel300477, 2022-03-22:
Hi,

I been monitoring my JKBMS and i noticed for a long while that the charging and discharging current goes to to the oposite number after 32amps.

So if im charging it shows for exemple 20amps but after it passes 32amps it goes negative value. the same is happening with the discharge...

Anyone has the same issue?

miguel300477, 2022-03-22:
[image]

sebmueller, 2022-09-01:
Hello,
I got similar readings. Not sure what it means. After the Battery is full, the Values are pending +- 10A

[image]

syssi, 2022-09-01:
Do you use BLE (JK02/JK04) or a UART-TTL connection?

sebmueller, 2022-09-01:
I use a Bluetooth Connection.

geobitus, 2023-12-24:
Hi there
First of all thanks for sharing that great work.
I am using a BLE connection to an JK BMS,
I encountered the same issue and found that this sign switching is related to data decoding,
When eg. voltage_cellxx/resistance_cellxx all decimals are present and decoded correctly the sign is correct,
When from voltage_cellxx/resistance_cellxx some decimal are missing eg. 3.38 instead of 3.885 etc..the sign is wrong, 
Hope this found will help.

jblance, 2023-12-29:
thanks for your feedback, if you can catch that failure in a debug run it would be helpfull

geobitus, 2024-01-02:
Happy one for all.
That was the worst feedback ever from my side, not enough tested....:-(
Digging deeper I found that the sign switching is related to '["LittleHex2Short:r /1000", 2 , "Current_Charge" , "A"]' found in /mppsolar/protocols/jjkv11.py,
My empirically tested solution to get rid of this behave was to change LittleHex2Short to BigHex2Short and add some math after, can not reproduce here because I don't want to produce to anyone hiccups or hart attacks :-)
Hope this help.

[image]