jblance/mpp-solar #206: QPIGS and Invalid response CRC

HyperHubs, 2022-04-01:
I have a "Powmr POW-HVM 6KW" inverter. I connected it via rs232 and most of the commands from the PI30 protocol work fine for me, but for the QPIGS command I get a message about an incorrect crc in the response.
Could someone advise what I should change to make it work?

below DEBUG:

root@DomoRaspi:/usr/local/lib/python3.9/dist-packages/mppsolar/protocols# mpp-solar -p /dev/ttyUSB0 -b 2400 -P PI30 -c QPIGS -D
2022-04-01 13:21:47,167:INFO:__init__:main@236: Solar Device Command Utility, version: 0.7.76, remove brackets in ved protocol
2022-04-01 13:21:47,167:INFO:__init__:main@339: Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=None)" using protocol "PI30"
2022-04-01 13:21:47,172:DEBUG:__init__:main@343: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2022-04-01 13:21:47,173:DEBUG:device:__init__@31: __init__ args ()
2022-04-01 13:21:47,173:DEBUG:device:__init__@32: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB0', 'protocol': 'PI30', 'baud': 2400, 'porttype': None, 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None}
2022-04-01 13:21:47,174:DEBUG:device:get_port_type@86: port matches ttyusb
2022-04-01 13:21:47,174:INFO:device:set_port@160: Using serialio for communications
2022-04-01 13:21:47,189:DEBUG:device:set_protocol@96: Protocol PI30
2022-04-01 13:21:47,256:DEBUG:device:__init__@36: __init__ name unnamed, port <mppsolar.io.serialio.SerialIO object at 0x7659e220>, protocol <mppsolar.protocols.pi30.pi30 object at 0x7659e268>
2022-04-01 13:21:47,256:DEBUG:__init__:main@390: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x7659e2c8>, 'QPIGS', 'QPIGS', 'screen', None, None)]
2022-04-01 13:21:47,257:INFO:__init__:main@395: Looping 1 commands
2022-04-01 13:21:47,257:INFO:__init__:main@405: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.serialio.SerialIO object at 0x7659e220>, protocol: <mppsolar.protocols.pi30.pi30 object at 0x7659e268> for command: QPIGS, tag: QPIGS, outputs: screen
2022-04-01 13:21:47,258:INFO:device:run_command@270: Running command QPIGS
2022-04-01 13:21:47,258:INFO:abstractprotocol:get_full_command@31: Using protocol b'PI30' with 36 commands
2022-04-01 13:21:47,258:DEBUG:protocol_helpers:crcPI@299: Calculating CRC for b'QPIGS'
2022-04-01 13:21:47,259:DEBUG:protocol_helpers:crcPI@349: Generated CRC 0xb7 0xa9 0xb7a9
2022-04-01 13:21:47,260:DEBUG:abstractprotocol:get_full_command@38: full command: b'QPIGS\xb7\xa9\r'
2022-04-01 13:21:47,260:INFO:device:run_command@296: full command b'QPIGS\xb7\xa9\r' for command QPIGS
2022-04-01 13:21:47,260:DEBUG:abstractprotocol:get_command_defn@42: Processing command 'QPIGS'
2022-04-01 13:21:47,261:DEBUG:abstractprotocol:get_command_defn@44: Found command QPIGS in protocol b'PI30'
2022-04-01 13:21:47,261:DEBUG:serialio:send_and_receive@19: port /dev/ttyUSB0, baudrate 2400
2022-04-01 13:21:47,266:DEBUG:serialio:send_and_receive@22: Executing command via serialio...
2022-04-01 13:21:47,772:DEBUG:serialio:send_and_receive@30: serial response was: b'(236.0 49.9 236.0 49.9 0000 0000 000 315 01.30 000 000 0028 00.0 000.0 00.00 00000 00010000 00 00 000\x000 010\x0f\x00\r'
2022-04-01 13:21:48,834:DEBUG:device:run_command@315: Send and Receive Response b'(236.0 49.9 236.0 49.9 0000 0000 000 315 01.30 000 000 0028 00.0 000.0 00.00 00000 00010000 00 00 000\x000 010\x0f\x00\r'
2022-04-01 13:21:48,835:INFO:abstractprotocol:decode@175: response passed to decode: b'(236.0 49.9 236.0 49.9 0000 0000 000 315 01.30 000 000 0028 00.0 000.0 00.00 00000 00010000 00 00 000\x000 010\x0f\x00\r'
2022-04-01 13:21:48,836:DEBUG:protocol_helpers:crcPI@299: Calculating CRC for b'(236.0 49.9 236.0 49.9 0000 0000 000 315 01.30 000 000 0028 00.0 000.0 00.00 00000 00010000 00 00 000\x000 010'
2022-04-01 13:21:48,838:DEBUG:protocol_helpers:crcPI@349: Generated CRC 0x20 0x80 0x2080
2022-04-01 13:21:48,838:INFO:abstractprotocol:decode@179: Invalid response CRC
2022-04-01 13:21:48,839:INFO:device:run_command@334: Decoded response {'ERROR': ['Invalid response CRC', '']}
2022-04-01 13:21:48,839:DEBUG:__init__:main@409: results: {'ERROR': ['Invalid response CRC', '']}
2022-04-01 13:21:48,840:INFO:__init__:get_outputs@26: attempting to create output processor: screen
2022-04-01 13:21:48,846:DEBUG:screen:__init__@16: processor.screen __init__ kwargs {}
2022-04-01 13:21:48,847:DEBUG:__init__:main@415: Using output filter: None
2022-04-01 13:21:48,847:INFO:screen:output@19: Using output processor: screen
2022-04-01 13:21:48,848:DEBUG:screen:output@20: kwargs {'data': {'ERROR': ['Invalid response CRC', '']}, 'tag': 'QPIGS', 'mqtt_broker': 'localhost', 'mqtt_port': 1883, 'mqtt_user': None, 'mqtt_pass': None, 'mqtt_topic': 'mpp-solar', 'filter': None, 'excl_filter': None, 'keep_case': False}
Parameter                       Value           Unit
error                           Invalid response CRC
2022-04-01 13:21:48,849:DEBUG:__init__:main@435: Not daemon, so not looping

jblance, 2022-04-07:
can you upgrade to the latest version - if you are using pip then uninstall `mpp-solar` and install `mppsolar` (note lack of dash)

rucati, 2022-05-20:
Hi,
Suddenly I am dealing with the same issue, all commands go fine, but the needed QPIGS fails. I am running 0.12.5 on Rpi.

sudo mpp-solar -p /dev/ttyUSB0 -c QPIGS -D
2022-05-20 20:40:16,734:INFO:__init__:main@218: Solar Device Command Utility, version: 0.12.5, tweak jkbleio for getInfo and to return first 300 bytes
2022-05-20 20:40:16,735:DEBUG:__init__:main@232: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x765762d0>, _isConnected=False)
2022-05-20 20:40:16,735:DEBUG:__init__:main@234: udp port 5555
2022-05-20 20:40:16,736:DEBUG:__init__:main@237: Using Mongo None with mppsolar
2022-05-20 20:40:16,736:INFO:__init__:main@338: Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=None)" using protocol "PI30"
2022-05-20 20:40:16,742:DEBUG:__init__:main@341: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2022-05-20 20:40:16,743:DEBUG:device:__init__@33: __init__ args ()
2022-05-20 20:40:16,743:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB0', 'protocol': 'PI30', 'baud': 2400, 'porttype': None, 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x765762d0>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2022-05-20 20:40:16,744:DEBUG:__init__:get_port_type@65: port matches ttyusb
2022-05-20 20:40:16,744:INFO:__init__:get_port@107: Using serialio for communications
2022-05-20 20:40:16,760:DEBUG:__init__:get_protocol@12: Protocol PI30
2022-05-20 20:40:16,767:DEBUG:device:__init__@39: __init__ name unnamed, port <mppsolar.inout.serialio.SerialIO object at 0x7600b030>, protocol <mppsolar.protocols.pi30.pi30 object at 0x7600b1d0>
2022-05-20 20:40:16,767:DEBUG:__init__:main@390: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x75ffe5d0>, 'QPIGS', 'QPIGS', 'screen', None, None)]
2022-05-20 20:40:16,767:INFO:__init__:main@395: Looping 1 commands
2022-05-20 20:40:16,768:INFO:__init__:main@406: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.serialio.SerialIO object at 0x7600b030>, protocol: <mppsolar.protocols.pi30.pi30 object at 0x7600b1d0> for command: QPIGS, tag: QPIGS, outputs: screen
2022-05-20 20:40:16,768:INFO:device:run_command@274: Running command QPIGS
2022-05-20 20:40:16,769:INFO:abstractprotocol:get_full_command@50: Using protocol b'PI30' with 40 commands
2022-05-20 20:40:16,769:DEBUG:protocol_helpers:crcPI@313: Calculating CRC for b'QPIGS'
2022-05-20 20:40:16,769:DEBUG:protocol_helpers:crcPI@363: Generated CRC 0xb7 0xa9 0xb7a9
2022-05-20 20:40:16,770:DEBUG:abstractprotocol:get_full_command@56: full command: b'QPIGS\xb7\xa9\r'
2022-05-20 20:40:16,770:INFO:device:run_command@303: full command b'QPIGS\xb7\xa9\r' for command QPIGS
2022-05-20 20:40:16,770:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'QPIGS'
2022-05-20 20:40:16,771:DEBUG:abstractprotocol:get_command_defn@62: Found command QPIGS in protocol b'PI30'
2022-05-20 20:40:16,771:DEBUG:serialio:send_and_receive@19: port /dev/ttyUSB0, baudrate 2400
2022-05-20 20:40:16,776:DEBUG:serialio:send_and_receive@22: Executing command via serialio...
2022-05-20 20:40:17,737:DEBUG:serialio:send_and_receive@30: serial response was: b'(\x002\x004\x002\x00.\x008\x00 \x005\x000\x00.\x000\x00 \x002\x004\x002\x00.\x008\x00 \x005\x000\x00.\x000\x00 \x000\x000\x000\x000\x00 \x000\x000\x000\x000\x00 \x000\x000\x000\x00 \x003\x007\x001\x00 \x002\x004\x00.\x008\x000\x00 \x000\x000\x000\x00 \x000\x004\x005\x00 \x000\x000\x004\x007\x00 \x000\x000\x00.\x000\x00 \x001\x001\x009\x00.\x009\x00 \x000\x000\x00.\x000\x000\x00 \x000\x000\x000\x000\x000\x00 \x000\x000\x000\x001\x000\x001\x001\x001\x00 \x000\x000\x00 \x000\x000\x00 \x000\x000\x000\x00\x00\x00\x00\x00 \x00\x00\x001\x000\x00{\x00\x91\x00\r'
2022-05-20 20:40:17,740:DEBUG:device:run_command@322: Send and Receive Response b'(\x002\x004\x002\x00.\x008\x00 \x005\x000\x00.\x000\x00 \x002\x004\x002\x00.\x008\x00 \x005\x000\x00.\x000\x00 \x000\x000\x000\x000\x00 \x000\x000\x000\x000\x00 \x000\x000\x000\x00 \x003\x007\x001\x00 \x002\x004\x00.\x008\x000\x00 \x000\x000\x000\x00 \x000\x004\x005\x00 \x000\x000\x004\x007\x00 \x000\x000\x00.\x000\x00 \x001\x001\x009\x00.\x009\x00 \x000\x000\x00.\x000\x000\x00 \x000\x000\x000\x000\x000\x00 \x000\x000\x000\x001\x000\x001\x001\x001\x00 \x000\x000\x00 \x000\x000\x00 \x000\x000\x000\x00\x00\x00\x00\x00 \x00\x00\x001\x000\x00{\x00\x91\x00\r'
2022-05-20 20:40:17,741:INFO:abstractprotocol:decode@196: response passed to decode: b'(\x002\x004\x002\x00.\x008\x00 \x005\x000\x00.\x000\x00 \x002\x004\x002\x00.\x008\x00 \x005\x000\x00.\x000\x00 \x000\x000\x000\x000\x00 \x000\x000\x000\x000\x00 \x000\x000\x000\x00 \x003\x007\x001\x00 \x002\x004\x00.\x008\x000\x00 \x000\x000\x000\x00 \x000\x004\x005\x00 \x000\x000\x004\x007\x00 \x000\x000\x00.\x000\x00 \x001\x001\x009\x00.\x009\x00 \x000\x000\x00.\x000\x000\x00 \x000\x000\x000\x000\x000\x00 \x000\x000\x000\x001\x000\x001\x001\x001\x00 \x000\x000\x00 \x000\x000\x00 \x000\x000\x000\x00\x00\x00\x00\x00 \x00\x00\x001\x000\x00{\x00\x91\x00\r'
2022-05-20 20:40:17,741:DEBUG:protocol_helpers:crcPI@313: Calculating CRC for b'(\x002\x004\x002\x00.\x008\x00 \x005\x000\x00.\x000\x00 \x002\x004\x002\x00.\x008\x00 \x005\x000\x00.\x000\x00 \x000\x000\x000\x000\x00 \x000\x000\x000\x000\x00 \x000\x000 [... cut ...]
2022-05-20 20:40:17,750:DEBUG:screen:output@20: kwargs {'data': {'ERROR': ['Invalid response CRC', '']}, 'tag': 'QPIGS', 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x765762d0>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar', 'filter': None, 'excl_filter': None, 'keep_case': False}
Parameter                       Value           Unit
error                           Invalid response CRC
2022-05-20 20:40:17,751:DEBUG:__init__:main@438: Not daemon, so not looping

HyperHubs, 2022-07-15:
I upgraded to the latest version. unfortunately, when sending the QPIGS command, I still get an incorrect CRC :(
On various forums I read that some inverters return more data or inverter ends line with a LF instead of a CR.
can you help to resolve my problem? what can i do to diagnose what the problem is?

this is my last query :
 /usr/local/bin/mpp-solar -p /dev/ttyUSB1 -c QPIGS -D
2022-07-15 21:32:58,778:INFO:__init__:main@218: Solar Device Command Utility, version: 0.12.5, tweak jkbleio for getInfo and to return first 300 bytes
2022-07-15 21:32:58,778:DEBUG:__init__:main@232: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x766a7fe8>, _isConnected=False)
2022-07-15 21:32:58,779:DEBUG:__init__:main@234: udp port 5555
2022-07-15 21:32:58,779:DEBUG:__init__:main@237: Using Mongo None with mppsolar
2022-07-15 21:32:58,779:INFO:__init__:main@337: Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB1 (porttype=None)" using protocol "PI30"
2022-07-15 21:32:58,786:DEBUG:__init__:main@341: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2022-07-15 21:32:58,787:DEBUG:device:__init__@33: __init__ args ()
2022-07-15 21:32:58,787:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB1', 'protocol': 'PI30', 'baud': 2400, 'porttype': None, 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x766a7fe8>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2022-07-15 21:32:58,787:DEBUG:__init__:get_port_type@65: port matches ttyusb
2022-07-15 21:32:58,788:INFO:__init__:get_port@107: Using serialio for communications
2022-07-15 21:32:58,799:DEBUG:__init__:get_protocol@12: Protocol PI30
2022-07-15 21:32:58,805:DEBUG:device:__init__@38: __init__ name unnamed, port <mppsolar.inout.serialio.SerialIO object at 0x76092fa0>, protocol <mppsolar.protocols.pi30.pi30 object at 0x75fbf940>
2022-07-15 21:32:58,806:DEBUG:__init__:main@390: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x76092f88>, 'QPIGS', 'QPIGS', 'screen', None, None)]
2022-07-15 21:32:58,806:INFO:__init__:main@395: Looping 1 commands
2022-07-15 21:32:58,807:INFO:__init__:main@405: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.serialio.SerialIO object at 0x76092fa0>, protocol: <mppsolar.protocols.pi30.pi30 object at 0x75fbf940> for command: QPIGS, tag: QPIGS, outputs: screen
2022-07-15 21:32:58,807:INFO:device:run_command@274: Running command QPIGS
2022-07-15 21:32:58,807:INFO:abstractprotocol:get_full_command@50: Using protocol b'PI30' with 40 commands
2022-07-15 21:32:58,807:DEBUG:protocol_helpers:crcPI@313: Calculating CRC for b'QPIGS'
2022-07-15 21:32:58,808:DEBUG:protocol_helpers:crcPI@363: Generated CRC 0xb7 0xa9 0xb7a9
2022-07-15 21:32:58,808:DEBUG:abstractprotocol:get_full_command@56: full command: b'QPIGS\xb7\xa9\r'
2022-07-15 21:32:58,808:INFO:device:run_command@303: full command b'QPIGS\xb7\xa9\r' for command QPIGS
2022-07-15 21:32:58,809:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'QPIGS'
2022-07-15 21:32:58,809:DEBUG:abstractprotocol:get_command_defn@62: Found command QPIGS in protocol b'PI30'
2022-07-15 21:32:58,809:DEBUG:serialio:send_and_receive@19: port /dev/ttyUSB1, baudrate 2400
2022-07-15 21:32:58,813:DEBUG:serialio:send_and_receive@22: Executing command via serialio...
2022-07-15 21:32:59,319:DEBUG:serialio:send_and_receive@30: serial response was: b'(219.2 50.0 231.0 50.0 0993 0789 016 387 52.30 000 061 0037 00.0 000.0 00.00 00019 00010000 00 00 000\x020 010\x98\x00\r'
2022-07-15 21:33:00,456:DEBUG:device:run_command@322: Send and Receive Response b'(219.2 50.0 231.0 50.0 0993 0789 016 387 52.30 000 061 0037 00.0 000.0 00.00 00019 00010000 00 00 000\x020 010\x98\x00\r'
2022-07-15 21:33:00,457:INFO:abstractprotocol:decode@196: response passed to decode: b'(219.2 50.0 231.0 50.0 0993 0789 016 387 52.30 000 061 0037 00.0 000.0 00.00 00019 00010000 00 00 000\x020 010\x98\x00\r'
2022-07-15 21:33:00,457:DEBUG:protocol_helpers:crcPI@313: Calculating CRC for b'(219.2 50.0 231.0 50.0 0993 0789 016 387 52.30 000 061 0037 00.0 000.0 00.00 00019 00010000 00 00 000\x020 010'
2022-07-15 21:33:00,459:DEBUG:protocol_helpers:crcPI@363: Generated CRC 0x3c 0x70 0x3c70
2022-07-15 21:33:00,459:INFO:abstractprotocol:decode@200: Invalid response CRC
2022-07-15 21:33:00,459:INFO:device:run_command@341: Decoded response {'ERROR': ['Invalid response CRC', '']}
2022-07-15 21:33:00,460:DEBUG:__init__:main@409: results: {'ERROR': ['Invalid response CRC', '']}
2022-07-15 21:33:00,460:INFO:__init__:get_output@34: attempting to create output processor: screen
2022-07-15 21:33:00,464:DEBUG:screen:__init__@16: processor.screen __init__ kwargs {}
2022-07-15 21:33:00,464:DEBUG:__init__:main@415: Using output filter: None
2022-07-15 21:33:00,465:INFO:screen:output@19: Using output processor: screen
2022-07-15 21:33:00,465:DEBUG:screen:output@20: kwargs {'data': {'ERROR': ['Invalid response CRC', '']}, 'tag': 'QPIGS', 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x766a7fe8>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar', 'filter': None, 'excl_filter': None, 'keep_case': False}
Parameter                       Value           Unit
error                           Invalid response CRC
2022-07-15 21:33:00,466:DEBUG:__init__:main@438: Not daemon, so not looping

Dudhr, 2022-08-01:
interesting, this "invalid CRC response" happens on my pi on USB0, but works on USB1
found it by coincidence. i had another device  plugged all the time, so usb-serial cable was on USB1. then i cleaned my setup, only solar on pi - it won't work....
Later i reverted it to USB1 by plugging something to usb before usb-serial and it works again on USB1
I don't know the reason for that :/

jblance, 2023-01-08:
looks resolved?

closed 2023-01-08 by jblance