jblance/mpp-solar #208: Installing on a Raspberry Pi

gcalianu, 2022-04-05:
When I have tried to install on a Raspberry Pi 3 I  have encountered the following error (Python 3.6.5).

```
pi@raspberrypi:~ $ sudo pip install mpp-solar
Collecting mpp-solar
  Using cached https://files.pythonhosted.org/packages/c2/36/fad21f11cdadabbde9ff6a031975ed96a56bb8f8c64120885afe7664e50b/mpp-solar-0.4.2.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-SaIO7O/mpp-solar/setup.py", line 18, in <module>
        with open(path.join(here, 'pypi_README.md'), encoding='utf-8') as f:
      File "/usr/lib/python2.7/codecs.py", line 896, in open
        file = __builtin__.open(filename, mode, buffering)
    IOError: [Errno 2] No such file or directory: '/tmp/pip-build-SaIO7O/mpp-solar/pypi_README.md'

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-SaIO7O/mpp-solar/

```
Any clue about this?
Thank you.

jblance, 2022-04-05:
It looks like python2.7 is getting in the way
Try using pip3 for the install (may need to install it first)

jblance, 2022-04-05:
Also looks like a very old version. Maybe try -f (I think) with the install
to grab the current version

gcalianu, 2022-04-06:
Thank you for your answer. I used the newest version of Raspberry OS and mpp-solar has installed without problems.

closed 2022-04-07 by jblance