jblance/mpp-solar #209: Using mmp-solar with PIP 8048MAX

gcalianu, 2022-04-06:
I tried mpp-solar with PIP 8048MAX which have two inputs for PVs. It is working but --getstatus shows the values only for one input. There is any chance to read the other input?
Thank you.

jblance, 2022-04-07:
try `-c QPGS0` command as well

Buelo77, 2022-10-04:
Hi,
i have a Debian running, but no success with the USB connection. Maybe someone has a hint?

fhemadmin@Hausautomation:~$ lsusb 
Bus 002 Device 002: ID 8087:8000 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8008 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 003 Device 003: ID 0665:5161 Cypress Semiconductor USB to Serial
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

fhemadmin@Hausautomation:~$ ls /dev/
autofs         console          dvd        initctl       mem     ptmx    sda1      sr0     tty10  tty18  tty25  tty32  tty4   tty47  tty54  tty61  ttyS2    vcs3   vcsa3  vcsu3        vhost-net
block          core             dvdrw      input         mqueue  ptp0    sda2      stderr  tty11  tty19  tty26  tty33  tty40  tty48  tty55  tty62  ttyS3    vcs4   vcsa4  vcsu4        vhost-vsock
bsg            cpu              fd         kmsg          net     pts     sda5      stdin   tty12  tty2   tty27  tty34  tty41  tty49  tty56  tty63  uhid     vcs5   vcsa5  vcsu5        watchdog
btrfs-control  cpu_dma_latency  full       kvm           null    random  sg0       stdout  tty13  tty20  tty28  tty35  tty42  tty5   tty57  tty7   uinput   vcs6   vcsa6  vcsu6        watchdog0
bus            cuse             fuse       log           nvram   rfkill  sg1       tpm0    tty14  tty21  tty29  tty36  tty43  tty50  tty58  tty8   urandom  vcs7   vcsa7  vcsu7        watchdog1
cdrom          disk             hpet       loop-control  port    rtc     shm       tty     tty15  tty22  tty3   tty37  tty44  tty51  tty59  tty9   vcs      vcsa   vcsu   vfio         zero
cdrw           dri              hugepages  mapper        ppp     rtc0    snapshot  tty0    tty16  tty23  tty30  tty38  tty45  tty52  tty6   ttyS0  vcs1     vcsa1  vcsu1  vga_arbiter
char           drm_dp_aux0      hwrng      mei0          psaux   sda     snd       tty1    tty17  tty24  tty31  tty39  tty46  tty53  tty60  ttyS1  vcs2     vcsa2  vcsu2  vhci

syssi, 2022-10-04:
Please unplug the USB-to-serial adapter and plug it in again. Check the output of `dmesg`. You should see something like this:

```
[19001.104396] usbserial: USB Serial support registered for ch341-uart
[19001.104413] ch341 2-11:1.0: ch341-uart converter detected
[19001.104791] usb 2-11: ch341-uart converter now attached to ttyUSB0
```

As soon you know the device name check the permissions of the device:

```
ls -l /dev/ttyUSB0
```

I assume you have to add your linux user to the group `dialout` by

```
sudo adduser username dialout
```

Buelo77, 2022-10-04:
I have found this:
T:  Bus=03 Lev=01 Prnt=01 Port=09 Cnt=01 Dev#=  3 Spd=12  MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS= 8 #Cfgs=  1
P:  Vendor=0665 ProdID=5161 Rev=00.02
S:  Manufacturer=STMicroelectronics
S:  Product=HID in FS Mode
C:  #Ifs= 0 Cfg#= 0 Atr= MxPwr=
cat: '/sys/bus/usb/devices/usb3/3-10/3-*:?.*/bInterfaceNumber': Datei oder Verzeichnis nicht gefunden
cat: '/sys/bus/usb/devices/usb3/3-10/3-*:?.*/bAlternateSetting': Datei oder Verzeichnis nicht gefunden
cat: '/sys/bus/usb/devices/usb3/3-10/3-*:?.*/bNumEndpoints': Datei oder Verzeichnis nicht gefunden
cat: '/sys/bus/usb/devices/usb3/3-10/3-*:?.*/bInterfaceClass': Datei oder Verzeichnis nicht gefunden
cat: '/sys/bus/usb/devices/usb3/3-10/3-*:?.*/bInterfaceSubClass': Datei oder Verzeichnis nicht gefunden
cat: '/sys/bus/usb/devices/usb3/3-10/3-*:?.*/bInterfaceProtocol': Datei oder Verzeichnis nicht gefunden
/usr/bin/usb-devices: 79: printf: (none): expected numeric value
I:  If#=0x Alt= 0 #EPs= 0 Cls=() Sub= Prot= Driver=

[    1.997059] usb 3-10: new full-speed USB device number 3 using xhci_hcd
[    2.148065] usb 3-10: New USB device found, idVendor=0665, idProduct=5161, bcdDevice= 0.02
[    2.148068] usb 3-10: New USB device strings: Mfr=3, Product=1, SerialNumber=0
[    2.148069] usb 3-10: Product: HID in FS Mode
[    2.148071] usb 3-10: Manufacturer: STMicroelectronics
[    2.149819] usb 3-10: can't set config #1, error -71

Buelo77, 2022-10-04:
fhemadmin@Hausautomation:~$ ls -l /dev/ttyUSB0
ls: Zugriff auf '/dev/ttyUSB0' nicht möglich: Datei oder Verzeichnis nicht gefunden
fhemadmin@Hausautomation:~$ 

fhemadmin@Hausautomation:~$ sudo adduser fhemadmin dialout
Füge Benutzer »fhemadmin« der Gruppe »dialout« hinzu ...
Benutzer fhemadmin wird zur Gruppe dialout hinzugefügt.
Fertig.

syssi, 2022-10-04:
Oh. I was on the wrong track. I did assume you were using an ordinary USB-to-Serial adapter. It looks like your inverter provides USB connectivity and exposes a HID device instead a serial interface device.

87524, 2022-10-17:
Hi this is the result on dual MPPT tracker Easun 8kw (Axpert MAX8000) with normal USB cable and QPGS0 command.
I still cannot make QPIGS2 to work and get errors.
 And I changed 4 USB date cables until it worked . Looks like not all are the same quality. I was testing with the Watch power local installation and the cable that worked on it also worked with the mpp-solar

#  mpp-solar -p /dev/hidraw0 -P PI30MAX  -c QPGS0
Command: QPGS0 - Parallel Information inquiry
------------------------------------------------------------
Parameter                     	Value           Unit
parallel_instance_number      	Not valid      	    
serial_number                 	92932203100842 	    
work_mode                     	Battery Mode   	    
fault_code                    	No fault       	    
grid_voltage                  	231.6          	V   
grid_frequency                	49.98          	Hz  
ac_output_voltage             	229.7          	V   
ac_output_frequency           	49.95          	Hz  
ac_output_apparent_power      	368            	VA  
ac_output_active_power        	163            	W   
load_percentage               	4              	%   
battery_voltage               	52.9           	V   
battery_charging_current      	0              	A   
battery_capacity              	47             	%   
pv1_input_voltage             	0.0            	V   
total_charging_current        	0              	A   
total_ac_output_apparent_power	368            	VA  
total_output_active_power     	163            	W   
total_ac_output_percentage    	2              	%   
is_scc_ok                     	0              	bool
is_ac_charging                	0              	bool
is_scc_charging               	0              	bool
is_battery_over_voltage       	0              	bool
is_battery_under_voltage      	0              	bool
is_line_lost                  	0              	bool
is_load_on                    	1              	bool
is_configuration_changed      	0              	bool
output_mode                   	single machine 	    
charger_source_priority       	Utility first  	    
max_charger_current           	30             	A   
max_charger_range             	120            	A   
max_ac_charger_current        	2              	A   
pv1_input_current             	0              	A   
battery_discharge_current     	4              	A   
pv2_input_voltage             	0.0            	V   
pv2_input_current             	0              	A

87524, 2022-10-17:
> Oh. I was on the wrong track. I did assume you were using an ordinary USB-to-Serial adapter. It looks like your inverter provides USB connectivity and exposes a HID device instead a serial interface device.

Hi after reading all issues and testing on my setup it looks like those using ttyUSB have QPIGS2 working and those like me with hidraw0 (usb connection to the inverter) get errors

`root@raspberry:~# mpp-solar -p /dev/hidraw0 -P PI30MAX -c QPIGS2 -D
2022-10-17 19:06:59,799:INFO:__init__:main@218: Solar Device Command Utility, version: 0.12.5, tweak jkbleio for getInfo and to return first 300 bytes
2022-10-17 19:06:59,803:DEBUG:__init__:main@232: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0xf755a9e8>, _isConnected=False)
2022-10-17 19:06:59,803:DEBUG:__init__:main@234: udp port 5555
2022-10-17 19:06:59,804:DEBUG:__init__:main@237: Using Mongo None with mppsolar
2022-10-17 19:06:59,804:INFO:__init__:main@337: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI30MAX"
2022-10-17 19:06:59,809:DEBUG:__init__:main@341: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2022-10-17 19:06:59,809:DEBUG:device:__init__@33: __init__ args ()
2022-10-17 19:06:59,811:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': '/dev/hidraw0', 'protocol': 'PI30MAX', 'baud': 2400, 'porttype': None, 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0xf755a9e8>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2022-10-17 19:06:59,811:DEBUG:__init__:get_port_type@38: port matches hidraw
2022-10-17 19:06:59,811:INFO:__init__:get_port@89: Using hidrawio for communications
2022-10-17 19:06:59,813:DEBUG:__init__:get_protocol@12: Protocol PI30MAX
2022-10-17 19:06:59,831:DEBUG:device:__init__@38: __init__ name unnamed, port <mppsolar.inout.hidrawio.HIDRawIO object at 0xf6e506d0>, protocol <mppsolar.protocols.pi30max.pi30max object at 0xf6e506e8>
2022-10-17 19:06:59,836:DEBUG:__init__:main@390: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0xf6e50640>, 'QPIGS2', 'QPIGS2', 'screen', None, None)]
2022-10-17 19:06:59,841:INFO:__init__:main@395: Looping 1 commands
2022-10-17 19:06:59,842:INFO:__init__:main@405: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.hidrawio.HIDRawIO object at 0xf6e506d0>, protocol: <mppsolar.protocols.pi30max.pi30max object at 0xf6e506e8> for command: QPIGS2, tag: QPIGS2, outputs: screen
2022-10-17 19:06:59,842:INFO:device:run_command@274: Running command QPIGS2
2022-10-17 19:06:59,842:INFO:abstractprotocol:get_full_command@50: Using protocol b'PI30MAX' with 30 commands
2022-10-17 19:06:59,842:DEBUG:protocol_helpers:crcPI@313: Calculating CRC for b'QPIGS2'
2022-10-17 19:06:59,842:DEBUG:protocol_helpers:crcPI@363: Generated CRC 0x68 0x2d 0x682d
2022-10-17 19:06:59,843:DEBUG:abstractprotocol:get_full_command@56: full command: b'QPIGS2h-\r'
2022-10-17 19:06:59,843:INFO:device:run_command@303: full command b'QPIGS2h-\r' for command QPIGS2
2022-10-17 19:06:59,845:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'QPIGS2'
2022-10-17 19:06:59,845:DEBUG:abstractprotocol:get_command_defn@62: Found command QPIGS2 in protocol b'PI30MAX'
2022-10-17 19:06:59,903:DEBUG:hidrawio:send_and_receive@29: length of to_send: 9
2022-10-17 19:06:59,903:DEBUG:hidrawio:send_and_receive@40: 2 chunk send
2022-10-17 19:07:01,013:DEBUG:hidrawio:send_and_receive@69: usb response was: b'(NAKss\r'
2022-10-17 19:07:01,015:DEBUG:device:run_command@322: Send and Receive Response b'(NAKss\r'
2022-10-17 19:07:01,016:INFO:abstractprotocol:decode@196: response passed to decode: b'(NAKss\r'
2022-10-17 19:07:01,017:INFO:abstractprotocol:decode@200: NAK
2022-10-17 19:07:01,017:INFO:device:run_command@341: Decoded response {'ERROR': ['NAK', '']}
2022-10-17 19:07:01,017:DEBUG:__init__:main@409: results: {'ERROR': ['NAK', '']}
2022-10-17 19:07:01,018:INFO:__init__:get_output@34: attempting to create output processor: screen
2022-10-17 19:07:01,019:DEBUG:screen:__init__@16: processor.screen __init__ kwargs {}
2022-10-17 19:07:01,019:DEBUG:__init__:main@415: Using output filter: None
2022-10-17 19:07:01,020:INFO:screen:output@19: Using output processor: screen
2022-10-17 19:07:01,020:DEBUG:screen:output@20: kwargs {'data': {'ERROR': ['NAK', '']}, 'tag': 'QPIGS2', 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0xf755a9e8>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar', 'filter': None, 'excl_filter': None, 'keep_case': False}
Parameter                     	Value           Unit
error                         	NAK            	    
2022-10-17 19:07:01,021:DEBUG:__init__:main@438: Not daemon, so not looping
`

closed 2023-01-08 by jblance