jblance/mpp-solar #210: Why is _command and _command_description removed from json_mqtt?

volkerjaenisch, 2022-04-09:
Dear @jblance !

I do not understand why you drop _command and _command_description in json_mqtt output mode.
The code documentation mentions influxDB. But now there are separate influxDB outputs available?
IMHO "json_mqtt" should emit the same information as json_mqtt but over mqtt transport?

I think this part of the code is broken.

```@python
        # Remove command and _command_description
        cmd = data.pop("_command", None)
        data.pop("_command_description", None)
        data.pop("raw_response", None)
        if tag is None:
            tag = cmd
```

The local variable "tag" is set but never used.

Also when using "-o json" or "-o json_mqtt" the "-T" cmdline switch does not work as documented:

(inverter-GtHWsQ7f-py3.9) ubuntu@ubuntu:~/workspace/inverter$ mpp-solar -p /dev/hidraw0 -c QPGS0 -q localhost -o json -T "huhu"

{"_command": "QPGS0", "_command_description": "Parallel Information inquiry", "parallel_instance_number": "valid", "serial_number": "92931805102502", "work_mode": "Battery Mode", "fault_code": "No fault", "grid_voltage": 231.6, "grid_frequency": 49.99, "ac_output_voltage": 229.8, "ac_output_frequency": 49.99, "ac_output_apparent_power": 137, "ac_output_active_power": 45, "load_percentage": 2, "battery_voltage": 53.2, "battery_charging_current": 0, "battery_capacity": 98, "pv_input_voltage": 0.0, "total_charging_current": 0, "total_ac_output_apparent_power": 981, "total_output_active_power": 555, "total_ac_output_percentage": 2, "is_scc_ok": 0, "is_ac_charging": 0, "is_scc_charging": 0, "is_battery_over_voltage": 0, "is_battery_under_voltage": 0, "is_line_lost": 0, "is_load_on": 1, "is_configuration_changed": 1, "output_mode": "Phase 1 of 3 phase output", "charger_source_priority": "Solar first", "max_charger_current": 80, "max_charger_range": 140, "max_ac_charger_current": 10, "pv_input_current": 0, "battery_discharge_current": 1}

What is your opinion on this? Shall I fix the errors and come up with a PR?

Cheers,
Volker

jblance, 2022-04-10:
> (inverter-GtHWsQ7f-py3.9) ***@***.***:~/workspace/inverter$ mpp-solar

Not sure why I remove those in that output, it might have had a reason once.

Happy for you to fix that.

Not sure what the tag option would do for these outputs - do you have a use
case?

Cheers
John

volkerjaenisch, 2022-04-10:
Dear @jblance
> Not sure what the tag option would do for these outputs - do you have a use case?
This is a bit longer story that may be interesting for other mpp-solar users too. Mostly for such which like relational databases.

I am working with the new  [postgres-output](https://github.com/phemmer/telegraf/tree/postgres) plugin for telegraf. This is not in the telegraf trunk, yet. But a PR is open and will finalize eventually.
This output is really what I like since postgres is quite memory efficient in comparison e.g. with influxdb. Not possible to run influxdb2 on a RaspberryPI 4 2GB RAM without getting into memory issues.

If I choose mpp-solar influxdb2 output style as input for telegraf and postgres output the braindead telegraf gives each measure a different timestamp and I end up with a postgres-table with 40 columns of which in each line is only one filled. 

Therefore I like go with mpp-solar json_mqtt then use the exellent telegraf json_consumer and ta tat ta taaa with postgres output I get :

```
           time            |  host  |   topic   | ac_output_active_power | ac_output_apparent_power | ac_output_frequency | ac_output_voltage | battery_capacity | battery_charging_current | battery_discharge_current | battery_voltage | charger_source_priority | fault_code | grid_frequency | grid_voltage | is_ac_charging | is_battery_over_voltage | is_battery_under_voltage | is_configuration_changed | is_line_lost | is_load_on | is_scc_charging | is_scc_ok | load_percentage | max_ac_charger_current | max_charger_current | max_charger_range |        output_mode        | parallel_instance_number | pv_input_current | pv_input_voltage | serial_number  | total_ac_output_apparent_power | total_ac_output_percentage | total_charging_current | total_output_active_power |  work_mode   
----------------------------+--------+-----------+------------------------+--------------------------+---------------------+-------------------+------------------+--------------------------+---------------------------+-----------------+-------------------------+------------+----------------+--------------+----------------+-------------------------+--------------------------+--------------------------+--------------+------------+-----------------+-----------+-----------------+------------------------+---------------------+-------------------+---------------------------+--------------------------+------------------+------------------+----------------+--------------------------------+----------------------------+------------------------+---------------------------+--------------
 2022-04-09 19:48:05.011615 | ubuntu | mpp-solar |                     91 |                      207 |               49.99 |             230.1 |               99 |                        0 |                         1 |            53.3 | Solar first             | No fault   |          50.03 |        231.8 |              0 |                       0 |                        0 |                        1 |            0 |          1 |               0 |         0 |               4 |                     10 |                  80 |               140 | Phase 1 of 3 phase output | valid                    |                0 |                0 | 92931805102502 |                            963 |                          3 |                      0 |                       613 | Battery Mode
 2022-04-09 19:48:08.568067 | ubuntu | mpp-solar |                    366 |                      473 |               49.98 |             225.7 |              100 |                        0 |                         7 |            53.4 | Solar first             | No fault   |          50.01 |        232.4 |              0 |                       0 |                        0 |                        1 |            0 |          1 |               0 |         0 |               5 |                     10 |                  80 |               140 | Phase 2 of 3 phase output | valid                    |                0 |                0 | 92931805100802 |                            954 |                          3 |                      0 |                       637 | Battery Mode
 2022-04-09 19:48:12.12329  | ubuntu | mpp-solar |                    110 |                      161 |               50.05 |             230.1 |               97 |                        0 |                         2 |            53.1 | Solar first             | No fault   |          50.01 |        231.2 |              0 |                       0 |                        0 |                        1 |            0 |          1 |               0 |         0 |               3 |                     10 |                  80 |               140 | Phase 3 of 3 phase output | valid                    |                0 |                0 | 92931805100790 |                            851 |                          2 |                      0 |                       535 | Battery Mode
 2022-04-09 19:48:15.679758 | ubuntu | mpp-solar |                     60 |                      161 |               50.04 |               230 |               97 |                        0 |                         1 |            53.1 | Solar first             | No fault   |          50.02 |        232.9 |              0 |                       0 |                        0 |                        1 |            0 |          1 |               0 |         0 |               3 |                     10 |                  80 |               1 [... cut ...]
- The directory name "io" is not a good choice since it collides with the internal io namespace of python3.8. 
- Having a main() method in __init__.py gives also trouble with some IDEs.
- Having a pyproject .toml is a really good idea, even better if it were understandable by poetry which uses also a pyproject .toml. Many poeple are using poetry these days for building python packages as well as deploy them.

Is it ok if I fix this too? I will make sure that there will be no compatibility issues. 

Cheers,
Volker

jblance, 2022-04-10:
Happy for contributions and your suggestions are all good as far as I can
see.

Thanks

volkerjaenisch, 2022-04-11:
Dear @jblance 
I have a working version now. It lives in [](https://github.com/volkerjaenisch/mpp-solar/tree/reconstruction) it works as expected, but maybe you want to have a look before I make a pull request.

Cheers,
Volker

jblance, 2022-04-20:
If you make a PR it is easier to see the changes and I can comment on it / assess it / just merge...

thanks again for your efforts

volkerjaenisch, 2022-04-23:
Sorry John!

Was swept away by work. I did a pull request. If you are unhapy with it 
please feel free to critizise.

Cheers,

Volker

Message ID: ***@***.***>

-- 
=========================================================
    inqbus Scientific Computing    Dr.  Volker Jaenisch
    Hungerbichlweg 3               +49 (8860) 9222 7 92
    86977 Burggenhttps://inqbus.de
=========================================================

jblance, 2022-04-25:
Thanks - looks good, just a couple of questions / comments

closed 2022-05-09 by jblance

jblance, 2022-10-11:
Raise a pr and I can review and comment there

Thanks for your efforts!

PR #216 (2022-05-09): Reconstruction
Fixed #210