jblance/mpp-solar #214: Issues after partial update of Ubuntu

dustinzinnj, 2022-04-20:
Tried to get my system updated, what a bad idea. Something broke the conversion of the output into influxdb (v1.8.10) so nothing is being added to the database.  I tried removing mppsolar with pip, cleaning out all of influxdb, and removing all mqtt and starting fresh. 

I even tried to edit the "mqtt.py" files under /usr/local/bin to see if I could fix the issue myself but none of my changes seemed to matter to the output. 

Screenshots attached
[image]
[image]
[image]
[image]

jblance, 2022-04-20:
try taking the space of the tag in the cron line "TEST "

dustinzinnj, 2022-04-20:
I've tried that, and even removed the entire -T for the tag. I added that space thinking influxdb wanted to see a space before it would actually add a new measurement to the db.

jblance, 2022-04-20:
Was the error (telegraf one) the same?

I'm no telegraf expert. Do you no the format that telegraf is expecting? It
may have changed with a newer version...

dustinzinnj, 2022-04-20:
Not a clue, I only ever followed your instructions on here to get it working, along with your help back when we found whatever the issue was with my LV2424 setup. Is there some way I can get you that information?

dustinzinnj, 2022-04-20:
Or maybe some way to install known working versions that work well with your current release?

jblance, 2022-04-20:
Prolly better to fix it to work with newer versions. Let me google around
and see what I can find

dustinzinnj, 2022-04-20:
Let me know whatever you need for logs and info. If I could get the "mppt.py" or similar files to change when I change the setup inside the files, it would probably work with some odd form. It doesn't seem to ever change the actual output no matter what I change, even down to ruining the code..  It may also be that telegraf and mqtt updated, but influx is still on 1.8, when they're now at version 2?

dustinzinnj, 2022-04-20:
Looks like the output into influxdb is a "-" instead of any real information. 

"9.201, 162.17.229.201,192.168.0.5 - grafana [20/Apr/2022:11:34:10 -0400] "POST /query?db=mppsolar&epoch=ms HTTP/1.1 {'q': 'SELECT mean("nvalue") FROM "GS" WHERE ("setting" = 'pv1_input_power') AND time >= 1648049578863ms GROUP BY time(3>
 - grafana [20/Apr/2022:11:34:13 -0400] "POST /write?db=mppsolar HTTP/1.1 " 204 0 "-" "Telegraf/1.22.1 Go/1.17.7" 54df8693-c0bf-11ec-9eb9-10e7c6acde81 8073"

I also see "telegraf[43319]: 2022-04-20T01:55:02Z E! **[inputs.mqtt_consumer**] Error in plugin: metric parse error: expected field at 1:6: TEST" under my /etc/telegraf/telegraf.d when I run cat on it. It's also in the live systemctl status. That plugin name, "mqtt_consumer" isnt a file I see listed nor one I made following your steps. Should I try to make the file and place it somewhere?

jblance, 2022-04-20:
Have you configured telegraf - similar to https://github.com/jblance/mpp-solar/blob/master/docs/MQTT_Influx_Grafana.md#configure-telegraf ?
The above link also details the influx setup needed - are these steps all done?

Note the telegraf docs have a message format like below so a tag is required and should turn up as the measurement (with no space)

```
weather,location=us-midwest temperature=82 1465839830100400200
  |    -------------------- --------------  |
  |             |             |             |
  |             |             |             |
+-----------+--------+-+---------+-+---------+
|measurement|,tag_set| |field_set| |timestamp|
+-----------+--------+-+---------+-+---------+
```

jblance, 2022-04-20:
> Looks like the output into influxdb is a "-" instead of any real information.
>
> ```
> "9.201, 162.17.229.201,192.168.0.5 - grafana [20/Apr/2022:11:34:10 -0400] "POST /query?db=mppsolar&epoch=ms [...]

This might be just a function of the web api

> I also see "telegraf[43319]: 2022-04-20T01:55:02Z E! **[inputs.mqtt_consumer**] Error in plugin: metric parse error: expected field at 1:6: TEST" under my /etc/telegraf/telegraf.d when I run cat on [...]

inputs.mqtt_consumer is a section in the config file, either `/etc/telegraf/telegraf.conf` or if you followed the docs in this repo `/etc/telegraf/telegraf.d/mqtt-input.conf`

dustinzinnj, 2022-04-21:
So I finally got it sort of working, but some tag/fields don't seem to work completely. 

Some Examples, along with the entire ET command on PI18 doesn't seem to work.
"
Apr 20 22:58:03 dustinz-solar telegraf[138908]: 2022-04-21T02:58:03Z E! [inputs.mqtt_consumer] Error in plugin: metric parse error: expected tag at 1:42: "SOLAR,setting=dc/ac_power_direction,unit= value=AC-DC"
Apr 20 22:58:03 dustinz-solar telegraf[138908]: 2022-04-21T02:58:03Z E! [inputs.mqtt_consumer] Error in plugin: metric parse error: expected tag at 1:41: "SOLAR,setting=line_power_direction,unit= value=input"
Apr 20 22:58:03 dustinz-solar telegraf[138908]: 2022-04-21T02:58:03Z E! [inputs.mqtt_consumer] Error in plugin: metric parse error: expected tag at 1:38: "SOLAR,setting=local_parallel_id,unit= value=0"
"

using crontab or ssh of:
"sudo /usr/local/bin/mpp-solar -o influx_mqtt  -q LOCALHOST -p /dev/ttyUSB0 -P PI18 -c GS"

and these exact files
[image]
[image]
[image]
[image]

on these versions
InfluxDB shell version: 1.6.4 (the only other influx version other than 1.8.10 that I could find for ubuntu anymore)
Telegraf 1.22.1 (git: HEAD fc8301ae)
Requirement already satisfied: mppsolar in /usr/local/lib/python3.8/dist-packages (0.9.12)
Requirement already satisfied: pyyaml in /usr/lib/python3/dist-packages (from mppsolar) (5.3.1)
Requirement already satisfied: paho-mqtt in /usr/local/lib/python3.8/dist-packages (from mppsolar) (1.5.1)
Requirement already satisfied: pyserial in /usr/local/lib/python3.8/dist-packages (from mppsolar) (3.5)

dustinzinnj, 2022-04-21:
[image]

It does get into Grafana now, although some info is still MIA for whatever syntax problem there still is.

jblance, 2022-04-21:
I assume that you put the actual password in the outputs.influxdb section
looks like `units=` where there are no units is causing a problem

What version of mpp-solar are you using?

dustinzinnj, 2022-04-21:
Correct on the password. 

Seems to be up to date following your instructions, from a quite old version 0.7.49 at newest, and I updated via pip install mppsolar (which gives the versions out above 0.0.12)

jblance, 2022-04-21:
> gives the versions out above 0.9.12)

Ok great. I'll make some changes (remive units= if no units are available)
and upload to pypi.org so pip can get a new version and we can see if that
helps

dustinzinnj, 2022-04-21:
Okay, can I remove them just from those that don't have any somehow? Or do I have to just remove the entire units= entirely?

jblance, 2022-04-21:
Just the ones that don't have units.
There is logic in the output module that sets units="". Can change that to
make a message that doesn't include units

dustinzinnj, 2022-04-21:
If you want me to try a syntax or anything before sending out an update I can, although trying to edit it to just remove those that have no unit info, is probably above my knowledge anytime soon. I barely made it work this much, it took me a few days on and off...

jblance, 2022-04-21:
Prolly easier to just wait

jblance, 2022-04-22:
actually looking at the code, can you try influx2_mqtt output module (ie `-o influx2_mqtt` )

dustinzinnj, 2022-04-22:
This is how the influx output of "SELECT * FROM mppsolar" looks now:

"
1650603123165067277                                                                                                                                                                                                                                                      SOLAR                                                     dustinz-solar 22                                                                                                                                                                                                                                                                                                            mpp-solar
1650603123167764523                                                                                                                                                                                                                                                      SOLAR                                                     dustinz-solar                                                                                                            0                                                                                                                                                                                                  mpp-solar
1650603123168035487                                                                                                                                                                                                                                                      SOLAR                                                     dustinz-solar                                                                                                                                                           0                                                                                                                                                   mpp-solar
1650603123168254050                                                                                                                                                                                                                                                      SOLAR                                                     dustinz-solar                                                                                                                                                                                                         0                                                                                                     mpp-solar
1650603123168441026                                                                                                                                                                                                                                                      SOLAR                                                     dustinz-solar                                                                                                                                                                                                                                           0                                                                   mpp-solar
1650603123168637317                                                                                                                                                                                                                                                      SOLAR                                                     dustinz-solar                                                                                                                                                                                                                         0                                                                                     mpp-solar
1650603123168818033                                                                                                                                                                                                                                                      SOLAR                                                     dustinz-solar                                                                                                                                                                                                                                                           0                                                   mpp-solar
1650603123169015176                                                                                                                                                                                                                                                      SOLAR                                                     dustinz-solar                                                                                                                                                                                                                                                                             Nothing changed                   mpp-solar
1650603123169199795                                                                                                                                                                                                                          [... cut ...]
1650603123170282766                                                                                                                                                                                                                                                      SOLAR                                                     dustinz-solar                                                                     0                                                                                                                                                                                                                                         mpp-solar
>"

jblance, 2022-04-22:
What command are you using?
What do the mqtt messages look like?

dustinzinnj, 2022-04-22:
Command:
sudo mpp-solar -o influx2_mqtt -q 127.0.0.1 -p /dev/ttyUSB0 -P PI18 -c GS

No errors using 
"watch -n 1 systemctl telegraf"

dustinzinnj, 2022-04-22:
I also forgot to mention my shorthand (saved me an extra cronjob while troubleshooting, also stopped working. Which would have been:
sudo mpp-solar -o influx2_mqtt -q 127.0.0.1 -p /dev/ttyUSB0 -P PI18 -c GS,ET 

To send both GS and ET

jblance, 2022-04-22:
Subscribe to the mqtt broker and check the messages generated

dustinzinnj, 2022-04-22:
Looks totally legitimate as far as I can tell.. 

"dustinz@dustinz-solar:~$ mosquitto_sub -h localhost -v -t "#"
mppsolar mppsolar,command=GS grid_voltage=114.0
mppsolar mppsolar,command=GS grid_frequency=59.9
mppsolar mppsolar,command=GS ac_output_voltage=114.0
mppsolar mppsolar,command=GS ac_output_frequency=59.9
mppsolar mppsolar,command=GS ac_output_apparent_power=637
mppsolar mppsolar,command=GS ac_output_active_power=621
mppsolar mppsolar,command=GS output_load_percent=25
mppsolar mppsolar,command=GS battery_voltage=29.2
mppsolar mppsolar,command=GS battery_voltage_from_scc=0.0
mppsolar mppsolar,command=GS battery_voltage_from_scc2=0.0
mppsolar mppsolar,command=GS battery_discharge_current=0
mppsolar mppsolar,command=GS battery_charging_current=1
mppsolar mppsolar,command=GS battery_capacity=100
mppsolar mppsolar,command=GS inverter_heat_sink_temperature=22
mppsolar mppsolar,command=GS mppt1_charger_temperature=0
mppsolar mppsolar,command=GS mppt2_charger_temperature=0
mppsolar mppsolar,command=GS pv1_input_power=0
mppsolar mppsolar,command=GS pv2_input_power=0
mppsolar mppsolar,command=GS pv1_input_voltage=0.0
mppsolar mppsolar,command=GS pv2_input_voltage=0.0
mppsolar mppsolar,command=GS setting_value_configuration_state="Nothing changed"
mppsolar mppsolar,command=GS mppt1_charger_status="abnormal"
mppsolar mppsolar,command=GS mppt2_charger_status="abnormal"
mppsolar mppsolar,command=GS load_connection="connect"
mppsolar mppsolar,command=GS battery_power_direction="charge"
mppsolar mppsolar,command=GS dc/ac_power_direction="AC-DC"
mppsolar mppsolar,command=GS line_power_direction="input"
mppsolar mppsolar,command=GS local_parallel_id=0"

dustinzinnj, 2022-04-22:
output into influx shows this still in the db

"                                                                                          Nothing changed                   mppsolar
1650604777941778307                                                                                                                                                                                                                                                      GS                                                        dustinz-solar                                                                                       abnormal                                                                                                                                                                                                                mppsolar
1650604777941786354                                                                                                                                                                                                                                                      GS                                                        dustinz-solar                                                                                                                                      abnormal                                                                                                                                                                 mppsolar
1650604777941797087                                                                                                                                                                                                                                                      GS                                                        dustinz-solar                                                     connect                                                                                                                                                                                                                                                   mppsolar
1650604777941807000                                                                                                                                                           charge                                                                                     GS                                                        dustinz-solar                                                                                                                                                                                                                                                                                                               mppsolar
1650604777941851569                                                                                                                                                                                                                                                      GS      AC-DC                                             dustinz-solar                                                                                                                                                                                                                                                                                                               mppsolar
1650604777941857048                                                                                                                                                                                                                                                      GS                                                        dustinz-solar                                input                                                                                                                                                                                                                                                                          mppsolar
1650604777941864410                                                                                                                                                                                                                                                      GS                                                        dustinz-solar                                                                     0                                                                                                                                                                                                                                         mppsolar
>"

dustinzinnj, 2022-04-22:
[image]

jblance, 2022-04-22:
Yeah looks correct to me

Might be the select is just getting the strange values

I'd try hooking grafana up and see what it gives (after running the command
a few times)

jblance, 2022-04-22:
Yeah it will be hard to read direct from the db

Try grafana, the graph setup  will be different to before

dustinzinnj, 2022-04-22:
[image]

[image]

Grafana sees the new measurement of mppsolar, but doesn't have any options to select after selecting command. Tried deleting and re-adding, also tried testing the db again just to see.

jblance, 2022-04-22:
The where and select clauses will be different. Remove them and browse for
options

dustinzinnj, 2022-04-22:
That's what I mean, I removed that entire panel and recreated a new one. I found the selections under the SELEC field(xx) section, but I have no data.. Maybe how I did it before was already broken?
[image]

jblance, 2022-04-22:
Maybe command = GS or similar is needed?

dustinzinnj, 2022-04-22:
There aren't any options showing under the command= section if I select in there, that used to be where the options for the real measurement was (IE PV1_input_power)

jblance, 2022-04-22:
Hmm. Might be worth cleaning up some of the measurements in influxdb and
start again so any of the old ones are not confusing the issue

dustinzinnj, 2022-04-22:
I'm thinking maybe this machine has some bad extra-old config somewhere even though it's updated. Certain modifications on the influx2 and influx_mqtt.py don't make certain changes correctly. The only way I can seem to get _any_ data into grafana through influx is with "f"{tag},setting={key} value={value},unit={unit}"," in the influx_mqtt.py setup. influx2 has different outputs somewhere and I'm certainly no programmer. 

I even made another new db, and still no major changes, things dont show in a drop down under the "setting=x" section, but if I type it out as it is in influx, the data appears. Nothing with certain unit values show up (charger status and some more). Probably all leads back to the damn LV2424 having that odd protocol and that I never updated long long ago.

jblance, 2022-04-22:
Ok sounds like an update to influx_mqtt to remove blank units is the best
way forward for you.

I'll look at that change tomorrow if I can

dustinzinnj, 2022-04-24:
Another note, definetly seems the versions of files are mismatched. The -c returns options for PI18 that I can't send out.
[image]

dustinzinnj, 2022-04-24:
Using nano as root, shows this under the pi18.py file in outputs of /usr/local/lib/python3.8/dist-packages/mppsolar/:
[image]

How do I just remove everything related to mppsolar and start a fresh perfectly clean install? Would that help me?

jblance, 2022-05-09:
> 
> How do I just remove everything related to mppsolar and start a fresh perfectly clean install? Would that help me?

How are you installing?

jblance, 2022-05-09:
sorry for the delay, this change (and a bunch of others) are in git version 0.12.2

closed 2023-01-08 by jblance