jblance/mpp-solar #219: Full BMS support for PI17

riogrande75, 2022-04-29:
With later firmware versions (>2017) of the infini-/mpp-solar 10+15kw inverters, full bms support was implemented.
Only supported BMS I'm aware of is pylontech via RS485 bus. You need a extra modbus-card (with modified fw on it) so the inverter can communicate with the bms. See [http://www.mppsolar.com/manual/ACCESSORIES%20(FOR%20MONITORING)/BMS%20RS485%20Card/RS485-Card-Box-for%20BMS-manual-20181106.pdf](url)
While pylontech's rs485 bms commandos are well know, the internal communication between the (bms-)modbus card and the inverter is secret.
To disclose the secret I disassamled the slave firmware of the 10k and noticed 2 PI17 commands that might be involved:
**BMS
BMSV**

The BMSV command just reads out the fw-version (date) from the plugged bms-card - so quite uninterresting.
```
BMSV: (BMSCARD:0000-00-00
BMSVersionTimeYear
BMSVersionTimeYear
BMSVersionTimeDay
```
The BMS commando seems to be more usefult to let any BMS communicate with the inverter. 
This is what I explored so far:
`BMS:
^D0520000,000,0,0000,00,0,0000,0000,0000,0,0,0000,0000
^D0520aaa,bbb,c,dddd,ee,f,gggg,hhhh,iiii,j,k,llll,mmmm

; aaa
; bbb
; c
; dddd
; ee
 
; j
; k 
; llll: BMS max charge curr (amps)
; mmmm: BMS max discharge curr (amps)

; ; P B M S a bbb c d e fff ggg hhh iiii jjjj<CRC><cr>
; 0 1 2 3 4 567 9 B D F01 345 789 BCDE 0123 45 6
; 0 1 2
; a : disconnect
; bbb : BMS SOC%
; c : BMS force AC charge
; d : Bat stop discharge flag
; e : Bat stop charge flag
; fff : BMS bat CV volt
; ggg : BMS Bat float volt
; hhh : BMS bat cutoff volt
; iiii: BMS max charge curr (amps)
; jjjj: BMS max discharge curr (amps)
`
Anyone here interrested in getting together this 2 scripts (mpp-solar + jkbms)?

jblance, 2022-05-03:
Seems like you could achieve something with a specific output module, i.e
- command to query BMS 
- gets needed results
- outputs formated result to Inverter 

this type of output hasnt been implemented yet, but shouldnt be a problem (with a lot of testing)
I'm (slowly) working on a more flexible process (powermon command in latest versions) that will allow for better configuration of inputs and outputs - seems like the above would work better with that, but the current mppsolar/jkbms commands should be able to made to work with a bunch of extra options

riogrande75, 2022-05-04:
Ok, thx.
I'll investigate commands further.
Some help with logfiles from other users would be very helpful.
Can any of the MPI10k users pls. send me the inverter's reply for the command `^P004BMS`?

riogrande75, 2022-08-17:
After a long research I've finally found out, how PI17/PI18 inverters (10k, 15k) get the BMS values.

Conclusion: You simply need to send a ^D054BMS... string periodically to the inverter.
Format looks like this:
[image]
I created a script to read bms data from a mqtt server and create the ^D054BMS string that can be sent to the inverter: [https://github.com/riogrande75/readmqttbms/blob/main/readmqttbms.php](url)
Sending this to the inverter let's the display show the "LI-bAt" and battery values read from BMS.
[image]
The inverter get's informed about BMS stoped charging/discharging and can react correctly:
[image]

@jblance Maybe you can implement this in your work - reading data from jk-bms (I do that currently with a extra ESP32 from [https://forum.drbacke.de/viewtopic.php?t=982](url)) and send it to mpp solar inverter.

gianfrdp, 2023-05-04:
> After a long research I've finally found out, how PI17/PI18 inverters (10k, 15k) get the BMS values.
> 
> Conclusion: You simply need to send a ^D054BMS... string periodically to the inverter.

Hello,
how php script sends string to inverter?
And which battery type needs to be selected on inverter to let this protocol work?

rene-dev, 2024-01-12:
I have MPI 10k, and successfully  used this command to set charge and discharge limits, and alarms.
converting them from pylontech battery connected via can.
Will share my code when its finished.
`^P004BMS` reply is:
`b'^D0520422,042,0,0000,01,0,0535,0525,0500,0,0,0500,0010\x15\xd3\r'`

PaulEPop, 2024-06-03:
It would be wonderfull, that we could communicate like a BMS to MPP-inverters!
Many BMS have a problem with inverters from MPP (12k / 15k)

riogrande75, 2024-06-03:
In fact BMS comms to MPP/FSP/Voltrinic hybrid inverters (10/15k) is quite easy. I run it with my php scripts since more than 2 years witout any issue. All you need is the information from my [post ](https://github.com/jblance/mpp-solar/issues/219#issuecomment-1217540276) above.

As far as I understand this project, it's "simply" pulling data from inverters and bms' for using this data in 3rd pty visualization. For bms comms to work, data from bms need to be processed and sent to the inverter.

@jblance Is bms->inverter comms on the 2do list, or will it never be implemented anyway. Off course, pulled data from bms (e.g. jk) would have to be temporarily stored on the machine and then be sent to the inverter while quering production data.
Off course I'd assist in implementing, even my python abilities are quite low.

jblance, 2024-06-04:
> As far as I understand this project, it's "simply" pulling data from inverters and bms' for using this data in 3rd pty visualization. For bms comms to work, data from bms need to be processed and sent to the inverter.
> 

This is pretty close - it simply runs a command (any inverter or other device command) and outputs it (to a variety of outputs including mqtt)

> @jblance Is bms->inverter comms on the 2do list, or will it never be implemented anyway. Off course, pulled data from bms (e.g. jk) would have to be temporarily stored on the machine and then be [...]

It is, probably in the powermon codebase though (as mppsolar is creaking at the seams a bit)
I havent figured out what the inverter logic is for BMS integration
is it:
1- just needs a 'command' that contains BMS data
2- needs to send a command (to the BMS) and then get a correctly formated response
3- something else
???

How does you php logic work?

riogrande75, 2024-06-04:
There is no start command, etc. for BMS comms to start working.
You simply send the ^D054BMS... command in cycles to the inverter. I do it every 5 seconds inbetween my queries for production data (^P003GS,...).
After a few seconds the inverter starts displaying "Li-Bat" in it's display and updates battery charging&discharging settings with the values presented by the BMS.

I created this [php script ](https://github.com/riogrande75/readmqttbms/blob/main/jkble.php)that pulls data from jk bms and prepares the BMS string in a shared mem obj as well as pushing values to mqtt for my hassio.
[My script](https://github.com/riogrande75/infinipoll10k/blob/master/infinipoll10k_bms.php) that reads all needed values from the inverter opens this shmop and sends the contents to the inverter (line 370ff).

There is in fact no communication in direction downwards from the inverter to bms. Obviously not needed.

PaulEPop, 2024-06-04:
Hi, i will use my words: Einfach Endgeil hier....
I have to modify your php script (I will try), because I'm using the direct connection via rs232 and I need just the BMS funktionality.

riogrande75, 2024-06-05:
> Hi, i will use my words: Einfach Endgeil hier.... I have to modify your php script (I will try), because I'm using the direct connection via rs232 and I need just the BMS funktionality.

Well, I guess that shouldn't be a big deal with the code I posted. Big difference to the query command 's is to prepare the BMS string's CRC (crc16-xmodem).

PaulEPop, 2024-06-06:
Do you get a response to the following command?
$serial->sendMessage($bmscmd.chr(hexdec($bmscrc1)).chr(hexdec($bmscrc2)).chr(0x0d));

In my case the log looks like this: (BMS1 without CRC, BMS2 with CRC)
2024-06-06 23:13:12: BMS_1: ^D054BMS0554,098,0,0000,00,0,0552,0552,1500,1,1,0504,150
2024-06-06 23:13:12: BMS_2: ^D054BMS0554,098,0,0000,00,0,0552,0552,1500,1,1,0504,1506#
string(0) ""

I'm sending just 'BMS_2'
Serial connection is running well,I will receive correct responses to ID requests.
I have a MPP-SOLAR MPI12K (PI17 Protokoll)

riogrande75, 2024-06-07:
Your string is too long, it's more than 54 (^D**054**) characters long.
Fault see pic.
[image]
Off course I don't see the CRC you send in your post.

PaulEPop, 2024-06-07:
I have changed your code in readmqttbms.php from:
shmop_write($sh_bms1, ",**00**", 23);   to 
shmop_write($sh_bms1, ",**0**", 23); 

Result
^D054BMS0539,077,1,0039,**0#**,0,0552,0552,1500,1,1,0504,150

PaulEPop, 2024-06-07:
I have now the following datapackage 
`$bmscmd = ^D054BMS0531,094,0,0019,0,0,0560,0560,1500,1,1,0504,2000` (from shmop_read...)

this package is going thrue your script

```
       if($bms){
                $bmscmd = shmop_read($sh_bms, 0, 56);
                $bmscrc = paddings(genCRC($bmscmd),4);
                $bmscrc1 = substr($bmscrc,-4,2);
                $bmscrc2 = substr($bmscrc,-2);
				if($debug) logging("BMS_1: ".$bmscmd); 
				if($debug) logging("BMS_2: ".$bmscmd.chr(hexdec($bmscrc1)).chr(hexdec($bmscrc2)).chr(0x0d));
                $serial->sendMessage($bmscmd.chr(hexdec($bmscrc1)).chr(hexdec($bmscrc2)).chr(0x0d));
```
output in console:
2024-06-07 21:42:00: BMS_1: ^D054BMS0531,093,0,0020,0,0,0560,0560,1500,1,1,0504,2000
2024-06-07 21:42:00: BMS_2: ^D054BMS0531,093,0,0020,0,0,0560,0560,1500,1,1,0504,2000Rt

riogrande75, 2024-06-08:
What FW version is on your inverter?
Can you post the string in 'hex to see if crc is ok?

PaulEPop, 2024-06-08:
2024-06-08 21:53:40: BMS_1: ^D054BMS0531,093,0,0015,0,0,0560,0560,1500,1,1,0504,2000
2024-06-08 21:53:40: BMS_2: ^D054BMS0531,093,0,0015,0,0,0560,0560,1500,1,1,0504,20007▒
2024-06-08 21:53:40: CRC: 37d80  and at the end 0d
for the BMS_1 string I have used Python to 'verhexen' .

riogrande75, 2024-06-10:
FW version is ok.
Looking at your output, it seems that "verhexen" of BMS_1 did not include CRC.

When I do it with your BMS string and my code the hex string looks like this.
5E44303534424D53303533312C3039302C302C303031352C302C302C303536302C303536302C313530302C312C312C303530342C32303030**EED90D**
Make sure that the character at the end is 0x**0**d and not simply "**d**"

PaulEPop, 2024-06-11:
The hex value is converted into a chr and transferred.
I use exactly your code for this.
`$serial->sendMessage($bmscmd.chr(hexdec($bmscrc1)).chr(hexdec($bmscrc2)).chr(0x0d));`
Further up I use the following code, to which I get the corresponding answer.
```
$serial->sendMessage("^P003PI".chr(0x0d)); //QPI Protocol ID abfragen 6byte
$antw = $serial->readPort();
```

My guess is that the MPI12k does not recognize the ^D054... command, I don't get a response to the ^P004BMS request either.

riogrande75, 2024-06-11:
Ahhh, a MPI 12k WP....
Ok, that could be the case, as the inverter has a built in BMS card/port and does not need to be able to receive BMS commands via serial port.

If you have a firmware update for your inverter, I can disassamble it and check if the command is really missing or it has been modified.

PaulEPop, 2024-06-11:
I have testet a snipset of pythoncode with serial.
I have got b'^0\x1b\xe3\r' as answer for the ^D054 command
With ^P003ID --> b'^D0251496162206100759000000@\x90\r'
Serial com is running well. D054 is unknown for MPI12k.
@riogrande75 : I have asked MPP-Solar for an Firmwareupdate.

Thank you very much for your kind help.

riogrande75, 2024-06-12:
Did you activate BMS and set battery mode to LiFe in solarpower?
Did you try sending the bms command cyclic for some minutes? Even my inverters need some time to recognize bms command and show it in display.

If you have a RS485 BMS port on your MPI 12k WP, why do you frickle arround with bms command instead of using a esp32 for directly connecting jk with the inv? You could also use a simple serial2eth converter and send reply's via bms port as a last resort.

PaulEPop, 2024-06-12:
I have tried everything, the commands was sendind more than 12h.
You mean [mppsolar-lib-protocol-to-jk-bms-rs485](https://github.com/sdragos/mppsolar-lib-protocol-to-jk-bms-rs485)?
The BMS isn't a JK, it' a Seplos. So I cant use this project direct and I'm not a softwareengineer just a verry little hobby coder with limited skills. :)

riogrande75, 2024-06-12:
We're OT for a long time 🫣

Pls. leave this git issue and read how to [connect a seplos directly to your inverter](https://www.akkudoktor.net/forum/bms-seplos/seplos-v16-10e-und-voltronic-wechselrichter/#post-156195). SEPLOS "speaks" pylontech protocol, it can be connected with a simple cable, and the correct setup, off course.

drnry, 2025-04-07:
Hi, i'm using the MPI 12k WP aswell.
The inverter has a bat-type option (only settable via the display) i set it to "USE" (user defined battery)

I'm able to send it the D054 command - the inverter then starts showing the following effects:
cut-off and re-discharge value settings disappear from the inverters display - instead "soc-weak" and "soc-back" values can be set for grid available and not-available.

**BMS Voltage**: display keeps showing value measured by the inverter but Inverter gives the bat-low-warning when a lower than cut-off-voltage is being sent
**Battery percent**: display starts showing "SOC" and the transmitted soc-value in an alternating manner. soc-values below "soc-weak" will cause bat-low warning.
**Charge/Discharge**: i don't notice any effects sending the direction flag
**current**: i don't notice any effects sending this value
**BMS warning code**: i have only been sending 0 so far
**BMS force charge**: not noticing any effect
**BMS cv voltage**: is being set correctly by the inverter but also sets float voltage to the same value
**BMS float voltage**: not noticing any effect
**BMS MaxChgCurrent**: is being set correctly
**BMS StopDiscFlag**: works as expected
**BMS StopChaFlag**: works as expected
**BMS CutOffVoltage**: works as expected
**BMS MaxDisChgCurr**: works, but needs to be shifted one decimal point to right

i should add, that i'm not on the factory firmware. I received a fw-upgrade because i always had issues with completely wrong kWh generation values. they didn't go away..and i dont care about it anymore. I now have a different issue that the battery does not charge correctly when max feed power is above 0. But this is probably not connected to using the D054 command and could be connected to the fw upgrade.

Anyway @riogrande75 thank you for figuring out this command and if you could disassmble fw files this would of course be interesting! Here you find a fw package i received in january '24 for the MPI 12k WP:

[Reflash 12K Master.zip](https://github.com/user-attachments/files/19631395/Reflash.12K.Master.zip)
[Reflash 12K Remote.zip](https://github.com/user-attachments/files/19631394/Reflash.12K.Remote.zip)
the "remote" is flashing an STM32F107VCT6 on a PCB managing all comm-inputs and display controller and forwards everything to the Master. This is why the flashing order has to be Master,Slave,Remote.
https://maximumsolar.online/product/mpi-12k_15k-wp-remote-board/