jblance/mpp-solar #224: Getting the jblance respository working with JK-B2A8S20P JK-BMS Boards

EasilyBoredEngineer, 2022-05-09:
Hi there - 

@jblance thankyou for this repository.

I'm currently evaluating the JK-B2A8S20P board, hardware version 9.X, software version 9.01M3. I'm going to document my progress here so that others can benefit. 

Apologies if some of this is clunky, it's my first time using github.

I've taken a video of the board, but it's still uploading, but for the moment, some insights:-

1. My version of the board has three attached serial ports. One is marked RS485, the other is marked GPS and the third is a multi pin connector for the optional display.
2. I haven't yet done much work with the GPS port, but I believe it's straight TTL.
3. The RS485 and Display port use RS485 natively - as opposed to previous board versions which required an external OEM adaptor.
4. The RS485 port uses a 3 pin 1.25mm pitch JST connector.
5. Both the display and the RS485 port trace back to a YD3082 RS485 driver chip each (see at [this link](https://www.x-on.com.au/mpn/gatemode/yd3082eesa)
6. The 'board side' of the RS485 chips then trace to a [pi 122U31 galvanic isolator chip](https://taoic.oss-cn-hangzhou.aliyuncs.com/temp/product/hstz_1576814805000.pdf), which appears to be a bit similar in design to the more well known ADUM1201 IC's.
7. I've probed the TTL side of both RS485 chips. 
8. The display RS485 is driven by a repetitive pattern of data at what seems to be 9600 baud.
9. The 'RS485' port, although it has all the components necessary, doesn't seem to be 'switched on' on my board - oscilloscope shows no data whatsoever on that line.
10. The board is also fitted with a bluetooth module, the same as other jk-bms versions. A picture of the [bluetooth module has been linked by other users](https://raw.githubusercontent.com/syssi/esphome-jk-bms/main/images/jk-bd6a17s6p/005-labeled.jpg).
11. My work to date has been heavily influenced by the discussion "[Cannot connect to JK BMS/Active Balancer](https://github.com/jblance/mpp-solar/issues/170#)", in which @bullar points out that if your receiving device has a bluetooth stack <4.2, this breaks the repository. As a result, given that I have only a rpi 3 B+ (bluetooth version 4.1) and a linux laptop (bluetooth version 4.1), I've soldered a YP-05 TTL to USB converter to the TTL ports on the bluetooth module while I await an ESP32. This has given me usable I/O capabilities. That process will be covered in an upcoming linked video.

Cheers - EBE

EasilyBoredEngineer, 2022-05-09:
I've generated the following hex dump from the unit. To generate it, I've gone through the various pages on the android app. It's captured the settings page, the info page and the status page. I haven't yet made any attempts to capture the process of changing settings.

1. Have confirmed that it's possible to open this log file using HxD editor (if anyone can suggest a better editor, please go ahead.). 
2. Have confirmed that the hex file contains data relevant to the 'getInfo' command, including thename of the BMS etc (this is a command that I've noted doesn't work properly with this BMS and will need mods).
3. Have confirmed that the capture includes data relevant to 'getCellData' command.

[putty4.log](https://github.com/jblance/mpp-solar/files/8648714/putty4.log)

EasilyBoredEngineer, 2022-05-09:
As previously mentioned, have followed the discussion at [Cannot connect to JK BMS/Active Balancer](https://github.com/jblance/mpp-solar/issues/170#), with thanks particularly to [@bullar](https://github.com/bullar) and [@syssi](https://github.com/syssi). I experienced many of the same issues mentioned in that discussion.
Syssi [proposed a pull](https://github.com/jblance/mpp-solar/pull/221) to implement the suggested change - which includes using the FFE1 characteristic instead of the FFE3, and also some changes to the format of JK02.py, both of which I implemented.

With these changes implemented, I can connect using 'getInfo', but obviously the code needs more changes if it is to be suitable for the JK-B2A8S20P, as it outputs garbage characters and carriage returns in the reply. I'll do some work using the hex dump above and try and work out where the differences lie - but for completeness here's the command I used..

`jkbms -p /dev/ttyUSB0 -P JK02 -b 115200 -c getInfo -D`

and here's the output I received (noted right away that the header and record_type seem to be mangled - offset 2 bytes?)..

`2022-05-09 15:10:28,207:INFO:__init__:main@195: Solar Device Command Utility, version: 0.9.12, doc updates
2022-05-09 15:10:28,208:DEBUG:__init__:main@204: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='jkbms', mqttc=<paho.mqtt.client.Client object at 0x765d2f58>, _isConnected=False)
2022-05-09 15:10:28,209:INFO:__init__:main@299: Creating device "unnamed" (type: "jkbms") on port "/dev/ttyUSB0 (porttype=None)" using protocol "JK02"
2022-05-09 15:10:28,215:DEBUG:__init__:main@303: device_class <class 'mppsolar.devices.jkbms.jkbms'>
2022-05-09 15:10:28,216:DEBUG:device:__init__@33: __init__ args ()
2022-05-09 15:10:28,217:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB0', 'protocol': 'JK02', 'baud': 115200, 'porttype': None, 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='jkbms', mqttc=<paho.mqtt.client.Client object at 0x765d2f58>, _isConnected=False)}
2022-05-09 15:10:28,217:DEBUG:__init__:get_port_type@65: port matches ttyusb
2022-05-09 15:10:28,218:INFO:__init__:get_port@107: Using serialio for communications
2022-05-09 15:10:28,231:DEBUG:__init__:get_protocol@12: Protocol JK02
2022-05-09 15:10:28,251:DEBUG:device:__init__@38: __init__ name unnamed, port <mppsolar.io.serialio.SerialIO object at 0x75f9f4a8>, protocol <mppsolar.protocols.jk02.jk02 object at 0x75f9f4c0>
2022-05-09 15:10:28,252:DEBUG:__init__:main@352: Commands [(<mppsolar.devices.jkbms.jkbms object at 0x75f9f418>, 'getInfo', 'getInfo', 'screen', None, None)]
2022-05-09 15:10:28,252:INFO:__init__:main@357: Looping 1 commands
2022-05-09 15:10:28,253:INFO:__init__:main@367: Getting results from device: jkbms device - name: unnamed, port: <mppsolar.io.serialio.SerialIO object at 0x75f9f4a8>, protocol: <mppsolar.protocols.jk02.jk02 object at 0x75f9f4c0> for command: getInfo, tag: getInfo, outputs: screen
2022-05-09 15:10:28,253:INFO:device:run_command@271: Running command getInfo
2022-05-09 15:10:28,254:INFO:jkabstractprotocol:get_full_command@73: Using protocol b'JK02' with 3 commands
2022-05-09 15:10:28,254:DEBUG:jkabstractprotocol:get_command_defn@104: get_command_defn for: getInfo
2022-05-09 15:10:28,255:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'getInfo'
2022-05-09 15:10:28,255:DEBUG:abstractprotocol:get_command_defn@62: Found command getInfo in protocol b'JK02'
2022-05-09 15:10:28,256:DEBUG:jkabstractprotocol:get_full_command@78: self._command = getInfo
2022-05-09 15:10:28,256:DEBUG:jkabstractprotocol:get_full_command@89: cmd with SOR: bytearray(b'\xaaU\x90\xeb\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2022-05-09 15:10:28,257:DEBUG:jkabstractprotocol:get_full_command@97: cmd with command code: bytearray(b'\xaaU\x90\xeb\x97\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2022-05-09 15:10:28,257:DEBUG:jkabstractprotocol:get_full_command@99: cmd with crc: bytearray(b'\xaaU\x90\xeb\x97\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x11')
2022-05-09 15:10:28,258:INFO:device:run_command@298: full command bytearray(b'\xaaU\x90\xeb\x97\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x11') for command getInfo
2022-05-09 15:10:28,258:DEBUG:jkabstractprotocol:get_command_defn@104: get_command_defn for: getInfo
2022-05-09 15:10:28,258:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'getInfo'
2022-05-09 15:10:28,259:DEBUG:abstractprotocol:get_command_defn@62: Found command getInfo in protocol b'JK02'
2022-05-09 15:10:28,259:DEBUG:serialio:send_and_receive@19: port /dev/ttyUSB0, baudrate 115200
2022-05-09 15:10:28,262:DEBUG:serialio:send_and_receive@22: Executing command via serialio...
2022-05-09 15:10:29,753:DEBUG:serialio:send_and_receive@30: serial response was: b'\x00mU\xaa\xeb\x90\x02\xd1\xea\x0c\xeb\x0c\xea\x0c\xea\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0f\x00\x00\x00\xeb\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xab3\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xc6\x00\xbe\x00\xd3\x00\x04\x00\x00\x00\x00\x00\x03l\x02\x00 [... cut ...]
device_passcode                 ë

manufacturing_date
serial_number
user_data
setup_passcode                  «3Æ
unknown_value_in_response_16    b'\x00\x00\x00\x00\x00\x01\x00\x05\x00\x00`R\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x81U'`

syssi, 2022-05-09:
Please edit your post and add **three** backticks (`) in front and at the end of each code block. Did I understand your setup correctly: You have connected a FTDI to the RX/TX pins of the bluetooth module. To talk to the BMS there the JK02 protocol can be used?

jblance, 2022-05-09:
this maybe a symptom of the mtu increase. there some kinda poor logic to look for the start of the record which looks to not work now, so getting rolled together messages

EasilyBoredEngineer, 2022-05-09:
> Did I understand your setup correctly: You have connected a FTDI to the RX/TX pins of the bluetooth module. To talk to the BMS there the JK02 protocol can be used?

Yes, I have successfully used the JK02 protocol to talk to the BMS by soldering to the Tx and Rx of the bluetooth module. I'll post a video once it's uploaded.

jblance, 2022-05-09:
maybe a fix in 0.12.1 - a lot of untested changes at the moment, so be careful.

EasilyBoredEngineer, 2022-05-09:
Given that the output of 'getInfo' was mangled, I moved on to 'getCellData', and executed the following command:-

``` jkbms -p /dev/ttyUSB0 -P JK02 -b 115200 -c getCellData -D```

the output received was more encouraging (note in my particular setup I'm presently only using 4 cells, so only expecting data from the first 4 cells). The result seems to be accurate, but offset by 2 bytes:-

```2022-05-09 15:32:24,016:INFO:__init__:main@195: Solar Device Command Utility, version: 0.9.12, doc updates
2022-05-09 15:32:24,017:DEBUG:__init__:main@204: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='jkbms', mqttc=<paho.mqtt.client.Client object at 0x7660ef58>, _isConnected=False)
2022-05-09 15:32:24,018:INFO:__init__:main@299: Creating device "unnamed" (type: "jkbms") on port "/dev/ttyUSB0 (porttype=None)" using protocol "JK02"
2022-05-09 15:32:24,027:DEBUG:__init__:main@303: device_class <class 'mppsolar.devices.jkbms.jkbms'>
2022-05-09 15:32:24,028:DEBUG:device:__init__@33: __init__ args ()
2022-05-09 15:32:24,029:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB0', 'protocol': 'JK02', 'baud': 115200, 'porttype': None, 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='jkbms', mqttc=<paho.mqtt.client.Client object at 0x7660ef58>, _isConnected=False)}
2022-05-09 15:32:24,031:DEBUG:__init__:get_port_type@65: port matches ttyusb
2022-05-09 15:32:24,032:INFO:__init__:get_port@107: Using serialio for communications
2022-05-09 15:32:24,054:DEBUG:__init__:get_protocol@12: Protocol JK02
2022-05-09 15:32:24,092:DEBUG:device:__init__@38: __init__ name unnamed, port <mppsolar.io.serialio.SerialIO object at 0x75fda4a8>, protocol <mppsolar.protocols.jk02.jk02 object at 0x75fda4c0>
2022-05-09 15:32:24,093:DEBUG:__init__:main@352: Commands [(<mppsolar.devices.jkbms.jkbms object at 0x75fda418>, 'getCellData', 'getCellData', 'screen', None, None)]
2022-05-09 15:32:24,094:INFO:__init__:main@357: Looping 1 commands
2022-05-09 15:32:24,095:INFO:__init__:main@367: Getting results from device: jkbms device - name: unnamed, port: <mppsolar.io.serialio.SerialIO object at 0x75fda4a8>, protocol: <mppsolar.protocols.jk02.jk02 object at 0x75fda4c0> for command: getCellData, tag: getCellData, outputs: screen
2022-05-09 15:32:24,096:INFO:device:run_command@271: Running command getCellData
2022-05-09 15:32:24,096:INFO:jkabstractprotocol:get_full_command@73: Using protocol b'JK02' with 3 commands
2022-05-09 15:32:24,097:DEBUG:jkabstractprotocol:get_command_defn@104: get_command_defn for: getCellData
2022-05-09 15:32:24,097:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'getCellData'
2022-05-09 15:32:24,098:DEBUG:abstractprotocol:get_command_defn@62: Found command getCellData in protocol b'JK02'
2022-05-09 15:32:24,098:DEBUG:jkabstractprotocol:get_full_command@78: self._command = getCellData
2022-05-09 15:32:24,099:DEBUG:jkabstractprotocol:get_full_command@89: cmd with SOR: bytearray(b'\xaaU\x90\xeb\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2022-05-09 15:32:24,099:DEBUG:jkabstractprotocol:get_full_command@97: cmd with command code: bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2022-05-09 15:32:24,100:DEBUG:jkabstractprotocol:get_full_command@99: cmd with crc: bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10')
2022-05-09 15:32:24,100:INFO:device:run_command@298: full command bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10') for command getCellData
2022-05-09 15:32:24,101:DEBUG:jkabstractprotocol:get_command_defn@104: get_command_defn for: getCellData
2022-05-09 15:32:24,101:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'getCellData'
2022-05-09 15:32:24,101:DEBUG:abstractprotocol:get_command_defn@62: Found command getCellData in protocol b'JK02'
2022-05-09 15:32:24,102:DEBUG:serialio:send_and_receive@19: port /dev/ttyUSB0, baudrate 115200
2022-05-09 15:32:24,106:DEBUG:serialio:send_and_receive@22: Executing command via serialio...
2022-05-09 15:32:25,276:DEBUG:serialio:send_and_receive@30: serial response was: b'\x00\xb6U\xaa\xeb\x90\x02\xc8\xe9\x0c\xe9\x0c\xe9\x0c\xe9\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0f\x00\x00\x00\xe9\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xa43\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xc3\x00\xbe\x00\xd0\x00\x04\x00\x00\x00\x00\x00\x03l\x02\x00\xc0E\x04\x00\x00\x00\x00\x00\xeb\xe0\x01\x00\x00\x00\x00\x00\x08C#\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x07\x00\x00\x00\x00\x00#\x05\x00\x00\x00\x00\xbbd;@\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x05\x00\x00X\x85\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xaaU\xaa\xeb\x90\x02\xc9\xe9\x0c\xe9\x0c\xe8\x0c\xe9\x0c\x00\x00\x00\x00\x00\x0 [... cut ...] \x00\x00\x00\x00\x00\x00\x00\x00\x00\x07\x00\x00\x00\x00\x00#\x05\x00\x00\x00\x00\xbbd;@\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x05\x00\x00\\\x85\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xaaU'
```

EasilyBoredEngineer, 2022-05-09:
In a quick and nasty attempt to fix the 2 byte offset on 'getCellData', I changed the code in jk02.py, adding a 2 byte discard before the header - see below:-

```NEW_COMMANDS = {
    "getCellData": {
        "name": "getCellData",
        "command_code": "96",
        "record_type": "2",
        "description": "BLE Cell Data inquiry",
        "help": " -- queries the ble device for the cell data",
        "type": "QUERY",
        "response_type": "POSITIONAL",
        "response": [
            **["discard", 2, "discard", ""],**
            ["Hex2Str", 4, "Header", ""],
            ["Hex2Str", 1, "Record_Type", ""],
            ["Hex2Int", 1, "Record_Counter", ""],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell01", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell02", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell03", "V"],```

This brought everything back into alignment - output below - works - but has a strange error at the end:-

```2022-05-09 15:41:13,122:INFO:__init__:main@195: Solar Device Command Utility, version: 0.9.12, doc updates
2022-05-09 15:41:13,122:DEBUG:__init__:main@204: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='jkbms', mqttc=<paho.mqtt.client.Client object at 0x7659ff58>, _isConnected=False)
2022-05-09 15:41:13,123:INFO:__init__:main@299: Creating device "unnamed" (type: "jkbms") on port "/dev/ttyUSB0 (porttype=None)" using protocol "JK02"
2022-05-09 15:41:13,129:DEBUG:__init__:main@303: device_class <class 'mppsolar.devices.jkbms.jkbms'>
2022-05-09 15:41:13,130:DEBUG:device:__init__@33: __init__ args ()
2022-05-09 15:41:13,130:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB0', 'protocol': 'JK02', 'baud': 115200, 'porttype': None, 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='jkbms', mqttc=<paho.mqtt.client.Client object at 0x7659ff58>, _isConnected=False)}
2022-05-09 15:41:13,131:DEBUG:__init__:get_port_type@65: port matches ttyusb
2022-05-09 15:41:13,131:INFO:__init__:get_port@107: Using serialio for communications
2022-05-09 15:41:13,144:DEBUG:__init__:get_protocol@12: Protocol JK02
2022-05-09 15:41:13,164:DEBUG:device:__init__@38: __init__ name unnamed, port <mppsolar.io.serialio.SerialIO object at 0x75f6b4a8>, protocol <mppsolar.protocols.jk02.jk02 object at 0x75f6b4c0>
2022-05-09 15:41:13,165:DEBUG:__init__:main@352: Commands [(<mppsolar.devices.jkbms.jkbms object at 0x75f6b418>, 'getCellData', 'getCellData', 'screen', None, None)]
2022-05-09 15:41:13,165:INFO:__init__:main@357: Looping 1 commands
2022-05-09 15:41:13,165:INFO:__init__:main@367: Getting results from device: jkbms device - name: unnamed, port: <mppsolar.io.serialio.SerialIO object at 0x75f6b4a8>, protocol: <mppsolar.protocols.jk02.jk02 object at 0x75f6b4c0> for command: getCellData, tag: getCellData, outputs: screen
2022-05-09 15:41:13,166:INFO:device:run_command@271: Running command getCellData
2022-05-09 15:41:13,166:INFO:jkabstractprotocol:get_full_command@73: Using protocol b'JK02' with 3 commands
2022-05-09 15:41:13,166:DEBUG:jkabstractprotocol:get_command_defn@104: get_command_defn for: getCellData
2022-05-09 15:41:13,167:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'getCellData'
2022-05-09 15:41:13,167:DEBUG:abstractprotocol:get_command_defn@62: Found command getCellData in protocol b'JK02'
2022-05-09 15:41:13,167:DEBUG:jkabstractprotocol:get_full_command@78: self._command = getCellData
2022-05-09 15:41:13,168:DEBUG:jkabstractprotocol:get_full_command@89: cmd with SOR: bytearray(b'\xaaU\x90\xeb\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2022-05-09 15:41:13,168:DEBUG:jkabstractprotocol:get_full_command@97: cmd with command code: bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2022-05-09 15:41:13,168:DEBUG:jkabstractprotocol:get_full_command@99: cmd with crc: bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10')
2022-05-09 15:41:13,168:INFO:device:run_command@298: full command bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10') for command getCellData
2022-05-09 15:41:13,169:DEBUG:jkabstractprotocol:get_command_defn@104: get_command_defn for: getCellData
2022-05-09 15:41:13,169:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'getCellData'
2022-05-09 15:41:13,169:DEBUG:abstractprotocol:get_command_defn@62: Found command getCellData in protocol b'JK02'
2022-05-09 15:41:13,170:DEBUG:serialio:send_and_receive@19: port /dev/ttyUSB0, baudrate 115200
2022-05-09 15:41:13,173:DEBUG:serialio:send_and_receive@22: Executing command via serialio...
2022-05-09 15:41:14,577:DEBUG:serialio:send_and_receive@30: serial response was: b'\x00\x18U\xaa\xeb\x90\x02\x97\xeb\x0c\xe9\x0c\xe9\x0c\xe9\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0f\x00\x00\x00\xeb\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xaa3\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xc2\x00\xba\x00\xce\x00\x04\x00\x00\x00\x00\x00\x03l\x02\x00\xc0E\x04\x00\x00\x00\x00\x00\xeb\xe0\x01\x00\x00\x00\x00\x00\x15E#\x00\x00\ [... cut ...] 0\x00\x00\x00\x00\x00\x00\x00\x00\x00\x07\x00\x00\x00\x00\x00#\x05\x00\x00\x00\x00\xbbd;@\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x05\x00\x00\xe0\x99\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00)U'
```

EasilyBoredEngineer, 2022-05-09:
My change brought everything back into alignment - output below - works - but has a strange error at the end, so I suspect it's simply quick and dirty, with a deeper underlying cause.:-

```2022-05-09 15:41:13,122:INFO:__init__:main@195: Solar Device Command Utility, version: 0.9.12, doc updates
2022-05-09 15:41:13,122:DEBUG:__init__:main@204: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='jkbms', mqttc=<paho.mqtt.client.Client object at 0x7659ff58>, _isConnected=False)
2022-05-09 15:41:13,123:INFO:__init__:main@299: Creating device "unnamed" (type: "jkbms") on port "/dev/ttyUSB0 (porttype=None)" using protocol "JK02"
2022-05-09 15:41:13,129:DEBUG:__init__:main@303: device_class <class 'mppsolar.devices.jkbms.jkbms'>
2022-05-09 15:41:13,130:DEBUG:device:__init__@33: __init__ args ()
2022-05-09 15:41:13,130:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB0', 'protocol': 'JK02', 'baud': 115200, 'porttype': None, 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='jkbms', mqttc=<paho.mqtt.client.Client object at 0x7659ff58>, _isConnected=False)}
2022-05-09 15:41:13,131:DEBUG:__init__:get_port_type@65: port matches ttyusb
2022-05-09 15:41:13,131:INFO:__init__:get_port@107: Using serialio for communications
2022-05-09 15:41:13,144:DEBUG:__init__:get_protocol@12: Protocol JK02
2022-05-09 15:41:13,164:DEBUG:device:__init__@38: __init__ name unnamed, port <mppsolar.io.serialio.SerialIO object at 0x75f6b4a8>, protocol <mppsolar.protocols.jk02.jk02 object at 0x75f6b4c0>
2022-05-09 15:41:13,165:DEBUG:__init__:main@352: Commands [(<mppsolar.devices.jkbms.jkbms object at 0x75f6b418>, 'getCellData', 'getCellData', 'screen', None, None)]
2022-05-09 15:41:13,165:INFO:__init__:main@357: Looping 1 commands
2022-05-09 15:41:13,165:INFO:__init__:main@367: Getting results from device: jkbms device - name: unnamed, port: <mppsolar.io.serialio.SerialIO object at 0x75f6b4a8>, protocol: <mppsolar.protocols.jk02.jk02 object at 0x75f6b4c0> for command: getCellData, tag: getCellData, outputs: screen
2022-05-09 15:41:13,166:INFO:device:run_command@271: Running command getCellData
2022-05-09 15:41:13,166:INFO:jkabstractprotocol:get_full_command@73: Using protocol b'JK02' with 3 commands
2022-05-09 15:41:13,166:DEBUG:jkabstractprotocol:get_command_defn@104: get_command_defn for: getCellData
2022-05-09 15:41:13,167:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'getCellData'
2022-05-09 15:41:13,167:DEBUG:abstractprotocol:get_command_defn@62: Found command getCellData in protocol b'JK02'
2022-05-09 15:41:13,167:DEBUG:jkabstractprotocol:get_full_command@78: self._command = getCellData
2022-05-09 15:41:13,168:DEBUG:jkabstractprotocol:get_full_command@89: cmd with SOR: bytearray(b'\xaaU\x90\xeb\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2022-05-09 15:41:13,168:DEBUG:jkabstractprotocol:get_full_command@97: cmd with command code: bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2022-05-09 15:41:13,168:DEBUG:jkabstractprotocol:get_full_command@99: cmd with crc: bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10')
2022-05-09 15:41:13,168:INFO:device:run_command@298: full command bytearray(b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10') for command getCellData
2022-05-09 15:41:13,169:DEBUG:jkabstractprotocol:get_command_defn@104: get_command_defn for: getCellData
2022-05-09 15:41:13,169:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'getCellData'
2022-05-09 15:41:13,169:DEBUG:abstractprotocol:get_command_defn@62: Found command getCellData in protocol b'JK02'
2022-05-09 15:41:13,170:DEBUG:serialio:send_and_receive@19: port /dev/ttyUSB0, baudrate 115200
2022-05-09 15:41:13,173:DEBUG:serialio:send_and_receive@22: Executing command via serialio...
2022-05-09 15:41:14,577:DEBUG:serialio:send_and_receive@30: serial response was: b'\x00\x18U\xaa\xeb\x90\x02\x97\xeb\x0c\xe9\x0c\xe9\x0c\xe9\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0f\x00\x00\x00\xeb\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xaa3\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xc2\x00\xba\x00\xce\x00\x04\x00\x00\x00\x00\x00\x03l\x02\x00\xc0E\x04\x00\x00\x00\x00\x00\xeb\xe0\x01\x00\x00\x00\x00\x00\x15E#\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x07\x00\x00\x00\x00\x00#\x05\x00\x00\x00\x00\xbbd;@\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x05\x00\x00\xd7\x99\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x1eU\xaa\xeb\x90\x02\x98\xe8\x0c\xe8\x0c\xe8\x0c\xe9\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0f\x00\x00\x00\xe8\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00 [... cut ...] 0\x00\x00\x00\x00\x00\x00\x00\x00\x00\x07\x00\x00\x00\x00\x00#\x05\x00\x00\x00\x00\xbbd;@\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x05\x00\x00\xe0\x99\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00)U'
```

jblance, 2022-05-09:
the remainder was the start of another record

try latest commit and see where it gets you (you may need to manually clear your edits)

EasilyBoredEngineer, 2022-05-09:
@jblance - I had initially installed the stable version, as I said I'm new to git.. when I try to install the commit using the instructions in the how-to:-

````sudo pip3 install -e "git+https://github.com/jblance/mpp-solar.git#egg=mppsolar"````

it errors out with:-

"    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    ModuleNotFoundError: No module named 'setuptools'"

I've checked, and setupTools is definitely installed - any insights?

jblance, 2022-05-09:
Try with out sudo

jblance, 2022-05-09:
Or use sudo to globally install setuptools

syssi, 2022-05-09:
> this maybe a symptom of the mtu increase. there some kinda poor logic to look for the start of the record which looks to not work now, so getting rolled together messages

@EasilyBoredEngineer doesn't use bluetooth at the moment.The issue is unrelated to the changed MTU. It looks like the new frames are 10 bytes shorter (290 bytes) instead of the usual 300 (cellInfo) / 320 bytes (deviceInfo).

EasilyBoredEngineer, 2022-05-09:
@jblance ... yeah nope setupTools is installed as sudo. I'm now completely lost.

trying the above command without sudo errored.... have checked setup.py, appears repo is called mppsolar so I've no idea what's going on here.. 

pi@raspberrypi:~ $ pip3 install -e "git+https://github.com/jblance/mpp-solar.git#egg=mppsolar"
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Obtaining mppsolar from git+https://github.com/jblance/mpp-solar.git#egg=mppsolar
  Updating ./src/mppsolar clone
  Running command git fetch -q --tags
  error: cannot open .git/FETCH_HEAD: Permission denied
WARNING: Discarding git+https://github.com/jblance/mpp-solar.git#egg=mppsolar. Command errored out with exit status 255: git fetch -q --tags Check the logs for full command output.
ERROR: Could not find a version that satisfies the requirement mppsolar (unavailable)
ERROR: No matching distribution found for mppsolar (unavailable)

jblance, 2022-05-09:
So you use `sudo apt-get install python3-setuptools` to install?

EasilyBoredEngineer, 2022-05-09:
@jblance yes I did..

``` sudo apt-get install python3-setuptools
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-setuptools is already the newest version (52.0.0-4).
The following package was automatically installed and is no longer required:
  libfuse2
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.```

jblance, 2022-05-09:
Ok I chucked a new version on pypi so you should be able to update from pip

syssi, 2022-05-09:
I've checked the `getInfo` respons above:

> 2022-05-09 15:10:29,753:DEBUG:serialio:send_and_receive@30: serial response was: [...]

This response isn't valid becaue of a wrong `data_type`:

```
\xaa\xeb\x90\x02
                      ^^ indicates a cellInfo frame 
```

A response starting with `0xaa 0xeb 0x90 0x03` is required.

jblance, 2022-05-09:
> 2022-05-09 ***@***.***: serial

I think I noticed the check type function wasn't resetting the data. I
didn't restore coz I wasn't sure if it would break. Will restore that in  a
bit to see if it helps

jblance, 2022-05-09:
Did the debug have a "wrong type of record" message anywhere?

EasilyBoredEngineer, 2022-05-09:
given that I need to have the jkbms app open in order to use the 'solder off the back of bluetooth module' hack it is more than possible that there's messages passing across the serial port other than those directly generated by the command itself.. this could well be why getCellData works but getInfo doesn't - as the app itself may be on the status page when I'm running the command.

EasilyBoredEngineer, 2022-05-09:
accidentally closed....

jblance, 2022-05-09:
Latest commit has the check record reset re-enabled

syssi, 2022-05-09:
I've extracted the frames from the `putty4.log`: [putty4-frames-hex.txt](https://github.com/jblance/mpp-solar/files/8649676/putty4-frames-hex.txt)

Cell info (`0x02`) frame example (300 bytes):

```
0xaa 0xeb 0x90 0x02 0x95 0xed 0x0c 0xea 0x0c 0xe9 0x0c 0xe8 0x0c 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x0f 0x00 0x00 0x00 0xe9 0x0c 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0xa5 0x33 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0xcc 0x00 0xc8 0x00 0xda 0x00 0x04 0x00 0x00 0x00 0x00 0x00 0x03 0x6c 0x02 0x00 0xc0 0x45 0x04 0x00 0x00 0x00 0x00 0x00 0xeb 0xe0 0x01 0x00 0x00 0x00 0x00 0x00 0xc3 0x2f 0x23 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x07 0x00 0x00 0x00 0x00 0x00 0x23 0x05 0x00 0x00 0x00 0x00 0xbb 0x64 0x3b 0x40 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x00 0x05 0x00 0x00 0xa3 0xc4 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0xc9 0x55
```

Settings (`0x01`) frame example (320 bytes):

```
0xaa 0xeb 0x90 0x01 0x9d 0x58 0x02 0x00 0x00 0x28 0x0a 0x00 0x00 0xb8 0x0b 0x00 0x00 0x10 0x0e 0x00 0x00 0x48 0x0d 0x00 0x00 0x05 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0xc4 0x09 0x00 0x00 0xa8 0x61 0x00 0x00 0x1e 0x00 0x00 0x00 0x3c 0x00 0x00 0x00 0x40 0x0d 0x03 0x00 0x2c 0x01 0x00 0x00 0x3c 0x00 0x00 0x00 0x3c 0x00 0x00 0x00 0xd0 0x07 0x00 0x00 0xbc 0x02 0x00 0x00 0x58 0x02 0x00 0x00 0xbc 0x02 0x00 0x00 0x58 0x02 0x00 0x00 0x38 0xff 0xff 0xff 0x9c 0xff 0xff 0xff 0x84 0x03 0x00 0x00 0xbc 0x02 0x00 0x00 0x08 0x00 0x00 0x00 0x01 0x00 0x00 0x00 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0xc0 0x45 0x04 0x00 0xdc 0x05 0x00 0x00 0xb8 0x0b 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0xd5 0xaa 0x55 0x90 0xeb 0xc8 0x01 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x44 0x55
```

Device info (`0x03`) frame example (320 bytes):

```
0xaa 0xeb 0x90 0x03 0xb8 0x4a 0x4b 0x2d 0x42 0x32 0x41 0x38 0x53 0x32 0x30 0x50 0x00 0x00 0x00 0x00 0x00 0x39 0x2e 0x58 0x00 0x00 0x00 0x00 0x00 0x39 0x2e 0x30 0x31 0x4d 0x33 0x00 0x00 0x08 0x2f 0x23 0x00 0x1d 0x00 0x00 0x00 0x4a 0x4b 0x2d 0x42 0x32 0x41 0x38 0x53 0x32 0x30 0x50 0x00 0x00 0x00 0x00 0x00 0x31 0x32 0x33 0x34 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x32 0x31 0x31 0x31 0x30 0x39 0x00 0x00 0x31 0x30 0x39 0x31 0x35 0x31 0x35 0x31 0x35 0x33 0x00 0x30 0x30 0x30 0x30 0x00 0x30 0x31 0x32 0x33 0x34 0x35 0x36 0x37 0x38 0x39 0x30 0x31 0x32 0x61 0x00 0x00 0x31 0x32 0x33 0x34 0x35 0x36 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x08 0xaa 0x55 0x90 0xeb 0xc8 0x01 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x44 0x55
```

EasilyBoredEngineer, 2022-05-09:
> Latest commit has the check record reset re-enabled

Great - thanks, and thanks for reopening, @jblance. I'll track down the change, implement it on my version and double check that my 'bluetooth hack' isn't a receive only boondoggle.. which is a hunch I'm chasing right now. Hopefully the hex dump is nonetheless useful -

syssi, 2022-05-09:
@EasilyBoredEngineer The hex dump is super useful. Thanks! The length of the frames are equal to the frames of other JK-BMS models. I assume most of your issues are because of the piggybacking. I would be happy if you are able to identify a pin of the bluetooth module to shutdown/stop the module without unsoldering.

EasilyBoredEngineer, 2022-05-09:
@syssi roger that. I'll do some more ferreting around tomorrow (I'm in oz so it's end of day now). I wish I could find a model number or name of the bluetooth module so that I could get a better understanding of how it works. The only info I have is [this picture](https://raw.githubusercontent.com/syssi/esphome-jk-bms/main/images/jk-bd6a17s6p/005-labeled.jpg) and the [datasheet for the BK3432 IC](https://developer.tuya.com/en/docs/iot/bk3432-datasheet?id=K9wmepyr04ob1) it uses.
I may well try pulling the RST pin low tomorrow and see if that helps, as this (seemingly programmatic) slack of a functional wired serial port is problematic for these new boards.

syssi, 2022-05-09:
> 3. The RS485 and Display port use RS485 natively - as opposed to previous board versions which required an external OEM adaptor.

I talked to the seller and he told me the external OEM adapter is still required. Are you sure about the purpose of the parts between the UART of the processor and the connector? I assume the BMS can still be connected to an ESP (UART TTL) directly.

Smoeller0, 2022-05-09:
Hello
The Bluetooth module ist a JDY-18

I solder wires to my Module and connect a serial port. So im able to use serial port and still Bluetooth. 
When anyone need this info i can share.

Smoeller0, 2022-05-09:
My English is not great
Datasheet to the Module
http://myosuploads3.banggood.com/products/20190515/20190515050356JDY-18-V1.5Englishmanual.pdf

 getting no response unless the android app is active. This is true until u connect the pin State to VCC on the Module.

EasilyBoredEngineer, 2022-05-09:
> I talked to the seller and he told me the external OEM adapter is still required. Are you sure about the purpose of the parts between the UART of the processor and the connector? I assume the BMS can still be connected to an ESP (UART TTL) directly.

@syssi - yes absolutely 100 percent certain. Seller (minerals) said the same to me, but I checked with the manufacturer originally and he told me these ports are now native RS485, no TTL. The latter contention is supported by the board itself - there are two TTL to RS485 converters inline between the processor and the port. It would make zero sense to have them there if the seller is correct.

> My English is not great Datasheet to the Module http://myosuploads3.banggood.com/products/20190515/20190515050356JDY-18-V1.5Englishmanual.pdf
> 
> getting no response unless the android app is active. This is true until u connect the pin State to VCC on the Module.

@Smoeller0 your English is fine - and you're a legend :) Thankyou for this really great information.

Smoeller0, 2022-05-09:
I connected the BMS to an Raspberry and use nodered for read the Bms datas into an Influxdb and show it on graphana and it works since many days.

EasilyBoredEngineer, 2022-05-09:
@smoeller0 - thanks for that info. Out of interest, which BMS and Version are you using, please? Do you have a photo of your soldering work?

EasilyBoredEngineer, 2022-05-09:
@Smoeller0 - by the looks of that document, this is a different module. Have a look at the picture I posted above - different number of pins, it seems :( But thanks for your efforts, though - I appreciate it.

syssi, 2022-05-09:
This is the bluetooth module & BMS @Smoeller0 is talking about:

https://raw.githubusercontent.com/syssi/esphome-jk-bms/main/images/jk-bd6a17s6p/004.jpg
https://raw.githubusercontent.com/syssi/esphome-jk-bms/main/images/jk-bd6a17s6p/005-labeled.jpg
https://raw.githubusercontent.com/syssi/esphome-jk-bms/main/images/jk-bd6a17s6p/bk3432-qfn32.png

EasilyBoredEngineer, 2022-05-09:
@syssi yes my BMS is using the same module as your photos - this does not seem to be the JDY-18 that has been linked by @Smoeller0 - it has a different layout and different number of pins - please see attached screenshot from the JDY-18 datasheet
[image]

syssi, 2022-05-09:
@Smoeller0 Could you help to identify a possible "state pin" at this bluetooth module version? I assume if one of the pins (P10, P11, P12 or P13) isn't pulled the BMS doesn't respond to requests? @EasilyBoredEngineer you could try to measure the pins while the app is connect vs. the app isn't connected. 

https://raw.githubusercontent.com/syssi/esphome-jk-bms/main/images/jk-bd6a17s6p/005-labeled.jpg

Smoeller0, 2022-05-09:
Oh they update again. The Layout looks identical so try p11 to vcc

Smoeller0, 2022-05-09:
[image]

I use the new version as well and even not realize this ;)

EasilyBoredEngineer, 2022-05-09:
Finally my video of the board and a discussion particularly around the comms componentry has finished uploading - [JK-B2A8S20P board description and interfacing](https://youtu.be/2Pi0x7HxuH4).

syssi, 2022-05-09:
I've seen you own the display. I would be happy if you could provide a traffic capture of the display traffic some day too. :-)

EasilyBoredEngineer, 2022-05-09:
@syssi - sure thing, we can do that. Hey I saw that you mentioned on [the thread you linked above](https://github.com/syssi/esphome-jk-bms/issues/49) that you selected the wrong baud when you had earlier done traffic captures - small trick for you - if you try different bauds using realTerm, you'll see the red error indicator will flash really regularly if you're on the wrong baud - due to framing errors. So you can rule out the wrong baud pretty quickly...

@Smoeller0 thanks for your photos!
[image]

EasilyBoredEngineer, 2022-05-10:
Unfortunately there will be no more updates on this. Connecting VCC to pin 11 bricked my board.

sanwizard, 2022-05-10:
JK should be funding your work, and buy you new equipment.

EasilyBoredEngineer, 2022-05-10:
@sanwizard - that was a false alarm. It seems the onboard 3.3v regulator can't supply enough current for the ADUM1201 I installed to try and isolate the board (only 1.4mA). When removed, luckily the board commence working again :)

EasilyBoredEngineer, 2022-05-10:
Ok... the verdict with piggybacking off the bluetooth is...

1. Indeed, applying VCC to pin 11 does prevent you having to keep the jkbms app open on your mobile phone - but you still need to open the app at least once to 'get the data flowing' - so the app must send some sort of magic wakeup command (more on that later).
2. With the VCC to Pin 11 workaround, once you have opened the app, you can then close it and the same 'getCellData' frame just keeps repeating over and over and over again.. any commands sent just seem to be ignored, and you just keep getting the same 02 record repeating over and over again.
3. I swapped the RX and TX on my ADUM1201 and did a capture of the 'transmitted data', to try and see what happens at various points. I think you will possibly find this very interesting.

1. When I press the 'start' button on my BMS display, the following hex is sent on the TX line:- 00 4F 4B 0D 0A
2. When I open the JK-BMS app, the following hex is sent by the app:- AA 55 90 EB 97 00 E8 13 9E 26 AF C5 72 44 BC 6D 78 50 66 51 AA 55 90 EB 96 00 2F 66 8F 8E 4F A0 34 03 7C 72 20 46 12 4E
3. I think the second longer string of HEX is 'the magic wakeup command'.
4. When I scroll through the pages in the app, the following hex strings are sent:-

- Settings Page:- AA5590EB9600BD7B74BEF738119DE6038B4FCCE6
- Control Page:- AA5590EB96004216A7D08D9B7D9E10369D383572
- Back to Status Page:- AA5590EB9600D0CC6027807BE37E273FCED8C762

>

EasilyBoredEngineer, 2022-05-10:
Mucking around a bit further, even sending the hex strings I quoted above directly to the module via my FTDI doesn't do anything. I'm beginning to suspect the bluetooth module might assert a digital pin when it's ready to send data to the cpu, hence why I can't get the CPU to listen to anything I'm sending it.

I'm giving up on this line of enquiry (piggyybacking on bluetooth) for the moment. I'll wait for my ESP32 module and provide further updates then.
[captureWhole.txt](https://github.com/jblance/mpp-solar/files/8657427/captureWhole.txt)

EasilyBoredEngineer, 2022-05-11:
@jblance - I've found your [early discussion about the JK02 protocol](https://github.com/jblance/mpp-solar/issues/114).. I'm wondering if you can please refer me to any further work you've done on that protocol - I just simply want to understand how the CRC works and how the commands bytes that all seem to begin with AA 55 90 EB work.

As an example... my 'opening the bluetooth app' captures seem to constantly change from the example string I listed above, which is :- 

```AA 55 90 EB 97 00 E8 13 9E 26 AF C5 72 44 BC 6D 78 50 66 51 AA 55 90 EB 96 00 2F 66 8F 8E 4F A0 34 03 7C 72 20 46 12 4E```

The AA 55 90 EB 97 00 part always stays the same, but the rest of the string always changes - whereas I would have thought that simply opening the app would be the same 'command' sent from the app to the CPU everytime... trying to get a handle on what's happening there. Any help appreciated 👍

EasilyBoredEngineer, 2022-05-11:
As an FYI, here's a few more findings.

I think making up for the lack of a functional serial port by piggybacking the bluetooth module is just a complete fools errand at this stage - 

You can successfully use it to 'intercept' data going past on its way between the app and the bms microcontroller, and grab that data and interpret it using the @jblance JK02 library, in terms of using the uart pins on the module to SEND commands to the microcontroller just doesn't seem to work at all, whatsoever. They're just ignored.

I can see the commands from the app fly past, but I can't get any commands I inject on the line listened to - neither before I start the app, nor after.

I've tried to eliminate any CRC problem by sending the string "\xaa\x55\x90\xeb\x97\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x11" directly, and also tried to send hex dumps obtained by sniffing the turn on of the BMS and its interactions with the app, but I just can't get it to answer anything.

I've double checked that my FTDI's are functional (checked on the scope, and also used a different FTDI), they are. There must be something else that tells the microcontroller when to accept commands (eg a digital pin pulled low?). 

I also followed a hunch and completely disabled the bluetooth module by pulling its _RST pin low, still no dice.

I just don't know. Haven't been able to crack it - if anyone has any ideas, please let me know.

Nonetheless we do have a number of learnings:-

1. The bluetooth board on the JK-B2A8S20P uses 115,200 baud.
2. The bluetooth board can be disabled completely by pulling the _RST pin low ([see a labelled picture of the board here](https://raw.githubusercontent.com/syssi/esphome-jk-bms/main/images/jk-bd6a17s6p/005-labeled.jpg), from another user).
3. as @Smoeller0 pointed out, you can keep the board awake even if the bluetooth app is closed, by pulling the P11 pin high, but you must open the JK-bms app at least once to get the cell data flowing.

I'm now going to move on to sniffing and obtaining hex dumps from the screen interface for @syssi

EasilyBoredEngineer, 2022-05-11:
@syssi - please find hex dump from the data that goes to the display attached. 9600 baud seems to give the most consistent, patterned data - so I think that's the baud of that port.
[ScreenCapture9600.txt](https://github.com/jblance/mpp-solar/files/8666700/ScreenCapture9600.txt)

syssi, 2022-05-11:
Do you own a logic analyzer to monitor multiple I/Os (RX, TX, P10, P11, P12, P13) of the bluetooth module in parallel?

EasilyBoredEngineer, 2022-05-11:
> Do you own a logic analyzer to monitor multiple I/Os (RX, TX, P10, P11, P12, P13) of the bluetooth module in parallel?

No. I unfortunately don't have a particularly well equipped lab anymore

syssi, 2022-05-11:
@EasilyBoredEngineer Could you help to identify the start byte and end byte of a display frame by timing? It's hard to guess. :-)

I will provide a logic analyzer capture of my `JK-B1A24S-15P-V10.1.2` later the week.

EasilyBoredEngineer, 2022-05-11:
@syssi 
> Are you sure about the purpose of the parts between the UART of the processor and the connector? I assume the BMS can still be connected to an ESP (UART TTL) directly.
You and I are both right, kinda - but you're definitely more right than me. 

The key difference between this and previous versions seems to be that port marked RS485 used to be native TTL, and you'd then connect their dongle to convert it to RS485. So there's no more 'hooking up to the serial port' as it's now speaking at RS485 levels instead of TTL.

On this new board, the TTL from the BMS processor is converted to RS485 by that circuitry I pointed out - **but only at the physical level**. I had assumed it was a 'clever chip' that decoded the protocol, but it's not - it's just a dumb driver / receiver like a MAX232 but for RS485..

So you connect the RS485 cable directly to the port, then at the other end of your comms line you attach their RS485 dongle and it does the translation - at the BMS end, the protocol itself is handled by the firmware in the BMS microcontroller. That's something I hadn't understood until I read up more on the YD3082 - so the problem is likely to be that the RS485 isn't dead, I'm just a very big dill and I'm not speaking RS485anese to it.... so it's not answering me. Time to crack out the arduino, methinks...

The dongle they sell for this bms is at [the following url](https://www.aliexpress.com/item/1005003967587421.html?pdp_npi=2%40dis%21USD%2128.83%2117.30%21%21%21%21%21%40212a645216520572202352181ec975%21%21im)

syssi, 2022-05-11:
This screenshots should help a bit. Please try to pull up `P11` and `P11`. 

No connection / idle: 

[image]

Connection + 0x97 request + periodic responses

[image]

The colors and pin labels should match with the sigrok screenshot:

[image]

Raw capture (sigrok format) + sigrok UI config (labels + colors):

[jkbms-ble-connected-capture.zip](https://github.com/jblance/mpp-solar/files/8667235/jkbms-ble-connected-capture.zip)

EasilyBoredEngineer, 2022-05-11:
@syssi - legend :) 

Syssi,  could you please share the connection and command hex string that you are sending to receive this output?

Thankyou!

syssi, 2022-05-11:
I didn't try to inject traffic. I just captured what's happening between the bluetooth module and the BMS if I connect using the app. But I'm pretty sure if we do the same (pulling P11 and P12) the BMS should respond to our requests.

The app sends to commands to the BMS.

1. The first command is the `0x97` which retrieves the device info.
2. The second command is the `0x96` which enables the periodic cell info frame.

Please don't copy the values from the following screenshots because my logic analyzer (~10€) isn't accurate. It's fine if you null the full message like mpp-solar does. It's well tested:

https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms_ble/jk_bms_ble.cpp#L1054-L1075
[image]
[image]

syssi, 2022-05-11:
> So you connect the RS485 cable directly to the port, then at the other end of your comms line you attach their RS485 dongle and it does the translation - at the BMS end, the protocol itself is [...]

A cheap solution to connect a ESP8266 / Arduino to the native RS485 of your BMS is something like this:

https://de.aliexpress.com/item/1005002970555534.html

Try to avoid buying this version (called "DI DE RE RO") if you don't want to care about flow control:

https://de.aliexpress.com/item/1005003114276251.html

If you want to connect the BMS to your computer this is a good choice:

https://de.aliexpress.com/item/1005003032516569.html

Smoeller0, 2022-05-11:
need any help to read data from bms? 
In my system it is working. i read data from the bms via serial.
[image]

[21 more replies]

closed 2022-05-15 by EasilyBoredEngineer