jblance/mpp-solar #226: Making this repo work with a JK-BD6A20S10P

axyaxyfr, 2022-05-14:
I would very much like my Home Assistant or even better my three Growatt-5000's to see my batteries. I have two 16S batteries in parallel, both of them with this BMS.
I got hold of a rpi4 but here I couldn't get the BLE to work reliably. I also have a rpi3, and here I get a response. 

My output is this:
ps@raspberrypi:~ $ jkbms -p C8:47:8C:F7:88:FE -D -P JK02 -c getInfo
2022-05-14 19:09:48,916:INFO:__init__:main@218: Solar Device Command Utility, version: 0.12.2, influx_mqtt remove blank units
2022-05-14 19:09:48,917:DEBUG:__init__:main@232: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='jkbms', mqttc=<paho.mqtt.client.Client object at 0x7658ff40>, _isConnected=False)
2022-05-14 19:09:48,918:DEBUG:__init__:main@234: udp port 5555
2022-05-14 19:09:48,918:DEBUG:__init__:main@237: Using Mongo None with mppsolar
2022-05-14 19:09:48,918:INFO:__init__:main@337: Creating device "unnamed" (type: "jkbms") on port "C8:47:8C:F7:88:FE (porttype=None)" using protocol "JK02"
2022-05-14 19:09:48,925:DEBUG:__init__:main@341: device_class <class 'mppsolar.devices.jkbms.jkbms'>
2022-05-14 19:09:48,926:DEBUG:device:__init__@33: __init__ args ()
2022-05-14 19:09:48,926:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': 'C8:47:8C:F7:88:FE', 'protocol': 'JK02', 'baud': 2400, 'porttype': None, 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='jkbms', mqttc=<paho.mqtt.client.Client object at 0x7658ff40>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2022-05-14 19:09:48,927:DEBUG:__init__:get_port_type@50: port matches jkble ':'
2022-05-14 19:09:48,927:INFO:__init__:get_port@101: Using jkbleio for communications
2022-05-14 19:09:48,986:DEBUG:__init__:get_protocol@12: Protocol JK02
2022-05-14 19:09:48,998:DEBUG:device:__init__@38: __init__ name unnamed, port <mppsolar.inout.jkbleio.JkBleIO object at 0x75f5b5f8>, protocol <mppsolar.protocols.jk02.jk02 object at 0x75f5b610>
2022-05-14 19:09:48,999:DEBUG:__init__:main@390: Commands [(<mppsolar.devices.jkbms.jkbms object at 0x75f5b568>, 'getInfo', 'getInfo', 'screen', None, None)]
2022-05-14 19:09:48,999:INFO:__init__:main@395: Looping 1 commands
2022-05-14 19:09:49,000:INFO:__init__:main@405: Getting results from device: jkbms device - name: unnamed, port: <mppsolar.inout.jkbleio.JkBleIO object at 0x75f5b5f8>, protocol: <mppsolar.protocols.jk02.jk02 object at 0x75f5b610> for command: getInfo, tag: getInfo, outputs: screen
2022-05-14 19:09:49,000:INFO:device:run_command@274: Running command getInfo
2022-05-14 19:09:49,000:INFO:jkabstractprotocol:get_full_command@73: Using protocol b'JK02' with 3 commands
2022-05-14 19:09:49,001:DEBUG:jkabstractprotocol:get_command_defn@106: get_command_defn for: getInfo
2022-05-14 19:09:49,001:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'getInfo'
2022-05-14 19:09:49,001:DEBUG:abstractprotocol:get_command_defn@62: Found command getInfo in protocol b'JK02'
2022-05-14 19:09:49,002:DEBUG:jkabstractprotocol:get_full_command@80: self._command = getInfo
2022-05-14 19:09:49,002:DEBUG:jkabstractprotocol:get_full_command@91: cmd with SOR: bytearray(b'\xaaU\x90\xeb\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2022-05-14 19:09:49,002:DEBUG:jkabstractprotocol:get_full_command@99: cmd with command code: bytearray(b'\xaaU\x90\xeb\x97\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2022-05-14 19:09:49,003:DEBUG:jkabstractprotocol:get_full_command@101: cmd with crc: bytearray(b'\xaaU\x90\xeb\x97\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x11')
2022-05-14 19:09:49,003:INFO:device:run_command@303: full command bytearray(b'\xaaU\x90\xeb\x97\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x11') for command getInfo
2022-05-14 19:09:49,003:DEBUG:jkabstractprotocol:get_command_defn@106: get_command_defn for: getInfo
2022-05-14 19:09:49,004:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'getInfo'
2022-05-14 19:09:49,004:DEBUG:abstractprotocol:get_command_defn@62: Found command getInfo in protocol b'JK02'
2022-05-14 19:09:49,004:INFO:jkabstractprotocol:get_full_command@73: Using protocol b'JK02' with 3 commands
2022-05-14 19:09:49,005:DEBUG:jkabstractprotocol:get_command_defn@106: get_command_defn for: getInfo
2022-05-14 19:09:49,005:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'getInfo'
2022-05-14 19:09:49,005:DEBUG:abstractprotocol:get_command_defn@62: Found command getInfo in protocol b'JK02'
2022-05-14 19:09:49,005:DEBUG:jkabstractprotocol:get_full_command@80: self._command = getInfo
2022-05-14 19:09:49,006:DEBUG:jkabstractprotocol:get_full_command@91: cmd with SOR: bytearray(b'\xaaU\x90\xeb\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2022-05-14 19:09:49,006:DEBUG:jkabstractprotocol:get_full_command@99: cmd with command code: bytearray(b'\xaaU\x90\xeb\x97\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2022-05-14 19:09:49,006:DEBUG:jkabstractprotocol:get_full_command@101: cmd with crc: bytearray(b'\xaaU\x90\xeb\x97\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x11')
2022-05-14 19:09:49,007:INFO:jkbleio:send_and_receive@29: full command bytearray(b'\xaaU\x90\xeb\x97\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x11') for command getInfo
2022-05-14 19:09:49,007:DEBUG:jkabstractprotocol:get_command_defn@106: get_command_defn for: getInfo
2022-05-14 19:09:49,007:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'getInfo'
2022-05-14 19:09:49,008:DEBUG:abstractprotocol:get_command_defn@62: Found command getInfo in protocol b'JK02'
2022-05-14 19:09:49,008:DEBUG:jkbleio:send_and_receive@33: expected record type 3 for command getInfo
2022-05-14 19:09:49,008:INFO:jkbleio:ble_connect@57: Attempting to connect to C8:47:8C:F7:88:FE
2022-05-14 19:09:49,974:INFO:jkbleio:ble_get_data@88: Connected to b''
2022-05-14 19:09:50,463:INFO:jkbleio:ble_get_data@102: Read characteristic: Characteristic <ffe1>, handle 12
2022-05-14 19:09:50,464:INFO:jkbleio:ble_get_data@110: Enable 0x0b handle
2022-05-14 19:09:50,466:INFO:jkbleio:ble_get_data@113: Enable read handle
2022-05-14 19:09:50,468:INFO:jkbleio:ble_get_data@117: Write getInfo to read handle
2022-05-14 19:09:50,596:DEBUG:jkbledelegate:handleNotification@27: From handle: 0x12 Got 128 bytes of data
2022-05-14 19:09:50,598:DEBUG:jkbledelegate:handleNotification@29: Pre wipe to start bytearray(b'U\xaa\xeb\x90\x03!JK-BD6A20S10P\x00\x00\x008.X\x00\x00\x00\x00\x008.0.8G\x00\x00\x10E\xaa\x00\x01\x00\x00\x00SN 22018\x00\x00\x00\x00\x00\x00\x00\x001234\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00220104\x00\x001111504169\x000000\x00Input Userdata\x00\x00123456\x00\x00\x00\x00')
2022-05-14 19:09:50,599:DEBUG:jkbledelegate:handleNotification@31: Post wipe to start bytearray(b'U\xaa\xeb\x90\x03!JK-BD6A20S10P\x00\x00\x008.X\x00\x00\x00\x00\x008.0.8G\x00\x00\x10E\xaa\x00\x01\x00\x00\x00SN 22018\x00\x00\x00\x00\x00\x00\x00\x001234\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00220104\x00\x001111504169\x000000\x00Input Userdata\x00\x00123456\x00\x00\x00\x00')
2022-05-14 19:09:50,600:DEBUG:jkabstractprotocol:is_record_correct_type@162: Record is type 3
2022-05-14 19:09:50,601:DEBUG:jkabstractprotocol:is_record_start@147: SOR found in record
2022-05-14 19:09:50,616:DEBUG:jkbledelegate:handleNotification@27: From handle: 0x12 Got 128 bytes of data
2022-05-14 19:09:50,617:DEBUG:jkbledelegate:handleNotification@29: Pre wipe to start bytearray(b'U\xaa\xeb\x90\x03!JK-BD6A20S10P\x00\x00\x008.X\x00\x00\x00\x00\x008.0.8G\x00\x00\x10E\xaa\x00\x01\x00\x00\x00SN 22018\x00\x00\x00\x00\x00\x00\x00\x001234\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00220104\x00\x001111504169\x000000\x00Input Userdata\x00\x00123456\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
[... cut ...]
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/devices/device.py", line 316, in run_command
    raw_response = self._port.send_and_receive(
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/inout/jkbleio.py", line 37, in send_and_receive
    response = self.ble_get_data(full_command)
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/inout/jkbleio.py", line 145, in ble_get_data
    log.debug(f"Record now {self.record} len {len(self.record)}")
TypeError: object of type 'NoneType' has no len()

Any suggestions?
TIA
Silden

syssi, 2022-05-14:
What's the exact model, hard- und software version of your BMS? Could you provide a screenshot of the android app (info page).

jblance, 2022-05-14:
That looks like a bug. I'll need to add a fix

axyaxyfr, 2022-05-15:
Here's the screenshot.
[image]

jblance, 2022-05-15:
Ive uploaded a new version that may fix this issue (v0.12.5)
Not tested as I cant get to me testing environment at the moment

Give it a go at let me know what your results are

Regards
John

axyaxyfr, 2022-05-15:
Thanks for the quick reply and update. The output still looks the same:

Successfully installed mppsolar-0.12.5
ps@raspberrypi:~ $ jkbms -p C8:47:8C:F7:88:FE -D -P JK02 -c getInfo
2022-05-15 07:42:09,954:INFO:__init__:main@218: Solar Device Command Utility, version: 0.12.5, tweak jkbleio for getInfo and to return first 300 bytes
2022-05-15 07:42:09,954:DEBUG:__init__:main@232: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='jkbms', mqttc=<paho.mqtt.client.Client object at 0x76656f40>, _isConnected=False)
2022-05-15 07:42:09,955:DEBUG:__init__:main@234: udp port 5555
2022-05-15 07:42:09,955:DEBUG:__init__:main@237: Using Mongo None with mppsolar
2022-05-15 07:42:09,955:INFO:__init__:main@337: Creating device "unnamed" (type: "jkbms") on port "C8:47:8C:F7:88:FE (porttype=None)" using protocol "JK02"
2022-05-15 07:42:09,962:DEBUG:__init__:main@341: device_class <class 'mppsolar.devices.jkbms.jkbms'>
2022-05-15 07:42:09,962:DEBUG:device:__init__@33: __init__ args ()
2022-05-15 07:42:09,963:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': 'C8:47:8C:F7:88:FE', 'protocol': 'JK02', 'baud': 2400, 'porttype': None, 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='jkbms', mqttc=<paho.mqtt.client.Client object at 0x76656f40>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2022-05-15 07:42:09,963:DEBUG:__init__:get_port_type@50: port matches jkble ':'
2022-05-15 07:42:09,963:INFO:__init__:get_port@101: Using jkbleio for communications
2022-05-15 07:42:10,023:DEBUG:__init__:get_protocol@12: Protocol JK02
2022-05-15 07:42:10,034:DEBUG:device:__init__@38: __init__ name unnamed, port <mppsolar.inout.jkbleio.JkBleIO object at 0x760225f8>, protocol <mppsolar.protocols.jk02.jk02 object at 0x76022610>
2022-05-15 07:42:10,035:DEBUG:__init__:main@390: Commands [(<mppsolar.devices.jkbms.jkbms object at 0x76022568>, 'getInfo', 'getInfo', 'screen', None, None)]
2022-05-15 07:42:10,035:INFO:__init__:main@395: Looping 1 commands
2022-05-15 07:42:10,035:INFO:__init__:main@405: Getting results from device: jkbms device - name: unnamed, port: <mppsolar.inout.jkbleio.JkBleIO object at 0x760225f8>, protocol: <mppsolar.protocols.jk02.jk02 object at 0x76022610> for command: getInfo, tag: getInfo, outputs: screen
2022-05-15 07:42:10,036:INFO:device:run_command@274: Running command getInfo
2022-05-15 07:42:10,036:INFO:jkabstractprotocol:get_full_command@73: Using protocol b'JK02' with 3 commands
2022-05-15 07:42:10,036:DEBUG:jkabstractprotocol:get_command_defn@106: get_command_defn for: getInfo
2022-05-15 07:42:10,036:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'getInfo'
2022-05-15 07:42:10,037:DEBUG:abstractprotocol:get_command_defn@62: Found command getInfo in protocol b'JK02'
2022-05-15 07:42:10,037:DEBUG:jkabstractprotocol:get_full_command@80: self._command = getInfo
2022-05-15 07:42:10,037:DEBUG:jkabstractprotocol:get_full_command@91: cmd with SOR: bytearray(b'\xaaU\x90\xeb\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2022-05-15 07:42:10,037:DEBUG:jkabstractprotocol:get_full_command@99: cmd with command code: bytearray(b'\xaaU\x90\xeb\x97\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2022-05-15 07:42:10,038:DEBUG:jkabstractprotocol:get_full_command@101: cmd with crc: bytearray(b'\xaaU\x90\xeb\x97\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x11')
2022-05-15 07:42:10,038:INFO:device:run_command@303: full command bytearray(b'\xaaU\x90\xeb\x97\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x11') for command getInfo
2022-05-15 07:42:10,038:DEBUG:jkabstractprotocol:get_command_defn@106: get_command_defn for: getInfo
2022-05-15 07:42:10,038:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'getInfo'
2022-05-15 07:42:10,039:DEBUG:abstractprotocol:get_command_defn@62: Found command getInfo in protocol b'JK02'
2022-05-15 07:42:10,039:INFO:jkabstractprotocol:get_full_command@73: Using protocol b'JK02' with 3 commands
2022-05-15 07:42:10,039:DEBUG:jkabstractprotocol:get_command_defn@106: get_command_defn for: getInfo
2022-05-15 07:42:10,039:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'getInfo'
2022-05-15 07:42:10,039:DEBUG:abstractprotocol:get_command_defn@62: Found command getInfo in protocol b'JK02'
2022-05-15 07:42:10,040:DEBUG:jkabstractprotocol:get_full_command@80: self._command = getInfo
2022-05-15 07:42:10,040:DEBUG:jkabstractprotocol:get_full_command@91: cmd with SOR: bytearray(b'\xaaU\x90\xeb\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2022-05-15 07:42:10,040:DEBUG:jkabstractprotocol:get_full_command@99: cmd with command code: bytearray(b'\xaaU\x90\xeb\x97\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2022-05-15 07:42:10,040:DEBUG:jkabstractprotocol:get_full_command@101: cmd with crc: bytearray(b'\xaaU\x90\xeb\x97\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x11')
2022-05-15 07:42:10,041:INFO:jkbleio:send_and_receive@29: full command bytearray(b'\xaaU\x90\xeb\x97\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x11') for command getInfo
2022-05-15 07:42:10,041:DEBUG:jkabstractprotocol:get_command_defn@106: get_command_defn for: getInfo
2022-05-15 07:42:10,041:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'getInfo'
[... cut ...]
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/devices/device.py", line 316, in run_command
    raw_response = self._port.send_and_receive(
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/inout/jkbleio.py", line 37, in send_and_receive
    response = self.ble_get_data(full_command)
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/inout/jkbleio.py", line 150, in ble_get_data
    log.debug(f"Record now {self.record} len {len(self.record)}")
TypeError: object of type 'NoneType' has no len()

axyaxyfr, 2022-05-15:
But getCellData now looks better/right!
Parameter                     	Value           Unit
header                        	55aaeb90       	    
record_type                   	02             	    
record_counter                	36             	    
voltage_cell01                	3.312          	V   
voltage_cell02                	3.313          	V   
voltage_cell03                	3.312          	V   
voltage_cell04                	3.313          	V   
voltage_cell05                	3.313          	V   
voltage_cell06                	3.313          	V   
voltage_cell07                	3.31           	V   
voltage_cell08                	3.313          	V   
voltage_cell09                	3.313          	V   
voltage_cell10                	3.313          	V   
voltage_cell11                	3.313          	V   
voltage_cell12                	3.313          	V   
voltage_cell13                	3.313          	V   
voltage_cell14                	3.313          	V   
voltage_cell15                	3.313          	V   
voltage_cell16                	3.312          	V   
voltage_cell17                	0.0            	V   
voltage_cell18                	0.0            	V   
voltage_cell19                	0.0            	V   
voltage_cell20                	0.0            	V   
voltage_cell21                	0.0            	V   
voltage_cell22                	0.0            	V   
voltage_cell23                	0.0            	V   
voltage_cell24                	0.0            	V   
average_cell_voltage          	3.313          	V   
delta_cell_voltage            	0.001          	V   
current_balancer              	2.048          	    
resistance_cell01             	0.438          	Ohm 
resistance_cell02             	0.432          	Ohm 
resistance_cell03             	0.43           	Ohm 
resistance_cell04             	0.432          	Ohm 
resistance_cell05             	0.428          	Ohm 
resistance_cell06             	0.429          	Ohm 
resistance_cell07             	0.428          	Ohm 
resistance_cell08             	0.427          	Ohm 
resistance_cell09             	0.426          	Ohm 
resistance_cell10             	0.427          	Ohm 
resistance_cell11             	0.424          	Ohm 
resistance_cell12             	0.425          	Ohm 
resistance_cell13             	0.425          	Ohm 
resistance_cell14             	0.426          	Ohm 
resistance_cell15             	0.425          	Ohm 
resistance_cell16             	0.425          	Ohm 
resistance_cell17             	0.0            	Ohm 
resistance_cell18             	0.0            	Ohm 
resistance_cell19             	0.0            	Ohm 
resistance_cell20             	0.0            	Ohm 
resistance_cell21             	0.0            	Ohm 
resistance_cell22             	0.0            	Ohm 
resistance_cell23             	0.0            	Ohm 
resistance_cell24             	0.0            	Ohm 
battery_voltage               	53.002         	V   
battery_power                 	341758         	W   
balance_current               	6.448          	A   
battery_t1                    	24.5           	°C  
battery_t2                    	24.3           	°C  
mos_temp                      	25.0           	°C  
percent_remain                	64             	%   
capacity_remain               	155.503        	Ah  
nominal_capacity              	240.0          	Ah  
cycle_count                   	5              	    
cycle_capacity                	1296.095       	Ah  
time                          	129D16H11M57S  	    
current_charge                	6.66           	    
current_discharge             	0.0

axyaxyfr, 2022-05-15:
This is fan-fugu-tastic :-)
Now off to create an mqtt in my Home Assistant, and after that it's time to have a look at the Growatt's
Thanks a bunch.

Silden

closed 2022-05-15 by axyaxyfr