jblance/mpp-solar #227: JKBMS JK-B2A24S20P version 4.3.4

robyice, 2022-05-20:
I am guessing that JKBMS JK-B2A24S20P version 4.3.4 
Hardware Ver 10-.x-W

I am guessing that the new version of the JKBMS is not currently supported I have tried to communicate with it over Bluetooth and RS485 with no luck.

I can see that it tries to connect and send commands but no response is received.

preussal, 2022-08-26:
I have two BMS

BMS 1
JK-B2A24S15P
HW: 10.XW
SW 10.08
Bluetooth OK
RS485 or via TTL no answer

BMS 2
JK-B2A24S20P
HW: 10.XW
SW 10.07
Bluetooth no connect. maybe Raspi Problem
RS485 or via TTL I get an answer but the data are strange

Both BMS work perfectly via the app

jblance, 2022-11-09:
to help troubleshooting please add the debug flag `-D` to the command and post the output