jblance/mpp-solar #228: New JK BMS Hardware Model JK-B2A8S20P-H Erroneous Readings

bluetrepidation, 2022-06-05:
I was able to successfully poll this new hardware with the JK02 protocol. Model: JK-B2A8S20P-H Hardware: 11.XW Software: 11.12 Some of the readings don't match the JIKONG BMS app. See resistance_cell04 through 8, capacity_remain, and cycle_count?

`Command: getCellData - BLE Cell Data inquiry

------------------------------------------------------------

Parameter                     	Value           Unit

header                        	55aaeb90       	    

record_type                   	02             	    

record_counter                	47             	    

voltage_cell01                	3.186          	V   

voltage_cell02                	3.185          	V   

voltage_cell03                	3.186          	V   

voltage_cell04                	3.185          	V   

voltage_cell05                	3.185          	V   

voltage_cell06                	3.186          	V   

voltage_cell07                	3.185          	V   

voltage_cell08                	3.186          	V   

voltage_cell09                	0.0            	V   

voltage_cell10                	0.0            	V   

voltage_cell11                	0.0            	V   

voltage_cell12                	0.0            	V   

voltage_cell13                	0.0            	V   

voltage_cell14                	0.0            	V   

voltage_cell15                	0.0            	V   

voltage_cell16                	0.0            	V   

voltage_cell17                	0.0            	V   

voltage_cell18                	0.0            	V   

voltage_cell19                	0.0            	V   

voltage_cell20                	0.0            	V   

voltage_cell21                	0.0            	V   

voltage_cell22                	0.0            	V   

voltage_cell23                	0.0            	V   

voltage_cell24                	0.0            	V   

average_cell_voltage          	0.0            	V   

delta_cell_voltage            	0.0            	V   

current_balancer              	0.0            	    

resistance_cell01             	0.0            	Ohm 

resistance_cell02             	0.0            	Ohm 

resistance_cell03             	0.0            	Ohm 

resistance_cell04             	0.255          	Ohm 

resistance_cell05             	0.0            	Ohm 

resistance_cell06             	3.186          	Ohm 

resistance_cell07             	0.002          	Ohm 

resistance_cell08             	0.007          	Ohm 

resistance_cell09             	0.049          	Ohm 

resistance_cell10             	0.049          	Ohm 

resistance_cell11             	0.049          	Ohm 

resistance_cell12             	0.049          	Ohm 

resistance_cell13             	0.049          	Ohm 

resistance_cell14             	0.049          	Ohm 

resistance_cell15             	0.049          	Ohm 

resistance_cell16             	0.048          	Ohm 

resistance_cell17             	0.0            	Ohm 

resistance_cell18             	0.0            	Ohm 

resistance_cell19             	0.0            	Ohm 

resistance_cell20             	0.0            	Ohm 

resistance_cell21             	0.0            	Ohm 

resistance_cell22             	0.0            	Ohm 

resistance_cell23             	0.0            	Ohm 

resistance_cell24             	0.0            	Ohm 

battery_voltage               	0.0            	V   

battery_power                 	0              	W   

balance_current               	0.0            	A   

battery_t1                    	0.0            	°C  

battery_t2                    	0.0            	°C  

mos_temp                      	0.0            	°C  

percent_remain                	0              	%   

capacity_remain               	17694.72       	Ah  

nominal_capacity              	0.0            	Ah  

cycle_count                   	25485          	    

cycle_capacity                	14.883         	Ah  

time                          	157D0H22M41S   	    

current_charge                	25.6           	    

current_discharge             	0.0            	    `

Debug attached.
[BMS_Debug.txt](https://github.com/jblance/mpp-solar/files/8840961/BMS_Debug.txt)

jblance, 2022-06-07:
looks like the data is offset from the definitions
is it possible to do another debug and a screenshot of the app as close in time as possible?

bluetrepidation, 2022-06-08:
Hello. It looks like the app hijacks the bluetooth connection so I get a timeout error. I took a screenshot, closed the app, then ran the python app and captured the files below. Hopefully within a few seconds is fast enough? See attached.
[BMS_Debug_2.txt](https://github.com/jblance/mpp-solar/files/8865125/BMS_Debug_2.txt)
[image]

stcan, 2022-11-07:
Hi,

I have a JK-B2A8S20P where the first voltage values of the single cells still fit well, but later values do not.
Is there already a solution for the problem?  I have installed the latest version (pip3 install -e "git+https://github.com/jblance/mpp-solar.git#egg=mppsolar")

I would appreciate any help.

Thank you,
steve

pi@raspberrypi: jkbms -p c8:47:8c:a5:c3:68 -b 115200 -P JK02 -c getInfo 
Command: getInfo - BLE Device Information inquiry
------------------------------------------------------------
Parameter                     	Value           Unit
header                        	55aaeb90       	    
record_type                   	03             	    
record_counter              	87             	    
device_model                	JK_B2A8S20P    	    
hardware_version          	11.XW          	    
software_version           	11.17H         	    
up_time                       	36D4H10M0S     	    
power-on_times             	4              	    
device_name                 	bms01          	    
device_passcode           	xxxxxxxxx        	    
manufacturing_date       	221002         	    
serial_number                	2062015280     	    
passcode                      	0000           	    
user_data                     	Input Userdata 	    
setup_passcode            	xxxxxxxxx 

       	    
pi@raspberrypi: jkbms -p c8:47:8c:e5:be:34 -b 115200 -P JK02 
Command: getCellData - BLE Cell Data inquiry
------------------------------------------------------------
Parameter                     	        Value           Unit
header                        	        55aaeb90       	    
record_type                   	        02             	    
record_counter                	92             	    
voltage_cell01                	3.214          	V   
voltage_cell02                	3.214          	V   
voltage_cell03                	3.215          	V   
voltage_cell04                	3.214          	V   
voltage_cell05                	3.214          	V   
voltage_cell06                	3.215          	V   
voltage_cell07                	3.214          	V   
voltage_cell08                	3.214          	V   
voltage_cell09                	0.0            	V   
voltage_cell10                	0.0            	V   
voltage_cell11                	0.0            	V   
voltage_cell12                	0.0            	V   
voltage_cell13                	0.0            	V   
voltage_cell14                	0.0            	V   
voltage_cell15                	0.0            	V   
voltage_cell16                	0.0            	V   
voltage_cell17                	0.0            	V   
voltage_cell18                	0.0            	V   
voltage_cell19                	0.0            	V   
voltage_cell20                	0.0            	V   
voltage_cell21                	0.0            	V   
voltage_cell22                	0.0            	V   
voltage_cell23                	0.0            	V   
voltage_cell24                	0.0            	V   
average_cell_voltage          	0.0            	V   
delta_cell_voltage            	0.0            	V   
current_balancer              	0.0            	    
resistance_cell01             	0.0            	Ohm 
resistance_cell02             	0.0            	Ohm 
resistance_cell03             	0.0            	Ohm 
resistance_cell04             	0.255          	Ohm 
resistance_cell05             	0.0            	Ohm 
resistance_cell06             	3.214          	Ohm 
resistance_cell07             	0.001          	Ohm 
resistance_cell08             	0.256          	Ohm 
resistance_cell09             	0.048          	Ohm 
resistance_cell10             	0.049          	Ohm 
resistance_cell11             	0.046          	Ohm 
resistance_cell12             	0.045          	Ohm 
resistance_cell13             	0.046          	Ohm 
resistance_cell14             	0.048          	Ohm 
resistance_cell15             	0.048          	Ohm 
resistance_cell16             	0.049          	Ohm 
resistance_cell17             	0.0            	Ohm 
resistance_cell18             	0.0            	Ohm 
resistance_cell19             	0.0            	Ohm 
resistance_cell20             	0.0            	Ohm 
resistance_cell21             	0.0            	Ohm 
resistance_cell22             	0.0            	Ohm 
resistance_cell23             	0.0            	Ohm 
resistance_cell24             	0.0            	Ohm 
battery_voltage               	0.0            	V   
battery_power                 	0              	W   
balance_current               	0.0            	A   
battery_t1                    	        0.0            	°C  
battery_t2                    	        0.0            	°C  
mos_temp                      	        0.0            	°C  
percent_remain                	0              	%   
capacity_remain               	8781.824       	Ah  
nominal_capacity              	0.0            	Ah  
cycle_count                   	        25713          	    
cycle_capacity                	0.0            	Ah  
time                          	        93D7H10M53S    	    
current_charge                	25.6           	    
current_discharge             	0.0

syssi, 2022-11-07:
The payload/offset has changed a bit because 32 cells are supported now: https://github.com/syssi/esphome-jk-bms/blob/main/docs/pdus/JK-B2A8S20P-11.17-BLE-JK02_32S.txt

jblance, 2022-11-07:
not supported yet
can you post a debug (particularly the bit that contains the response from
the inverter) and i'll get a new protocol definition for testing

syssi, 2022-11-07:
This is my implementation of the new payload: https://github.com/syssi/esphome-jk-bms/commit/6f44d189c949837a40a1455ee85df65119d8f8ec

stcan, 2022-11-08:
Thank you for the hint with the 32 cells... I added 8 cells in jk02.py and the read values fit perfectly !!!

Thanks a lot.

      ```
 ["LittleHex2Short:r/1000", 2, "Voltage_Cell15", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell16", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell17", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell18", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell19", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell20", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell21", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell22", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell23", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell24", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell25", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell26", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell27", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell28", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell29", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell30", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell31", "V"],
            ["LittleHex2Short:r/1000", 2, "Voltage_Cell32", "V"],
            ["discard", 4, "discard1", ""],
            ["LittleHex2Short:r/1000", 2, "Average_Cell_Voltage", "V"],
            ["LittleHex2Short:r/1000", 2, "Delta_Cell_Voltage", "V"],
            ["LittleHex2Short:r/1000", 2, "Current_Balancer", ""],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell01", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell02", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell03", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell04", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell05", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell06", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell07", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell08", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell09", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell10", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell11", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell12", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell13", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell14", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell15", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell16", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell17", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell18", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell19", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell20", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell21", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell22", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell23", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell24", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell25", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell26", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell27", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell28", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell29", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell30", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell31", "Ohm"],
            ["LittleHex2Short:r/1000", 2, "Resistance_Cell32", "Ohm"],
```

jkbms -p c8:47:8c:e5:be:34 -b 115200 -P JK02 -c getCellData
Command: getCellData - BLE Cell Data inquiry
------------------------------------------------------------
Parameter                       Value           Unit
header                          55aaeb90
record_type                     02
record_counter                  136
voltage_cell01                  3.21            V
voltage_cell02                  3.212           V
[... 54 lines cut ...]
resistance_cell22               0.0             Ohm
resistance_cell23               0.0             Ohm
resistance_cell24               0.0             Ohm
resistance_cell25               0.0             Ohm
resistance_cell26               0.0             Ohm
resistance_cell27               0.0             Ohm
resistance_cell28               0.0             Ohm
resistance_cell29               0.0             Ohm
resistance_cell30               0.0             Ohm
resistance_cell31               0.0             Ohm
resistance_cell32               0.0             Ohm
battery_voltage                 25.68           V
battery_power                   0               W
balance_current                 0.0             A
battery_t1                      11.9            °C
battery_t2                      11.7            °C
mos_temp                        0.0             °C
percent_remain                  65              %
capacity_remain                 151.165         Ah
nominal_capacity                230.0           Ah
cycle_count                     0
cycle_capacity                  178.097         Ah
time                            37D1H43M35S
current_charge                  0.003
current_discharge               0.0

jblance, 2022-11-08:
ive had a first crack at adding this as jkv11 protocol (in version 0.14.1+)
not sure the protocol definition is completely correct - did you adjust
anything other that adding the  extra cells?
if you get a chance to test (or send me your edited version) i can ensure
it is added to the code

stcan, 2022-11-09:
I have not changed anything except adding 8 cells in the voltage and resistance section.
This cron job has run 24h so far without any problems:
***** /usr/local/bin/jkbms -p c8:47:8c:e5:be:34 -b 115200 -P JK02 -c getCellData -o mqtt -q 192.168.0.55 > /dev/null 2>&1

The edited file: [jk02.zip](https://github.com/jblance/mpp-solar/files/9974586/jk02.zip)

closed 2023-01-08 by jblance