jblance/mpp-solar #229: JK-BD6A17S8P - problem with reading cells voltages and other stuff

paolss, 2022-06-23:
jkbms -p C8:47:8C:E2:A6:29 --getstatus -D

.....
.....
bytearray(b'U\xaa\xeb\x90\x02c\xcc\x0c\xce\x0c\xce\x0c\xcb\x0c\xcd\x0c\xcb\x0c\xcc\x0c\xcc\x0c\xcc\x0c\xcb\x0c\xcd\x0c\xcc\x0c\xcb\x0c\xcd\x0c\xcd\x0c\xcd\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0\x00\xff\xff\x00\x00\xcc\x0c\x03\x00\x00\x03\x8c\x00\x86\x00\x8a\x00\x86\x00\x8b\x00\x85\x00\x8b\x00\x85\x00\x8a\x00\x85\x00\x89\x00\x86\x00\x8a\x00\x85\x00\x8a\x00\x86\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xc5\xcc\x00\x00\xb9\xdb\n\x00\xf9\xca')
2022-06-24 01:18:59,968:DEBUG:jkbledelegate:handleNotification@31: Post wipe to start bytearray(b'U\xaa\xeb\x90\x02c\xcc\x0c\xce\x0c\xce\x0c\xcb\x0c\xcd\x0c\xcb\x0c\xcc\x0c\xcc\x0c\xcc\x0c\xcb\x0c\xcd\x0c\xcc\x0c\xcb\x0c\xcd\x0c\xcd\x0c\xcd\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\xff\x00\x00\xcc\x0c\x03\x00\x00\x03\x8c\x00\x86\x00\x8a\x00\x86\x00\x8b\x00\x85\x00\x8b\x00\x85\x00\x8a\x00\x85\x00\x89\x00\x86\x00\x8a\x00\x85\x00\x8a\x00\x86\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xc5\xcc\x00\x00\xb9\xdb\n\x00\xf9\xca')
2022-06-24 01:18:59,968:DEBUG:jkabstractprotocol:is_record_correct_type@162: Record is type 2
2022-06-24 01:18:59,969:DEBUG:jkabstractprotocol:is_record_start@147: SOR found in record
2022-06-24 01:19:00,002:DEBUG:jkbledelegate:handleNotification@27: From handle: 0x12 Got 22 bytes of data
2022-06-24 01:19:00,002:DEBUG:jkbledelegate:handleNotification@29: Pre wipe to start bytearray(b'U\xaa\xeb\x90\x02c\xcc\x0c\xce\x0c\xce\x0c\xcb\x0c\xcd\x0c\xcb\x0c\xcc\x0c\xcc\x0c\xcc\x0c\xcb\x0c\xcd\x0c\xcc\x0c\xcb\x0c\xcd\x0c\xcd\x0c\xcd\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\xff\x00\x00\xcc\x0c\x03\x00\x00\x03\x8c\x00\x86\x00\x8a\x00\x86\x00\x8b\x00\x85\x00\x8b\x00\x85\x00\x8a\x00\x85\x00\x89\x00\x86\x00\x8a\x00\x85\x00\x8a\x00\x86\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xc5\xcc\x00\x00\xb9\xdb\n\x00\xf9\xca\xff\xff\x12\x01\x04\x01;\x01\x00\x00\x00\x00\x006\xaeU\x02\x00\xc0E\x04\x00')
2022-06-24 01:19:00,002:DEBUG:jkbledelegate:handleNotification@31: Post wipe to start bytearray(b'U\xaa\xeb\x90\x02c\xcc\x0c\xce\x0c\xce\x0c\xcb\x0c\xcd\x0c\xcb\x0c\xcc\x0c\xcc\x0c\xcc\x0c\xcb\x0c\xcd\x0c\xcc\x0c\xcb\x0c\xcd\x0c\xcd\x0c\xcd\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\xff\x00\x00\xcc\x0c\x03\x00\x00\x03\x8c\x00\x86\x00\x8a\x00\x86\x00\x8b\x00\x85\x00\x8b\x00\x85\x00\x8a\x00\x85\x00\x89\x00\x86\x00\x8a\x00\x85\x00\x8a\x00\x86\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xc5\xcc\x00\x00\xb9\xdb\n\x00\xf9\xca\xff\xff\x12\x01\x04\x01;\x01\x00\x00\x00\x00\x006\xaeU\x02\x00\xc0E\x04\x00')
2022-06-24 01:19:00,003:DEBUG:jkabstractprotocol:is_record_correct_type@162: Record is type 2
2022-06-24 01:19:00,003:DEBUG:jkabstractprotocol:is_record_start@147: SOR found in record
2022-06-24 01:19:00,152:DEBUG:jkbledelegate:handleNotification@27: From handle: 0x12 Got 128 bytes of data
2022-06-24 01:19:00,153:DEBUG:jkbledelegate:handleNotification@29: Pre wipe to start bytearray(b'U\xaa\xeb\x90\x02c\xcc\x0c\xce\x0c\xce\x0c\xcb\x0c\xcd\x0c\xcb\x0c\xcc\x0c\xcc\x0c\xcc\x0c\xcb\x0c\xcd\x0c\xcc\x0c\xcb\x0c\xcd\x0c\xcd\x0c\xcd\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\xff\x00\x00\xcc\x0c\x03\x00\x00\x03\x8c\x00\x86\x00\x8a\x00\x86\x00\x8b\x00\x85\x00\x8b\x00\x85\x00\x8a\x00\x85\x00\x89\x00\x86\x00\x8a\x00\x85\x00\x8a\x00\x86\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xc5\xcc\x00\x00\xb9\xdb\n\x00\xf9\xca\xff\xff\x12\x01\x04\x01;\x01\x00\x00\x00\x00\x006\xaeU\x02\x00\xc0E\x04\x00\x00\x00\x00\x00q_\x03\x00d\x00\x05\t\xb9\xb7\x01\x00\x01\x01\xd9\x06\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x07\x00\x01\x00\x00\x00\xc9\x02\x00\x00w\x00\x84\xf2>@\x00\x00\x00\x00\xe2\x04\x00\x00\x00\x00\x00\x01\x00\x05\x00\x00=-\x11\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')

2022-06-24 01:19:01,608:DEBUG:jkabstractprotocol:get_command_defn@106: get_command_defn for: getCellData
2022-06-24 01:19:01,608:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'getCellData'
2022-06-24 01:19:01,609:DEBUG:abstractprotocol:get_command_defn@62: Found command getCellData in protocol b'JK04'
2022-06-24 01:19:01,609:INFO:abstractprotocol:decode@236: Processing response of type POSITIONAL
2022-06-24 01:19:01,609:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['Hex2Str', 4, 'Header', '']
2022-06-24 01:19:01,610:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['Hex2Str', 1, 'Record Type', '']
2022-06-24 01:19:01,610:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['Hex2Int', 1, 'Record Counter', '']
2022-06-24 01:19:01,610:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['LittleHex2Float', 4, 'Voltage Cell01', 'V']
2022-06-24 01:19:01,611:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['LittleHex2Float', 4, 'Voltage Cell02', 'V']
2022-06-24 01:19:01,611:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['LittleHex2Float', 4, 'Voltage Cell03', 'V']
2022-06-24 01:19:01,611:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['LittleHex2Float', 4, 'Voltage Cell04', 'V']
2022-06-24 01:19:01,611:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['LittleHex2Float', 4, 'Voltage Cell05', 'V']
2022-06-24 01:19:01,612:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['LittleHex2Float', 4, 'Voltage Cell06', 'V']
2022-06-24 01:19:01,612:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['LittleHex2Float', 4, 'Voltage Cell07', 'V']
2022-06-24 01:19:01,612:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['LittleHex2Float', 4, 'Voltage Cell08', 'V']
2022-06-24 01:19:01,613:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['LittleHex2Float', 4, 'Voltage Cell09', 'V']
2022-06-24 01:19:01,613:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['LittleHex2Float', 4, 'Voltage Cell10', 'V']
2022-06-24 01:19:01,613:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['LittleHex2Float', 4, 'Voltage Cell11', 'V']
2022-06-24 01:19:01,613:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['LittleHex2Float', 4, 'Voltage Cell12', 'V']
2022-06-24 01:19:01,614:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['LittleHex2Float', 4, 'Voltage Cell13', 'V']
2022-06-24 01:19:01,614:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['LittleHex2Float', 4, 'Voltage Cell14', 'V']
2022-06-24 01:19:01,614:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['LittleHex2Float', 4, 'Voltage Cell15', 'V']
2022-06-24 01:19:01,615:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['LittleHex2Float', 4, 'Voltage Cell16', 'V']
2022-06-24 01:19:01,615:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['LittleHex2Float', 4, 'Voltage Cell17', 'V']
2022-06-24 01:19:01,615:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['LittleHex2Float', 4, 'Voltage Cell18', 'V']
2022-06-24 01:19:01,615:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['LittleHex2Float', 4, 'Voltage Cell19', 'V']
2022-06-24 01:19:01,616:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['LittleHex2Float', 4, 'Voltage Cell20', 'V']
2022-06-24 01:19:01,616:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['LittleHex2Float', 4, 'Voltage Cell21', 'V']
2022-06-24 01:19:01,616:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['LittleHex2Float', 4, 'Voltage Cell22', 'V']
2022-06-24 01:19:01,617:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['LittleHex2Float', 4, 'Voltage Cell23', 'V']
2022-06-24 01:19:01,617:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['LittleHex2Float', 4, 'Voltage Cell24', 'V']
2022-06-24 01:19:01,617:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['LittleHex2Float', 4, 'Resistance Cell01', 'Ohm']
2022-06-24 01:19:01,617:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['LittleHex2Float', 4, 'Resistance Cell02', 'Ohm']
2022-06-24 01:19:01,618:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['LittleHex2Float', 4, 'Resistance Cell03', 'Ohm']
[... cut ...]
delta_cell_voltage              2.350988701644575e-38   V   
highest_cell                    1                   
lowest_cell                     1                   
flags                           0000                
uptime                          0D0H0M0S            
checksum                        bc                  
highest_cell_voltage            3.1592955277925957e-31  V   
lowest_cell_voltage             3.1592955277925957e-31  V   
2022-06-24 01:19:01,713:DEBUG:__init__:main@438: Not daemon, so not looping

syssi, 2022-06-24:
It looks like the JK04 protocol was used to parse the response. If you have a recent BMS mode the correct protocol should be JK02. Please try to issue the same command and select the JK02 protocol using the protocol option parameter.

paolss, 2022-06-24:
Yes thank You - didnt checked that JK04 is default one.
Also is there possible to set some timeout waiting for data so that when jkbms (hangs) it can be killed after some timout waiting data from BMS ? 
Because many times there is problem with reading data from BMS over BLE - jkbms apps just waiting for data like 10 , 20 or 30 minutes...

jblance, 2022-06-26:
i thought it did time out. the jkbms code needs work, i'll try to find the error and fix it

closed 2024-05-07 by jblance