jblance/mpp-solar #233: USB open error [errno 2]

meccip, 2022-07-12:
Hello,
I am using mppsolar with Easun SV4 connected with USB cable to Raspberry 3B+. Protocol P17 and P18. I send data to nodered with json_mqtt output.
After some working time (random time) connection with mppsolar stop working. **If i unplug /plug usb cable connection is good again**.

**I use config file** 
[GSdata]
protocol=PI17
port=/dev/hidraw0
baud=2400
command=GS
outputs=json_mqtt

[workmode]
protocol=PI18
port=/dev/hidraw0
baund=2400
command=MOD
outputs=json_mqtt

**my error in command line:**
pi@raspberrypi:~ $ sudo mpp-solar -p /dev/hidraw0 -P PI17 -c GS -I
2022-07-12 19:51:18,502:INFO:__init__:main@223: Solar Device Command Utility, version: 0.13.0, add postgres support from skoky
2022-07-12 19:51:18,503:INFO:__init__:main@347: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI17"
2022-07-12 19:51:18,514:INFO:__init__:get_port@89: Using hidrawio for communications
2022-07-12 19:51:18,529:INFO:__init__:main@404: Looping 1 commands
2022-07-12 19:51:18,529:INFO:__init__:main@415: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.hidrawio.HIDRawIO object at 0x75e44eb0>, protocol: <mppsolar.protocols.pi17.pi17 object at 0x75e5d8f0> for command: GS, tag: GS, outputs: screen
2022-07-12 19:51:18,530:INFO:device:run_command@274: Running command GS
2022-07-12 19:51:18,531:INFO:pi17:get_full_command@914: Using protocol b'PI17' with 47 commands
2022-07-12 19:51:18,531:INFO:device:run_command@303: full command b'^P003GS\r' for command GS
2022-07-12 19:51:18,532:INFO:__init__:get_output@35: attempting to create output processor: screen
2022-07-12 19:51:18,536:INFO:screen:output@19: Using output processor: screen
Parameter                     	Value           Unit
error                         	USB open error: [Errno 2] No such file or directory: '/dev/hidraw0'

jblance, 2022-07-12:
> 2022-07-12 ***@***.***: Solar Device Command
> 2022-07-12 ***@***.***: Creating device "unnamed"
> 2022-07-12 ***@***.***: Using hidrawio for
> 2022-07-12 ***@***.***: Looping 1 commands
> 2022-07-12 ***@***.***: Getting results from
> 2022-07-12 ***@***.***: Running command GS
> 2022-07-12 ***@***.***: Using protocol
> 2022-07-12 ***@***.***: full command
> 2022-07-12 ***@***.***: attempting to create

it looks like the usb device is 'disappearing' are there any system
messages at the time (dmesg or /var/log/messages)

jblance, 2022-07-12:
> 000000000 crc f0
> [186829.209764] w1_master_driver w1_bus_master1: Family 0 for
> 00.407000000000.f0 is not registered.
> [186892.722687] w1_master_driver w1_bus_master1: Attaching one wire slave
> 00.c07000000000 crc 7c
> [186892.745011] w1_master_driver w1_bus_master1: Family 0 for
> 00.c07000000000.7c is not registered.
> [186968.804448] w1_master_driver w1_bus_master1: Attaching one wire slave
> 00.207000000000 crc 95
> [186968.814198] w1_master_driver w1_bus_master1: Family 0 for
> 00.207000000000.95 is not registered.
> [187019.544851] w1_master_driver w1_bus_master1: Attaching one wire slave
> 00.a07000000000 crc 19
> ..........
> ..........
> [199561.745330] w1_master_driver w1_bus_master1: Family 0 for
> 00.6f7000000000.92 is not registered.
> [199625.216694] w1_master_driver w1_bus_master1: Attaching one wire slave
> 00.ef7000000000 crc 1e
> [199625.225478] w1_master_driver w1_bus_master1: Family 0 for
> 00.ef7000000000.1e is not registered.
> [199665.006682] w1_master_driver w1_bus_master1: Attaching one wire slave
> 00.1f7000000000 crc 6a
> [199665.017914] w1_master_driver w1_bus_master1: Family 0 for
> 00.1f7000000000.6a is not registered.
> [199717.126678] w1_master_driver w1_bus_master1: Attaching one wire slave
> 00.9f7000000000 crc e6
> [199717.140723] w1_master_driver w1_bus_master1: Family 0 for
> 00.9f7000000000.e6 is not registered.
> [199766.286352] w1_master_driver w1_bus_master1: Attaching one wire slave
> 00.5f7000000000 crc 2c
> [199766.295070] w1_master_driver w1_bus_master1: Family 0 for
> 00.5f7000000000.2c is not registered.
> [199805.626455] w1_master_driver w1_bus_master1: Attaching one wire slave
> 00.df7000000000 crc a0
> [199805.636269] w1_master_driver w1_bus_master1: Family 0 for
> 00.df7000000000.a0 is not registered.
>
> this is a part of what is listed after enter dmesg in command line. but
> mppsolar is working at this time

see if you can get a dmesg the next time it fails

meccip, 2022-07-12:
ok

meccip, 2022-07-13:
[237683.456407] w1_master_driver w1_bus_master1: Attaching one wire slave 00.8b8800000000 crc f5
[237683.465157] w1_master_driver w1_bus_master1: Family 0 for 00.8b8800000000.f5 is not registered.
[237748.146376] w1_master_driver w1_bus_master1: Attaching one wire slave 00.4b8800000000 crc 3f
[237748.156367] w1_master_driver w1_bus_master1: Family 0 for 00.4b8800000000.3f is not registered.
[237787.456800] w1_master_driver w1_bus_master1: Attaching one wire slave 00.cb8800000000 crc b3
[237787.465879] w1_master_driver w1_bus_master1: Family 0 for 00.cb8800000000.b3 is not registered.
[237800.723708] ch341-uart ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[237800.723762] ch341-uart ttyUSB1: usb_serial_generic_read_bulk_callback - urb stopped: -32
[237800.723805] ch341-uart ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[237800.723819] ch341-uart ttyUSB1: usb_serial_generic_read_bulk_callback - urb stopped: -32
[237800.746828] usb 1-1.5.1-port2: cannot reset (err = -32)
[237800.746932] usb 1-1.5.1-port2: cannot reset (err = -32)
[237800.747057] usb 1-1.5.1-port2: cannot reset (err = -32)
[237800.747179] usb 1-1.5.1-port2: cannot reset (err = -32)
[237800.747306] usb 1-1.5.1-port2: cannot reset (err = -32)
[237800.747314] usb 1-1.5.1-port2: Cannot enable. Maybe the USB cable is bad?
[237800.747435] usb 1-1.5.1-port2: cannot disable (err = -32)
[237800.747575] usb 1-1.5.1-port2: cannot reset (err = -32)
[237800.747679] usb 1-1.5.1-port2: cannot reset (err = -32)
[237800.747803] usb 1-1.5.1-port2: cannot reset (err = -32)
[237800.747930] usb 1-1.5.1-port2: cannot reset (err = -32)
[237800.748056] usb 1-1.5.1-port2: cannot reset (err = -32)
[237800.748063] usb 1-1.5.1-port2: Cannot enable. Maybe the USB cable is bad?
[237800.748178] usb 1-1.5.1-port2: cannot disable (err = -32)
[237800.748305] usb 1-1.5.1-port2: cannot reset (err = -32)
[237800.748428] usb 1-1.5.1-port2: cannot reset (err = -32)
[237800.748554] usb 1-1.5.1-port2: cannot reset (err = -32)
[237800.748682] usb 1-1.5.1-port2: cannot reset (err = -32)
[237800.748804] usb 1-1.5.1-port2: cannot reset (err = -32)
[237800.748812] usb 1-1.5.1-port2: Cannot enable. Maybe the USB cable is bad?
[237800.748930] usb 1-1.5.1-port2: cannot disable (err = -32)
[237800.749056] usb 1-1.5.1-port2: cannot reset (err = -32)
[237800.749179] usb 1-1.5.1-port2: cannot reset (err = -32)
[237800.749305] usb 1-1.5.1-port2: cannot reset (err = -32)
[237800.749434] usb 1-1.5.1-port2: cannot reset (err = -32)
[237800.749576] usb 1-1.5.1-port2: cannot reset (err = -32)
[237800.749583] usb 1-1.5.1-port2: Cannot enable. Maybe the USB cable is bad?
[237800.749682] usb 1-1.5.1-port2: cannot disable (err = -32)
[237800.749804] usb 1-1.5.1-port2: cannot disable (err = -32)
[237800.750431] hub 1-1.5.1:1.0: hub_ext_port_status failed (err = -32)
[237800.840817] usb 1-1.5: USB disconnect, device number 62
[237800.840837] usb 1-1.5.1: USB disconnect, device number 63
[237800.840849] usb 1-1.5.1.2: USB disconnect, device number 67
[237800.841599] usb 1-1.5.1.3: ch341_read_int_callback - usb_submit_urb failed: -19
[237800.841619] usb 1-1.5.2: ch341_read_int_callback - usb_submit_urb failed: -19
[237800.842962] usb 1-1.5.1.3: USB disconnect, device number 68
[237800.843271] usb 1-1.5.1.3: failed to send control message: -19
[237800.847555] ch341-uart ttyUSB1: ch341-uart converter now disconnected from ttyUSB1
[237800.847669] ch341 1-1.5.1.3:1.0: device disconnected
[237800.852421] usb 1-1.5.2: USB disconnect, device number 64
[237800.852835] usb 1-1.5.2: failed to send control message: -19
[237800.853907] ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
[237800.854004] ch341 1-1.5.2:1.0: device disconnected
[237801.151580] usb 1-1.5: new full-speed USB device number 69 using dwc_otg
[237801.285355] usb 1-1.5: New USB device found, idVendor=0a05, idProduct=7211, bcdDevice= 1.00
[237801.285374] usb 1-1.5: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[237801.285383] usb 1-1.5: Product: USB2.0 HUB
[237801.286391] hub 1-1.5:1.0: USB hub found
[237801.286958] hub 1-1.5:1.0: 4 ports detected
[237801.881708] usb 1-1.5.1: new full-speed USB device number 70 using dwc_otg
[237802.016028] usb 1-1.5.1: New USB device found, idVendor=0a05, idProduct=7211, bcdDevice= 1.00
[237802.016051] usb 1-1.5.1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[237802.016061] usb 1-1.5.1: Product: USB2.0 HUB
[237802.017204] hub 1-1.5.1:1.0: USB hub found
[237802.017864] hub 1-1.5.1:1.0: 4 ports detected
[237802.121722] usb 1-1.5.2: new full-speed USB device number 71 using dwc_otg
[237802.264842] usb 1-1.5.2: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.54
[237802.264860] usb 1-1.5.2: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[237802.264869] usb 1-1.5.2: Product: USB2.0-Ser!
[237802.265734] ch341 1-1.5.2:1.0: ch341-uart converter detected
[... 1 lines cut ...]
[237802.267699] usb 1-1.5.2: ch341-uart converter now attached to ttyUSB0
[237802.571622] usb 1-1.5.1.3: new full-speed USB device number 72 using dwc_otg
[237802.704967] usb 1-1.5.1.3: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.54
[237802.704986] usb 1-1.5.1.3: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[237802.704994] usb 1-1.5.1.3: Product: USB2.0-Ser!
[237802.706013] ch341 1-1.5.1.3:1.0: ch341-uart converter detected
[... cut ...]
[237973.547929] usb 1-1.5.1.2: Manufacturer: STMicroelectronics
[237973.557618] hid-generic 0003:0665:5161.000F: hiddev96,hidraw0: USB HID v1.11 Device [STMicroelectronics HID in FS Mode] on usb-3f980000.usb-1.5.1.2/input0
[238018.976263] w1_master_driver w1_bus_master1: Attaching one wire slave 00.1b8800000000 crc e4
[238018.984487] w1_master_driver w1_bus_master1: Family 0 for 00.1b8800000000.e4 is not registered.
pi@raspberrypi:~ $ 

**this time connection fails and then start again after few minutes**

meccip, 2022-07-13:
easun and 2 pzem AC meter are connected to a HUB. Then HUB connected to raspberry. From the error message i think the HUB is disconnected from system  (ttyUSB0, USB1 - PZEM RS485 connections)

meccip, 2022-08-12:
I use this with node red. So, I made a function if nothing is received from inverter or an error is received, after 30 sec. will send a usb reset request for /dev/hidraw0 raspberry port. I made a counter to see how many times is reseting (7-8 times per day).

So, I believe the case is closed.

closed 2023-01-08 by jblance