jblance/mpp-solar #238: Connection Issues

realSater, 2022-07-23:
Updating mpp to the latest version using pip3 (from mpp-solar to mppsolar) makes the connection super unstable.

here is an example of the issues i am facing with the new version:
```
solar@solar:~ $ sudo mpp-solar -c QMUCHGCR -P PI30 -p /dev/hidraw0
Parameter                       Value           Unit
error                           Response to short
solar@solar:~ $ sudo mpp-solar -c QMUCHGCR -P PI30 -p /dev/hidraw0
Parameter                       Value           Unit
error                           NAK
solar@solar:~ $ sudo mpp-solar -c QMUCHGCR -P PI30 -p /dev/hidraw0
Parameter                       Value           Unit
error                           Response to short
solar@solar:~ $ sudo mpp-solar -c QMUCHGCR -P PI30 -p /dev/hidraw0
Parameter                       Value           Unit
error                           NAK
solar@solar:~ $ sudo mpp-solar -c QMUCHGCR -P PI30 -p /dev/hidraw0
Traceback (most recent call last):
  File "/usr/local/bin/mpp-solar", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/__init__.py", line 408, in main
    results = _device.run_command(command=_command)
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/devices/device.py", line 316, in run_command
    raw_response = self._port.send_and_receive(
  File "/usr/local/lib/python3.9/dist-packages/mppsolar/inout/hidrawio.py", line 52, in send_and_receive
    os.write(usb0, send)
BrokenPipeError: [Errno 32] Broken pipe
solar@solar:~ $ sudo mpp-solar -c QMUCHGCR -P PI30 -p /dev/hidraw0
Parameter                       Value           Unit
error                           Response to short
solar@solar:~ $ sudo mpp-solar -c QMUCHGCR -P PI30 -p /dev/hidraw0
Parameter                       Value           Unit
error                           Response to short
solar@solar:~ $ sudo mpp-solar -c QMUCHGCR -P PI30 -p /dev/hidraw0
Parameter                       Value           Unit
error                           NAK
solar@solar:~ $ sudo mpp-solar -c QMUCHGCR -P PI30 -p /dev/hidraw0
Command: QMUCHGCR - Max Utility Charging Current Options inquiry
------------------------------------------------------------
Parameter                       Value           Unit
max_utility_charging_current    010             A
unknown_value_in_response_1     020
```

I am simply running the command over and over till i received actual data at the end as you can see above.

is there anyway I can modify mppsolar to make a delay when communicating through serialio? "assuming this is the case".

Please let me know what do you think of this weird behaviour.

jblance, 2022-07-23:
Can you do a debug run and post the results.

Do you remember what version you were using before?

realSater, 2022-07-23:
I do not remember the version but i installed "mpp-solar" using pip3 2 days ago, and just today i switched to "mppsolar" after reading somewhere in the "issues tab" that this is the version with the latest code updates.

here's what the debug says:
```
solar@solar:~ $ sudo mpp-solar -c QMUCHGCR -P PI30 -p /dev/hidraw0 -D
2022-07-23 13:23:05,558:INFO:__init__:main@218: Solar Device Command Utility, version: 0.12.5, tweak jkbleio for getInfo and to return first 300 bytes
2022-07-23 13:23:05,559:DEBUG:__init__:main@232: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7fa7363d00>, _isConnected=False)
2022-07-23 13:23:05,560:DEBUG:__init__:main@234: udp port 5555
2022-07-23 13:23:05,560:DEBUG:__init__:main@237: Using Mongo None with mppsolar
2022-07-23 13:23:05,561:INFO:__init__:main@337: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI30"
2022-07-23 13:23:05,568:DEBUG:__init__:main@341: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2022-07-23 13:23:05,569:DEBUG:device:__init__@33: __init__ args ()
2022-07-23 13:23:05,569:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': '/dev/hidraw0', 'protocol': 'PI30', 'baud': 2400, 'porttype': None, 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7fa7363d00>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2022-07-23 13:23:05,570:DEBUG:__init__:get_port_type@38: port matches hidraw
2022-07-23 13:23:05,570:INFO:__init__:get_port@89: Using hidrawio for communications
2022-07-23 13:23:05,573:DEBUG:__init__:get_protocol@12: Protocol PI30
2022-07-23 13:23:05,579:DEBUG:device:__init__@38: __init__ name unnamed, port <mppsolar.inout.hidrawio.HIDRawIO object at 0x7fa6a8ec10>, protocol <mppsolar.protocols.pi30.pi30 object at 0x7fa6a8ec40>
2022-07-23 13:23:05,579:DEBUG:__init__:main@390: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x7fa6a8eaf0>, 'QMUCHGCR', 'QMUCHGCR', 'screen', None, None)]
2022-07-23 13:23:05,580:INFO:__init__:main@395: Looping 1 commands
2022-07-23 13:23:05,580:INFO:__init__:main@405: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.hidrawio.HIDRawIO object at 0x7fa6a8ec10>, protocol: <mppsolar.protocols.pi30.pi30 object at 0x7fa6a8ec40> for command: QMUCHGCR, tag: QMUCHGCR, outputs: screen
2022-07-23 13:23:05,581:INFO:device:run_command@274: Running command QMUCHGCR
2022-07-23 13:23:05,581:INFO:abstractprotocol:get_full_command@50: Using protocol b'PI30' with 40 commands
2022-07-23 13:23:05,581:DEBUG:protocol_helpers:crcPI@313: Calculating CRC for b'QMUCHGCR'
2022-07-23 13:23:05,582:DEBUG:protocol_helpers:crcPI@363: Generated CRC 0x26 0x34 0x2634
2022-07-23 13:23:05,582:DEBUG:abstractprotocol:get_full_command@56: full command: b'QMUCHGCR&4\r'
2022-07-23 13:23:05,582:INFO:device:run_command@303: full command b'QMUCHGCR&4\r' for command QMUCHGCR
2022-07-23 13:23:05,583:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'QMUCHGCR'
2022-07-23 13:23:05,583:DEBUG:abstractprotocol:get_command_defn@62: Found command QMUCHGCR in protocol b'PI30'
2022-07-23 13:23:05,640:DEBUG:hidrawio:send_and_receive@29: length of to_send: 11
2022-07-23 13:23:05,641:DEBUG:hidrawio:send_and_receive@47: multiple chunk send
2022-07-23 13:23:05,641:DEBUG:hidrawio:send_and_receive@50: send: b'QMUCHGCR', to_send: b'&4\r'
2022-07-23 13:23:05,995:DEBUG:hidrawio:send_and_receive@47: multiple chunk send
2022-07-23 13:23:05,995:DEBUG:hidrawio:send_and_receive@50: send: b'&4\r', to_send: b''
2022-07-23 13:23:06,750:DEBUG:hidrawio:send_and_receive@69: usb response was: b'(NAKss\r'
2022-07-23 13:23:06,751:DEBUG:device:run_command@322: Send and Receive Response b'(NAKss\r'
2022-07-23 13:23:06,751:INFO:abstractprotocol:decode@196: response passed to decode: b'(NAKss\r'
2022-07-23 13:23:06,751:INFO:abstractprotocol:decode@200: NAK
2022-07-23 13:23:06,751:INFO:device:run_command@341: Decoded response {'ERROR': ['NAK', '']}
2022-07-23 13:23:06,752:DEBUG:__init__:main@409: results: {'ERROR': ['NAK', '']}
2022-07-23 13:23:06,752:INFO:__init__:get_output@34: attempting to create output processor: screen
2022-07-23 13:23:06,755:DEBUG:screen:__init__@16: processor.screen __init__ kwargs {}
2022-07-23 13:23:06,756:DEBUG:__init__:main@415: Using output filter: None
2022-07-23 13:23:06,756:INFO:screen:output@19: Using output processor: screen
2022-07-23 13:23:06,756:DEBUG:screen:output@20: kwargs {'data': {'ERROR': ['NAK', '']}, 'tag': 'QMUCHGCR', 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7fa7363d00>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar', 'filter': None, 'excl_filter': None, 'keep_case': False}
Parameter                       Value           Unit
error                           NAK
2022-07-23 13:23:06,757:DEBUG:__init__:main@438: Not daemon, so not looping
```

realSater, 2022-07-23:
my simple conf:

```
[SETUP]
pause=5
mqtt_broker=localhost

[PI30]
protocol=PI30
port=/dev/hidraw0
baud=2400
command=QFLAG,QMCHGCR,QMOD,QMUCHGCR,QPIGS,QPIRI
outputs=json_mqtt
tag=PI30
```

realSater, 2022-07-23:
Okay I figured out the problem. Although I am not 100% convinced, but it solved the issue...

in the mpp-solar.conf file, I was using the configuration shown above?

I simply changed it to the below:

```
[SETUP]
pause=5
mqtt_broker=127.0.0.1
mqtt_port=1883

[PI30]
protocol=PI30
port=/dev/hidraw0
baud=2400
command=QFLAG#QMCHGCR#QMOD#QMUCHGCR#QPIGS#QPIRI
outputs=json_mqtt
```

and now it works flawlessly.

One note that the "127.0.0.1" value change for mqtt has nothing to do with the fix, I just wanted to convert all the connections from "localhost" into 127.0.0.1 to be certain that nothing is interfering on the localhost address.

Thank you!

closed 2022-07-23 by realSater

jblance, 2022-10-11:
That error was different to the others. Can you try again to get the
"response too short" error in debug mode

Do you have the service running as well?