jblance/mpp-solar #239: Add PBAT0 1 2 commands

ouinouin, 2022-07-23:
Hi , owning actually a superwatt inverter, having a 1V offset, i m on my way to correct that, 
unfortunately, the commands PBAT 0 1 2 to correct those issues are not implemented in your command :+1: 
here are some interresting discussions on the subject :
[forums aeva](https://forums.aeva.asn.au/viewtopic.php?p=71631)

attached two documents found on different forums explainin the commands, basically doing a two points curve calibration.
(mileage may vary on which inverters this will work as usual with chinese ecosystems...)
[Uploading product-application_pdf-20190724092401-8519-pip-gk-mk-protocol.pdf…]()

jblance, 2023-02-01:
sorry cant seem to access document and failed to find info on this command

ouinouin, 2023-04-07:
hi , late answer (missed notification )

i basically follewed the instructions given in the forum lin k : 

f your inverter reads too HIGH:

by .05 V: BTA-01      42 54 41 2D 30 31 82 3C 0D
by 0.1 V: BTA-02      42 54 41 2D 30 32 B2 5F 0D
by 0.2 V: BTA-04      42 54 41 2D 30 34 D2 99 0D
by 0.3 V: BTA-06      42 54 41 2D 30 36 F2 DB 0D
by 0.4 V: BTA-08      42 54 41 2D 30 38 13 15 0D
by .45 V: BTA-09      42 54 41 2D 30 39 03 34 0D

If your inverter reads too LOW:

by .05 V: BTA+01      42 54 41 2B 30 31 30 9C 0D
by 0.1 V: BTA+02      42 54 41 2B 30 32 00 FF 0D
by 0.2 V: BTA+04      42 54 41 2B 30 34 60 39 0D
by 0.3 V: BTA+06      42 54 41 2B 30 36 40 7B 0D
by 0.4 V: BTA+08      42 54 41 2B 30 38 A1 B5 0D
by .45 V: BTA+09      42 54 41 2B 30 39 B1 94 0D

another quote from this forum : 
 There are two ways I can see to save the settings: use the undocumented PSAVE command (no parameters), or a command that changes a variable in EEPROM set 1. POPV (Put OutPut Voltage) is probably the easiest of these, and can be sent from Watchpower or other monitoring programs. As with any EEPROM change, you have to actually make a difference for it to be actually saved to EEPROM, otherwise the firmware avoids EEPROM wear by not changing a value that doesn't seem to need changing. So you could change the output voltage to 220 V, then back to 230 V, for example. Note that POPV is a command that has to be done when the inverter switch is OFF. The inverter won't switch off completely if you have a charging source (PV or AC-in), and even if neither of these are present, it takes some 30 seconds (from memory) before it really switches off. Be generous with your timing; you don't want to be writing to EEPROM just as the power goes down. It could corrupt all the settings, including calibration values that you aren't even aware of.

i followed their explaination on the link given on the first post , works great on my sperwatt inverter.

ouinouin, 2023-04-07:
edit : the missing doc..... from the first post  [product-application_pdf-20190724092401-8519-pip-gk-mk-protocol.pdf](https://github.com/jblance/mpp-solar/files/11180246/product-application_pdf-20190724092401-8519-pip-gk-mk-protocol.pdf)

jblance, 2023-06-07:
just reviewing this again, the BTA commands are only in the index and are missing from the body of the document?

jblance, 2023-06-07:
added bta and psave to latest dev code

closed 2023-06-07 by jblance