jblance/mpp-solar #240: Silly Question

realSater, 2022-07-24:
I am able to connect to the PI30 inverter without any issues, I am able to fetch the data normally, everything is working just fine using mqtt --> telegraf --> influx --> grafana.

I have a non mpp-solar question, if that's fine to ask.

The inverter is showing the output load in watts on the inverter screen (ex: 432w) - this value changes dynamically based on how much load I am having regarding house equipment.

This value does not show in any of the values that mpp-solar is retrieving from the inverter.

List of commands that I am fetching are: `QFLAG#QMCHGCR#QMOD#QMUCHGCR#QPIGS#QPIRI`

Whether I fetch every command alone or not, I don't see the output load in watts.

the only thing I can relate to, is a parameter coming from the inverter which shows "ac_output_load" and it comes in numbers that doesn't exceed 20.

so after doing some research i figured out that this number is in %. so it shows 3% or 16% load etc..
is there any way that I can convert the percentage into watt?

I tried to google this before i post here, but I have a missing unit value which is the output ampere which is also missing from the data that mpp-solar is fetching, but it shows fine on the inverter screen.

I really appreciate anyone's help on this.

jblance, 2022-07-24:
Have you tried QPGS0?

realSater, 2022-07-24:
```
2022-07-24 09:27:47,480:INFO:__init__:main@223: Solar Device Command Utility, version: 0.13.0, add postgres support from skoky
2022-07-24 09:27:47,481:DEBUG:__init__:main@237: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f84b412e0>, _isConnected=False)
2022-07-24 09:27:47,482:DEBUG:__init__:main@239: udp port 5555
2022-07-24 09:27:47,483:DEBUG:__init__:main@241: Using Postgres None
2022-07-24 09:27:47,483:DEBUG:__init__:main@244: Using Mongo None with mppsolar
2022-07-24 09:27:47,484:INFO:__init__:main@346: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI30"
2022-07-24 09:27:47,492:DEBUG:__init__:main@350: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
2022-07-24 09:27:47,493:DEBUG:device:__init__@33: __init__ args ()
2022-07-24 09:27:47,493:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': '/dev/hidraw0', 'protocol': 'PI30', 'baud': 2400, 'porttype': None, 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f84b412e0>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2022-07-24 09:27:47,494:DEBUG:__init__:get_port_type@38: port matches hidraw
2022-07-24 09:27:47,494:INFO:__init__:get_port@89: Using hidrawio for communications
2022-07-24 09:27:47,498:DEBUG:__init__:get_protocol@12: Protocol PI30
2022-07-24 09:27:47,504:DEBUG:device:__init__@38: __init__ name unnamed, port <mppsolar.inout.hidrawio.HIDRawIO object at 0x7f842900a0>, protocol <mppsolar.protocols.pi30.pi30 object at 0x7f84284c70>
2022-07-24 09:27:47,505:DEBUG:__init__:main@399: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x7f84290040>, 'QPGS0', 'QPGS0', 'screen', None, None)]
2022-07-24 09:27:47,505:INFO:__init__:main@404: Looping 1 commands
2022-07-24 09:27:47,506:INFO:__init__:main@414: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.inout.hidrawio.HIDRawIO object at 0x7f842900a0>, protocol: <mppsolar.protocols.pi30.pi30 object at 0x7f84284c70> for command: QPGS0, tag: QPGS0, outputs: screen
2022-07-24 09:27:47,506:INFO:device:run_command@274: Running command QPGS0
2022-07-24 09:27:47,506:INFO:abstractprotocol:get_full_command@50: Using protocol b'PI30' with 40 commands
2022-07-24 09:27:47,506:DEBUG:protocol_helpers:crcPI@313: Calculating CRC for b'QPGS0'
2022-07-24 09:27:47,507:DEBUG:protocol_helpers:crcPI@363: Generated CRC 0x3f 0xda 0x3fda
2022-07-24 09:27:47,507:DEBUG:abstractprotocol:get_full_command@56: full command: b'QPGS0?\xda\r'
2022-07-24 09:27:47,507:INFO:device:run_command@303: full command b'QPGS0?\xda\r' for command QPGS0
2022-07-24 09:27:47,508:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'QPGS0'
2022-07-24 09:27:47,508:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: F
2022-07-24 09:27:47,509:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: MCHGC
2022-07-24 09:27:47,510:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: MNCHGC
2022-07-24 09:27:47,511:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: MUCHGC
2022-07-24 09:27:47,512:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PBCV
2022-07-24 09:27:47,512:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PBDV
2022-07-24 09:27:47,513:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PBFT
2022-07-24 09:27:47,514:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PBT
2022-07-24 09:27:47,515:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PCP
2022-07-24 09:27:47,516:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PCVV
2022-07-24 09:27:47,517:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PE
2022-07-24 09:27:47,518:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PD
2022-07-24 09:27:47,519:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PGR
2022-07-24 09:27:47,519:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: POP
2022-07-24 09:27:47,520:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: POPLG
2022-07-24 09:27:47,521:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: POPM
2022-07-24 09:27:47,522:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PPCP
2022-07-24 09:27:47,523:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PPVOKC
2022-07-24 09:27:47,524:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PSDV
2022-07-24 09:27:47,525:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PSPB
2022-07-24 09:27:47,526:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PBATCD
2022-07-24 09:27:47,527:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: DAT
2022-07-24 09:27:47,528:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: PBATMAXDISC
2022-07-24 09:27:47,529:DEBUG:abstractprotocol:get_command_defn@66: Regex commands _command: QPGS
2022-07-24 09:27:47,530:DEBUG:abstractprotocol:get_command_defn@70: Matched: QPGS0 to: QPGS value: 0
2022-07-24 09:27:47,531:DEBUG:hidrawio:send_and_receive@29: length of to_send: 8
[... cut ...] qtt.client.Client object at 0x7f84b412e0>, _isConnected=False), 'udp_port': '5555', 'postgres_url': None, 'mongo_url': None, 'mongo_db': 'mppsolar', 'filter': None, 'excl_filter': None, 'keep_case': False}
Parameter                       Value           Unit
error                           Inverter returned the command string for QPGS0 - the inverter didnt recognise this command
2022-07-24 09:27:48,292:DEBUG:__init__:main@448: Not daemon, so not looping
```

I think the command is not supported.

jblance, 2022-07-24:
> 2022-07-24 ***@***.***: Solar Device Command Utility, version: 0.13.0, add postgres support from skoky
> 2022-07-24 ***@***.***: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f84b412e0>, _isConnected=False)
> 2022-07-24 ***@***.***: udp port 5555
> 2022-07-24 ***@***.***: Using Postgres None
> 2022-07-24 ***@***.***: Using Mongo None with mppsolar
> 2022-07-24 ***@***.***: Creating device "unnamed" (type: "mppsolar") on port "/dev/hidraw0 (porttype=None)" using protocol "PI30"
> 2022-07-24 ***@***.***: device_class <class 'mppsolar.devices.mppsolar.mppsolar'>
> 2022-07-24 ***@***.***: __init__ args ()
> 2022-07-24 ***@***.***: __init__ kwargs {'name': 'unnamed', 'port': '/dev/hidraw0', 'protocol': 'PI30', 'baud': 2400, 'porttype': None, 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f84b412e0>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
> 2022-07-24 ***@***.***: port matches hidraw
> 2022-07-24 ***@***.***: Using hidrawio for communications
> 2022-07-24 ***@***.***: Protocol PI30
> 2022-07-24 ***@***.***: __init__ name unnamed, port <mppsolar.inout.hidrawio.HIDRawIO object at 0x7f842900a0>, protocol <mppsolar.protocols.pi30.pi30 object at 0x7f84284c70>
> 2022-07-24 ***@***.***: Commands [(<mppsolar.devices.mppsolar.mppsolar object at 0x7f84290040>, 'QPGS0', 'QPGS0', 'screen', None, None)]
> 2022-07-24 ***@***.***: Looping 1 commands
> [... cut ...] ne, results_topic='mpp-solar', mqttc=<paho.mqtt.client.Client object at 0x7f84b412e0>, _isConnected=False), 'udp_port': '5555', 'postgres_url': None, 'mongo_url': None, 'mongo_db': 'mppsolar', 'filter': None, 'excl_filter': None, 'keep_case': False}
> 2022-07-24 ***@***.***: Not daemon, so not looping

Bummer

What inverter do you have again?

realSater, 2022-07-24:
Brand: Crown Micro
Model: Durant
Model Number: DT-2KW-M

2000W inverter

Unfortunately I cannot seem to find the exact model name on the internet 😂, there are other models with lower Watt. - makes me think it is discontinued...
 
It is made in China 🇨🇳 

But on mpp-solar when i use QPI it shows:

`protocol_id                     PI30`

jblance, 2022-07-24:
Try command Q1

realSater, 2022-07-24:
Q1 doesn't work either

realSater, 2022-07-24:
Q2 fetched the below:

```
solar@solar:~ $ sudo mpp-solar -P PI30 -c Q2 -p /dev/hidraw0
Command: Q2 - No description found
------------------------------------------------------------
Parameter                       Value           Unit
warning                         No definition for command Q2 in protocol b'PI30'
response                        (010 020 030 040 050 060&ê
```

jblance, 2022-07-24:
Ok. That's not a command I've seen before. But it looks like "valid charge
amounts" or similar

This means there are unknown commands are implemented

jblance, 2022-07-24:
Does it come with any software ?

realSater, 2022-07-24:
I am have knowledge in what I am doing, would you like me to bruteforce the commands?

if so can you give me the general letters that usually all manufacturers set in their inverters?

like:
letter Q
letter S
letter V
letter P
letter 1

etc...

jblance, 2022-07-24:
Does Q3 or Q4 etc give any response?

realSater, 2022-07-24:
> Does it come with any software ?
> […](#)

unfortunately it doesn't, i barely ran it using watchpower on a windows machine

jblance, 2022-07-24:
If you do `-c` without a command it will show all the available commands
for pi30

Generally commands start with Q for queries

realSater, 2022-07-24:
> If you do `-c` without a command it will show all the available commands for pi30 Generally commands start with Q for queries

great! do you know if there is a maximum length for the commands? say a maximum of 5 letters?

jblance, 2022-07-24:
I've not seen a max length and some are quite long

realSater, 2022-07-24:
are you suggesting that the length can be more than 8 characters? including the letter Q?

jblance, 2022-07-24:
Some commands include the date and time so are longer than 8 chars in total

Most generic  queries are 5 chars or less tho

realSater, 2022-07-24:
can we leave this thread open till I get back with the results?

jblance, 2022-07-24:
Yup

realSater, 2022-07-24:
This thread is going to be long... I hope you don't mind.
I will start posting the results once they finish.

so far i noticed that if you issue any command simultaneously it sometimes can generate those weird output data i mentioned up like:

```
Command: Q2 - No description found
------------------------------------------------------------
Parameter                       Value           Unit
warning                         No definition for command Q2 in protocol b'PI30'
response                        (010 020 030 040 050 060&ê
```

But if you stop the mpp service and run the commands one by one without any other interference, you will get a correct response whether the command is invalid or not.

So far I have checked the below:

Qn where n is a number from 0-9
Qc where c is a capital letter from A-Z
Qcc where both c are capital letters from A-Z
Qcn where c is a number from A-Z and n is a number from 0-9
Qnc where n is a number from 0-9 and where c is a number from A-Z
Qnn where both n are numbers from 0-9

c stands for character and n stands for number 🤣 

I still have:
Qccc
Qncc
Qcnc
Qccn
Qcccc
Qnccc
Qcncc
Qccnc
Qcccn

Will keep you updated

realSater, 2022-10-19:
I am closing this thread.

I was calling `QFLAG#QMCHGCR#QMOD#QMUCHGCR#QPIGS#QPIRI` which is absolutely not necessary and a waste of pull requests from the inverter to mpp-solar.

Everyone who is getting such error or any kind of error like "response too short" is simply due to misconfiguration of the number of times mpp-solar is pulling data from any inverter.

You cannot run mpp-solar simultaneously (you will see errors when running 2 commands at the exact same time or during data pull time)

simply make sure that everything is stopped, there are no cronjobs making mpp-solar run every minute, or any software like monit or such, then trigger your command WITHOUT `#` or `,` because they are useless and have no use (at least to my experience)

closed 2022-10-19 by realSater