jblance/mpp-solar #241: Will a USB connection via FTDI USB RS485 adapter to JK BMS work with this project?

frederickjh, 2022-07-25:
Hello,
I am not sure if this would work but thought I would ask if a setup using a RS485 to USB adapter as [shown here](https://solar-assistant.io/help/battery/jk-bms), would work with jblance/mppsolar?

On the old [jbalance/jkbms](https://github.com/jblance/jkbms) project page it says, "Python library to talk to a JK-B1A24S or B2A24S battery monitor via **bluetooth (BLE)**." Which says it was only via Bluetooth Low Energy.

In [usage.md](https://github.com/jblance/mpp-solar/blob/master/docs/usage.md) it says in the [JKBMS Usage section](https://github.com/jblance/mpp-solar/blob/master/docs/usage.md#jkbms-usage),:
`-p PORT, --port PORT  Specifies the device communications port (/dev/ttyUSB0 [default], /dev/hidraw0, test, ...)`
which would lead one to believe that it is possible as USB ports are given as examples.

In the README.md it says in the [Protocol / Command References section](https://github.com/jblance/mpp-solar/tree/master/docs#protocol--command-references) : 
`JK-B1A24S-RS485-protocol-english.pdf | JKBMS RS485 protocol (un-implemented)`
which makes me unsure if it is supported.

So, before I go and buy a FTDI USB RS485 adapter for this can you tell me if this would work or not?
Thanks for your time and help!
Frederick

preussal, 2022-07-29:
Hi I bought two of the RS485 adapters and RS485 USB Dongels. 
It doesn't seem to work at the moment.

**jkbms -p /dev/ttyUSB1 -n "Battery2" -PJK485 -D**

```
2022-07-29 13:08:56,031:INFO:__init__:main@223: Solar Device Command Utility, version: 0.13.0, add postgres support from skoky
2022-07-29 13:08:56,032:DEBUG:__init__:main@237: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='jkbms', mqttc=<paho.mqtt.client.Client object at 0xffff7fc41c40>, _isConnected=False)
2022-07-29 13:08:56,032:DEBUG:__init__:main@239: udp port 5555
2022-07-29 13:08:56,032:DEBUG:__init__:main@241: Using Postgres None
2022-07-29 13:08:56,032:DEBUG:__init__:main@244: Using Mongo None with mppsolar
2022-07-29 13:08:56,032:INFO:__init__:main@346: Creating device "Battery2" (type: "jkbms") on port "/dev/ttyUSB1 (porttype=None)" using protocol "JK485"
2022-07-29 13:08:56,037:DEBUG:__init__:main@350: device_class <class 'mppsolar.devices.jkbms.jkbms'>
2022-07-29 13:08:56,037:DEBUG:device:__init__@33: __init__ args ()
2022-07-29 13:08:56,037:DEBUG:device:__init__@34: __init__ kwargs {'name': 'Battery2', 'port': '/dev/ttyUSB1', 'protocol': 'JK485', 'baud': 2400, 'porttype': None, 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='jkbms', mqttc=<paho.mqtt.client.Client object at 0xffff7fc41c40>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2022-07-29 13:08:56,037:DEBUG:__init__:get_port_type@65: port matches ttyusb
2022-07-29 13:08:56,038:INFO:__init__:get_port@107: Using serialio for communications
2022-07-29 13:08:56,047:DEBUG:__init__:get_protocol@12: Protocol JK485
2022-07-29 13:08:56,050:DEBUG:device:__init__@38: __init__ name Battery2, port <mppsolar.inout.serialio.SerialIO object at 0xffff7fc2d0a0>, protocol <mppsolar.protocols.jk485.jk485 object at 0xffff7fc2d850>
2022-07-29 13:08:56,051:DEBUG:__init__:main@399: Commands [(<mppsolar.devices.jkbms.jkbms object at 0xffff7fc22b20>, '', '', 'screen', None, None)]
2022-07-29 13:08:56,051:INFO:__init__:main@404: Looping 1 commands
2022-07-29 13:08:56,051:INFO:__init__:main@414: Getting results from device: jkbms device - name: Battery2, port: <mppsolar.inout.serialio.SerialIO object at 0xffff7fc2d0a0>, protocol: <mppsolar.protocols.jk485.jk485 object at 0xffff7fc2d850> for command: , tag: , outputs: screen
2022-07-29 13:08:56,051:INFO:device:run_command@274: Running command 
2022-07-29 13:08:56,051:INFO:jk485:get_full_command@97: Using protocol b'JK485' with 1 commands
2022-07-29 13:08:56,051:DEBUG:jkabstractprotocol:get_command_defn@106: get_command_defn for: getBalancerData
2022-07-29 13:08:56,052:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'getBalancerData'
2022-07-29 13:08:56,052:DEBUG:abstractprotocol:get_command_defn@62: Found command getBalancerData in protocol b'JK485'
2022-07-29 13:08:56,052:DEBUG:jk485:get_full_command@111: cmd with header: bytearray(b'U\xaa\x00\x00\x00\x00\x00')
2022-07-29 13:08:56,052:DEBUG:jk485:get_full_command@114: cmd with header + slave address: bytearray(b'U\xaa\x01\x00\x00\x00\x00')
2022-07-29 13:08:56,052:DEBUG:jk485:get_full_command@119: cmd with command code and frame data: bytearray(b'U\xaa\x01\xff\x00\x00\x00')
2022-07-29 13:08:56,052:DEBUG:jk485:get_full_command@122: cmd with crc: bytearray(b'U\xaa\x01\xff\x00\x00\xff')
2022-07-29 13:08:56,052:INFO:device:run_command@303: full command bytearray(b'U\xaa\x01\xff\x00\x00\xff') for command getBalancerData
2022-07-29 13:08:56,053:DEBUG:jkabstractprotocol:get_command_defn@106: get_command_defn for: getBalancerData
2022-07-29 13:08:56,053:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'getBalancerData'
2022-07-29 13:08:56,053:DEBUG:abstractprotocol:get_command_defn@62: Found command getBalancerData in protocol b'JK485'
2022-07-29 13:08:56,053:DEBUG:serialio:send_and_receive@19: port /dev/ttyUSB1, baudrate 2400
2022-07-29 13:08:56,061:DEBUG:serialio:send_and_receive@22: Executing command via serialio...
2022-07-29 13:08:57,164:DEBUG:serialio:send_and_receive@30: serial response was: b''
2022-07-29 13:08:57,169:DEBUG:device:run_command@322: Send and Receive Response b''
2022-07-29 13:08:57,169:INFO:abstractprotocol:decode@196: response passed to decode: b''
2022-07-29 13:08:57,169:DEBUG:jkabstractprotocol:get_command_defn@106: get_command_defn for: getBalancerData
2022-07-29 13:08:57,170:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'getBalancerData'
2022-07-29 13:08:57,170:DEBUG:abstractprotocol:get_command_defn@62: Found command getBalancerData in protocol b'JK485'
2022-07-29 13:08:57,171:INFO:abstractprotocol:decode@236: Processing response of type POSITIONAL
2022-07-29 13:08:57,171:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['Hex2Str', 2, 'Header', '']
2022-07-29 13:08:57,171:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['Hex2Str', 1, 'Slave Address', '']
2022-07-29 13:08:57,171:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['Hex2Str', 1, 'Command Code', '']
2022-07-29 13:08:57,172:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['BigHex2Short:r/100', 2, 'Total Battery Voltage', 'V']
2022-07-29 13:08:57,172:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['BigHex2Short:r/1000', 2, 'Average Cell Voltage', 'V']
2022-07-29 13:08:57,172:DEBUG:jkabstractprotocol:get_responses@131: Got defn: ['Hex2Int', 1, 'Number of Cells', '']
[... cut ...]
  File "/root/venv/mppsolar/src/mppsolar/mppsolar/protocols/abstractprotocol.py", line 460, in decode
    data_name, value, data_units = self.process_response(
  File "/root/venv/mppsolar/src/mppsolar/mppsolar/protocols/abstractprotocol.py", line 177, in process_response
    r = eval(format_string)
  File "<string>", line 1, in <module>
  File "/root/venv/mppsolar/src/mppsolar/mppsolar/protocols/protocol_helpers.py", line 95, in Hex2Int
    answer = hexString[0]
IndexError: index out of range
```

syssi, 2022-07-29:
Could you provide the model name of your BMS and a photo of your wiring?

preussal, 2022-07-29:
Everything works with Bluetooth

**jkbms -p "C8:47:8C:E2:90:FA" -b 148000 -n "Battery #1" -P JK02 -c getInfo**
```

Command: getInfo - BLE Device Information inquiry
------------------------------------------------------------
Parameter                     	Value           Unit
header                        	55aaeb90       	    
record_type                   	03             	    
record_counter bytearray(b';') 
device_model                  	JK-B2A24S15P   	    
hardware_version              	10.XW          	    
software_version              	10.08          	    
up_time                       	2D20H4M60S     	    
power-on_times bytearray(b'\x03\x00\x00\x00') 
device_name                   	Battery #1     	    
device_passcode               	1234           	    
manufacturing_date            	220726         	    
serial_number                 	2042102267     	    
passcode                      	
user_data                     	Battery #1a    	    
setup_passcode
```
**jkbms -p "C8:47:8C:E2:90:FA" -b 148000 -n "Battery #1" -P JK02 -c getCellData**

```
Command: getCellData - BLE Cell Data inquiry
------------------------------------------------------------
Parameter                     	Value           Unit
header                        	55aaeb90       	    
record_type                   	02             	    
record_counter bytearray(b'@') 
voltage_cell01                	3.382          	V   
voltage_cell02                	3.382          	V   
voltage_cell03                	3.382          	V   
voltage_cell04                	3.383          	V   
voltage_cell05                	3.382          	V   
voltage_cell06                	3.383          	V   
voltage_cell07                	3.385          	V   
voltage_cell08                	3.383          	V   
voltage_cell09                	3.382          	V   
voltage_cell10                	3.385          	V   
voltage_cell11                	3.377          	V   
voltage_cell12                	3.378          	V   
voltage_cell13                	3.383          	V   
voltage_cell14                	3.382          	V   
voltage_cell15                	3.383          	V   
voltage_cell16                	3.381          	V   
voltage_cell17                	0.0            	V   
voltage_cell18                	0.0            	V   
voltage_cell19                	0.0            	V   
voltage_cell20                	0.0            	V   
voltage_cell21                	0.0            	V   
voltage_cell22                	0.0            	V   
voltage_cell23                	0.0            	V   
voltage_cell24                	0.0            	V   
average_cell_voltage          	3.382          	V   
delta_cell_voltage            	0.008          	V   
current_balancer              	2.566          	    
resistance_cell01             	0.065          	Ohm 
resistance_cell02             	0.061          	Ohm 
resistance_cell03             	0.064          	Ohm 
resistance_cell04             	0.064          	Ohm 
resistance_cell05             	0.065          	Ohm 
resistance_cell06             	0.064          	Ohm 
resistance_cell07             	0.066          	Ohm 
resistance_cell08             	0.07           	Ohm 
resistance_cell09             	0.064          	Ohm 
resistance_cell10             	0.062          	Ohm 
resistance_cell11             	0.063          	Ohm 
resistance_cell12             	0.063          	Ohm 
resistance_cell13             	0.062          	Ohm 
resistance_cell14             	0.062          	Ohm 
resistance_cell15             	0.064          	Ohm 
resistance_cell16             	0.07           	Ohm 
resistance_cell17             	0.0            	Ohm 
resistance_cell18             	0.0            	Ohm 
resistance_cell19             	0.0            	Ohm 
resistance_cell20             	0.0            	Ohm 
resistance_cell21             	0.0            	Ohm 
resistance_cell22             	0.0            	Ohm 
resistance_cell23             	0.0            	Ohm 
resistance_cell24             	0.0            	Ohm 
battery_voltage               	54.114         	V   
battery_power                 	0              	W   
balance_current               	0.0            	A   
battery_t1                    	32.3           	°C  
battery_t2                    	32.2           	°C  
mos_temp                      	36.0           	°C  
percent_remain bytearray(b'b') %
capacity_remain               	276.962        	Ah  
nominal_capacity              	280.0          	Ah  
cycle_count                   	0              	    
cycle_capacity                	0.482          	Ah  
time                          	2D20H38M18S    	    
current_charge                	0.003          	    
current_discharge             	0.0          
```

preussal, 2022-07-29:
i Try the other one now

preussal, 2022-07-29:
Interesting, I exchanged A and B. 
So Cabling
Yellow -> B
White -> A

**jkbms -p /dev/ttyUSB1 -n "Battery2" -P JK485 -D**
```
2022-07-29 14:53:17,196:INFO:__init__:main@223: Solar Device Command Utility, version: 0.13.0, add postgres support from skoky
2022-07-29 14:53:17,197:DEBUG:__init__:main@237: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='jkbms', mqttc=<paho.mqtt.client.Client object at 0xffff82b2ac40>, _isConnected=False)
2022-07-29 14:53:17,197:DEBUG:__init__:main@239: udp port 5555
2022-07-29 14:53:17,197:DEBUG:__init__:main@241: Using Postgres None
2022-07-29 14:53:17,197:DEBUG:__init__:main@244: Using Mongo None with mppsolar
2022-07-29 14:53:17,197:INFO:__init__:main@346: Creating device "Battery2" (type: "jkbms") on port "/dev/ttyUSB1 (porttype=None)" using protocol "JK485"
2022-07-29 14:53:17,201:DEBUG:__init__:main@350: device_class <class 'mppsolar.devices.jkbms.jkbms'>
2022-07-29 14:53:17,202:DEBUG:device:__init__@33: __init__ args ()
2022-07-29 14:53:17,202:DEBUG:device:__init__@34: __init__ kwargs {'name': 'Battery2', 'port': '/dev/ttyUSB1', 'protocol': 'JK485', 'baud': 2400, 'porttype': None, 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='jkbms', mqttc=<paho.mqtt.client.Client object at 0xffff82b2ac40>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2022-07-29 14:53:17,202:DEBUG:__init__:get_port_type@65: port matches ttyusb
2022-07-29 14:53:17,202:INFO:__init__:get_port@107: Using serialio for communications
2022-07-29 14:53:17,210:DEBUG:__init__:get_protocol@12: Protocol JK485
2022-07-29 14:53:17,214:DEBUG:device:__init__@38: __init__ name Battery2, port <mppsolar.inout.serialio.SerialIO object at 0xffff82b160a0>, protocol <mppsolar.protocols.jk485.jk485 object at 0xffff82b16850>
2022-07-29 14:53:17,214:DEBUG:__init__:main@399: Commands [(<mppsolar.devices.jkbms.jkbms object at 0xffff82b0ceb0>, '', '', 'screen', None, None)]
2022-07-29 14:53:17,214:INFO:__init__:main@404: Looping 1 commands
2022-07-29 14:53:17,215:INFO:__init__:main@414: Getting results from device: jkbms device - name: Battery2, port: <mppsolar.inout.serialio.SerialIO object at 0xffff82b160a0>, protocol: <mppsolar.protocols.jk485.jk485 object at 0xffff82b16850> for command: , tag: , outputs: screen
2022-07-29 14:53:17,215:INFO:device:run_command@274: Running command 
2022-07-29 14:53:17,215:INFO:jk485:get_full_command@97: Using protocol b'JK485' with 1 commands
2022-07-29 14:53:17,215:DEBUG:jkabstractprotocol:get_command_defn@106: get_command_defn for: getBalancerData
2022-07-29 14:53:17,215:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'getBalancerData'
2022-07-29 14:53:17,215:DEBUG:abstractprotocol:get_command_defn@62: Found command getBalancerData in protocol b'JK485'
2022-07-29 14:53:17,215:DEBUG:jk485:get_full_command@111: cmd with header: bytearray(b'U\xaa\x00\x00\x00\x00\x00')
2022-07-29 14:53:17,215:DEBUG:jk485:get_full_command@114: cmd with header + slave address: bytearray(b'U\xaa\x01\x00\x00\x00\x00')
2022-07-29 14:53:17,215:DEBUG:jk485:get_full_command@119: cmd with command code and frame data: bytearray(b'U\xaa\x01\xff\x00\x00\x00')
2022-07-29 14:53:17,216:DEBUG:jk485:get_full_command@122: cmd with crc: bytearray(b'U\xaa\x01\xff\x00\x00\xff')
2022-07-29 14:53:17,216:INFO:device:run_command@303: full command bytearray(b'U\xaa\x01\xff\x00\x00\xff') for command getBalancerData
2022-07-29 14:53:17,216:DEBUG:jkabstractprotocol:get_command_defn@106: get_command_defn for: getBalancerData
2022-07-29 14:53:17,216:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'getBalancerData'
2022-07-29 14:53:17,216:DEBUG:abstractprotocol:get_command_defn@62: Found command getBalancerData in protocol b'JK485'
2022-07-29 14:53:17,216:DEBUG:serialio:send_and_receive@19: port /dev/ttyUSB1, baudrate 2400
2022-07-29 14:53:17,223:DEBUG:serialio:send_and_receive@22: Executing command via serialio...
2022-07-29 14:53:18,368:DEBUG:serialio:send_and_receive@30: serial response was: b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU [... cut ...] a5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xa5\xff\x01'	    
2022-07-29 14:53:18,417:DEBUG:__init__:main@448: Not daemon, so not looping
```

preussal, 2022-07-29:
Open JK RS485
[image]

I can Try with this tow
[image]
[image]

frederickjh, 2022-07-29:
Here is [info](https://github.com/syssi/esphome-jk-bms?spm=a2g0o.detail.1000023.3.7f2a4553Zejmmw#schematics) from another project on connecting a TTL to the JK BMS.

frederickjh, 2022-07-29:
@preussal  And this [info](https://solar-assistant.io/help/battery/jk-bms) from the Solar Assistant page about connecting the JK BMS.

frederickjh, 2022-07-29:
> @preussal I saw somewhere someone complaining that the JK BMS RS485 adapters were flaky and did not always work well and some times not at all. They switched to a TTL level converter and connected it directly to the RS485 port on the JK BMS.

I meant a TTL to serial converter not a TTL to RS485 could be directly connected to the JK BMS RS485 connector on the BMS instead of using their "RS485 box" and a RS485 to USB serial adapter.

frederickjh, 2022-07-29:
@preussal Another [link](https://github.com/PurpleAlien/jk-bms_grafana). This one idenitfies the pins on the TTL out socket (RS485).

syssi, 2022-07-29:
I support @frederickjh proposal. Feel free to remove both RS485 converters if you don't need long wires between your BMS and the computer.

This is your setup at the moment:

```
JK-BMS RS485 connector (UART-TTL 3.3V) <--> (UART-TTL 3.3V) JK-RS485 Module (RS485 -7V/12V logic level) <--> (RS485 -7V/12V logic level) RS485 Converter Module (UART-TTL 3.3V ) <--> (UART-TTL 3.3V ) FTDI-USB
```

This is the simplest solution by avoiding RS485:

```
JK-BMS RS485 connector (UART-TTL 3.3V) <--> (UART-TTL 3.3V ) FTDI-USB

```

preussal, 2022-08-26:
> I support @frederickjh proposal. Feel free to remove both RS485 converters if you don't need long wires between your BMS and the computer.
>
> This is your setup at the moment:
>
> ```
> JK-BMS RS485 [...]

Hi, 
So I have now tried 4 - 5 different TTL Uart adapter but I only get data garbage.
I also built a BMS TTL -> TTL / RS485 -> RS485 / USB. But here, too, I only get data garbage.

I'm still waiting for two TTL Fido USB adapter. Let's see what they bring.
I also tried every adapter with various Baud Rates because I saw a baud rate from 115200 in another forum.

preussal, 2022-08-26:
I have now found an adapter where I can at least get something back from a BMS
```
[Fri Aug 26 22:54:46 2022] usb 1-1.1: new full-speed USB device number 13 using xhci_hcd
[Fri Aug 26 22:54:46 2022] usb 1-1.1: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.64
[Fri Aug 26 22:54:46 2022] usb 1-1.1: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[Fri Aug 26 22:54:46 2022] usb 1-1.1: Product: USB Serial
[Fri Aug 26 22:54:46 2022] ch341 1-1.1:1.0: ch341-uart converter detected
[Fri Aug 26 22:54:46 2022] usb 1-1.1: ch341-uart converter now attached to ttyUSB1
```

BMS 1
device_model                  	JK-B2A24S15P   	    
hardware_version              	10.XW          	    
software_version              	10.08          	 

```
jkbms -p /dev/ttyUSB1 -P JK232 -D
2022-08-26 23:01:07,463:INFO:__init__:main@223: Solar Device Command Utility, version: 0.13.1, add hass force update support from sebmueller
2022-08-26 23:01:07,463:DEBUG:__init__:main@237: MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='jkbms', mqttc=<paho.mqtt.client.Client object at 0xffff9eb5cc70>, _isConnected=False)
2022-08-26 23:01:07,464:DEBUG:__init__:main@239: udp port 5555
2022-08-26 23:01:07,464:DEBUG:__init__:main@241: Using Postgres None
2022-08-26 23:01:07,464:DEBUG:__init__:main@244: Using Mongo None with mppsolar
2022-08-26 23:01:07,464:INFO:__init__:main@346: Creating device "unnamed" (type: "jkbms") on port "/dev/ttyUSB1 (porttype=None)" using protocol "JK232"
2022-08-26 23:01:07,469:DEBUG:__init__:main@350: device_class <class 'mppsolar.devices.jkbms.jkbms'>
2022-08-26 23:01:07,469:DEBUG:device:__init__@33: __init__ args ()
2022-08-26 23:01:07,469:DEBUG:device:__init__@34: __init__ kwargs {'name': 'unnamed', 'port': '/dev/ttyUSB1', 'protocol': 'JK232', 'baud': 2400, 'porttype': None, 'mqtt_broker': MqttBroker(name='localhost', port=1883, username=None, password=None, results_topic='jkbms', mqttc=<paho.mqtt.client.Client object at 0xffff9eb5cc70>, _isConnected=False), 'udp_port': '5555', 'mongo_url': None, 'mongo_db': 'mppsolar'}
2022-08-26 23:01:07,469:DEBUG:__init__:get_port_type@65: port matches ttyusb
2022-08-26 23:01:07,469:INFO:__init__:get_port@107: Using serialio for communications
2022-08-26 23:01:07,478:DEBUG:__init__:get_protocol@12: Protocol JK232
2022-08-26 23:01:07,481:DEBUG:device:__init__@38: __init__ name unnamed, port <mppsolar.inout.serialio.SerialIO object at 0xffff9ead0f40>, protocol <mppsolar.protocols.jk232.jk232 object at 0xffff9ead0fa0>
2022-08-26 23:01:07,481:DEBUG:__init__:main@399: Commands [(<mppsolar.devices.jkbms.jkbms object at 0xffff9ead0d00>, '', '', 'screen', None, None)]
2022-08-26 23:01:07,481:INFO:__init__:main@404: Looping 1 commands
2022-08-26 23:01:07,481:INFO:__init__:main@414: Getting results from device: jkbms device - name: unnamed, port: <mppsolar.inout.serialio.SerialIO object at 0xffff9ead0f40>, protocol: <mppsolar.protocols.jk232.jk232 object at 0xffff9ead0fa0> for command: , tag: , outputs: screen
2022-08-26 23:01:07,481:INFO:device:run_command@274: Running command 
2022-08-26 23:01:07,482:INFO:jk232:get_full_command@82: Using protocol b'JK232' with 1 commands
2022-08-26 23:01:07,482:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'getBalancerData'
2022-08-26 23:01:07,482:DEBUG:abstractprotocol:get_command_defn@62: Found command getBalancerData in protocol b'JK232'
2022-08-26 23:01:07,482:DEBUG:jk232:get_full_command@99: cmd with start bit: bytearray(b'\xdd\x00\x00\x00\x00\x00\x00')
2022-08-26 23:01:07,482:DEBUG:jk232:get_full_command@121: cmd with crc: bytearray(b'\xdd\xa5\x03\x00\xff\xfdw')
2022-08-26 23:01:07,482:INFO:device:run_command@303: full command bytearray(b'\xdd\xa5\x03\x00\xff\xfdw') for command getBalancerData
2022-08-26 23:01:07,482:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'getBalancerData'
2022-08-26 23:01:07,482:DEBUG:abstractprotocol:get_command_defn@62: Found command getBalancerData in protocol b'JK232'
2022-08-26 23:01:07,482:DEBUG:serialio:send_and_receive@19: port /dev/ttyUSB1, baudrate 2400
2022-08-26 23:01:07,491:DEBUG:serialio:send_and_receive@22: Executing command via serialio...
2022-08-26 23:01:08,593:DEBUG:serialio:send_and_receive@30: serial response was: b''
2022-08-26 23:01:08,596:DEBUG:device:run_command@322: Send and Receive Response b''
2022-08-26 23:01:08,597:INFO:abstractprotocol:decode@196: response passed to decode: b''
2022-08-26 23:01:08,597:DEBUG:abstractprotocol:get_command_defn@60: Processing command 'getBalancerData'
2022-08-26 23:01:08,597:DEBUG:abstractprotocol:get_command_defn@62: Found command getBalancerData in protocol b'JK232'
2022-08-26 23:01:08,597:INFO:abstractprotocol:decode@236: Processing response of type POSITIONAL
2022-08-26 23:01:08,598:DEBUG:jk232:get_responses@149: get_responses: responses [b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'']
2022-08-26 23:01:08,598:DEBUG:abstractprotocol:decode@240: trimmed and split responses: [b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'', b'']
2022-08-26 23:01:08,598:DEBUG:abstractprotocol:decode@409: Processing POSITIONAL type responses
2022-08-26 23:01:08,599:DEBUG:abstractprotocol:decode@431: Got defn ['Hex2Str', 1, 'Start Byte', '']
[... cut ...]
  File "/root/venv/mppsolar/lib/python3.8/site-packages/mppsolar/protocols/abstractprotocol.py", line 460, in decode
    data_name, value, data_units = self.process_response(
  File "/root/venv/mppsolar/lib/python3.8/site-packages/mppsolar/protocols/abstractprotocol.py", line 177, in process_response
    r = eval(format_string)
  File "<string>", line 1, in <module>
  File "/root/venv/mppsolar/lib/python3.8/site-packages/mppsolar/protocols/protocol_helpers.py", line 95, in Hex2Int
    answer = hexString[0]
IndexError: index out of range
```

meccip, 2022-09-14:
any news about this. it is working with ftdi cable ??

preussal, 2022-09-15:
Hi Meccip,

I opened a new ticket for my problem. 
https://github.com/jblance/mpp-solar/issues/257
Since I have a defective BMS where TTL doesn't work. 
And a second where TTL  works fine, but I don't get any data output.

With a TTL test script, I get data from the BMS.

With the TTL to RS485 adapters you have to be careful because there are two unsettling versions. Once with 9600 baud and once with 115200 baud. And this is not described on the RS485 adapters. And the 9600 only works for the older ones and the 115200.

The TTL variant with a TTL to USB seems best to me.
Since the reading takes too long via Bluetooth and then the BMS also makes a PIEP sound when data is queried, and then the channel is also blocked for the app

meccip, 2022-09-15:
Hello,
I have B2A24S15P Hard V10.XW Soft V10.08 made in 2022 09 10I use it with Bluetooth communication to get data to node red, but is very slow (slow, blocking the app, sometimes freeze and need bluetooth restart)
I try some flow in node with no success. I use a FT232RL industrial adaptor and I use Arduino as serial ttl adapter.
I measure between RX and TX and I have 1.3 Mohm
How do you test to see if TTL is working ?What do you use ?Can you give an exemple ?

preussal, 2022-09-16:
Hi Meccip,

i used the Script from here 
https://github.com/PurpleAlien/jk-bms_grafana

But my B2A24S15P BMS is not working. I measured around 2.5 megaohm.
With my JK-B2A24S20P I have 60 kilo ohm and I get data with the script

https://github.com/PurpleAlien/jk-bms_grafana/issues/2#issuecomment-1236952947

sw25481, 2023-05-14:
I do not think this is always a hardware issue.  I have a B2A24S20P and the RS485 dongle plugged into the GPS port on the BMS (as shown on the SolarAssistant page).  Works flawlessly with SolarAssistant and with the script from this project https://github.com/fah/jk-bms.  Connection details are

    bms = serial.Serial('/dev/ttyUSB1')
    bms.baudrate = 115200
    bms.timeout  = 0.2

but I get the same issues as you all with this project connecting on RS485