jblance/mpp-solar #243: [FR] add undocumented commands for inverters

Saentist, 2022-07-26:
There is a lot of commands with are supported by inverter but **mpp-solar** do not know how to describe
ex:
```
1.
Command: QVFW - Main CPU firmware version inquiry
------------------------------------------------------------
Parameter                       Value           Unit
main_cpu_firmware_version       VERFW:00041.15

2.
QVFW2
Parameter                       Value           Unit
error                           NAK

3.
Command: QVFW3 - No description found
------------------------------------------------------------
Parameter                       Value           Unit
warning                         No definition for command QVFW3 in protocol b'PI30'
response                        (VERFW:00002.40a

4.
QVFW4
Parameter                       Value           Unit
error                           NAK

5.
Command: QMN - No description found
------------------------------------------------------------
Parameter                       Value           Unit
warning                         No definition for command QMN in protocol b'PI30'
response                        (VMIII-5000¨'  <<<<Inverter Type

6.
Command: QBEQI - No description found
------------------------------------------------------------
Parameter                       Value           Unit
warning                         No definition for command QBEQI in protocol b'PI30'
response                        (0 060 030 010 030 58.40 000 120 0 0000HÞ

7.
Command: QMUCHGCR - Max Utility Charging Current Options inquiry
------------------------------------------------------------
Parameter                       Value           Unit
max_utility_charging_current    002             A
unknown_value_in_response_1     010
unknown_value_in_response_2     020
unknown_value_in_response_3     030
unknown_value_in_response_4     040
unknown_value_in_response_5     050
unknown_value_in_response_6     060
```
QVFW = OK
QVFW2, QVFW3, QVFW4, QMN, QBEQI , QMUCHGCR = KO

№ 2 and 4 don't have a formatting

other interesting QUERY commands
```
QMN (some product identification)
QSID (advanced version of QID)
QBEQI (battery charging parameters)
QET: Query total PV generated energy
QEYyyyy: Query PV generated energy of year
QEMyyyymm: Query PV generated energy of month
QEDyyyymmdd: Query PV generated energy of day
QLT: Query total output load energy
QLYyyyy: Query output load energy of year
QLMyyyymm: Query output load energy of monthQLMyyyymm: Query output load energy of month
QLDyyyymmdd: Query output load energy of dayQLDyyyymmdd: Query output load energy of day
```

jblance, 2022-07-26:
Try pi30max protocol

Saentist, 2022-07-26:
```
QOPPT<cr>: The device output source priority time order inquiry
Computer: QOPPT<CRC><cr>
Device: (M M M M M M M M M M M M M M M M M M M M M M M M N O O
O<CRC><cr>
M: 24 hour correspond to the output source priority (0: Utility first, 1: Solar first, 2: SBU)
N: device output source priority
O: selection of output source priority order

Example:
Computer: QOPPT<CRC><cr>
Device: (0 0 0 0 0 2 2 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 1 2<CRC><cr>
Means: the device output source priority time order is SBU from 5 to 6, and output source priority
is Utility first.
```
```
QCHPT<cr>: The device charger source priority time order inquiry
Computer: QCHPT<CRC><cr>
Device: (M M M M M M M M M M M M M M M M M M M M M M M M N O O
O<CRC><cr>
M: 24 hour correspond to the charger source priority (1: Solar first, 2: Solar + Utility, 3: Only solar
charging permitted)
N: device charger source priority
O: selection of o charger source priority order
Example:
Computer: QCHPT<CRC><cr>
Device: (1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 2 2 2 2 2 2 1 0<CRC><cr>
Means: the device charger source priority time order is solar + utility from 20 to 23, and charger
source priority is Solar first.
```

Saentist, 2022-07-26:
> Try pi30max protocol
Same situation

jblance, 2023-01-08:
yes unfortunately the inverter developers do not change the protocol id when they change the protocol contents
maybe i need to come up with another way of distinquishing the various inverters and which 'protocol' to use

closed 2023-01-08 by jblance

Saentist, 2023-01-09:
> yes unfortunately the inverter developers do not change the protocol id when they change the protocol contents maybe i need to come up with another way of distinquishing the various inverters and which 'protocol' to use

`QPI#QMN#QGMN`
Will filter lot more by `protocol + inverter series + model number`
This can be used to create database of inverters and supported commands

```bash
mpp-solar -p /dev/hidraw1 -o raw -c QPI#QMN#QGMN
Command: QPI - Protocol ID inquiry
------------------------------------------------------------
raw_response                    '(PI30\x9a\x0b\r'
Command: QMN - No description found
------------------------------------------------------------
raw_response                    "(VMIII-5000\xa8'\r"
Command: QGMN - No description found
------------------------------------------------------------
raw_response                    '(037aZ\r'

```