jblance/mpp-solar #246: Telegraf parsing question

klamanch, 2022-07-29:
Greetings jblance
Thank you for the excellent work on this.
I recently ran across your software and wanted to implement it for my LV5048. I had a standalone thinkcentre M600 x86 I had laying around with debian 11 server on it.
I followed the instructions and have the mqtt broker running and mpp-solar sending QPIGS messages via cron. No issues there. I also installed the telegraf, influxdb and grafana apps without issue.
I am stuck with telegraf sending an errorr that I cannot seem to get around. I searched profusely on everything I could find regarding telegraf and have been unsuccessful over this past week at figuring it out.
● telegraf.service - The plugin-driven server agent for reporting metrics into InfluxDB
     Loaded: loaded (/lib/systemd/system/telegraf.service; enabled; vendor preset: enabled)
     Active: active (running) since Fri 2022-07-29 12:22:39 MDT; 25s ago
       Docs: https://github.com/influxdata/telegraf
   Main PID: 12614 (telegraf)
      Tasks: 9 (limit: 9405)
     Memory: 26.7M
        CPU: 413ms
     CGroup: /system.slice/telegraf.service
             └─12614 /usr/bin/telegraf -config /etc/telegraf/telegraf.conf -config-directory /etc/telegraf/telegraf.d

Jul 29 12:23:02 debian-server telegraf[12614]: 2022-07-29T18:23:02Z E! [inputs.mqtt_consumer] Error in plugin: metric parse error: expected tag at 1:2: "0"
Jul 29 12:23:02 debian-server telegraf[12614]: 2022-07-29T18:23:02Z E! [inputs.mqtt_consumer] Error in plugin: metric parse error: expected tag at 1:2: "A"
Jul 29 12:23:02 debian-server telegraf[12614]: 2022-07-29T18:23:02Z E! [inputs.mqtt_consumer] Error in plugin: metric parse error: expected tag at 1:4: "301"
Jul 29 12:23:02 debian-server telegraf[12614]: 2022-07-29T18:23:02Z E! [inputs.mqtt_consumer] Error in plugin: metric parse error: expected tag at 1:2: "W"
Jul 29 12:23:02 debian-server telegraf[12614]: 2022-07-29T18:23:02Z E! [inputs.mqtt_consumer] Error in plugin: metric parse error: expected tag at 1:2: "0"
Jul 29 12:23:02 debian-server telegraf[12614]: 2022-07-29T18:23:02Z E! [inputs.mqtt_consumer] Error in plugin: metric parse error: expected tag at 1:5: "bool"
Jul 29 12:23:02 debian-server telegraf[12614]: 2022-07-29T18:23:02Z E! [inputs.mqtt_consumer] Error in plugin: metric parse error: expected tag at 1:2: "1"
Jul 29 12:23:02 debian-server telegraf[12614]: 2022-07-29T18:23:02Z E! [inputs.mqtt_consumer] Error in plugin: metric parse error: expected tag at 1:5: "bool"
Jul 29 12:23:02 debian-server telegraf[12614]: 2022-07-29T18:23:02Z E! [inputs.mqtt_consumer] Error in plugin: metric parse error: expected tag at 1:2: "0"
Jul 29 12:23:02 debian-server telegraf[12614]: 2022-07-29T18:23:02Z E! [inputs.mqtt_consumer] Error in plugin: metric parse error: expected tag at 1:5: "bool"
~
If I run QMOD in cron, telegraf doesn't like the space between "value" and "line" 

The cron.out file doesn't show any errors:
  
GNU nano 5.4                                                                                   cron.out                                                                                             
2022-07-29 12:36:01,979:INFO:__init__:main@236: Solar Device Command Utility, version: 0.7.76, remove brackets in ved protocol
2022-07-29 12:36:01,979:INFO:__init__:main@339: Creating device "unnamed" (type: "mppsolar") on port "/dev/ttyUSB0 (porttype=None)" using protocol "PI30"
2022-07-29 12:36:01,982:INFO:device:set_port@160: Using serialio for communications
2022-07-29 12:36:01,995:INFO:__init__:main@395: Looping 1 commands
2022-07-29 12:36:01,996:INFO:__init__:main@405: Getting results from device: mppsolar device - name: unnamed, port: <mppsolar.io.serialio.SerialIO object at 0x7fb302d9cfd0>, protocol: <mppsolar.pro>
2022-07-29 12:36:01,996:INFO:device:run_command@270: Running command QPIGS
2022-07-29 12:36:01,996:INFO:abstractprotocol:get_full_command@31: Using protocol b'PI30' with 36 commands
2022-07-29 12:36:01,996:INFO:device:run_command@296: full command b'QPIGS\xb7\xa9\r' for command QPIGS
2022-07-29 12:36:02,515:INFO:abstractprotocol:decode@175: response passed to decode: b'(123.4 59.9 119.9 59.9 0491 0471 019 209 52.90 001 051 0042 0007 081.1 52.97 00002 1010010 00 00 00391 010\xb6>
2022-07-29 12:36:02,518:INFO:abstractprotocol:decode@215: Processing response of type DEFAULT
2022-07-29 12:36:02,519:INFO:abstractprotocol:decode@226: Processing DEFAULT type responses
2022-07-29 12:36:02,520:INFO:device:run_command@334: Decoded response {'raw_response': ['(123.4 59.9 119.9 59.9 0491 0471 019 209 52.90 001 051 0042 0007 081.1 52.97 00002 1010010 00 00 00391 010¶\>
2022-07-29 12:36:02,521:INFO:__init__:get_outputs@26: attempting to create output processor: mqtt
2022-07-29 12:36:02,671:INFO:mqtt:output@59: Using output processor: mqtt

I know you stated that mpp-solar was only proved on ubuntu 20 and RP's. Any help appreciated.
best regards

jblance, 2022-07-30:
> 2022-07-29 ***@***.***: Solar Device Command
> 2022-07-29 ***@***.***: Creating device "unnamed"
> 2022-07-29 ***@***.***: Using serialio for
> 2022-07-29 ***@***.***: Looping 1 commands
> 2022-07-29 ***@***.***: Getting results from
> 2022-07-29 ***@***.***: Running command QPIGS
> 2022-07-29 ***@***.***: Using
> 2022-07-29 ***@***.***: full command
> 2022-07-29 ***@***.***: response passed
> 2022-07-29 ***@***.***: Processing
> 2022-07-29 ***@***.***: Decoded response
> 2022-07-29 ***@***.***: attempting to create
> 2022-07-29 ***@***.***: Using output processor: mqtt

You'll need to use one of the influx output processors. The base mqtt
doesn't format the numbers correctly for telegraf

klamanch, 2022-07-30:
Thank you for the reply.
Would there a better mqtt broker to use?
I don't know enough about which telegraf output processor to use unless you would have some insight.
It seems strange no-one has run across this in the discussion group or elsewhere.
Thank you again for the help.

jblance, 2022-07-30:
Within mppsolar there are a bunch of output processors.
Try running the command on the commandline (after disabling cron) until you
get what you want

Using `-o` without specifying the output processor will give a list.

Suggest influx_mqtt as a starting point

klamanch, 2022-07-30:
Thank you for your patience. It must get annoying at times. 
In looking through the issues I found the command in #214 similar and used a variation of the command in https://github.com/jblance/mpp-solar/issues/214#issuecomment-1106019014
mpp-solar -o influx2_mqtt -q 127.0.0.1 -p /dev/ttyUSB0 -P PI30 (as root) to work and telegraf runs without errors.
Influx also recognizes the data from mpp-solar. I can now connect with grafana. 
If anyone is interested- mpp-solar runs on debian 11 server also. 
Thanks!

closed 2022-08-05 by jblance