jblance/mpp-solar #251: jkbms & MQTT continuous polling

sigxcpu76, 2022-08-22:
Is there any possibility to perfom continuous polling and sending to MQTT? 
By continuous polling I'm referring the phone app working mode.
The BLE setup is extremely slow and prone to error (sometimes I get a stacktrace instead of data).

maxxximatoze, 2022-11-30:
I have the same problem, getting data over ble is very slow and not very reliable (up to 50% error). Don't know if it can be faster but it would be nice...

sigxcpu76, 2022-12-13:
I've switched to the serial interface and everything is smooth, except that some data is missing (but can be inferred).

sugar012, 2023-04-22:
How did you do it @sigxcpu76 ?

maxxximatoze, 2023-04-23:
> I have the same problem, getting data over ble is very slow and not very reliable (up to 50% error). Don't know if it can be faster but it would be nice...

for those interested in getting data over bluetooth every X seconds, here is a python script I have found on github, and slitglty modified to my needs. It is outputing data every 5 seconds in a file indefinitely. You need a good bluetooth connection (short distance). For me it works great, 720 sample per hour and almost no error.
[Jkbms.py.txt](https://github.com/jblance/mpp-solar/files/11303198/Jkbms.py.txt)

sugar012, 2023-04-23:
> 

Thank you, i'm trying it out since the RS485 seems to be not working at all.
I've [read ](https://github.com/jblance/mpp-solar/issues/228#issuecomment-1307615480) from another post that the new 11.XW hardware now supports 32 cells so the readings are a bit off. How do you think i can integrate this modification in your script?
I got this output now: 

{'model_nbr': 'BK-BLE-1.0', 

'device_info': 
{'hw_rev': '11.XW', 
'sw_rev': '11.26', 
'uptime': 1218900, 
'vendor_id': 'JK_B2A24S15P', 
'manufacturing_date': '230409'}, 
'last_update': 1682243168.0821383, 
'settings': {'cell_uvp': 2.8000000000000003, 
'cell_uvpr': 2.85, 
'cell_ovp': 3.66, 
'cell_ovpr': 3.64, 
'balance_trigger_voltage': 0.003, 
'power_off_voltage': 2.5, 
'max_charge_current': 120.0, 
'max_discharge_current': 150.0, 
'max_balance_current': 120.0, 
'cell_count': 16, 
'charging_switch': True, 
'discharging_switch': True, 
'balancing_switch': True}, 
'cell_info': 

{'voltages': [3.378, 3.362, 3.359, 3.356, 3.356, 3.361, 3.36, 3.359, 3.356, 3.355, 3.355, 3.356, 3.358, 3.356, 3.359, 3.364], 
'average_cell_voltage': 0.0, 
'delta_cell_voltage': 0.0, 
'max_voltage_cell': 0, 
'min_voltage_cell': 0, 
'resistances': [0.0, 0.0, 0.0, 65.535, 0.0, 3.359, 0.023, 0.768, 0.061, 0.062, 0.06, 0.06, 0.059000000000000004, 0.059000000000000004, 0.058, 0.058], 
'total_voltage': 0.0, 
'current': 0.0, 
'temperature_sensor_1': 0.0, 
'temperature_sensor_2': 0.0, 
'temperature_mos': 0.0, 
'balancing_current': 0.0, 
'balancing_action': 0.0, 
'battery_soc': 0, 
'capacity_remain': 19857.408, 
'capacity_nominal': 0.0, 
'cycle_count': 53745, 
'cycle_capacity': 2736.213, 
'charging_switch_enabled': False, 
'discharging_switch_enabled': False, 
'balancing_active': False, 
'error_bitmask_16': '0x0', 
'error_bitmask_2': '0000000000000000', 
'power': 0.0}, 

'warnings': 
{'resistance_too_high': False, 
'cell_count_wrong': False, 
'charge_overtemp': False, 
'charge_undertemp': False, 
'discharge_overtemp': False, 
'cell_overvoltage': False, 
'cell_undervoltage': False, 
'charge_overcurrent': False, 
'discharge_overcurrent': False}}

maxxximatoze, 2023-04-23:
Maybe the protcol has slightly changed between my v10 hardware version and your V11, I think it can be corrected because the script seems to get some good values like the cells voltage and other stuff.

I think your best bet is to modify this section to try to get the good values where they are missing:

TRANSLATE_DEVICE_INFO = [
    [["device_info", "hw_rev"], 22, "8s"],
    [["device_info", "sw_rev"], 30, "8s"],
    [["device_info", "uptime"], 38, "<L"],
    [["device_info", "vendor_id"], 6, "16s"],
    [["device_info", "manufacturing_date"], 78, "8s"],
]

TRANSLATE_SETTINGS = [
    [["settings", "cell_uvp"], 10, "<L", 0.001],
    [["settings", "cell_uvpr"], 14, "<L", 0.001],
    [["settings", "cell_ovp"], 18, "<L", 0.001],
    [["settings", "cell_ovpr"], 22, "<L", 0.001],
    [["settings", "balance_trigger_voltage"], 26, "<L", 0.001],
    [["settings", "power_off_voltage"], 46, "<L", 0.001],
    [["settings", "max_charge_current"], 50, "<L", 0.001],
    [["settings", "max_discharge_current"], 62, "<L", 0.001],
    [["settings", "max_balance_current"], 50, "<L", 0.001],
    [["settings", "cell_count"], 114, "<L"],
    [["settings", "charging_switch"], 118, "4?"],
    [["settings", "discharging_switch"], 122, "4?"],
    [["settings", "balancing_switch"], 126, "4?"],
]

TRANSLATE_CELL_INFO = [
    [["cell_info", "voltages", 16], 6, "<H", 0.001],
    [["cell_info", "average_cell_voltage"], 58, "<H", 0.001],
    [["cell_info", "delta_cell_voltage"], 60, "<H", 0.001],
    [["cell_info", "max_voltage_cell"], 62, "<B"],
    [["cell_info", "min_voltage_cell"], 63, "<B"],
    [["cell_info", "resistances", 16], 64, "<H", 0.001],
    [["cell_info", "total_voltage"], 118, "<H", 0.001],
    [["cell_info", "current"], 126, "<l", 0.001],
    [["cell_info", "temperature_sensor_1"], 130, "<H", 0.1],
    [["cell_info", "temperature_sensor_2"], 132, "<H", 0.1],
    [["cell_info", "temperature_mos"], 134, "<H", 0.1],
    [["cell_info", "balancing_current"], 138, "<H", 0.001],
    [["cell_info", "balancing_action"], 140, "<B", 0.001],
    [["cell_info", "battery_soc"], 141, "B"],
    [["cell_info", "capacity_remain"], 142, "<L", 0.001],
    [["cell_info", "capacity_nominal"], 146, "<L", 0.001],
    [["cell_info", "cycle_count"], 150, "<L"],
    [["cell_info", "cycle_capacity"], 154, "<L", 0.001],
    [["cell_info", "charging_switch_enabled"], 166, "1?"],
    [["cell_info", "discharging_switch_enabled"], 167, "1?"],
    [["cell_info", "balancing_active"], 191, "1?"],
]

maxxximatoze, 2023-04-23:
I also have a spare V11 hardware with V11.25 software, maybe i'll give it a try to see if I have the same behavior than yours.

sugar012, 2023-04-23:
> Maybe the protcol has slightly changed between my v10 hardware version and your V11, I think it can be corrected because the script seems to get some good values like the cells voltage and other [...]

Yep, i tried already simply adding "8" starting from "average_cell_voltage" ) but with no luck

sugar012, 2023-04-23:
> I also have a spare V11 hardware with V11.25 software, maybe i'll give it a try to see if I have the same behavior than yours.

from row 21 there's this, it seems to be implemented already?:

FRAME_VERSION_JK04 = 0x01
FRAME_VERSION_JK02 = 0x02
FRAME_VERSION_JK02_32S = 0x03
PROTOCOL_VERSION_JK02 = 0x02

protocol_version = PROTOCOL_VERSION_JK02

maxxximatoze, 2023-04-24:
> from row 21 there's this, it seems to be implemented already?:
>
> FRAME_VERSION_JK04 = 0x01 FRAME_VERSION_JK02 = 0x02 FRAME_VERSION_JK02_32S = 0x03 PROTOCOL_VERSION_JK02 = 0x02
>
> protocol_version = [...]

Do you have tried modifying protocol_version = PROTOCOL_VERSION_JK02 to protocol_version = PROTOCOL_VERSION_JK02_32S ? Does it works ?

sugar012, 2023-04-24:
Yes i tried. I think it's the FRAME parte that Need to be modified

sugar012, 2023-05-05:
> Do you have tried modifying protocol_version = PROTOCOL_VERSION_JK02 to protocol_version = PROTOCOL_VERSION_JK02_32S ? Does it works ?

any luck with this ?

alexanderfitu, 2023-07-16:
Is there any further development on this? I have been unsseucfeul in getting the above script to work, as my device uses JK04 protocotol version and I cannot figure out how to define a new protocol in the linked script. If anyone has a source for that script, I may be able to fix it with the right documentation. 

I am looping the script round and round to control charging and low cell volts on an EV motorbike, but the delay causes issues at low battery voltage when sag is not detected. A live stream like the JKBMS app would alleviate this.

maxxximatoze, 2023-07-16:
> Is there any further development on this? I have been unsseucfeul in getting the above script to work, as my device uses JK04 protocotol version and I cannot figure out how to define a new protocol [...]

Here is the version I am using right now for for some time and it works really well. It is configured to stay connected forever and to write data to a file every 5s, but you can modify it to your needs. Don't know if it works with JK04, not tried yet. Give some feedback here if you get some succes with it, it could help others...

import os
import asyncio
from bleak import BleakScanner, BleakClient
import time
from logging import info, debug
import logging
from struct import unpack_from, calcsize
import threading

logging.basicConfig(level=logging.INFO)
#logging.basicConfig(level=logging.DEBUG)

# zero means parse all incoming data (every second)
CELL_INFO_REFRESH_S = 0
DEVICE_INFO_REFRESH_S = 43200  # every 12 Hours
CHAR_HANDLE = "0000ffe1-0000-1000-8000-00805f9b34fb"
MODEL_NBR_UUID = "00002a24-0000-1000-8000-00805f9b34fb"

COMMAND_CELL_INFO = 0x96
COMMAND_DEVICE_INFO = 0x97

FRAME_VERSION_JK04 = 0x01
FRAME_VERSION_JK02 = 0x02
FRAME_VERSION_JK02_32S = 0x03
PROTOCOL_VERSION_JK02 = 0x02

protocol_version = PROTOCOL_VERSION_JK02

MIN_RESPONSE_SIZE = 300
MAX_RESPONSE_SIZE = 320

TRANSLATE_DEVICE_INFO = [
    [["device_info", "hw_rev"], 22, "8s"],
    [["device_info", "sw_rev"], 30, "8s"],
    [["device_info", "uptime"], 38, "<L"],
    [["device_info", "vendor_id"], 6, "16s"],
    [["device_info", "manufacturing_date"], 78, "8s"],
]

TRANSLATE_SETTINGS = [
    [["settings", "cell_uvp"], 10, "<L", 0.001],
    [["settings", "cell_uvpr"], 14, "<L", 0.001],
    [["settings", "cell_ovp"], 18, "<L", 0.001],
    [["settings", "cell_ovpr"], 22, "<L", 0.001],
    [["settings", "balance_trigger_voltage"], 26, "<L", 0.001],
    [["settings", "power_off_voltage"], 46, "<L", 0.001],
    [["settings", "max_charge_current"], 50, "<L", 0.001],
    [["settings", "max_discharge_current"], 62, "<L", 0.001],
    [["settings", "max_balance_current"], 50, "<L", 0.001],
    [["settings", "cell_count"], 114, "<L"],
    [["settings", "charging_switch"], 118, "4?"],
    [["settings", "discharging_switch"], 122, "4?"],
    [["settings", "balancing_switch"], 126, "4?"],
]

TRANSLATE_CELL_INFO = [
    [["cell_info", "voltages", 16], 6, "<H", 0.001],
    [["cell_info", "average_cell_voltage"], 58, "<H", 0.001],
    [["cell_info", "delta_cell_voltage"], 60, "<H", 0.001],
    [["cell_info", "max_voltage_cell"], 62, "<B"],
    [["cell_info", "min_voltage_cell"], 63, "<B"],
    [["cell_info", "resistances", 16], 64, "<H", 0.001],
    [["cell_info", "total_voltage"], 118, "<H", 0.001],
    [["cell_info", "current"], 126, "<l", 0.001],
    [["cell_info", "temperature_sensor_1"], 130, "<H", 0.1],
    [["cell_info", "temperature_sensor_2"], 132, "<H", 0.1],
[... 308 lines cut ...]
    #while True:
        #print("asdf")
        #bms_status = {}
        #print(jk.get_status())
        #out = jk.get_status()
        #file = open('jkbmsout.txt', 'wb')
        #pickle.dump(out, file)
        #file.close()
        #with open('jkbmsout.txt', 'w', encoding='utf-8' ) as my_file:
        # my_file.write(text.decode('utf-8'))
        # my_file.close()
        #clear = lambda: os.system('clear')
        #clear()
        #time.sleep(1)

if __name__ == "__main__":
    jk = JkBmsBle("C8:47:8C:E1:FA:99")
    jk.start_scraping()
    while True:
        out = jk.get_status()
        print(out)
        with open('/root/jkbms/tmpramfs/jkbmsout.txt', 'w') as file:
            file.write(str(out))
        #time.sleep(0.5)
        time.sleep(5)

alexanderfitu, 2023-07-16:
Thanks!

This version, and the one linked, generate the following error, I have updated my MAC ID to my BMS. 

error while connecting to bt: Characteristic 00002a24-0000-1000-8000-00805f9b34fb was not found!

I have no way of identifying my devices "MODEL_NBR_UUID"

If I copy CHAR_HANDLE  ID to MODEL_NBR_UUID, it then generates the following error:

Could not read characteristic handle 9: Protocol Error 0x02: Read Not Permitted

I can only assume the CHAR_HANDLE value is wrong as well. 

I have dug through the JKBMS and MPP_Solar codebases to see how they are identifying them, but as far as I am aware, they are defining each protcol manually, wheras I think this script is trying to decipher the protocol based on the answer.

alexanderfitu, 2023-07-16:
I now understand that MODEL_NBR_UUID is the UUID address of the devices model name. 
and that CHAR_HANDLE is the UUID of the charachter mapping?

in any case, all variations did not produce a ususable result.

Zuikkis, 2023-09-05:
> for those interested in getting data over bluetooth every X seconds, here is a python script I have found on github, and slitglty modified to my needs. It is outputing data every 5 seconds in a file indefinitely. You need a good bluetooth connection (short distance). For me it works great, 720 samples per hour and almost no error. [Jkbms.py.txt](https://github.com/jblance/mpp-solar/files/11303289/Jkbms.py.txt)

Hey thanks for this, it works very well with my JK-B1A24S15P. :)

Zuikkis, 2023-09-09:
Small improvement to the above jkbms.py. I was wondering why it keeps reconnecting, you could hear constant "beep, beep" every few seconds from the BMS. And when looking at the console I saw this line when it reconnected:

`INFO:root:error while connecting to bt: 'bool' object is not callable
`
The error is near line 330:

`while client.is_connected and self.run():
`

self.run is a bool, not a function! So you after removing the parenthesis, error no longer appears and the script stays connected to the BMS forever. Works very well now, and no longer beeps annoyingly. :)

Mr-EJ, 2023-10-12:
What if I have multiple BMSes

Do I put each bms MAC on a 'jk = JkBmsBle("XX:XX:XX:XX:XX:XX")' line?

maxxximatoze, 2024-08-27:
For those who want to use this simple python script to get data from their jkbms V11.XX hardware, here is the corrected version that I'm currently using flawlessly to collect my data every 5 seconds. The script for V10.XX hardware version is also included if needed.
[JkbmsV10XXandV11XX.zip](https://github.com/user-attachments/files/16762740/JkbmsV10XXandV11XX.zip)

alexanderfitu, 2024-10-13:
> For those who want to use this simple python script to get data from their jkbms V11.XX hardware, here is the corrected version that I'm currently using flawlessly to collect my data every 5 [...]

Thanks this works very well, It is exactly what I have been looking for. 

Out of interest, what do you do with the txt file creates? Dump into a DB or something else?

My old script, queries 2x BMS, then adds all the data into one array, then writes the data to a file, then cycles over and over again. Now that I have "live" data I have to decide how to best deal with it :)

maxxximatoze, 2024-10-13:
> Thanks this works very well, It is exactly what I have been looking for.
>
> Out of interest, what do you do with the txt file creates? Dump into a DB or something else?
>
> My old script, queries 2x BMS, [...]

Personally, I'm parsing the output txt file with bash script to do a lot of things, such as a log file, graphs, "realtime" status and alarms (sms and mail) without any external and complicated software, just gnu commands, better for very long term run imho...

alexanderfitu, 2024-10-13:
> Personally, I'm parsing the output txt file with bash script to do a lot of things, such as a log file, graphs, "realtime" status and alarms (sms and mail) without any external and complicated [...]

Great. That is essentially what I am doing ATM. I may use it to write some values to the LCD instead of the bash script that I have that runs in a loop. 

Is it possible to connect to 2x BMS simulatenously? Can I run 2 instances or input 2x macs in the script?

maxxximatoze, 2024-10-13:
You'll need two instances of the python script in parallel or you'll need to modify the script to handle the two bms at once.

Zuikkis, 2025-03-23:
I have a JK-B1A24S15P that I has worked successfully with this script, first with 14S LiPo pack, then with 16S LiFeP04, and now I have it in 20S LiPo.

It seems the total voltage overflows at 65.536V. :) Here's a graph from my home automation:

[image]

This is fixed by changing line:

    [["cell_info", "total_voltage"], 118, "<H", 0.001],

into:

    [["cell_info", "total_voltage"], 118, "<L", 0.001],

So total_voltage is read as 32bit integer, not 16bit..